evcc-io/evcc #15561: Ladestandsanzeige Hyundai (SOC Poll)

Efferd1201, 2024-08-21:
### Describe the bug

Seit dem update auf 0.130.0 sowie anschließend 0.130.1 wird der Ladestand des Hyundai Kona nicht mehr regelmäßig abgefragt.
Bis dato war eine regelmäßige Abfrage alle 120 Minuten vorhanden. Die Abfrage wurde immer durchgeführt, also auch wenn der Wagen nicht mit der Wallbox verbunden war.
Sobald der Wagen mit der Wallbox verbunden wird, werden die Daten vom Hyundai regelmäßig (alle 120 Minuten)

### Steps to reproduce

update von 0.130.0 auf 0.130.1

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: hyundai
    title: Hyundai Kona
    user: xxxxxxxxxxxxxx
    password: "xxxxx"
    capacity: 64.0
    language: de

loadpoints:
- title: Garage
  charger: wallbox5
  vehicle: my_car
  mode: pv
  phases: 3
  soc:
    poll:
      mode: always
      interval: 120m
```

### Log details

```text
[site ] DEBUG 2024/08/21 09:55:47 ----
[lp-1 ] DEBUG 2024/08/21 09:55:47 charge power: 0W
[site ] DEBUG 2024/08/21 09:55:47 pv power: 1965W
[site ] DEBUG 2024/08/21 09:55:47 grid meter: -1659W
[site ] DEBUG 2024/08/21 09:55:47 site power: -1659W
[lp-1 ] DEBUG 2024/08/21 09:55:47 charger status: A
[lp-1 ] DEBUG 2024/08/21 09:55:47 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 09:55:47 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 09:55:47 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 09:55:47 vehicle range: 388km
[site ] DEBUG 2024/08/21 09:56:02 ----
[lp-1 ] DEBUG 2024/08/21 09:56:02 charge power: 0W
[site ] DEBUG 2024/08/21 09:56:02 pv power: 1989W
[site ] DEBUG 2024/08/21 09:56:02 grid meter: -1679W
[site ] DEBUG 2024/08/21 09:56:02 site power: -1679W
[lp-1 ] DEBUG 2024/08/21 09:56:02 charger status: A
[lp-1 ] DEBUG 2024/08/21 09:56:02 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 09:56:02 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 09:56:02 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 09:56:02 vehicle range: 388km
[site ] DEBUG 2024/08/21 09:56:17 ----
[lp-1 ] DEBUG 2024/08/21 09:56:17 charge power: 0W
[site ] DEBUG 2024/08/21 09:56:17 pv power: 2013W
[site ] DEBUG 2024/08/21 09:56:17 grid meter: -1703W
[site ] DEBUG 2024/08/21 09:56:17 site power: -1703W
[lp-1 ] DEBUG 2024/08/21 09:56:17 charger status: A
[lp-1 ] DEBUG 2024/08/21 09:56:17 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 09:56:17 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 09:56:17 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 09:56:17 vehicle range: 388km
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.2

andig, 2024-08-21:
> Seit dem update auf 0.130.0 sowie anschließend 0.130.1 wird der Ladestand des Hyundai Kona nicht mehr regelmäßig abgefragt.

Ist im Logfile nicht erkennbar. Bitte vollständiges Log.

closed 2024-08-21 by andig

Efferd1201, 2024-08-21:
[lp-1 ] DEBUG 2024/08/21 10:37:47 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:37:47 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:37:47 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:37:47 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:37:47 vehicle range: 388km
[mqtt ] TRACE 2024/08/21 10:37:53 recv topic/wechselrichter1/leistung_ac: '1148'
[mqtt ] TRACE 2024/08/21 10:37:54 recv topic/wechselrichter2/leistung_ac: '658'
[mqtt ] TRACE 2024/08/21 10:37:55 recv topic/smartmeter/netzleistung: '644'
[site ] DEBUG 2024/08/21 10:38:02 ----
[lp-1 ] DEBUG 2024/08/21 10:38:02 charge power: 0W
[site ] DEBUG 2024/08/21 10:38:02 pv power: 1806W
[site ] DEBUG 2024/08/21 10:38:02 grid meter: 644W
[site ] DEBUG 2024/08/21 10:38:02 site power: 644W
[lp-1 ] DEBUG 2024/08/21 10:38:02 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:38:02 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:38:02 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:38:02 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:38:02 vehicle range: 388km
[mqtt ] TRACE 2024/08/21 10:38:08 recv topic/wechselrichter1/leistung_ac: '1095'
[mqtt ] TRACE 2024/08/21 10:38:08 recv topic/wechselrichter2/leistung_ac: '624'
[mqtt ] TRACE 2024/08/21 10:38:09 recv topic/smartmeter/netzleistung: '-1321'
[site ] DEBUG 2024/08/21 10:38:17 ----
[lp-1 ] DEBUG 2024/08/21 10:38:17 charge power: 0W
[site ] DEBUG 2024/08/21 10:38:17 pv power: 1719W
[site ] DEBUG 2024/08/21 10:38:17 grid meter: -1321W
[site ] DEBUG 2024/08/21 10:38:17 site power: -1321W
[lp-1 ] DEBUG 2024/08/21 10:38:17 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:38:17 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:38:17 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:38:17 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:38:17 vehicle range: 388km
[mqtt ] TRACE 2024/08/21 10:38:23 recv topic/wechselrichter1/leistung_ac: '1013'
[mqtt ] TRACE 2024/08/21 10:38:24 recv topic/wechselrichter2/leistung_ac: '579'
[mqtt ] TRACE 2024/08/21 10:38:25 recv topic/smartmeter/netzleistung: '871'
[site ] DEBUG 2024/08/21 10:38:32 ----
[lp-1 ] DEBUG 2024/08/21 10:38:32 charge power: 0W
[site ] DEBUG 2024/08/21 10:38:32 pv power: 1592W
[site ] DEBUG 2024/08/21 10:38:32 grid meter: 871W
[site ] DEBUG 2024/08/21 10:38:32 site power: 871W
[lp-1 ] DEBUG 2024/08/21 10:38:32 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:38:32 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:38:32 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:38:32 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:38:32 vehicle range: 388km
[mqtt ] TRACE 2024/08/21 10:38:39 recv topic/wechselrichter1/leistung_ac: '953'
[mqtt ] TRACE 2024/08/21 10:38:40 recv topic/wechselrichter2/leistung_ac: '538'
[mqtt ] TRACE 2024/08/21 10:38:42 recv topic/smartmeter/netzleistung: '996'
[site ] DEBUG 2024/08/21 10:38:47 ----
[lp-1 ] DEBUG 2024/08/21 10:38:47 charge power: 0W
[site ] DEBUG 2024/08/21 10:38:47 pv power: 1491W
[site ] DEBUG 2024/08/21 10:38:47 grid meter: 996W
[site ] DEBUG 2024/08/21 10:38:47 site power: 996W
[lp-1 ] DEBUG 2024/08/21 10:38:47 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:38:47 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:38:47 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:38:47 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:38:47 vehicle range: 388km
[mqtt ] TRACE 2024/08/21 10:38:53 recv topic/wechselrichter1/leistung_ac: '873'
[mqtt ] TRACE 2024/08/21 10:38:54 recv topic/wechselrichter2/leistung_ac: '497'
[mqtt ] TRACE 2024/08/21 10:38:54 recv topic/smartmeter/netzleistung: '1088'
[site ] DEBUG 2024/08/21 10:39:02 ----
[lp-1 ] DEBUG 2024/08/21 10:39:02 charge power: 0W
[site ] DEBUG 2024/08/21 10:39:02 pv power: 1370W
[site ] DEBUG 2024/08/21 10:39:02 grid meter: 1088W
[site ] DEBUG 2024/08/21 10:39:02 site power: 1088W
[lp-1 ] DEBUG 2024/08/21 10:39:02 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:39:02 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:39:02 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:39:02 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:39:02 vehicle range: 388km
[... 905 lines cut ...]
[mqtt ] TRACE 2024/08/21 10:56:24 recv topic/wechselrichter2/leistung_ac: '528'
[mqtt ] TRACE 2024/08/21 10:56:26 recv topic/smartmeter/netzleistung: '-1108'
[site ] DEBUG 2024/08/21 10:56:32 ----
[lp-1 ] DEBUG 2024/08/21 10:56:32 charge power: 0W
[site ] DEBUG 2024/08/21 10:56:32 pv power: 1440W
[site ] DEBUG 2024/08/21 10:56:32 grid meter: -1108W
[site ] DEBUG 2024/08/21 10:56:32 site power: -1108W
[lp-1 ] DEBUG 2024/08/21 10:56:32 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:56:32 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:56:32 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:56:32 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:56:32 vehicle range: 388km
[mqtt ] TRACE 2024/08/21 10:56:38 recv topic/wechselrichter1/leistung_ac: '927'
[mqtt ] TRACE 2024/08/21 10:56:40 recv topic/wechselrichter2/leistung_ac: '536'
[mqtt ] TRACE 2024/08/21 10:56:41 recv topic/smartmeter/netzleistung: '-1119'
[site ] DEBUG 2024/08/21 10:56:47 ----
[lp-1 ] DEBUG 2024/08/21 10:56:47 charge power: 0W
[site ] DEBUG 2024/08/21 10:56:47 pv power: 1463W
[site ] DEBUG 2024/08/21 10:56:47 grid meter: -1119W
[site ] DEBUG 2024/08/21 10:56:47 site power: -1119W
[lp-1 ] DEBUG 2024/08/21 10:56:47 charger status: A
[lp-1 ] DEBUG 2024/08/21 10:56:47 soc estimated: 0.00% (vehicle: 0.00%)
[lp-1 ] DEBUG 2024/08/21 10:56:47 vehicle soc: 0%
[lp-1 ] DEBUG 2024/08/21 10:56:47 vehicle soc limit: 80%
[lp-1 ] DEBUG 2024/08/21 10:56:47 vehicle range: 388km

Efferd1201, 2024-08-21:
Hi,

 

ich bin neu im Github Forum, bitte entschuldigt meine Unwissenheit. Ich habe noch eine umfangreichere Log eingestellt. Mir ist nur aufgefallen,

dass der Eintrag vorher als geschlossen markiert wurde. Ist der Punkt somit erledigt und das Problem ist bekannt oder muss ich etwas machen?

 

 

Gruß

 

Andre Manssen

 

Gesendet: Mittwoch, 21. August 2024 um 10:19 Uhr

Efferd1201, 2024-08-21:
Sorry, das ist natürlich richtig, wusste bis gerade leider nicht wie ich längere Logs bekomme. 
Ich hoffe mit der folgenden Log könnt ihr was anfangen. 
[evcc-20240821-193034-trace.log](https://github.com/user-attachments/files/16695990/evcc-20240821-193034-trace.log)