evcc-io/evcc #15571: Auswertung Ladeenergie

leika65, 2024-08-21:
Hi,
is it possible to insert the current year in the "Charging energy evaluation" information?

[image]

Thanks for your effort

naltatis, 2024-08-21:
Wir planen generell, die Ladevorgänge noch besser auswertbar zu machen. Die aktuelle tabellarische Darstellung ist funktional, hat aber noch Luft nach oben 😄. 

Der Wunsch hier eine "Dieses Jahr" Stufe einzuführen ist technisch allerdings super einfach. Daher würd ich das als Quickwin schon vorher hier machen. Weiterführende Themen ("Letztes Jahr", ) dann zu gegebener Zeit im Kontext der Ladevorgänge.

leika65, 2024-08-21:
Great.
Thank you very much for the quick implementation