evcc-io/evcc #15577: Problems after switching on "home battery discharge lock", seems not to switch back, all energy since then from electricity grid

Hstrohi, 2024-08-21:
### Describe the bug

After activating once the switch "home battery discharge lock" in the webui and switched it back afterwards to "not on" it seems that the battery does not provide any power for usage of the home at all. Since playing around with the switch battery stays on 100% constant for days now and all power that is needed is provided from the electricity grid or the pv collectors (when energy is available there).
Did a config check but nothing was changed there and values look ok.
Running a solaredge-hybrid with LG Battery environment. One thing that is maybe a little bit special, the installation was extend last year and has 2 inverters running that are connected. Just wanted to mention that, maybe that gets important when it comes to battery behviour steering.

Tried to set back the option via command line 
evcc -l trace meter --battery-mode  normal
please take a look at trace file below.

### Steps to reproduce

1. Update to the latest version 0.130.2
2. Switch on the "home battery discharge lock"
3. Switch off the "home battery discharge lock"
4. Keep an eye on battery status and power consumption, battery stays on 100%
...

### Configuration details

```yaml
meters:
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.188.45
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.188.45
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: solaredge-hybrid
  id: 1
  host: 192.168.188.45
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery3
  capacity: 6
....
site:
  title: HomeSweetHome
  meters:
    grid: grid1
    battery:
    - battery3
  residualPower: 100
```

### Log details

```text
root@raspberrypi:~# evcc -l trace meter --battery-mode  normal
[main  ] INFO 2024/08/21 17:09:04 evcc 0.130.2
[main  ] INFO 2024/08/21 17:09:04 using config file: /root/evcc.yaml
[db    ] INFO 2024/08/21 17:09:04 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2024/08/21 17:09:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/08/21 17:09:04 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/21 17:09:04 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/21 17:09:04 SELECT * FROM `settings` 7 <nil>
[db    ] TRACE 2024/08/21 17:09:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/08/21 17:09:04 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[... 499 lines cut ...]
[sunspec] TRACE 2024/08/21 17:09:18 modbus: recv 00 10 00 00 00 07 01 03 04 02 c7 00 20
[sunspec] TRACE 2024/08/21 17:09:18 modbus: send 00 11 00 00 00 06 01 03 a1 02 00 02
[sunspec] TRACE 2024/08/21 17:09:18 modbus: recv 00 11 00 00 00 07 01 03 04 02 c8 00 2c
[sunspec] TRACE 2024/08/21 17:09:18 modbus: send 00 12 00 00 00 06 01 03 a1 30 00 02
[sunspec] TRACE 2024/08/21 17:09:18 modbus: recv 00 12 00 00 00 07 01 03 04 02 c9 00 07
[sunspec] TRACE 2024/08/21 17:09:18 modbus: send 00 13 00 00 00 06 01 03 a1 39 00 02
[sunspec] TRACE 2024/08/21 17:09:18 modbus: recv 00 13 00 00 00 07 01 03 04 ff ff 00 00
[sunspec] TRACE 2024/08/21 17:09:18 modbus: send 00 14 00 00 00 06 01 03 9c bb 00 42
[sunspec] TRACE 2024/08/21 17:09:18 modbus: recv 00 14 00 00 00 87 01 03 84 53 6f 6c 61 72 45 64 67 65 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 53 45 2d 4d 54 52 2d 33 59 2d 34 30 30 56 2d 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 45 78 70 6f 72 74 2b 49 6d 70 6f 72 74 00 00 00 37 39 00 00 00 00 00 00 00 00 00 00 00 00 00 00 36 30 36 34 38 39 33 38 38 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 cb
[sunspec] TRACE 2024/08/21 17:09:18 modbus: send 00 15 00 00 00 06 01 03 9c fe 00 69
[sunspec] TRACE 2024/08/21 17:09:18 modbus: recv 00 15 00 00 00 d5 01 03 d2 00 34 00 11 00 13 00 0f ff ff 5b 32 5b 32 5a fd 5a b2 00 00 00 00 00 00 00 00 ff fe 13 88 ff fe ff d6 ff 75 00 56 00 0b 00 00 01 4c 01 0c 00 96 00 19 00 00 fe b6 ff 1b ff 85 00 16 00 00 e9 0f f4 d4 e2 a1 e3 b9 ff fe 00 8f d8 2e 00 2c ff d5 00 96 91 34 00 3a 06 9b 01 12 e5 9a 00 e4 93 32 00 51 44 8d 00 4a cd 51 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 00 00 00 00 00
[modbus] TRACE 2024/08/21 17:09:18 modbus: send 00 01 00 00 00 06 01 06 e0 0d 00 07
[modbus] TRACE 2024/08/21 17:09:18 modbus: recv 00 01 00 00 00 06 01 06 e0 0d 00 07
[modbus] TRACE 2024/08/21 17:09:18 modbus: send 00 01 00 00 00 0b 01 10 e0 10 00 02 04 40 00 45 9c
[modbus] TRACE 2024/08/21 17:09:18 modbus: recv 00 01 00 00 00 06 01 10 e0 10 00 02
root@raspberrypi:~#
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.2

andig, 2024-08-21:
Normal wäre ja:

```yaml
source: sequence
set:
- source: const
  value: 7 # Maximize self-consumption
  set:
    source: modbus
    {{- include "modbus" . | indent 12 }}
    register:
      address: 0xE00D # StorageRemoteCtrl_CommandMode
      type: writesingle
      encoding: uint16
- source: const
  value: 5000 # W
  set:
    source: modbus
    {{- include "modbus" . | indent 12 }}
    register:
      address: 0xE010 # StorageRemoteCtrl_DischargeLimit
      type: writemultiple
      encoding: float32s
```

Ist daran etwas falsch?

Hstrohi, 2024-08-23:
Gibt es eine Möglichkeit die Entnahmesperre für die Battarie über die cmd zurückzusetzen/auszuschalten? Wäre gut wieder die Ressource nutzen zu können.

closed 2024-09-04 by github-actions[bot]