evcc-io/evcc #15579: Elli reports 3p charging while charging 1p

matzeruegge, 2024-08-21:
### Describe the bug

After upgrading from version 0.130.0 to version 0.130.2. Evcc shows 3 phase charging while charging 1 phase. Also the reported power seems to be real power * 3. 

### Steps to reproduce

1. Connect vehicle to charger with 1 phase extension cable
2. start charging by card authentication 
3. Check evcc gui
[image]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug # info | debug
levels:
  cache: error

# unique installation id
plant: ################################

interval: 30s # control cycle interval

meters:
- name: pv_dach
  type: template
  template: rct-power
  usage: pv
  host: HF-A21.fritz.box # IP-Adresse oder Hostname 

- name: battery
  type: template
  template: rct-power
  usage: battery
  host: HF-A21.fritz.box # IP-Adresse oder Hostname 
- name: grid
  type: template
  template: rct-power
  usage: grid
  host: HF-A21.fritz.box # IP-Adresse oder Hostname 

- name: pv_carport
  type: custom
  power:
    source: mqtt
    broker: localhost:1883
    topic: carport_solar/ac/power
    scale: 1
  energy:
    source: mqtt
    broker: localhost:1883
    topic: carport_solar/ac/yieldday
    scale: 1

vehicles:
- type: template
  template: vw 
  title: id7  
  user: #######################
  password: ################
  vin: #####################
  capacity: 77  
  timeout: 10s  
  name: ev4
[... 332 lines cut ...]
Aug 21 16:12:09 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:09 Send: d:_i:47859_Elli-Wallbox-2021A0QFZF notify 100 DeviceDiagnosisHeartbeatData
Aug 21 16:12:11 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:11 Send: d:_i:47859_Elli-Wallbox-2021A0QFZF notify 101 DeviceDiagnosisHeartbeatData
Aug 21 16:12:13 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:13 Send: d:_i:47859_Elli-Wallbox-2021A0QFZF notify 102 DeviceDiagnosisHeartbeatData
Aug 21 16:12:15 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:15 Send: d:_i:47859_Elli-Wallbox-2021A0QFZF notify 103 DeviceDiagnosisHeartbeatData
Aug 21 16:12:17 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:17 Send: d:_i:47859_Elli-Wallbox-2021A0QFZF notify 104 DeviceDiagnosisHeartbeatData
Aug 21 16:12:19 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:19 Send: d:_i:47859_Elli-Wallbox-2021A0QFZF notify 105 DeviceDiagnosisHeartbeatData
Aug 21 16:12:19 minix evcc[131067]: [site  ] DEBUG 2024/08/21 16:12:19 ----
Aug 21 16:12:19 minix evcc[131067]: [lp-1  ] DEBUG 2024/08/21 16:12:19 charge power: 0W
Aug 21 16:12:19 minix evcc[131067]: [site  ] DEBUG 2024/08/21 16:12:19 pv power: 5105W
Aug 21 16:12:19 minix evcc[131067]: [site  ] DEBUG 2024/08/21 16:12:19 battery soc: 97%
Aug 21 16:12:19 minix evcc[131067]: [site  ] DEBUG 2024/08/21 16:12:19 battery power: -3272W
Aug 21 16:12:19 minix evcc[131067]: [site  ] DEBUG 2024/08/21 16:12:19 grid meter: 0W
Aug 21 16:12:19 minix evcc[131067]: [site  ] DEBUG 2024/08/21 16:12:19 site power: -3272W
Aug 21 16:12:19 minix evcc[131067]: [lp-1  ] DEBUG 2024/08/21 16:12:19 charger status: B
Aug 21 16:12:19 minix evcc[131067]: [eebus ] DEBUG 2024/08/21 16:12:19 !! OpEV.LoadControlLimits set: {a false true 6}
Aug 21 16:12:19 minix evcc[131067]: [lp-1  ] DEBUG 2024/08/21 16:12:19 pv charge current: 4.74A = 0A + 4.74A (-3272W @ 3p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.2

VolkerK62, 2024-08-21:
Without charge meter, you must either set the phases (1p or 3p) within the loadpoint settings or, if your car can only charge 1p, you must configure `phases: 1` at the vehicle configuration.

If you use different cable (1p/3p) see dokumentation:
https://docs.evcc.io/en/docs/reference/configuration/loadpoints#phases
>In cases where 1p/3p charging is realised using the corresponding charging cable rather than a switch, this "automatic" behaviour causes problems. In this case, the phases value must be adjusted in the vehicle accordingly. Since this value cannot be changed via API, the following workaround can be used:
Configure the vehicle twice: once with phases: 1 and once with phases: 3. Depending on the desired charging mode, select the appropriate vehicle in the UI.

andig, 2024-08-21:
Die Ursache ist https://github.com/evcc-io/evcc/pull/15359. Wir müssen nochmal überlegen, ob uns das wirklich hilft.

DerAndereAndi, 2024-08-21:
In general: There are many situations where the box doesn't provide any measurement data and in such scenarios the ONLY workaround it to use an external meter as it is suggested in the documentation since 2 years.

There are some things possibly can be done, but with a lot of effort and it will NEVER fix the issue. If the box doesn't send measurements, there is no workaround possible. Ever.

@andig see my comment on Slack from Monday.

matzeruegge, 2024-08-21:
I understand you removed the measurement receiver for elli data. I was somewhat satisfied for at least the last 2 years with the implemented solution. (My wb is connected to the LAN) When I faced a problem around once a month I had to reboot the elli wb and it was working again. 
So I have to ask an electrician to install a more reliable measurement device or to downgrade and stick to 0.130.0. 
Maybe you can consider an option so that users can continue to use elli measurements on their own risk?

DerAndereAndi, 2024-08-21:
@matzeruegge The option is mentioned at the end of the post https://github.com/evcc-io/evcc/discussions/15367

ilion1, 2024-08-21:
My setup is working fine with 0.128.4 and wallbox is showing "Energy Manager" connected. When switching to 0.130.2 and changing evcc.yaml as described in the post mentioned above EVCC starts correctly but the wallbox is showing "Energy Manager" not connected. So when trying to charge EVCC cannot control the wallbox and logs the error "charge power: not available". 

Version 0.128.4 (working, Energy Manager connected)
```
chargers:
- type: template
  template: ellipro
  ski: D9A2 03D9 E918 F950 EC97 E593 7251 E179 79A3 905E
  ip: 192.168.189.34
  name: wallbox5
```
eebus trace after start:
```
[eebus ] INFO 2024/08/21 20:26:53 Local SKI: 1d5be7dbd5197baaed9616d379fc6d3d2dd5ebcc
[eebus ] DEBUG 2024/08/21 20:26:53 starting websocket server on :4712
[eebus ] DEBUG 2024/08/21 20:26:53 mdns: announce
[eebus ] DEBUG 2024/08/21 20:26:53 mdns: using zeroconf
[eebus ] DEBUG 2024/08/21 20:26:53 mdns: start search
[eebus ] TRACE 2024/08/21 20:26:53 registering ski: d9a203d9e918f950ec97e5937251e17979a3905e
[eebus ] DEBUG 2024/08/21 20:26:53 ski: d9a203d9e918f950ec97e5937251e17979a3905e name: Elli-Wallbox-2125A1G53O brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2125A1G53O register: false host: wallbox-2125A1G53O.local. port: 4712 addresses: [192.168.189.34]
[eebus ] DEBUG 2024/08/21 20:26:53 trying to connect to d9a203d9e918f950ec97e5937251e17979a3905e at wallbox-2125A1G53O.local.
[eebus ] DEBUG 2024/08/21 20:26:53 initiating connection to d9a203d9e918f950ec97e5937251e17979a3905e at wallbox-2125A1G53O.local.:4712/ship/
[eebus ] DEBUG 2024/08/21 20:26:53 connection to d9a203d9e918f950ec97e5937251e17979a3905e failed: dial tcp: lookup wallbox-2125A1G53O.local. on 100.100.100.100:53: no such host
[eebus ] DEBUG 2024/08/21 20:26:53 trying to connect to d9a203d9e918f950ec97e5937251e17979a3905e at 192.168.189.34
[eebus ] DEBUG 2024/08/21 20:26:53 initiating connection to d9a203d9e918f950ec97e5937251e17979a3905e at 192.168.189.34:4712/ship/
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 1
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 2
[eebus ] TRACE 2024/08/21 20:26:53 Send: d9a203d9e918f950ec97e5937251e17979a3905e ship init
[eebus ] TRACE 2024/08/21 20:26:53 Recv: d9a203d9e918f950ec97e5937251e17979a3905e ship init
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 3
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 6
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 7
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 8
[eebus ] TRACE 2024/08/21 20:26:53 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionHello":[{"phase":"ready"},{"waiting":60000}]}
[eebus ] TRACE 2024/08/21 20:26:53 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionHello":[{"phase":"ready"},{"waiting":60000}]}
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 13
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 19
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 19
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 22
[eebus ] TRACE 2024/08/21 20:26:53 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"messageProtocolHandshake":[{"handshakeType":"announceMax"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
[eebus ] TRACE 2024/08/21 20:26:53 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"messageProtocolHandshake":[{"handshakeType":"select"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 24
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 26
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 26
[eebus ] TRACE 2024/08/21 20:26:53 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 27
[eebus ] TRACE 2024/08/21 20:26:53 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"messageProtocolHandshake":[{"handshakeType":"select"},{"version":[{"major":1},{"minor":0}]},{"formats":[{"format":["JSON-UTF8"]}]}]}
[eebus ] TRACE 2024/08/21 20:26:53 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionPinState":[{"pinState":"none"}]}
[eebus ] TRACE 2024/08/21 20:26:54 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"connectionPinState":[{"pinState":"none"}]}
[eebus ] TRACE 2024/08/21 20:26:54 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 31
[eebus ] TRACE 2024/08/21 20:26:54 d9a203d9e918f950ec97e5937251e17979a3905e SHIP state changed to: 36
[eebus ] TRACE 2024/08/21 20:26:54 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"accessMethodsRequest":[]}
[eebus ] TRACE 2024/08/21 20:26:54 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"accessMethodsRequest":[]}
[eebus ] TRACE 2024/08/21 20:26:54 Send: d9a203d9e918f950ec97e5937251e17979a3905e {"accessMethods":[{"id":"EVCC-3639656234373032"}]}
[eebus ] TRACE 2024/08/21 20:26:54 Recv: d9a203d9e918f950ec97e5937251e17979a3905e {"accessMethods":[{"id":"Elli-Wallbox-2125A1G53O"},{"dnsSd_mDns":[]}]}
[... cut ...] seName":"limitationOfPowerProduction"},{"useCaseVersion":"1.0.0"},{"useCaseAvailable":true},{"scenarioSupport":[1,2,3,4]},{"useCaseDocumentSubRevision":"release"}]]}],[{"address":[{"device":"d:_n:EVCC_HEMS-EVCC-3639656234373032"},{"entity":[1]}]},{"actor":"MonitoringAppliance"},{"useCaseSupport":[[{"useCaseName":"monitoringOfGridConnectionPoint"},{"useCaseVersion":"1.0.0"},{"useCaseAvailable":true},{"scenarioSupport":[1,2,3,4,5,6,7]},{"useCaseDocumentSubRevision":"release"}]]}]]}]}]]}]}]}}]}
```

matzeruegge, 2024-08-21:
@DerAndereAndi Thanks a lot Andi! Perfect. Sorry for not reading till the end.

closed 2024-08-21 by matzeruegge

ilion1, 2024-08-21:
> @DerAndereAndi Thanks a lot Andi! Perfect. Sorry for not reading till the end.

@matzeruegge is it working for you? Have you tried to charge and can you manage the wallbox via EEBUS? How does your evcc.yaml section look like?

matzeruegge, 2024-08-21:
@ilion1 I made one short test with min pv and it looks good. 

`chargers:
- name: elliwb
  type: eebus
  ski: '0B4D A4C9 624A 2F8E 6F47 6B84 FD50 C4ED 0382 523E'
  meter: true
  chargedEnergy: true`

DerAndereAndi, 2024-08-23:
@ilion1 der Trace zeigt in beiden Fällen dass eine Verbindung besteht. Vielleicht ist einfach die Anzeige in der Web UI der Wallbox falsch?

andig, 2024-08-24:
@ilion1 to not confuse things- do you want me to extract your log into a new issue?

ilion1, 2024-08-24:
Thanks for following up. I have opened a new issue as requested with more [details](https://github.com/evcc-io/evcc/issues/15633):