evcc-io/evcc #15581: Pulsar Plus: `SetChargingProfile` gets rejected

dscho, 2024-08-21:
### Describe the bug

This is a bug. EVCC was working correctly before I upgraded from v0.128.4 to v0.130.1. After upgrading, `SetChargingProfile` always gets rejected.

There have been extremely similar reports elsewhere in this here bug tracker, essentially differing only in the model/brand of the charger.

### Steps to reproduce

1. Connect car to a Pulsar Plus charger
2. observe the log (and the errors in the UI)

### Configuration details

```yaml
log: debug
levels:
  cache: error
  ocpp: trace

meters:
- type: template
  template: senec-home
  usage: grid
  schema: https
  host: 192.168.178.84
  name: grid1
- type: template
  template: senec-home
  usage: pv
  schema: https
  host: 192.168.178.84
  name: bsh
- type: template
  template: senec-home
  usage: battery
  schema: https
  host: 192.168.178.84
  name: senec

chargers:
- type: template
  template: pulsarplus
  name: pulsar

vehicles:
- type: template
  name: id3
  title: id.3
  template: id
  [... login data ...]

loadpoints:
- title: Garage
  charger: pulsar
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: [... title ...]
  meters:
    grid: grid1
    pv:
    - bsh
    battery:
    - senec
  residualPower: 300
  maxGridSupplyWhileBatteryCharging: 50
```
[... 133 lines cut ...]
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle status: B
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle soc: 79%
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle soc limit: 80%
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 vehicle range: 327km
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 pv charge current: 5.61A = 0A + 5.61A (-3868W @ 3p)
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 pv disable timer reset
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] DEBUG 2024/08/20 12:01:12 pv timer inactive
Aug 20 12:01:12 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:12 send evcc: [2,"2640631095","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":1,"stackLevel":1,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-20T11:01:12Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
Aug 20 12:01:12 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:12 recv evcc: [3,"2640631095",{"status": "Rejected"}]
Aug 20 12:01:12 raspberrypi evcc[637]: [lp-1  ] ERROR 2024/08/20 12:01:12 max charge current 6A: set charging profile: Rejected
Aug 20 12:01:35 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:35 recv evcc: [2,"1317886357","Heartbeat",{}]
Aug 20 12:01:35 raspberrypi evcc[637]: [ocpp  ] TRACE 2024/08/20 12:01:35 send evcc: [3,"1317886357",{"currentTime":"2024-08-20T11:01:35Z"}]
Aug 20 12:01:35 raspberrypi evcc[637]: [site  ] DEBUG 2024/08/20 12:01:35 ----
[...]
```

</details>

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.1

andig, 2024-08-21:
Ursache ist vmtl https://github.com/evcc-io/evcc/pull/14888?

dscho, 2024-08-21:
> Ursache ist vmtl [#14888](https://github.com/evcc-io/evcc/pull/14888)?

@andig Hätte ich auch gesagt. Auch, dass das genau dieselbe Ursache ist wie bei den anderen Tickets, aber ich linke die jetzt nicht, sonst wird meine Antwort wieder als Off-Topic abgestraft.

andig, 2024-08-21:
Kannste gerne verlinken wenns dazu zugehört

andig, 2024-08-22:
Folgendes ist passiert:

    Ocpp: use ChargingProfileId:1
    Ocpp: use ChargingProfileKindAbsolute (#15240)
    OCPP: use TxDefaultProfile, add more apis (BC) (#14888)
    Revert "Ocpp: use setProfile=0 for suspend (#12459)"
    Ocpp: use setProfile=0 for suspend (#12459)
    Ocpp: add transaction id to TxProfile (#9766)
    Revert "Ocpp: use default transaction profile (#8549)"
    Ocpp: use default transaction profile (#8549)
    OCPP: use TxProfile and enable phase switching (#4946) (#4969)

In https://github.com/evcc-io/evcc/pull/14888 gabs eine Änderung von `TxProfile` auf `TxDefaultProfile`.
In #15240 wurde von `relative` auf `absolute` geändert.

Eine der beiden Änderungen sollte ursächlich sein. Könnte das jemand lokal (oder in Docker) testen?

dscho, 2024-08-22:
Also ich würde das ja gerne mit `git bisect` analysieren, aber ganz ehrlich, diese vielen "wip" commits in #14888 sind echt nicht sauber programmiert.

Ich kann mal versuchen, mit `git revert` zu arbeiten, aber erwarte aufgrund der Qualität und Struktur der Commits Merge-Konflikte, mit denen ich höchstwahrscheinlich dank meiner Unvertrautheit mit der Sprache Go sowie mit der Software-Architektur von EVCC überfordert sein werde.

In jedem Fall muss ich erst mal wieder Auto fahren, damit das laden kann.

andig, 2024-08-22:
Mit Verlaub, aber der Kommentar ist nicht angemessen. Zumal die WIPs beim Merge gesquashed werden. Ist also völlig unerheblich.

dscho, 2024-08-22:
> Mit Verlaub, aber der Kommentar ist nicht angemessen. Zumal die WIPs beim Merge gesquashed werden. Ist also völlig unerheblich.

Der Kommentar ging um die Möglichkeit, `git bisect` zu verwenden, was ich hier nicht kann.

Es tut mir leid, wenn dabei Gefühle verletzt wurden, das war nicht meine Absicht. Stattdessen wollte ich helfen, aber die Struktur der Commit-Historie, im Verband mit meinem Mangel an Go-Kenntnissen, macht mir das nahezu unmöglich.

premultiply, 2024-08-22:
Können wir uns bitte in den Issues einfach nur auf die Problemlösung konzentrieren?
Wer alles was nicht kann und will ist für die Sache irrelevant?
Im Zweifelsfall bitte einfach raushalten wenn es nichts sachdienliches beizutragen gibt.
Wenn das wieder nicht klappt mache ich hier einfach zu oder eine Diskussion draus und dann ist für mich das Thema wirklich erledigt.

Es gibt einen Herstellersupport, der IMMER die erste Anlaufstelle bei Merkwürdigkeiten in der Firmware ist.

andig, 2024-08-22:
> Der Kommentar ging um die Möglichkeit, git bisect zu verwenden, was ich hier nicht kann.

Das git bisect machst Du auf master. Da sollte es für den PR nur einen Commit geben.

> Es gibt einen Herstellersupport, der IMMER die erste Anlaufstelle bei Merkwürdigkeiten in der Firmware ist.

@premultiply das teile ich nicht. Hier ging es mal, jetzt nicht mehr. Die Änderungen sind bei uns und wir sollten raus finden welche Änderung das konkret ist.

premultiply, 2024-08-22:
Darum geht es hier. Und bitte nur darum.

premultiply, 2024-08-22:
@dscho Bitte mal ein neues Log mit dem aktuellen Nightly anhängen wo sich der Charger nicht im Zustand "Waiting for authentication" (remotestart: true) ohne existierende Transaktion befindet.

mrhund, 2024-08-22:
@dscho  Also ich habe das Problem mit meiner Pulsar nicht, ein paar andere Themen - aber das nicht.
Daher möchte ich anmerken, dass man im Log sieht, dass du Firmware Version **5.17.87** verwendest, ich habe schon Version **6.4.14**. Das könnte ein entscheidender Unterschied sein.

andig, 2024-08-22:
@premultiply 

```go
	return &types.ChargingProfile{
		ChargingProfileId:      c.chargingProfileId, // <-
		StackLevel:             c.stackLevel, // <-
		ChargingProfilePurpose: types.ChargingProfilePurposeTxDefaultProfile,
		ChargingProfileKind:    types.ChargingProfileKindAbsolute,
		ChargingSchedule: &types.ChargingSchedule{
			StartSchedule:          types.Now(),
			ChargingRateUnit:       c.chargingRateUnit,
			ChargingSchedulePeriod: []types.ChargingSchedulePeriod{period},
		},
	}
```

Warum kommen hier jeweils die _Maxima_ des im Charger möglichen zum Einsatz und wir verwenden nicht einfach 0 oder 1 wie zuvor?

premultiply, 2024-08-22:
Das Stacklevel bestimmt die Priorität des Profils. Damit keine anderen eventuell vorhandenen Profile hineinmischen verwenden wir die höchste Priorität. Andernfalls müsste man bei der Initialsierung erstmal alle vorhanden Profile löschen was wieder ganz andere Fehler und Unzulänglichkeiten der Firmwares ans Tageslicht spülen dürfte.

Die ChargingProfileId ist im Prinzip (nach Standard) völlig egal. Jedoch gab es dann direkt wieder Firmwaren die nicht mit Wert 0 zurecht kamen...
Damit man einen halbwegs gescheiten Defaut-Wert hat nehmen wir hier hilfsweise die höchste Zahl der zulässigen gleichzeitig installierten Profile - in der Hoffnung dass dies selbst bei verbaselter Stacklevel-Prio oder bei eingeschränktem Value-Range in der Firmware auch die höchste Prio bekommt und evcc andererseits immer die gleiche ChargingProfileId verwendet.
Speziell Charger die nur ein ChargingProfile gleichzeitig unterstützen sind da recht empfindlich.

dscho, 2024-08-22:
> Also ich habe das Problem mit meiner Pulsar nicht, ein paar andere Themen - aber das nicht.
> Daher möchte ich anmerken, dass man im Log sieht, dass du Firmware Version **5.17.87** verwendest, ich habe schon Version **6.4.14**. Das könnte ein entscheidender Unterschied sein.

Ich habe Auto-Update an, und sowohl App als auch Web UI behaupten, das sei up to date:

[image]

[image]

Ich glaube nicht, dass ich damit alleine bin.

andig, 2024-08-22:
...und deshalb müssen wir raus finden, welche der (an sich funktionierenden) Änderungen mit Deiner FW das Problem macht. Nutzt ja nix...

cdyna43, 2024-08-22:
Es gibt verschiedene Versionen der Pulsar Plus und verschiedene Firmwares. Bei mir ist auch die 5.17.87 die aktuellste. Habe das bei Wallbox verifiziert.

Ich habe auch genau das gleiche Problem wie dscho. 

Zurück zur 0.129.0 und alles funktioniert wieder.

mfuchs1984, 2024-08-22:
Recherchiert man online, gibt's einige Diskussionen um die Firmwareversionen, es scheint unklar zu sein, warum manche die Updates bekommen haben und andere nicht und alle scheinen so ihre Problemchen zu haben. https://community.home-assistant.io/t/wallbox-pulsar-plus-integration/200339/1074?page=54

Leider lassen sich aufgrund eines Fehlers der Website nichtmal die releasnotes aufrufen, soweit man was findet, wurden aber durchaus ocpp Themen adressiert. Evtl macht es unabhängig von der Lösung hier Sinn, mal beim Hersteller nach einer Klärung bzgl. der Versionen zu fragen.

premultiply, 2024-08-22:
> Ich habe auch genau das gleiche Problem wie dscho.

Und, was sagt der Hersteller dazu?

andig, 2024-08-22:
> Und, was sagt der Hersteller dazu?

Können wir uns das bitte sparen?

Spricht irgendwas dagegen, dass wir herausfinden, welche Änderung das Problem verursacht? Ohne wird es ohnehin schwierig, von irgendeinem Hersteller dieser Welt eine sinnvolle Antwort zu erhalten. 

Danke.

andig, 2024-08-22:
@dscho siehe https://github.com/evcc-io/evcc/issues/15581#issuecomment-2304104933

Alles Andere ist jetzt irrelevant.

premultiply, 2024-08-22:
Vielleicht kann uns auch @cdyna43 mit einem aktuellen Tracelog des aktuellen Nightly Build aushelfen?

cdyna43, 2024-08-22:
Wenn mir jemand für Dummies erklärt, was ich tun soll, gerne. Ich habe aber evcc zurück auf 0.129.0 gesetzt, damit es wieder funktioniert...

premultiply, 2024-08-22:
Schau mal hier: https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

highendsyl, 2024-08-23:
Ich habe das gleiche Problem mit Wallbox Commander2. Kann ich Hilfen?

andig, 2024-08-23:
Weiterhin https://github.com/evcc-io/evcc/issues/15581#issuecomment-2305180087.

Falls jemand compilieren kann würde es auch helfen, commits adf90ecccd44156da4016af7d2d7b1bbf116af81, d0b58ba883a28f49894e907ff14b37cf24b468c7, 1515a67fc0e7446a076e9c0690bdb98c76d122e3 und 568c5646a8f15d2ded18c1c77eec6c0471000315. Das sind jeweils die Commits vor/nach breaking Changes. Damit könnten wir die Ursache mal eingrenzen.

highendsyl, 2024-08-23:
[evcc.log](https://github.com/user-attachments/files/16727505/evcc.log)

andig, 2024-08-23:
@highendsyl wir waren nicht klar genug :(. Es braucht ocpp auf trace.

highendsyl, 2024-08-24:
1.000x Sorry: wie mache ich das? Ich helfe gerne und schnell, aber bin mega-junior mit Rpi

andig, 2024-08-24:
In dem du das einfach in der Benutzeroberfläche auswählst…

andig, 2024-08-24:
Oder in die Doku schaust: https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

cdyna43, 2024-08-24:
Das Problem von highendsyl (vermute ich) und mir ist, dass das hier "Es braucht ocpp auf trace" leider für uns (jedenfalls mich) nicht verständlich ist. Da hilft mir der Link auch nicht, da es wohl nicht um das normale Logfile geht.
Ich habe jetzt auf die 0.130.2 aktualisiert und der Fehler besteht. In EVCC die Logs auf Trace gestellt und den Download-Button gedrückt. Nur wo ist jetzt die Datei gespeichert...?

cdyna43, 2024-08-24:
[evcc-20240824-102649-trace.log.txt](https://github.com/user-attachments/files/16735949/evcc-20240824-102649-trace.log.txt)

andig, 2024-08-24:
Nein, nein und nein. Es braucht bitte alle relevanten Komponenten- site, loadpoint und ocpp.

highendsyl, 2024-08-24:
[evcc-20240824-103501-trace.log.txt](https://github.com/user-attachments/files/16735967/evcc-20240824-103501-trace.log.txt)

highendsyl, 2024-08-24:
Lieber Andig, ist das was du brauchst?

cdyna43, 2024-08-24:
Sorry. Dann noch einmal für mich: Ich wähle bei Log "Trace" aus und bei Bereich zur Sicherheit alles (habe nur die PV-Anlage, die Wallboxen und die Autos)?

mfuchs1984, 2024-08-24:
@highendsyl hast du remotestart: true in der charger config?

andig, 2024-08-24:
> Lieber Andig, ist das was du brauchst?

Jetzt bitte noch ab Start und ohne http- dann wirds leichter lesbar und vollständig :)

highendsyl, 2024-08-24:
[evcc-20240824-124018-trace.log.txt](https://github.com/user-attachments/files/16736266/evcc-20240824-124018-trace.log.txt)

andig, 2024-08-24:
Keine neuen Erkenntnisse. Die WB mag `SetChargingProfile` nicht. Es ist unklar, was genau sie daran nicht mag. Entweder:
- findet jemand raus welche Änderung genau die Box nicht mag
- gelingt das FW-Update von dem wir wissen, dass es funktioniert
- oder bewegt den Hersteller dazu eine Aussage zu treffen

Für uns gibts hier grad nix zu tun. Ich würde das Issue mangels neuer Erkenntnisse dann schließen.

MilkoKA, 2024-08-24:
Nur zur Info
Ich habe dauch das Problem mit der der Firmware Version 5.17.87. Mir wird ebenfalls keine neue Firmware angeboten und auf Nachfrage bei Wallbox lautete die Antword, dass bei meiner Seriennummer die Firmware Version 5.17.87 die aktuellste Firmware wäre.

Hoffentlich wird das Problem gefunden. Ich geh erstmal auf die 0.129.0 zurück.
Viel Glück

closed 2024-08-24 by andig

andig, 2024-08-24:
> Für uns gibts hier grad nix zu tun. Ich würde das Issue mangels neuer Erkenntnisse dann schließen.

Nur wenn jemand sucht...

matti1993, 2024-08-25:
> Nur zur Info Ich habe dad das Problem mit der Firmware Version 5.17.87. Mir wird ebenfalls keine neue Firmware angeboten und auf Nachfrage bei Wallbox lautete das Antword, dass bei meiner Seriennummer die Firmware Version 5.17.87 die aktuellste Firmware wäre.
> 
> Hoffentlich wird das Problem gefunden. Ich geh erstmal auf die 0.129.0 zurück. Viel Glück

Kann mir jemand erklären wie ich wieder auf 0.129.0 zurück komme?

MilkoKA, 2024-08-25:
Hi, falls mann die Firmwareversion abfragen kann, könnte man vielleicht sowas bauen wie:

If FW<6 = alteMetode else neu Metode

Ich denke bei der PulsarPlus gibte sowas wie Hardware Gen1 und Gen2. Gen1 wird wohl keine neuen Firmware bekommen.

Ich mache dazu aber ein Anfrage bei Wallbox.com
Gruß
Milko

andig, 2024-08-25:
Auch dafür müsste man rausfinden, was genau "alte Methode" heisst...

MilkoKA, 2024-08-25:
Ich dachte nur, da die 128.4 funktioniert und die 129.0 nicht mehr.
wurde denn auch in der 128.4 "SetChargingProfile" benutzt um von aus/pv/min+PV/schnell umzuschalten?

typxxi, 2024-08-25:
> @dscho Also ich habe das Problem mit meiner Pulsar nicht, ein paar andere Themen - aber das nicht. Daher möchte ich anmerken, dass man im Log sieht, dass du Firmware Version **5.17.87** verwendest, ich habe schon Version **6.4.14**. Das könnte ein entscheidender Unterschied sein.

Ich habe 5.17.87 und ich sehe kein Update, was ich durchführen kann, also bin ich auch abgeschnitten. 
Vielleicht auch noch mehr Nutzer, die sich alle hier finden. 

Wenn ich sicher bin, dass die Sonne lange scheinen wird, dann probiere ich morgen erneut einen Rollout der 130.4 , wobei ich nur brauch, dass die wieder Überschussladen kann und nicht blind 11 kW verballert. 
Aktuell haben wir 0,5 kWh Bezug am Tag und der letzte Versuch war kein Zuckerschlecken, sondenr hat 3 h gedauert und endete im Schnellladen spät am Abend ohne Sonne.

dscho, 2024-08-26:
> Falls jemand compilieren kann würde es auch helfen, commits [adf90ec](https://github.com/evcc-io/evcc/commit/adf90ecccd44156da4016af7d2d7b1bbf116af81), [d0b58ba](https://github.com/evcc-io/evcc/commit/d0b58ba883a28f49894e907ff14b37cf24b468c7), [1515a67](https://github.com/evcc-io/evcc/commit/1515a67fc0e7446a076e9c0690bdb98c76d122e3) und [568c564](https://github.com/evcc-io/evcc/commit/568c5646a8f15d2ded18c1c77eec6c0471000315). Das sind jeweils die Commits vor/nach breaking Changes. Damit könnten wir die Ursache mal eingrenzen.

Ich habe das mal gemacht:

[image]

Leider ist EVCC damit immer noch kaputt:

[image]

> If FW<6 = alteMetode else neu Metode

Das wäre verwandt mit der Best Practice, dass man so eine neue Methode erst mal hinter einem experimentellen Schalter versteckt, und dann abhängig von der Konfiguration Benutzer einlädt, das mal auszuprobieren. Mit so einem Schalter würde ich das wahrscheinlich auch mal wieder austesten, aber für's erste wurde ich etwas zu viel geschockt, als ich nach diversen manuellen `Revert`s und `Re-apply`s folgenden Fehler bekam, wo mir nur ein Backup, was ich vor einiger Zeit mehr aus Versehen als absichtlich gemacht hatte, heraushalf:

[image]

Ich war ja vor dieser Regression sehr glücklich mit EVCC. Hoffentlich kann ich das auch wieder werden.

dscho, 2024-08-26:
Vielleicht sollte dieses Issue wieder aufgemacht werden, der Fehler ist ja mitnichten behoben.

MilkoKA, 2024-08-26:
Hiermal die Antwort meiner Anfrage bei Wallbox.com warum wir kein Update erhalten:
Guten Tag,

vielen Dank, dass Sie sich an den Wallbox-Kundensupport gewandt haben. Wir schätzen Ihre Geduld, während wir Ihre Anfrage bezüglich des Updates auf Version 6.4.19 bearbeiten.

Wir können bestätigen, dass Version 6.4.19 tatsächlich verfügbar ist. Allerdings hängt die Verfügbarkeit dieses Updates vom Herstellungsdatum Ihrer Wallbox ab. Einheiten mit neueren Herstellungsdaten erhalten das Update zuerst. Seien Sie jedoch versichert, dass alle Wallbox-Einheiten letztendlich dieses Update erhalten werden, es kann jedoch etwas Zeit in Anspruch nehmen, je nach Herstellungsdatum Ihres Geräts.

Laut unseren Unterlagen wurde Ihre Wallbox am 19. Juli 2021 hergestellt. Das bedeutet, dass Ihr Gerät das Update erhalten wird, es jedoch etwas länger dauern kann.

Falls Sie weitere Fragen haben oder zusätzliche Unterstützung benötigen, lassen Sie es uns bitte wissen.

Mit freundlichen Grüßen

dscho, 2024-08-26:
@MilkoKA danke für das Update. Schade, dass wir bis zu dem ungenannten Datum kein aktuelles EVCC verwenden können.

typxxi, 2024-08-27:
> Ich dachte nur, da die 128.4 funktioniert und die 129.0 nicht mehr.
> wurde denn auch in der 128.4 "SetChargingProfile" benutzt um von aus/pv/min+PV/schnell umzuschalten?

das bezweifel ich, denn ich habe auch die 5.17.87 und 129.0 läuft hier, aber alles von 130.* bis 130.6 machte nur Ärger.

matti1993, 2024-08-27:
Ja so ist es bei mir auch 0.129.0 läuft alles Super

typxxi, 2024-08-27:
Macht ja keinen Sinn, hier bei EVCC den Druck, den wir haben, abzuladen, wenn am Ende der Hersteller genau wissen muss, was die Ursache ist, denn mit der neuen Firmware, die nur wenige haben dürften, soll der Fehler behoben sein laut @mrhund 

> @dscho Also ich habe das Problem mit meiner Pulsar nicht, ein paar andere Themen - aber das nicht. Daher möchte ich anmerken, dass man im Log sieht, dass du Firmware Version **5.17.87** verwendest, ich habe schon Version **6.4.14**. Das könnte ein entscheidender Unterschied sein.

Daher BITTE das Kontaktformular der WALLBOX PULSAR PLUS GEPLAGTEN kurz mit copy & paste ausfüllen und zwar [hier](https://support.wallbox.com/de/kontaktieren-sie-uns/)

Wenn wir uns nicht DORT BEI DENEN beschweren, passiert bei denen gar nix, sondern die kriechen in Barcelona langsam von 9 to 5 vor sich her - oder wer weiß, wohin die ihre Entwicklung mittlerweile ausgelagert haben.

Titel: 
WALLBOX PULSAR PLUS BRAUCHT Firmware 6.4.19

Dann einen Text, der besagt, dass die Pulsar Plus unbrauchbar ist und sich die Kundenbeschwerden auf evcc seit Tagen stapeln mit Verweis auf https://github.com/evcc-io/evcc/issues/15581#issuecomment-2304396454

Korrekt ausgefüllt und eingereicht sollte es dann diesen Hinweis geben
[image]

und damit wächst der Druck auf dem Kessel an der richtigen Stelle.
Bis zur neuen Firmware können wir nur auf 129.x zurückgehen und dort bleiben.
Ich habe nun 4 Versuche von 130.0 / 130.2 / 130.4 / 130.6 ERFOLGLOS hinter mich gebracht inkl. restore - und vielleicht war auch noch 130.5 um Mitternacht dabei, aber ohne Druck auf deren Kessel, da passiert nix.

Unsere Wallbox ist im Feb. 2023 gekauft worden und ging Ende Juli 2024 ans Netz zum 2 phasigen e-GOLF laden, aber nicht bei WALLBOX, sondern bei Lichtblick für 250€ gekauft.

dscho, 2024-08-27:
Alles gute Punkte, insbesondere dass auch ich mit Wallbox zusammen arbeiten sollte, was ich bislang noch nicht gemacht habe. Das gebe ich zu, dass ich das hätte tun sollen.

Bevor ich das mache, möchte ich aber wissen, was hier genau seit 0.130 so schief läuft. Denn sonst würde ich selber, so ich denn in den Schuhen eines überarbeiteten Wallbox Support Engineers stünde, reflexartig sagen: wieso, ging doch vor dem Update, muss also ein Software-Fehler sein. Das ist eben auch weiterhin nicht auszuschließen.

> wurde denn auch in der 128.4 "SetChargingProfile" benutzt um von aus/pv/min+PV/schnell umzuschalten?

Ja, in der Tat. Da gab es aber etliche Unterschiede in den Parametern, die da mitgegeben wurden (z.B. wurde `TxProfile` gesetzt anstatt `TxDefaultProfile`, und das ist nur eine von mehreren, grundsätzlichen Änderungen).

Ich muss mich mal in Go reinfuchsen und in Vite, damit ich da mal in meinem eigenen Fork Funktionen einbauen kann, mit denen diese geänderten Parameter hier zielgerichteter getestet werden können.

Um `SetChargingProfile` aufzurufen braucht es ja keine Sonne und nicht mal ein angestöpseltes Auto. Und es dauert auch unproduktiverweise jedes Mal ziemlich viel Zeit, wenn ich für jede Parameter -Änderung, egal wie minimal, den kompletten Code neu bauen muss und dann EVCC neu starten (denn da muss ich dann wieder minutenlang warten, bis der OCPP-Handschlag abgeschlossen ist).

Ich denke da z.B. an ein Eingabefeld, mit dem ich ein komplettes, JSON-formatiertes OCPP Kommando absetzen kann. Die Antwort kann ich ja im Log sehen, obwohl es auf lange Sicht noch viel schöner wäre, die Antwort im Web-Browser zu sehen (und das ganze als experimentelle Funktion mit ausreichenden Warnungen für alle Nutzer verfügbar zu machen, die solche Regressionen analysieren wollen, es wird ja erfahrungsgemäß nicht das letzte Mal sein, dass so etwas hilfreich wäre).

Falls jemand anderes hier schon an solcher Funktionalität gearbeitet hat, würde ich mir natürlich gerne die Zeit sparen, das noch einmal zu implementieren. Hat das schon jemand?

typxxi, 2024-08-27:
> Alles gute Punkte, insbesondere dass auch ich mit Wallbox zusammen arbeiten sollte, was ich bislang noch nicht gemacht habe. Das gebe ich zu, dass ich das hätte tun sollen.

nee, so war das ja an sich nicht gemeint, sondern nur das Zitat von mrhund.

Du hast ja Deinen Beitrag geleistet und das sauber dokumentiert, 
worauf mrhund uns ja die Hoffnung gab, dass die neue Firmware das Thema lösen würde, so dass nun jeder einfach die neue Firmware verlangen kann und sollte, weil die ein Problem löst.

Und wenn ich das recht verstand, geht es hier ja mal wieder **um einen nicht eingehaltenen Standard**, der nun in einer evcc Weiterentwicklung einen Fehler nur deshalb verursacht.  
Der Fehler ist ja nicht der evcc Weiterentwicklung geschuldet ist, sondern mangelnder Einhaltung des Standards auf Wallbox Seite. 

Da ja mrhund klar aufzeigen konnte, dass er nix von den bekannten / unseren Problemem an seiner Wallbox hat und er meinte, dass Du / wir eine veraltete Firmware hätten **(sieht ja auch mega alt aus  von 5.17.87 nicht zu 6.0 sondern  6.4.19)**

liegt der Hund also auf der **Wallbox Seite begraben**, die ja zumindest die Fehler genau kennen müssen, die sie von 5.17.87 bis zu 6.4.19 behoben haben und dafür sorgen, dass evcc dann unsere Fehler nicht mehr hat.

Von daher geht es um alle Betroffenen Wallbox Pulsar Plus Besitzer, die das Problem für sich selber lösen können, indem sie zuerst per Formular ein Update verlangen wie beschrieben (und in der Zwischenzeit auf evcc 129.x zurückgehen und dort bleiben, denn ich glaub kaum, dass es von evcc eine Lösung geben wird, wenn wir wissen, dass es ein Hersteller Thema ist).

Ich fand es auch mal spannend, wie träge so ein Hersteller selbst als Sponsor sein kann, der die Sache an sich unterstützt, denn er ist ja Sponsor. Da hätte ich gedacht, dass die evcc Entwickler einen roten Teppich in das Entwicklungslabor von Wallbox bekämen, um genau solche Situationen wie jetzt zu vermeiden, denn je früher Fehler entdeckt und verfolgt werden, je besser das Produkt. 

Schauen wir mal, wann eine inhaltliche Antwort kommt. Am Freitag ruf ich dann mal an, wenn nix kommen sollte.

premultiply, 2024-08-27:
> wie träge so ein Hersteller selbst als Sponsor sein kann, der die Sache an sich unterstützt, denn er ist ja Sponsor.

Wallbox.com ist leider kein Sponsor.

Wir freuen uns aber u. a. genau aus diesem Grund 

>Da hätte ich gedacht, dass die evcc Entwickler einen roten Teppich in das Entwicklungslabor von Wallbox bekämen, um genau solche Situationen wie jetzt zu vermeiden, denn je früher Fehler entdeckt und verfolgt werden, je besser das Produkt.

immer über Sponsoren und Unterstützer des Projekts.

typxxi, 2024-08-27:
> > wie träge so ein Hersteller selbst als Sponsor sein kann, der die Sache an sich unterstützt, denn er ist ja Sponsor.
> 
> Wallbox.com ist leider kein Sponsor.

Komisch, wir hatten die Wallbox for 1,5 Jahren gekauft, weil wir glaubten, der Hersteller würde zu der Liste der Unterstützer gehören, denn es war damals kein Sponsortoken erforderlich. Oder hat sich da was geändert gehabt ?

dscho, 2024-08-27:
> Ich hab mir dafür einen Workaroound gebaut https://github.com/mrhund/Wallbox_Pulsar-OCPP-Charge_Current-Fix

@mrhund Das ist interessant, danke, dass Du das geteilt hast. Damit kann ich ja eigentlich auch Workarounds in EVCC selber einbauen, diese Schleife muss ja gar nicht unbedingt in einem externen C# Projekt ausgeführt werden.

[43 more replies]