evcc-io/evcc #15591: OCCP Error with myenergi Zappi

stjaeger78, 2024-08-22:
### Describe the bug

Not starting with myenergi Zappi wallbox using OCCP. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  
# unique installation id
plant: xx

interval: 20s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: localhost
  port: 502
  usage: grid
  storageunit: 1
  modbus: tcpip
  timeout: 15s
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: localhost
  port: 502
  usage: pv
  storageunit: 1
  modbus: tcpip
  timeout: 15s
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: localhost
  port: 502
  usage: battery
  storageunit: 1
  modbus: tcpip
  timeout: 15s
  name: battery3
  
chargers:
- type: template
  template: ocpp
  connector: 1
  idtag: 1234
  stationid: xx
  connecttimeout: 5m
  timeout: 2m
  #getconfiguration: false
  #bootnotification: true
  chargingrateunit: A
  name: wallbox5

vehicles:
- type: template
  template: tesla
  title: Model Y
  icon: car
  accessToken: xx
  refreshToken: xx
  vin: xx
  capacity: 78
  mode: pv
  name: ev4
  
loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16

site:
  title: xx
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
  residualPower: 100
  
tariffs:
  currency: EUR
  grid:
    type: tibber
    token: xx
```

### Log details

```text
[main  ] FATAL 2024/08/19 20:02:32 cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'ocpp': the mandatory SmartCharging profile is not supported

[ocpp  ] ERROR 2024/08/22 08:31:37 ocpp message (1909250288): FormationViolation - iso8601: Unexpected character `T` ([3,"1909250288",{"status":"Accepted","connectorId":1,"scheduleStart":"Thu, 22 Aug 2024 06:31:37 GMT","chargingSchedule":{"duration":60,"startSchedule":"Thu, 22 Aug 2024 06:31:37 GMT","chargingRateUnit":"W","chargingSchedulePeriod":[]}}])
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.0

premultiply, 2024-08-22:
Log output is missing.
Run `evcc charger --log trace --diagnose`.

Try latest release and actual nightly build.

Reduce your charger config to
```
chargers:
- type: template
  template: ocpp
  name: wallbox5
```

Remember to setup your charger to local "autostart" mode or use local RFID auth!
Ref https://github.com/evcc-io/evcc/pull/14888 and https://docs.evcc.io/blog/2024/08/17/highlights-14a-enwg-ocpp-loadmanagement-elli#rfid-und-autorisierung

The charger in not compliant to ocpp spec.
Please contact the vendor to get it fixed.

mfuchs1984, 2024-08-22:
According to 
https://support.myenergi.com/hc/en-gb/articles/16907254510865-What-OCPP-messages-does-zappi-support
the zappi does not support the Smart charging profile

>We currently do not support the Smart Charging, or Reservation sections of the OCPP 1.6J specification. We'll update this page as changes to our service our made

andig, 2024-08-22:
Es werde ungültige Timestamps geliefert und daher abgewiesen. Die Zappi entspricht nicht der OCPP Spec. Bitte Ticket beim Hersteller aufmachen.

closed 2024-08-22 by andig