evcc-io/evcc #15592: Charger Delta AC MAX Smart (Modbus TCP) broken since v0.130.0

madmat17, 2024-08-22:
### Describe the bug

Since v0.130.0 evcc no longer recognizes when a vehicle is connected to the Delta AC MAX Smart in a Modbus TCP configuration.
This is due to the change in the file `delta.go` , lines 154 & 156 (https://github.com/evcc-io/evcc/compare/0.129.0...0.130.0#diff-6d9fb2ec3d3d401d69c4079b978d1c89d59bddfae6243ebeb45681a1ac125be5) where the EVSE state `3` is falsely translated into charger status A - should be charger status B, though.

### Steps to reproduce

1. Update evcc to v0.130.x
2. Connect vehicle to charger
3. Charger remains in status A

### Configuration details

```yaml
.
```

### Log details

```text
[lp-1 ] DEBUG 2024/08/22 08:44:17 charger status: A
[delta ] TRACE 2024/08/22 08:44:17 modbus: send 00 98 00 00 00 06 01 03 06 40 00 02
[delta ] TRACE 2024/08/22 08:44:18 modbus: recv 00 98 00 00 00 07 01 03 04 00 00 00 00
[modbus] TRACE 2024/08/22 08:44:27 modbus: send 00 6e 00 00 00 06 01 03 7d 40 00 02
[modbus] TRACE 2024/08/22 08:44:28 modbus: recv 00 6e 00 00 00 07 01 03 04 00 00 06 41
[modbus] TRACE 2024/08/22 08:44:28 modbus: send 00 6e 00 00 00 06 01 03 7d 6a 00 02
[modbus] TRACE 2024/08/22 08:44:29 modbus: recv 00 6e 00 00 00 07 01 03 04 00 0d bd a5
[modbus] TRACE 2024/08/22 08:44:29 modbus: send 00 79 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/08/22 08:44:30 modbus: recv 00 79 00 00 00 07 01 03 04 00 00 04 af
[modbus] TRACE 2024/08/22 08:44:30 modbus: send 00 8a 00 00 00 06 01 03 90 f9 00 02
[site ] DEBUG 2024/08/22 08:44:31 ----
[delta ] TRACE 2024/08/22 08:44:31 modbus: send 00 99 00 00 00 06 01 04 03 ed 00 02
[modbus] TRACE 2024/08/22 08:44:32 modbus: recv 00 8a 00 00 00 07 01 03 04 00 00 00 03
[modbus] TRACE 2024/08/22 08:44:32 modbus: send 00 8a 00 00 00 06 01 03 91 01 00 02
[delta ] TRACE 2024/08/22 08:44:32 modbus: recv 00 99 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/08/22 08:44:32 charge power: 0W
[modbus] TRACE 2024/08/22 08:44:33 modbus: recv 00 8a 00 00 00 07 01 03 04 00 06 05 af
[modbus] TRACE 2024/08/22 08:44:33 modbus: send 00 79 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/08/22 08:44:34 modbus: recv 00 79 00 00 00 07 01 03 04 00 03 bd c5
[modbus] TRACE 2024/08/22 08:44:34 modbus: send 00 6f 00 00 00 06 01 03 7d 40 00 02
[modbus] TRACE 2024/08/22 08:44:37 modbus: recv 00 6f 00 00 00 07 01 03 04 00 00 06 4d
[modbus] TRACE 2024/08/22 08:44:37 modbus: send 00 8a 00 00 00 06 01 03 90 f3 00 02
[modbus] TRACE 2024/08/22 08:44:37 modbus: recv 00 8a 00 00 00 07 01 03 04 00 00 00 4d
[modbus] TRACE 2024/08/22 08:44:37 modbus: send 00 79 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/08/22 08:44:39 modbus: recv 00 79 00 00 00 05 01 03 02 01 68
[modbus] TRACE 2024/08/22 08:44:39 modbus: send 00 6f 00 00 00 06 01 03 7d 6a 00 02
[modbus] TRACE 2024/08/22 08:44:39 modbus: recv 00 6f 00 00 00 07 01 03 04 00 0d bd a6
[modbus] TRACE 2024/08/22 08:44:39 modbus: send 00 8a 00 00 00 06 01 03 90 f5 00 02
[site ] DEBUG 2024/08/22 08:44:39 pv power: 1613W
[modbus] TRACE 2024/08/22 08:44:39 modbus: recv 00 8a 00 00 00 07 01 03 04 00 00 00 42
[modbus] TRACE 2024/08/22 08:44:39 modbus: send 00 7a 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/08/22 08:44:40 modbus: recv 00 7a 00 00 00 07 01 03 04 00 00 04 b3
[modbus] TRACE 2024/08/22 08:44:40 modbus: send 00 7b 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/08/22 08:44:41 modbus: recv 00 7b 00 00 00 07 01 03 04 00 00 04 b3
[modbus] TRACE 2024/08/22 08:44:41 modbus: send 00 8a 00 00 00 06 01 03 90 f7 00 02
[modbus] TRACE 2024/08/22 08:44:42 modbus: recv 00 8a 00 00 00 07 01 03 04 ff ff ff a0
[modbus] TRACE 2024/08/22 08:44:42 modbus: send 00 7a 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/08/22 08:44:42 modbus: recv 00 7a 00 00 00 07 01 03 04 00 03 bd c5
[modbus] TRACE 2024/08/22 08:44:42 modbus: send 00 7b 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/08/22 08:44:47 modbus: recv 00 7b 00 00 00 07 01 03 04 00 03 bd c5
[modbus] TRACE 2024/08/22 08:44:47 modbus: send 00 8b 00 00 00 06 01 03 90 f9 00 02
[modbus] TRACE 2024/08/22 08:44:48 modbus: recv 00 8b 00 00 00 07 01 03 04 ff ff ff fb
[modbus] TRACE 2024/08/22 08:44:48 modbus: send 00 7a 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/08/22 08:44:49 modbus: recv 00 7a 00 00 00 05 01 03 02 01 68
[modbus] TRACE 2024/08/22 08:44:49 modbus: send 00 7b 00 00 00 06 01 03 90 8c 00 01
[modbus] TRACE 2024/08/22 08:44:51 modbus: recv 00 7b 00 00 00 05 01 03 02 01 68
[modbus] TRACE 2024/08/22 08:44:51 modbus: send 00 8b 00 00 00 06 01 03 91 01 00 02
[site ] DEBUG 2024/08/22 08:44:51 battery soc: 36%
[site ] DEBUG 2024/08/22 08:44:51 battery power: -1203W
[modbus] TRACE 2024/08/22 08:44:52 modbus: recv 00 8b 00 00 00 07 01 03 04 00 06 05 af
[... 94 lines cut ...]
[modbus] TRACE 2024/08/22 08:45:32 modbus: recv 00 72 00 00 00 07 01 03 04 00 00 06 2d
[modbus] TRACE 2024/08/22 08:45:32 modbus: send 00 72 00 00 00 06 01 03 7d 6a 00 02
[delta ] TRACE 2024/08/22 08:45:32 modbus: recv 00 9f 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2024/08/22 08:45:32 charge power: 0W
[modbus] TRACE 2024/08/22 08:45:32 modbus: recv 00 72 00 00 00 07 01 03 04 00 0d bd a6
[modbus] TRACE 2024/08/22 08:45:32 modbus: send 00 7e 00 00 00 06 01 03 90 89 00 02
[modbus] TRACE 2024/08/22 08:45:33 modbus: recv 00 7e 00 00 00 07 01 03 04 00 00 04 a8
[modbus] TRACE 2024/08/22 08:45:33 modbus: send 00 90 00 00 00 06 01 03 90 f9 00 02
[modbus] TRACE 2024/08/22 08:45:34 modbus: recv 00 90 00 00 00 07 01 03 04 ff ff ff e4
[modbus] TRACE 2024/08/22 08:45:34 modbus: send 00 73 00 00 00 06 01 03 7d 40 00 02
[modbus] TRACE 2024/08/22 08:45:34 modbus: recv 00 73 00 00 00 07 01 03 04 00 00 06 2d
[modbus] TRACE 2024/08/22 08:45:34 modbus: send 00 7e 00 00 00 06 01 03 90 cc 00 02
[modbus] TRACE 2024/08/22 08:45:35 modbus: recv 00 7e 00 00 00 07 01 03 04 00 03 bd c5
[modbus] TRACE 2024/08/22 08:45:35 modbus: send 00 90 00 00 00 06 01 03 91 01 00 02
[modbus] TRACE 2024/08/22 08:45:35 modbus: recv 00 90 00 00 00 07 01 03 04 00 06 05 af
[modbus] TRACE 2024/08/22 08:45:35 modbus: send 00 73 00 00 00 06 01 03 7d 6a 00 02
```

### What type of operating system are you running?

Linux

### Version

0.130.1

madmat17, 2024-08-22:
Here's the log I just made after downgrading to `v0.129.0` - no physical changes (no unplug/plug of the vehicle, no charger reboot etc), only downgraded evcc and now charger state is recognized correctly.

Log:
```
[main ] INFO 2024/08/22 08:49:30 evcc 0.129.0
[main ] INFO 2024/08/22 08:49:30 using config file: /etc/evcc.yaml
[db ] INFO 2024/08/22 08:49:30 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/08/22 08:49:31 listening at :7070
[site ] INFO 2024/08/22 08:49:32 site config:
[site ] INFO 2024/08/22 08:49:32 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/08/22 08:49:32 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/08/22 08:49:32 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/22 08:49:32 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/08/22 08:49:32 vehicles:
[site ] INFO 2024/08/22 08:49:32 vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/08/22 08:49:32 loadpoint 1:
[lp-1 ] INFO 2024/08/22 08:49:32 mode: off
[lp-1 ] INFO 2024/08/22 08:49:32 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/08/22 08:49:32 meters: charge ✓
[lp-1 ] INFO 2024/08/22 08:49:32 charge: power ✓ energy ✗ currents ✗
[site ] DEBUG 2024/08/22 08:49:32 set buffer soc: 95
[site ] DEBUG 2024/08/22 08:49:32 set priority soc: 90
[site ] DEBUG 2024/08/22 08:49:32 set battery discharge control: true
[lp-1 ] DEBUG 2024/08/22 08:49:32 set smart cost limit: 0.05
[lp-1 ] DEBUG 2024/08/22 08:49:32 !! active phases: 3p = min(0p measured 0p vehicle 3p physical 0p charger)
[lp-1 ] DEBUG 2024/08/22 08:49:32 phase timer inactive
[lp-1 ] DEBUG 2024/08/22 08:49:32 pv timer inactive
[main ] INFO 2024/08/22 08:49:32 new version available: 0.130.2
[site ] DEBUG 2024/08/22 08:49:32 ----
[lp-1 ] DEBUG 2024/08/22 08:49:33 charge power: 0W
[site ] DEBUG 2024/08/22 08:49:34 pv power: 1842W
[site ] DEBUG 2024/08/22 08:49:37 battery soc: 38%
[site ] DEBUG 2024/08/22 08:49:37 battery power: -1514W
[site ] DEBUG 2024/08/22 08:49:37 grid meter: 16W
[site ] DEBUG 2024/08/22 08:49:39 grid currents: [-0.71 -0.61 0.95]A
[site ] DEBUG 2024/08/22 08:49:39 battery has priority at soc 38% (< 90%)
[site ] DEBUG 2024/08/22 08:49:39 site power: 116W
[lp-1 ] DEBUG 2024/08/22 08:49:41 charger status: B
[lp-1 ] INFO 2024/08/22 08:49:41 car connected
[lp-1 ] DEBUG 2024/08/22 08:49:41 vehicle api refresh
[lp-1 ] DEBUG 2024/08/22 08:49:41 pv timer elapse
[lp-1 ] DEBUG 2024/08/22 08:49:41 pv timer inactive
```

closed 2024-08-22 by premultiply

Felsblick, 2024-08-25:
@premultiply 
we have no a problem, looking back to https://github.com/evcc-io/evcc/issues/14890 it all started to get complicated. meanwhile I think there is no other way then to split up between the Delta AC Max basic (and again use the register 1001 where all worked fine) and the AC Max Smart (use the register 1000).
The Point is, that the AC max basic Reports 3 on Register 1000 when NO vehicle is plugged in! --> https://github.com/evcc-io/evcc/issues/14890#issuecomment-2256551150

What do you think?

madmat17, 2024-08-26:
@Felsblick hast du die Version von gestern getestet? Dort wird der Wert ‚3‘ aus dem Register 1001 in den Status B übersetzt.
Das muss mit der Smart- und mit der Basic-Variante funktionieren.

Felsblick, 2024-08-26:
@madmat17 ich habe es mit dem letzen release 0.130.4 getestet, dort wird Register 1000 ausgewertet und Wert 3 auf B gemappt, soweit richtig. Das funktioniert aber nur für die AC Max Smart. Bei der Basic muss Wert 3 auf Status A gemappt sein, da es dort bedeutet, das kein Fahrzeug eingesteckt ist wie ich es [hier](https://github.com/evcc-io/evcc/issues/14890#issuecomment-2256551150) schon einmal geschrieben habe (woraufhin @premultiply das mapping geändert hatte, was jetzt bei der Smart zu Problemen geführt hat...).

Daher der Lösungsvorschlag: Eigenes Template für die AC Max basic mit Nutzung des Registers 1001 mit dem Mapping aus evcc 0.128.4, was einwandfrei für die basic funktioniert hat.
Die AC Max smart behält das jetziges Template mit Register 1000.

premultiply, 2024-08-26:
Das ist kein Template.

Dann brauchen wir eine Weiche.

Wie kann man Anhand der Register die Wallboxen zuverlässig unterscheiden?

madmat17, 2024-08-26:
Ah - das ist natürlich nicht ganz optimal.

@premultiply das Register 130 gibt als String die Modellbezeichnung aus.
zB in meinem Fall "EIAW-E22KTSE5A04"

EIAW-E22KT**S**E5A04 = Smart
EIAW-E22KT**B**E5A04 = Basic

Es können auch andere Stellen in der der Modellnummer variieren (fixes Kabel vs. Dose, 11kW vs. 22 kW, EU-Variante vs. US- und andere Varianten), etc.
Leider konnte ich keine umfassende Liste der Produktnummern finden. Wäre ein Switch mit Auswertung der 11. Position im String ("B" oder "S") sinnvoll?

LG,
Mat

madmat17, 2024-08-26:
@premultiply: Ich hätte das das wie folgt gelöst: https://github.com/evcc-io/evcc/compare/master...madmat17:evcc_delta:master#diff-6d9fb2ec3d3d401d69c4079b978d1c89d59bddfae6243ebeb45681a1ac125be5

Da ich kein Entwickler bin und die kommende Tage nicht testen kann:
Könntest du kurz drüber schauen, bevor ich einen PR aufmache? Du siehst vermutlich gleich, wenn was nicht passt.

Danke!!

commit 956227a5 (2024-08-22): Delta: fix status mapping

Fixes #15592