evcc-io/evcc #15594: Add another SOC Limit/goal for a charging Limit "always" and the existing one "only with PV energy available"

p-manent, 2024-08-22:
Hello Andi,

I want to get a 80% (selectable) SOC every day, independent of PV energy available or not,  but get 100% (selectable) if I have PV energy that I cannot use.

What do you think about such a feature? 

At the moment I always restrict the charging goal manually (use either "Fast mode" (if necessary)  or "Min+PV" charging mode), and if the charging goal is reached, I set the goal to 100% manually and  switch to PV mode

It would be very cool if this could be done automatically :-) Then I can plug in in the evening, getting 80% during the night, and if I do use the car the next day it automatically uses only the PV mode to reach 100% (which is mostly not necessary but better to charge the car than giving away the energy for nearly nuts...)

At the GUI maybe a "combined" slider to set the charging goal "without PV" and the existing one absolutly  

[image]

like in the Battery window...

[image]

mfuchs1984, 2024-08-22:
Something similar should already be possible. Set the goal to 100 and use the charge plan to set the 80% goal for the next morning l.

p-manent, 2024-08-22:
yes, thanx! I did not see the "Ankunft" / Min-Ladung %" field :-) I guess this should work...

p-manent, 2024-08-22:
I will test it  :-)

mfuchs1984, 2024-08-22:
> yes, thanx! I did not see the "Ankunft" / Min-Ladung %" field :-) I guess this should work...

When you use Ankunft it will immediately charge with full speed to reach the goal and then use PV energy. 

In case you typically know when you want leave the charge point next time, you can also use the Abfahrt feature, it will then first use as much PV as possible and then charge the remaining percentage to reach the goal just before leaving. 

Btw, scheduled charging based on weekdays seems to be in work, see #5492.

p-manent, 2024-08-22:
Ok, If it´s always "full speed" then it´s only similar. Maybe you can extend "Ankunft" with a mode? This might be PV "mode" (based on the battery dialog setting, this can differ from PV Überschuss), Min+PV or Fast...

Regarding the Abfahrt: great, the "weekday" makes sense :-)

mfuchs1984, 2024-08-22:
The purpose of Ankunft is to ensure the Car reaches a certain charge level to maintain mobility, so it does not make sense to use PV mode since it would take an unpredictable amount of time to reach the configured minimum. After this minimum level is reached, it will continue in PV mode automatically. What difference would it make to charge slower in your usecase?

p-manent, 2024-08-22:
The difference is: more costs ;-)
If you can choose the mode you need have less energy from grid.
Example:
My battery can deliver only 5kW, so maybe 4.2k for the car. If there is not too much sun and the "fast" is 11kW, this will use grid power, too.
If I can set the "Ankunft-Mode" to Min+PV, the battery will be able to deliver 100% of the car load in the first minutes/hours  (if not blocked by the battery settings) and only use grid power for the remaining percent.

mfuchs1984, 2024-08-22:
Then I'd suggest to use Abfahrt, it will keep the PV mode as long as possible. With battery supported charging, this will utilize the battery. If there is still a gap, it will be filled in fast mode before departure. In case you configure tibber, Abfahrt uses the lowest price periods instead of charging directly before departure.

p-manent, 2024-08-22:
If you like just keep this ticket open because - without a compromise  / modification - neither Ankunft nor Abfahrt would be "optimal" for the use case...

p-manent, 2024-08-26:
reopen?

The "Ankunft" with "fast" loading is another use case. So "load with definable mode to x%" and then continue with PV only seems to be another case (for me)

andig, 2024-09-01:
There are no plans to add yet more limits. You can already use external automation to do so.

closed 2024-09-01 by andig