evcc-io/evcc #15597: Delta AC MAX Smart WB - maximale Ladeleistung 5,2 kW

futurini, 2024-08-22:
### Describe the bug

Die maximale Laistung der WB beträgt 5,2 kW. PV Leitung ist über 7kW die LEistung wir nicht gesteigert. über die 

### Steps to reproduce

1. Umschalten 1phasig <>3 phasig
[image]

### Configuration details

```yaml
loadpoints:
- title: Garage
  charger: wallbox2
  phases: 3
  mincurrent: 8
  maxcurrent: 20
  mode: minpv
```

### Log details

```text
[site  ] DEBUG 2024/08/22 11:57:58 ----
[lp-1  ] DEBUG 2024/08/22 11:57:59 charge power: 5287W
[site  ] DEBUG 2024/08/22 11:57:59 pv power: 7064W
[site  ] DEBUG 2024/08/22 11:57:59 battery soc: 100%
[site  ] DEBUG 2024/08/22 11:57:59 battery power: 0W
[site  ] DEBUG 2024/08/22 11:57:59 grid meter: -2728W
[site  ] DEBUG 2024/08/22 11:57:59 grid powers: [-572 -415 -1742]W
[site  ] DEBUG 2024/08/22 11:58:00 grid currents: [-4.76 -4.52 -8.57]A
[site  ] DEBUG 2024/08/22 11:58:00 site power: -2628W
[lp-1  ] DEBUG 2024/08/22 11:58:00 !! session: chargeRater.chargedEnergy=0.3 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/22 11:58:00 charger status: C
[lp-1  ] WARN 2024/08/22 11:58:02 charger logic error: disabled but charging
[site  ] DEBUG 2024/08/22 11:58:08 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.2

StefanSchoof, 2024-08-22:
Dein Satz ist nicht vollständig, so dass man raten muss was dein Problem ist.

Das Menü zum Umschalten ist die Einstellung wie deine Box angeschlossen ist und führt nicht eine Phasenumschaltung aus, (Wenn evcc das mit deine Box das könnte würde es dort eine dritte Option geben)

futurini, 2024-08-24:
Ich habe jetzt den Satz vervollständigt. Die Info habe ich jetzt verstanden. Die Box kann keine automatische Phasenumschaltung, dass war mir vorher bekannt. Bei schlechtem Wetter schalte ich manuell nur eine Phase ein und schließe dann das Auto an. Dann werden maximal 3,6kW geladen.

StefanSchoof, 2024-08-25:
Ich denke damit dir das evcc Team helfen kann braucht es mehr als der kleine Ausschnitt der Log. Man sollte zumindest sehen können was evcc als Limit für den charger berechnet.

Am besten sind Logs die beim Start von evcc beginnen und dann kurz danach das Problem zeigen.

futurini, 2024-08-27:
Nach Ausführlichen Test habe ich fest gestellt das in der WB der Ladestrom auf 8A Eingestellt war.

closed 2024-08-27 by futurini

capsmanrtg, 2025-02-16:
> Ich habe jetzt den Satz vervollständigt. Die Info habe ich jetzt verstanden. Die Box kann keine automatische Phasenumschaltung, dass war mir vorher bekannt. Bei schlechtem Wetter schalte ich manuell nur eine Phase ein und schließe dann das Auto an. Dann werden maximal 3,6kW geladen. Ich kann gerne weiter Test durchführen bzw. Information zur Verfügung stellen falls nötig.

Hey! Ich habe auch eine Delta AC Max. Wie kann man denn manuell nur eine Phase bei der einschalten? Das würde mich nämlich auch interessieren.

premultiply, 2025-02-16:
Wie bei jeder Wallbox: Lasttrennschalter für L2+L3 in die Zuleitung einbauen.