evcc-io/evcc #15599: Charging Power Level reported wrong and Charged Values are incorrect

markusgaller1978, 2024-08-22:
### Describe the bug

One of the latest evcc I observed that the values of the "charge Level" is wrongly reported.
This leads to a wrong value of charged capacity.
I am charging a Skoda Octavia just maximal charging is at 3.6kw and not more.

[image]
[image]

### Steps to reproduce

Every time by charging the car.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.home # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

mqtt:
  broker: 192.168.0.xxx:1883
  topic: evcc
  clientid: evcc
  user: ....
  password: ....

# unique installation id
plant: f2f9266b536181d7afad4f6e4605db48757accf5a6337c9567a31468481ca004
interval: 10s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.0.xxx
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 15s
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.0.xxx
  port: 502
  usage: pv
  modbus: tcpip
  timeout: 15s
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: 192.168.0.xxx
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 15s
  name: battery3

chargers:
- type: template
  template: ellipro
  ski: xxxx xxxx xxxx xxxx xxxx .....
[... 122 lines cut ...]
[site  ] DEBUG 2024/08/22 12:43:17 ----
[lp-1  ] DEBUG 2024/08/22 12:43:17 charge power: 11040W
[eebus ] DEBUG 2024/08/22 12:43:18 Send: d:_i:47859_Elli-Wallbox-2149A1UPID notify 2270 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/22 12:43:20 Send: d:_i:47859_Elli-Wallbox-2149A1UPID notify 2271 DeviceDiagnosisHeartbeatData
[site  ] DEBUG 2024/08/22 12:43:21 pv power: 5025W
[eebus ] DEBUG 2024/08/22 12:43:22 Send: d:_i:47859_Elli-Wallbox-2149A1UPID notify 2272 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/22 12:43:24 Send: d:_i:47859_Elli-Wallbox-2149A1UPID notify 2273 DeviceDiagnosisHeartbeatData
[site  ] DEBUG 2024/08/22 12:43:25 battery soc: 0%
[site  ] DEBUG 2024/08/22 12:43:25 battery power: 0W
[site  ] DEBUG 2024/08/22 12:43:26 grid meter: -487W
[eebus ] DEBUG 2024/08/22 12:43:26 Send: d:_i:47859_Elli-Wallbox-2149A1UPID notify 2274 DeviceDiagnosisHeartbeatData
[site  ] DEBUG 2024/08/22 12:43:27 grid currents: [9.67 -5.86 -5.65]A
[site  ] DEBUG 2024/08/22 12:43:28 site power: -487W
[lp-1  ] DEBUG 2024/08/22 12:43:28 !! session: chargeRater.chargedEnergy=10.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/22 12:43:28 charger status: C
[eebus ] DEBUG 2024/08/22 12:43:28 !! OpEV.LoadControlLimits set: {a false false 16}
[lp-1  ] DEBUG 2024/08/22 12:43:28 pv charge current: 16.7A = 16A + 0.706A (-487W @ 3p)
[site  ] DEBUG 2024/08/22 12:43:28 ----
[lp-1  ] DEBUG 2024/08/22 12:43:28 charge power: 11040W
[eebus ] DEBUG 2024/08/22 12:43:28 Send: d:_i:47859_Elli-Wallbox-2149A1UPID notify 2275 DeviceDiagnosisHeartbeatData
```

### What type of operating system are you running?

Linux

futurini, 2024-08-22:
please remove personal data from "Configuration details" 
Configuration details

vehicles:
    user: xxx
    password: xxx
    vin: xxx

StefanSchoof, 2024-08-22:
The root cause is the breaking change about the use of the meter data in the elli pro, see #15367

andig, 2024-08-22:
> please remove personal data from "Configuration details"

Consider password burned. Change, don't reuse!

andig, 2024-08-22:
And: configure Skoka as `phases: 1`.

andig, 2024-08-22:
We're discussing re-enabling the Elli Pro meter, stay tuned.

Aitschcux, 2024-08-25:
My Elli Pro Wallbox is currently running on EVCC 0.129.0 and is more stable than it has been for a long time. I am aware that you had to laboriously implement many workarounds for the wallbox and that a lot of energy and time was invested in this. My special thanks go to DerAndereAndi! I understand the reasons behind the current breaking change. However, reactivating the Elli Pro meter would be an option that I would welcome, as I cannot simply install an external meter. I could deal with the resulting errors.

IZ23, 2024-08-30:
> We're discussing re-enabling the Elli Pro meter, stay tuned.

any news?

andig, 2024-09-01:
Elli Pro meter support has been re-added in https://github.com/evcc-io/evcc/pull/15778. Nightly can be tested in 30min.

closed 2024-09-01 by andig