evcc-io/evcc #15616: Trouble PulsarPlus

JBN164, 2024-08-23:
### Describe the bug

I have configured EVCC on my HomeAssistant System.

I configured the YAML like its given in the EVCC documentation.
You find the YAML below.

[image]

The OCPP connection looked like it works.

But some errors accrued. You will find in Log Detail.

The Charger and other components were implemented in home assistant and worked fine. So I don`t think that there is some trouble with the wiring.
Yesterday only one error appeared. "waiting for local authentication". The LED of my wallbox was yellow, which means it is locked. EVCC said waiting for the vehicle and my Tesla said waiting for charger. Normally this is shown till I unlock the Wallbox via app. This is not working when the wallbox is connected via OCPP.
I tried `remote start: true` but it doesn't work.
The OCPP documentation of Pulsar Plus said:

> AuthEnabled in CSMS
> When the CPO has set the AuthEnabled to TRUE in the CSMS, the charger is unlocked and will not need authorization through an RFID card or PIN.
> By connecting the EV gun, the charger will start charging according to the car’s demand.
> When the AuthEnabled is set to TRUE, the charger status LED will remain Yellow.
> 

I tried to use `AuthEnabled: true` but this isn't a known command.

After trying this more failure appears, you find in Log Detail.

The UI looks like this:
[image]

### Steps to reproduce

1. open EVCC
2. change mode from off to PV, min+PV or "schnellladen"
3. waiting for vehicle +. failure appears
...

### Configuration details

```yaml
meters:
  - name: my_grid
    type: template
    template: tibber-pulse
    usage: grid
    token: hidden
    #homeid: hidden # optional

  - name: my_pv
    type: template
    template: sungrow-hybrid
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.130 # Hostname
    port: 502 # Port
    
  - name: my_pv2
    type: template
    template: hoymiles-opendtu
    usage: pv
    host: 192.168.178.168 # IP-Adresse oder Hostname
    
  - name: my_battery
    type: template
    template: sungrow-hybrid
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.130 # Hostname
    port: 502 # Port
    

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for documentation see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: my_charger
    type: template
    template: pulsarplus
    stationid: Parkplatz # Station ID des Ladepunktes.
    #remotestart: true #Ergänzt da Wallbox nicht mit laden beginnt geändert von true auf "Autostart"
    #idtag: evcc
    #AuthEnabled: true #https://support.wallbox.com/de/knowledge-base/ocpp-aktivierungs-und-einrichtungsanleitung/
        #Wenn der CPO die Option AuthEnabled im CSMS auf TRUE gesetzt hat, ist das Ladegerät entsperrt und benötigt keine Autorisierung durch eine RFID-Karte oder PIN.
        #Wenn Sie die EV-Pistole anschließen, beginnt das Ladegerät entsprechend dem Bedarf des Fahrzeugs zu laden.
        #Wenn AuthEnabled auf TRUE gesetzt ist, bleibt die Status-LED des Ladegeräts gelb.
```

### Log details

```text
[main ] INFO 2024/08/23 09:25:51 evcc 0.130.2
[main ] INFO 2024/08/23 09:25:51 using config file: /config/evcc.yaml
[db ] INFO 2024/08/23 09:25:51 using sqlite database: /data/evcc.db
[main ] INFO 2024/08/23 09:25:52 listening at :7070
[Parkplatz-1] DEBUG 2024/08/23 09:26:56 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2024/08/23 09:27:26 charge point connected: Parkplatz
[Parkplatz-1] DEBUG 2024/08/23 09:27:26 hijacking transaction: 2
[lp-1 ] DEBUG 2024/08/23 09:27:33 charge total import: 1666.102kWh
[site ] INFO 2024/08/23 09:27:33 site config:
[site ] INFO 2024/08/23 09:27:33 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/08/23 09:27:33 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2024/08/23 09:27:33 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/23 09:27:33 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/23 09:27:33 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/08/23 09:27:33 vehicles:
[site ] INFO 2024/08/23 09:27:33 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/08/23 09:27:33 loadpoint 1:
[lp-1 ] INFO 2024/08/23 09:27:33 mode: pv
[lp-1 ] INFO 2024/08/23 09:27:33 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/08/23 09:27:33 meters: charge ✓
[lp-1 ] INFO 2024/08/23 09:27:33 charge: power ✓ energy ✓ currents ✓
...
[lp-1 ] ERROR 2024/08/23 09:28:28 vehicle odometer: asleep
...
[site ] DEBUG 2024/08/23 09:39:33 ----
[lp-1 ] DEBUG 2024/08/23 09:39:33 charge power: 0W
[lp-1 ] DEBUG 2024/08/23 09:39:33 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:39:34 pv power: 2819W
[site ] DEBUG 2024/08/23 09:39:34 battery soc: 14%
[site ] DEBUG 2024/08/23 09:39:34 battery power: -1898W
[site ] DEBUG 2024/08/23 09:39:34 grid meter: 58W
[site ] DEBUG 2024/08/23 09:39:34 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:39:34 battery has priority at soc 14% (< 60%)
[site ] DEBUG 2024/08/23 09:39:34 site power: 158W
[lp-1 ] DEBUG 2024/08/23 09:39:34 charge voltages: [0 242 244]V
[lp-1 ] WARN 2024/08/23 09:39:34 invalid phase wiring between charge meter and charger
[lp-1 ] DEBUG 2024/08/23 09:39:34 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/23 09:39:34 charge total import: 1666.102kWh
[lp-1 ] DEBUG 2024/08/23 09:39:34 charger status: B
[lp-1 ] WARN 2024/08/23 09:39:34 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/23 09:39:34 charger disable
[lp-1 ] DEBUG 2024/08/23 09:39:36 charger enable
[lp-1 ] DEBUG 2024/08/23 09:39:36 wake-up timer: start
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

evcc 0.130.2

andig, 2024-08-23:
What is the issue?

premultiply, 2024-08-23:
> I tried remote start: true but it doesn't work.

The setting is named `remotestart: true`.

Make sure you restart evcc the change of the config and have the car disconnected.

JBN164, 2024-08-23:
> What is the issue?

The issue is that the **Charger didn't charge the vehicle**.
The EVCC UI says waiting for the vehicle and the vehicle is waiting to get power. 

I think it is because the wallbox is locked and EVCC can not unlock it. 

I addition to that there are some warnings I can't explain and don't know to fix it.

`[lp-1 ] WARN 2024/08/23 09:39:34 charger out of sync: expected enabled, got disabled`

`[lp-1 ] WARN 2024/08/23 09:39:34 invalid phase wiring between charge meter and charger`

`[lp-1 ] ERROR 2024/08/23 09:28:28 vehicle odometer: asleep`

JBN164, 2024-08-23:
> > I tried remote start: true but it doesn't work.
> 
> The setting is named `remotestart: true`.
> 
> Make sure you restart evcc after the change of the config and have the car disconnected.

I tried `remotestart: true` but it didn't changed anything :-(

andig, 2024-08-23:
Please share the log.

JBN164, 2024-08-23:
```
[main ] INFO 2024/08/23 09:25:51 evcc 0.130.2
[main ] INFO 2024/08/23 09:25:51 using config file: /config/evcc.yaml
[db ] INFO 2024/08/23 09:25:51 using sqlite database: /data/evcc.db
[main ] INFO 2024/08/23 09:25:52 listening at :7070
[Parkplatz-1] DEBUG 2024/08/23 09:26:56 waiting for chargepoint: 5m0s
[ocpp ] DEBUG 2024/08/23 09:27:26 charge point connected: Parkplatz
[Parkplatz-1] DEBUG 2024/08/23 09:27:26 hijacking transaction: 2
[lp-1 ] DEBUG 2024/08/23 09:27:33 charge total import: 1666.102kWh
[site ] INFO 2024/08/23 09:27:33 site config:
[site ] INFO 2024/08/23 09:27:33 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/08/23 09:27:33 grid: power ✓ energy ✗ currents ✓
[site ] INFO 2024/08/23 09:27:33 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/23 09:27:33 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/23 09:27:33 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/08/23 09:27:33 vehicles:
[site ] INFO 2024/08/23 09:27:33 vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/08/23 09:27:33 loadpoint 1:
[lp-1 ] INFO 2024/08/23 09:27:33 mode: pv
[lp-1 ] INFO 2024/08/23 09:27:33 charger: power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/08/23 09:27:33 meters: charge ✓
[lp-1 ] INFO 2024/08/23 09:27:33 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2024/08/23 09:27:33 set buffer soc: 95
[site ] DEBUG 2024/08/23 09:27:33 set priority soc: 60
[site ] DEBUG 2024/08/23 09:27:33 set battery discharge control: true
[lp-1 ] DEBUG 2024/08/23 09:27:33 set smart cost limit: 0.2
[lp-1 ] DEBUG 2024/08/23 09:27:33 phase timer inactive
[lp-1 ] DEBUG 2024/08/23 09:27:33 pv timer inactive
[site ] DEBUG 2024/08/23 09:27:33 ----
[lp-1 ] DEBUG 2024/08/23 09:27:33 charge power: 0W
[lp-1 ] DEBUG 2024/08/23 09:27:33 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:27:34 pv power: 2215W
[site ] DEBUG 2024/08/23 09:27:34 battery soc: 8%
[site ] DEBUG 2024/08/23 09:27:34 battery power: -1095W
[site ] DEBUG 2024/08/23 09:27:34 grid meter: -10W
[site ] DEBUG 2024/08/23 09:27:34 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:27:34 battery has priority at soc 8% (< 60%)
[site ] DEBUG 2024/08/23 09:27:34 site power: 90W
[lp-1 ] DEBUG 2024/08/23 09:27:34 charge voltages: [0 240 249]V
[lp-1 ] WARN 2024/08/23 09:27:34 invalid phase wiring between charge meter and charger
[lp-1 ] DEBUG 2024/08/23 09:27:34 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/23 09:27:34 charge total import: 1666.102kWh
[lp-1 ] DEBUG 2024/08/23 09:27:34 charger status: B
[lp-1 ] DEBUG 2024/08/23 09:27:34 charge start energy: 1.67e+03kWh
[lp-1 ] INFO 2024/08/23 09:27:34 car connected
[lp-1 ] DEBUG 2024/08/23 09:27:34 vehicle api refresh
[lp-1 ] DEBUG 2024/08/23 09:27:34 pv timer elapse
[lp-1 ] DEBUG 2024/08/23 09:27:34 pv timer inactive
[lp-1 ] DEBUG 2024/08/23 09:27:34 charge total import: 1666.102kWh
[lp-1 ] DEBUG 2024/08/23 09:27:35 max charge current: 16A
[lp-1 ] DEBUG 2024/08/23 09:27:36 charger enable
[lp-1 ] DEBUG 2024/08/23 09:27:36 wake-up timer: start
[lp-1 ] INFO 2024/08/23 09:28:23 vehicle updated: unknown -> Snowball
[lp-1 ] DEBUG 2024/08/23 09:28:27 set session soc limit: 95
[site ] DEBUG 2024/08/23 09:28:27 ----
[lp-1 ] DEBUG 2024/08/23 09:28:27 charge power: 0W
[lp-1 ] DEBUG 2024/08/23 09:28:27 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:28:28 pv power: 2556W
[site ] DEBUG 2024/08/23 09:28:28 battery soc: 8%
[site ] DEBUG 2024/08/23 09:28:28 battery power: -1463W
[site ] DEBUG 2024/08/23 09:28:28 grid meter: 17W
[site ] DEBUG 2024/08/23 09:28:28 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:28:28 battery has priority at soc 8% (< 60%)
[site ] DEBUG 2024/08/23 09:28:28 site power: 117W
[lp-1 ] ERROR 2024/08/23 09:28:28 vehicle odometer: asleep
[lp-1 ] DEBUG 2024/08/23 09:28:28 charge voltages: [0 240 249]V
[lp-1 ] WARN 2024/08/23 09:28:28 invalid phase wiring between charge meter and charger
[lp-1 ] DEBUG 2024/08/23 09:28:28 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/23 09:28:28 charge total import: 1666.102kWh
[lp-1 ] DEBUG 2024/08/23 09:28:28 charger status: B
[... 303 lines cut ...]
[site ] DEBUG 2024/08/23 09:45:35 site power: 70W
[lp-1 ] DEBUG 2024/08/23 09:45:35 charge voltages: [226 241 253]V
[lp-1 ] DEBUG 2024/08/23 09:45:35 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/23 09:45:35 charge total import: 1666.102kWh
[lp-1 ] DEBUG 2024/08/23 09:45:35 charger status: B
[lp-1 ] WARN 2024/08/23 09:45:35 charger out of sync: expected enabled, got disabled
[lp-1 ] DEBUG 2024/08/23 09:45:35 charger disable
[lp-1 ] DEBUG 2024/08/23 09:45:36 charger enable
[lp-1 ] DEBUG 2024/08/23 09:45:36 wake-up timer: start
[site ] DEBUG 2024/08/23 09:46:33 ----
[lp-1 ] DEBUG 2024/08/23 09:46:33 charge power: 0W
[lp-1 ] DEBUG 2024/08/23 09:46:33 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:46:34 pv power: 3036W
[site ] DEBUG 2024/08/23 09:46:35 battery soc: 20%
[site ] DEBUG 2024/08/23 09:46:35 battery power: -2545W
[site ] DEBUG 2024/08/23 09:46:35 grid meter: 21W
[site ] DEBUG 2024/08/23 09:46:35 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/23 09:46:35 battery has priority at soc 20% (< 60%)
[site ] DEBUG 2024/08/23 09:46:35 site power: 121W
[lp-1 ] DEBUG 2024/08/23 09:46:35 charge voltages: [226 242 253]V
[lp-1 ] DEBUG 2024/08/23 09:46:35 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/23 09:46:35 charge total import: 1666.102kWh
[lp-1 ] DEBUG 2024/08/23 09:46:35 charger status: B
[lp-1 ] DEBUG 2024/08/23 09:46:35 wake-up vehicle, attempts left: 4
```

andig, 2024-08-23:
Bitte ocpp: trace

JBN164, 2024-08-23:
```

[ocpp ] DEBUG 2024/08/23 12:11:35 charge point connected: Parkplatz
[ocpp ] TRACE 2024/08/23 12:11:36 send Parkplatz: [2,"316114569","GetConfiguration",{}]
[ocpp ] TRACE 2024/08/23 12:11:39 recv Parkplatz: [3,"316114569",{"unknownKey": [],"configurationKey": [{"readonly": false,"key": "AllowOfflineTxForUnknownId","value": "true"},{"readonly": false,"key": "AuthorizationCacheEnabled","value": "true"},{"readonly": false,"key": "AuthorizeRemoteTxRequests","value": "false"},{"readonly": false,"key": "BlinkRepeat","value": "10"},{"readonly": false,"key": "ClockAlignedDataInterval","value": "0"},{"readonly": false,"key": "ConnectionTimeOut","value": "120"},{"readonly": false,"key": "ConnectorPhaseRotation","value": "1.Unknown"},{"readonly": true,"key": "ConnectorPhaseRotationMaxLength","value": "1"},{"readonly": true,"key": "GetConfigurationMaxKeys","value": "50"},{"readonly": false,"key": "HeartbeatInterval","value": "60"},{"readonly": false,"key": "LightIntensity","value": "100"},{"readonly": false,"key": "LocalAuthorizeOffline","value": "true"},{"readonly": false,"key": "LocalPreAuthorize","value": "true"},{"readonly": false,"key": "MaxEnergyOnInvalidId","value": "5000"},{"readonly": false,"key": "MeterValuesAlignedData","value": "Energy.Active.Import.Register"},{"readonly": true,"key": "MeterValuesAlignedDataMaxLength","value": "10"},{"readonly": false,"key": "MeterValuesSampledData","value": "Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offered,SoC"},{"readonly": true,"key": "MeterValuesSampledDataMaxLength","value": "10"},{"readonly": false,"key": "MeterValueSampleInterval","value": "10"},{"readonly": false,"key": "MinimumStatusDuration","value": "1"},{"readonly": true,"key": "NumberOfConnectors","value": "1"},{"readonly": false,"key": "ResetRetries","value": "0"},{"readonly": false,"key": "StopTransactionOnEVSideDisconnect","value": "true"},{"readonly": false,"key": "StopTransactionOnInvalidId","value": "true"},{"readonly": false,"key": "StopTxnAlignedData","value": ""},{"readonly": true,"key": "StopTxnAlignedDataMaxLength","value": "0"},{"readonly": false,"key": "StopTxnSampledData","value": ""},{"readonly": true,"key": "StopTxnSampledDataMaxLength","value": "0"},{"readonly": true,"key": "SupportedFileTransferProtocols","value": ""},{"readonly": true,"key": "SupportedFeatureProfiles","value": "Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger"},{"readonly": true,"key": "SupportedFeatureProfilesMaxLength","value": "6"},{"readonly": false,"key": "TransactionMessageAttempts","value": "3"},{"readonly": false,"key": "TransactionMessageRetryInterval","value": "60"},{"readonly": false,"key": "UnlockConnectorOnEVSideDisconnect","value": "true"},{"readonly": false,"key": "WebSocketPingInterval","value": "30"},{"readonly": false,"key": "LocalAuthListEnabled","value": "true"},{"readonly": true,"key": "LocalAuthListMaxLength","value": "5000000"},{"readonly": true,"key": "SendLocalListMaxLength","value": "450"},{"readonly": true,"key": "ReserveConnectorZeroSupported","value": "false"},{"readonly": true,"key": "ChargeProfileMaxStackLevel","value": "3"},{"readonly": true,"key": "ChargingScheduleAllowedChargingRateUnit","value": "Current"},{"readonly": true,"key": "ChargingScheduleMaxPeriods","value": "100"},{"readonly": true,"key": "ConnectorSwitch3to1PhaseSupported","value": "false"},{"readonly": true,"key": "MaxChargingProfilesInstalled","value": "10"},{"readonly": false,"key": "chargingALimitConn1","value": "3"},{"readonly": false,"key": "AuthEnabled","value": "true"},{"readonly": false,"key": "AuthEnabledOffline","value": "true"},{"readonly": false,"key": "AuthDisabledIdTag","value": "NoAuthorization"},{"readonly": false,"key": "minSoC","value": "9"},{"readonly": false,"key": "maxSoC","value": "97"},{"readonly": true,"key": "SSID","value": "Baerenhoehle"},{"readonly": true,"key": "WiFiSignalStrength","value": "52"},{"readonly": true,"key": "SerialNumber","value": "209521"},{"readonly": true,"key": "PUK","value": "33618075"},{"readonly": false,"key": "WebSocketUrl","value": "ws://192.168.178.74:8887/"},{"readonly": false,"key": "WebSocketUser","value": "Parkplatz"},{"readonly": false,"key": "WebSocketPassword","value": ""},{"readonly": false,"key": "MeterValuesSkipWhenOffline","value": "false"} ]}]
[ocpp ] TRACE 2024/08/23 12:11:39 send Parkplatz: [2,"3839803363","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp ] TRACE 2024/08/23 12:11:39 recv Parkplatz: [3,"3839803363",{"status": "Accepted"}]
[ocpp ] TRACE 2024/08/23 12:11:39 recv Parkplatz: [2,"869807811","BootNotification",{"chargePointSerialNumber": "209521","chargePointVendor": "Wall Box Chargers","meterType": "Internal NON compliant","meterSerialNumber": "","chargePointModel": "PLP1-0-2-3","iccid": "","chargeBoxSerialNumber": "209521","firmwareVersion": "6.4.14","imsi": ""}]
[ocpp ] TRACE 2024/08/23 12:11:39 send Parkplatz: [3,"869807811",{"currentTime":"2024-08-23T10:11:39Z","interval":60,"status":"Accepted"}]
[ocpp ] TRACE 2024/08/23 12:11:39 send Parkplatz: [2,"2730364223","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Power.Active.Import"}]
[ocpp ] TRACE 2024/08/23 12:11:39 recv Parkplatz: [3,"2730364223",{"status": "Accepted"}]
[ocpp ] TRACE 2024/08/23 12:11:39 send Parkplatz: [2,"126169697","ChangeConfiguration",{"key":"MeterValuesSampledData","value":"Energy.Active.Import.Register"}]
[... cut ...] : "Voltage","unit": "V","value": "245.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "249.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A","value": "3"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "2078"} ]} ],"connectorId": 1}]
[ocpp ] TRACE 2024/08/23 12:13:25 send Parkplatz: [3,"1860865937",{}]
```

andig, 2024-08-23:
Really sorry- but we need a common log. Not either/or ;)

JBN164, 2024-08-23:
hopefully now I have the right Log-File

```
[site ] DEBUG 2024/08/23 12:30:43 battery soc: 98%
[site ] DEBUG 2024/08/23 12:30:43 battery power: -455W
[site ] DEBUG 2024/08/23 12:30:43 grid meter: -4007W
[site ] DEBUG 2024/08/23 12:30:43 grid currents: [0 0 0]A
[site ] DEBUG 2024/08/23 12:30:43 site power: -4362W
[lp-1 ] DEBUG 2024/08/23 12:30:43 charge voltages: [0 246 250]V
[lp-1 ] WARN 2024/08/23 12:30:43 invalid phase wiring between charge meter and charger
[lp-1 ] DEBUG 2024/08/23 12:30:43 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/23 12:30:43 charge total import: 1669.955kWh
[lp-1 ] DEBUG 2024/08/23 12:30:43 charger status: A
[lp-1 ] WARN 2024/08/23 12:30:43 charger out of sync: expected disabled, got enabled
[lp-1 ] DEBUG 2024/08/23 12:30:43 charger enable
[ocpp ] TRACE 2024/08/23 12:30:43 send Parkplatz: [2,"424083707","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-23T10:30:43Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}}]
[pulse ] TRACE 2024/08/23 12:30:44 {"id":"87bfcecb-63c7-4bb9-9c9c-3b97258f4bc1","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":4030,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3013,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2024/08/23 12:30:44 {"id":"41effc30-4b8e-4e79-b027-f039cb7f4458","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":4030,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3013,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[ocpp ] TRACE 2024/08/23 12:30:44 recv Parkplatz: [3,"424083707",{"status": "Accepted"}]
[lp-1 ] DEBUG 2024/08/23 12:30:44 charger disable
[lp-1 ] DEBUG 2024/08/23 12:30:44 wake-up timer: stop
[ocpp ] TRACE 2024/08/23 12:30:47 send Parkplatz: [2,"3367090546","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[pulse ] TRACE 2024/08/23 12:30:47 {"id":"41effc30-4b8e-4e79-b027-f039cb7f4458","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":4018,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3061,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2024/08/23 12:30:47 {"id":"87bfcecb-63c7-4bb9-9c9c-3b97258f4bc1","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":4018,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3061,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[ocpp ] TRACE 2024/08/23 12:30:48 recv Parkplatz: [3,"3367090546",{"status": "Accepted"}]
[ocpp ] TRACE 2024/08/23 12:30:48 recv Parkplatz: [2,"1735694177","MeterValues",{"meterValue": [{"timestamp": "2024-08-23T10:30:47Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "1669955.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "246.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "250.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A","value": "3"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "2078"} ]} ],"connectorId": 1}]
[ocpp ] TRACE 2024/08/23 12:30:48 send Parkplatz: [3,"1735694177",{}]
[pulse ] TRACE 2024/08/23 12:30:50 {"id":"87bfcecb-63c7-4bb9-9c9c-3b97258f4bc1","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":3997,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3085,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2024/08/23 12:30:50 {"id":"41effc30-4b8e-4e79-b027-f039cb7f4458","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":3997,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3085,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2024/08/23 12:30:53 {"id":"87bfcecb-63c7-4bb9-9c9c-3b97258f4bc1","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":4016,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3133,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[pulse ] TRACE 2024/08/23 12:30:55 {"id":"41effc30-4b8e-4e79-b027-f039cb7f4458","type":"next","payload":{"data":{"liveMeasurement":{"power":0,"powerProduction":4016,"lastMeterConsumption":4103.263,"lastMeterProduction":8387.3133,"currentL1":null,"currentL2":null,"currentL3":null}}}} server
[... cut ...] . 7"},"Power":{"v":230.8000031,"u":"W","d":1},"Voltage":{"v":28.29999924,"u":"V","d":1},"Current":{"v":8.159999847,"u":"A","d":2},"YieldDay":{"v":485,"u":"Wh","d":0},"YieldTotal":{"v":587.0939941,"u":"kWh","d":3}}},"INV":{"0":{"Temperature":{"v":38.09999847,"u":"°C","d":1}}},"events":1}],"total":{"Power":{"v":438.8999939,"u":"W","d":1},"YieldDay":{"v":977,"u":"Wh","d":0},"YieldTotal":{"v":1172.744995,"u":"kWh","d":2}},"hints":{"time_sync":false,"radio_problem":false,"default_password":true}}
```

andig, 2024-08-23:
Please share the _full_ log. It's impossible to guess what has led to an error when the log starts at the error. Also, please:

> What is the issue?

coderph0x, 2024-08-23:
The forst log says it was hijacking an existent transaction, that's probably why the remote start failed. Disconnect the car, restart evcc with `remotestart: true` set, and then connect the car again. That should do the trick. If not, full log as requested.

andig, 2024-08-24:
@coderph0x which log?

coderph0x, 2024-08-24:
The one you requested (?) 😅

andig, 2024-08-24:
> The one you requested

Da fehlte ja der spannende Teil :(

Die WB ist übrigens nur nur auf L2+3 angeschlossen:

```
[lp-1 ] DEBUG 2024/08/23 12:30:43 charge voltages: [0 246 250]V
[lp-1 ] WARN 2024/08/23 12:30:43 invalid phase wiring between charge meter and charger
```

Wir können hier gerne wieder auf machen wenn es eine konkrete Fehlerbeschreibung gibt.

closed 2024-08-24 by andig

JBN164, 2024-08-24:
> Da fehlte ja der spannende Teil :(
>
> Die WB ist übrigens nur nur auf L2+3 angeschlossen:
>
> ```
> [lp-1 ] DEBUG 2024/08/23 12:30:43 charge voltages: [0 246 250]V
> [lp-1 ] WARN 2024/08/23 12:30:43 invalid [...]

Die Wallbox ist auf allen Phasen angeschlossen. Ohne EVCC läuft sie bereits seit drei jähren ohne Probleme auf 11kW dreiphasig.

I found the solution for my problem that the wallbox didn't start charging.

I have installed the Carlo Ragazzi Smartmeter wich is connected via Bus to the PulsarPlus. With this it is possible to set smart charging in the Wallbox App. 
**But when the Wallbox is connected to OCPP you have to deactivate this function!!**

After this the wallbox start charging may car and EVCC showed it also in UI.

There should be a notice at the EVCC Homepage.

Thanks for your time @andig @coderph0x @premultiply

andig, 2024-08-24:
>Die Wallbox ist auf allen Phasen angeschlossen. Ohne EVCC läuft sie bereits seit drei jähren ohne Probleme auf 11kW dreiphasig.

Das sieht evcc allerdings nicht!

/cc @premultiply 

> There should be a notice at the EVCC Homepage.

Could you please suggest the exact wording and to which device/constellation it applies?

JBN164, 2024-08-28:
Sorry for delay.

I have a Wallbox Pulsar Plus Charger combined with a Wallbox Power Meter (Carlo Gavazzi) to use pv-overage.

> You have to deactivate the pv-overage function in the wallbox app. Otherwise the EVCC is not possible to start charging. 
> There is a notice in the settings of the wallbox app, that this function is not useable in ocpp mode but it is still activated and has to be turned off manually.

[image]

andig, 2024-08-29:
A PR to add this to the template would be great. Thank you!