evcc-io/evcc #15618: Victron Energy Grid Currents

tosog, 2024-08-23:
### Describe the bug

While verifying another bug, I came across a new error, which might be related to PR https://github.com/evcc-io/evcc/pull/15544 

I'm getting these errors: 
`grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'`
and:
`error executing query: sql: Scan error on column index 0, name \"SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh)\": converting NULL to float64 is unsupported`

### Steps to reproduce

error message is displayed as a red warning sign on the gui.
victron site: https://vrm.victronenergy.com/installation/181297/share/8fb1ce3a

### Configuration details

```yaml
meters:
  - name: grid
    type: template
    template: victron-energy
    usage: grid
    host: 192.168.1.137
  - name: pv
    type: template
    template: victron-energy
    usage: pv
    host: 192.168.1.137
  - name: battery
    type: template
    template: victron-energy
    usage: battery
    host: 192.168.1.137
```

### Log details

```text
[site ] DEBUG 2024/08/23 10:04:22 ----
[lp-1 ] DEBUG 2024/08/23 10:04:22 charge power: 0W
[lp-1 ] DEBUG 2024/08/23 10:04:22 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/08/23 10:04:22 charge power: 0W
[lp-2 ] DEBUG 2024/08/23 10:04:22 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/23 10:04:22 pv power: 4592W
[site ] DEBUG 2024/08/23 10:04:22 battery soc: 58%
[site ] DEBUG 2024/08/23 10:04:22 battery power: -1436W
[site ] DEBUG 2024/08/23 10:04:22 grid meter: 6W
[site ] ERROR 2024/08/23 10:04:22 grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'
[site ] DEBUG 2024/08/23 10:04:22 battery has priority at soc 58% (< 75%)
[site ] DEBUG 2024/08/23 10:04:22 site power: 6W
[lp-2 ] DEBUG 2024/08/23 10:04:22 charge total import: 2207.391kWh
[lp-2 ] DEBUG 2024/08/23 10:04:22 charger status: A
```

### What type of operating system are you running?

Linux

### Version

5253d6f9aa0a250f5118a4c18b093f5f34a13ed7

andig, 2024-08-23:
/cc @oscarminus @naltatis seems there are two issues here

oscarminus, 2024-08-23:
@tosog Could you tell me the vrm instance of your grid meter?

tosog, 2024-08-23:
@oscarminus I hope this is what you're looking for:
[image]

I should mention that I'm not using a real fronius or victron smartmeter. I'm simulating a fronius smartmeter using a forked project which can be found here: https://github.com/tosog/venus.dbus-nbsm-smartmeter
The grid data comes from script which connects directly to the smartmeter from the grid operator. The script can be found here: https://github.com/tosog/nbsm

I'm also still on v2.85 of Venus OS (never change a running system..)

oscarminus, 2024-08-23:
@tosog yes, that was the information i was looking for. It seems that the vrm instance may vary between various installations. I will check how we can deal with that. Perhaps we have to implement a new configuration parameter to define the grid meter instance. 

If you are compiling the code from scratch, you can change the lines "id: 30" in /templates/definition/meter/victron-energy.yaml to id: 40 (starting at line 53) as a quick fix.

naltatis, 2024-08-23:
The SQL error is unrelated. E.g. happens when you start evcc with an empty database. See fix: https://github.com/evcc-io/evcc/pull/15620

mdkeil, 2024-08-23:
Für was verwendet evcc die Ströme denn überhaupt?-- ausschlaggebend sollten ja die Leistungswerte sein. Ich frage, denn viele, die ein "simuliertes" GridMeter in VenusOS einbinden werden primär nur die Leistungswerte übergeben, da dies ausreichend ist. Sprich als Rückgabewert wird dann natürlich entsprechend 0A kommen.

closed 2024-08-23 by andig

tosog, 2024-08-26:
@mdkeil ja stimmt, ich simuliere die Stromstärke bei meinem Tool auch, indem ich einfach die Gesamtleistung durch die Spannung dividiere und das dann pro Phase drittle. Mein Netz-Smartmeter gibt mir diese Informationen leider nicht.

Warum evcc das benötigt? Es kann ja sein, dass man zwar unterhalb der Gesamtleistung ist, man aber trotzdem eine Phase überlastet. 

Beispiel: Ich habe zwei Wallboxen an einer 16A Sicherung und möchte einphasig per Überschuss laden, also 2x 16A = 7,4 kW maximal. Das wäre noch unterhalb der maximalen Last von 11 kW, aber sind beide Wallboxen identisch verdrahtet, wäre auf einer Phase 7,4 kW und die Sicherung würde fallen. Verdrahtet man die Wallboxen unterschiedlich (2. Wallbox auf L2 L3 L1), sind zwei Phasen zu je 3,7 kW belastet und es wäre wieder OK. EVCC kann das nicht wissen und benötigt daher die Infos von den grid meters 
Ich weiß nicht, wie evcc damit umgeht, wenn es diese Informationen nicht hat. Ich denke, dass so eine Konstellation nicht möglich sein wird und ich werde wohl weiterhin das Lastmanagement von go-e selbst verwenden müssen - dort kann ich angeben, wie die Leitungen verdrahtet sind.

premultiply, 2024-08-26:
evcc schaut nach dem maximalen Strom auf den drei Phasen als Führungsgröße.
Im Normalfall daher völlig problemlos und einfach, in speziellen Konstallation kann es im Gegensatz zu einer genau definierten (und richtig konfigurierten) Phasenzuordnung dadurch jedoch vorkommen dass evcc übervorsichtig ist und den Strom ggf. stärker reduziert als theoretisch im Optimalfall nötig.

Wir haben diesen Ansatz aber erstmal ganz bewusst so gewählt weil weniger komplex zu implementieren und auch deutlich einfacher zu konfigurieren.
Im Fall der Berücksichtigung der genauen Phasenzuordnung für jeden Ladepunkt ist das Fehlerpotenzial durch fehlerhafte Konfiguration erheblich größer. Noch viel komplizierter wird es dabei wenn auch noch Phasenumschaltung hinzu kommt und bei Fahrzeugen die nicht alle (richtig) konfigurierten Phasen auch wie erwartet nutzen.

tm-107, 2024-08-27:
Ich nutze zum Auslesen des EVU-Zählers Tasmota (https://ottelo.jimdofree.com/stromz%C3%A4hler-auslesen-tasmota/) und ein passendes Script zur Einbindung in Venus-OS (https://github.com/AchimKre/venus.dbus-tasmota-smartmeter).
In letzterem einfach in Zeile 149 die deviceinstance von 40 auf 30 ändern, Script neu starten und schon sind die Fehlermeldungen in evcc weg (und die Ströme werden angezeigt).
Venus-OS scheint sich daran bisher nicht zu stören.

Auch bei allen anderen in Venus-OS per Script integrierbaren "Fremdzählern" sollte die deviceinstance (VRM Instanz) problemlos änderbar sein.

ansysic, 2024-08-28:
After upgrading from 0.130.2 to .6 I now get the error messages regarding currents
`grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'`

Anything I need to configure differently in `evcc.yaml`?

```
meters:
  - type: template
    template: victron-energy
    usage: grid
    host: 192.168.1.179
    port: 502
    minsoc: 20
    maxsoc: 100
    name: grid1
  - type: template
    template: victron-energy
    usage: pv
    host: 192.168.1.179
    port: 502
    minsoc: 20
    maxsoc: 100
    name: pv2
  - type: template
    template: victron-energy
    usage: battery
    host: 192.168.1.179
    port: 502
    minsoc: 20
    maxsoc: 100
    name: battery3
```

[image]

andig, 2024-08-28:
Ja, siehe https://github.com/evcc-io/evcc/pull/15625.

ansysic, 2024-08-28:
Danke! Bedeutet das nicht auch gleichzeitig, dass alle Victron User nun den Fehler bekommen? Oder war nur meine Config "falsch" bzw. unvollständig?

oscarminus, 2024-08-28:
Den Fehler werden alle bekommen, die ihr Grid-Meter auf ID 40 haben. Bei mir z.B. ist der EM24 per RS485 angebunden und läuft auf ID 30. Bei der Implementierung musste man sich für eine ID als default entscheiden.

ansysic, 2024-08-28:
Danke euch allen!

PR #15625 (2024-08-23): Victron Energy: add param for grid-meter VRM instance
fixed #15618