evcc-io/evcc #15619: Ladestrom optimieren für Fiat 500e und andere Autos, die nicht 100% des Wallbox-Stroms abnehmen

seb-lie, 2024-08-23:
# English feature request (short)
Our Fiat 500e seems to not respect the current offered by the charger, but instead use a (fixed?) fraction of that to charge the car's battery system. You can see that in the graph below from 14:00, when the Fiat is plugged in and significantly more energy is "left over". 

The proposed fix is to add some logic (like a configurable factor per car) that will be multiplied into the charger's offered current to counter the car's internal mechanism. For example, we have 4kW available and the car uses 75%, then we tell the charger to offer 4kW / 75% = 5.3kW.

Potential issues: 
- It might get a little nasty with 1phase vs 3phase switching
- A fixed factor for compensation might be too simplistic

# German feature request (detailed)

Wir haben einen Tesla M3 und einen Fiat500e, beide hängen abwechselnd an einem go-e flex charger. Wir laden mit PV-Überschuss ab 85% Batterie (9.6kWh). Der Tesla hält sich anscheinend sehr genau an die Vorgaben des Chargers, und es wird kaum Strom eingespeist oder in die Batterie geladen (schwankend zwischen 100-500W). Nun zum Problem: Wenn der Fiat angeschlossen ist, ist deutlich mehr Strom "übrig" (zw. 600W und 1.7kW, abhängig vom der absoluten PV-Strom)

Ich habe ein aktuelles Dashboard angehängt mit Werten direkt aus evcc. Von 11-13:50 Uhr wurde der Tesla geladen, ab 14:20 dann der Fiat - dieses Auto scheint immer nur einen gewissen Bruchteil des von der angebotenen Wallbox Stromes abzunehmen. Das ist hauptsächlich doof, weil es dadurch langsamer lädt und z.B. bei PV-Überschussladen die Reichweite des Fahrzeugs eingeschränkt wird.

[image]

## Lösungsvorschlag
- Ein Faktor in der Konfiguration, der pro Auto eingestellt werden kann, und der extra "in den Ladepunkt" reingerechnet wird

## Alternativen
- Regeln basierend auf dem _abgenommenen_ Strom der jeweiligen Autos (aber trotzdem bei 16A bzw. eingestellten Höchstgrenzen kappen). Deutlich aufwändiger, aber wohl die eigentliche Lösung.

premultiply, 2024-08-23:
So ein Verhalten ist oft Benutzereinstellungen im Auto geschuldet. Schau da mal bitte genau nach. Leider gibt es da manchmal so Lademodi oder Profile die dann die Leistung beschränken.

evcc regelt das ansonsten in gewissen Grenzen schon aus, da es nicht auf die tatsächliche Ladeleistung sondern nur auf den Überschuss geschaut wird.

andig, 2024-08-23:
> Ein Faktor in der Konfiguration, der pro Auto eingestellt werden kann, und der extra "in den Ladepunkt" reingerechnet wird

evcc regelt auf "Delta am Netzanschlusspunkt". Das sollte auch bei Fiat funktionieren. Falls nein braucht es- wie immer- ein Logfile ;)

seb-lie, 2024-08-23:
Danke für die schnelle Rückmeldung. Ich habe eben evcc auf die neuste Version aktualisiert (von 0.125.0) und werde nach dem nächsten Laden das Logfile hochladen! Ich werde mir auch die Lademodi der Fiats anschauen, ob man da was beeinflußen kann.

seb-lie, 2024-08-23:
Im Fiat kann man anscheinend folgendes machen (aus https://youtu.be/AQ1inUJuGUE?t=37)

[image]

Das Auto ist gerade unterwegs, bin schon gespannt was da eingestellt ist!

sigma415, 2024-08-23:
Die Ladestufen-Einstellung im Fiat kann man auch gut nutzen, um bereits bei geringerem Überschuss zu laden (z.B. in den dunklen Monaten).
Allerdings unterstützt evcc das nicht von Haus aus und man muss evcc etwas austricksen, indem man vorgaukelt, es wären nur 1 oder 2 Phasen angeschlossen.
Siehe: #5779 und https://github.com/evcc-io/evcc/discussions/13699#discussioncomment-9327323
Länglich diskutiert haben wir das gewünschte Feature auch schon hier: #10136

seb-lie, 2024-08-23:
Bei uns war tatsächlich "Ladezustand 3" eingestellt - mit `5` klappt es wie sehnlich erwartet =)

[image]

Logfile-Hochladen spare ich allen. Vielen Dank für die schnelle Unterstützung @premultiply und @andig, und für die Wintertipps @sigma415 !

closed 2024-08-23 by seb-lie