evcc-io/evcc #15633: Elli control/connection via type EEBUS requires prior wallbox restart with version 1.30.2

ilion1, 2024-08-23:
### Describe the bug

When switching to type EEBUS instead of using ELLIPRO template to use the internal meter with version 1.30 the connection to the wallbox is only working when you restart the wallbox before you start EVCC. If you stop EVCC and restart it there is no error message in EVCC but the wallbox web interface shows "Energy Manager Status - Not Connected" and EVCC cannot control the wallbox.

### Steps to reproduce

1. Restart Elli Wallbox
2. Start EVCC --> connection is working
3. Stop EVCC, wait a few minutes and start EVCC again without prior restart of wallbox --> connection is not working
...
[eebus1.log](https://github.com/user-attachments/files/16732929/eebus1.log)
[eebus2.log](https://github.com/user-attachments/files/16732928/eebus2.log)

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
#  eebus: trace

# unique installation id
plant: xxxx

interval: 10s # control cycle interval

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.189.38  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.189.38  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.189.38  
  port: 8080  
  name: battery3

chargers:
- name: wallbox5
  type: eebus 
  ski: D9A2 03D9 E918 F950 EC97 E593 7251 E179 79A3 905E  
  meter: true
  chargedEnergy: false

loadpoints:
- title: Elli Wallbox
  charger: wallbox5
  mode: pv

site:
  title: Zu Hause
  residualPower: 100
  meters:
    grid: grid1
    pv:
[... 231 lines cut ...]
[http  ] TRACE 2024/08/23 21:05:14 GET http://192.168.189.38:8080/api/v1/status
[http  ] TRACE 2024/08/23 21:05:14 {"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":975,"Consumption_W":904,"Fac":50.01599884033203,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-43,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":843,"Production_W":0,"RSOC":86,"RemainingCapacity_Wh":35404,"SystemStatus":"OnGrid","Timestamp":"2024-08-23 21:05:14","USOC":85,"Uac":226,"Ubat":212,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site  ] DEBUG 2024/08/23 21:05:14 pv power: 0W
[http  ] TRACE 2024/08/23 21:05:14 GET http://192.168.189.38:8080/api/v1/status
[http  ] TRACE 2024/08/23 21:05:14 {"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":975,"Consumption_W":904,"Fac":50.01599884033203,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-43,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":843,"Production_W":0,"RSOC":86,"RemainingCapacity_Wh":35404,"SystemStatus":"OnGrid","Timestamp":"2024-08-23 21:05:14","USOC":85,"Uac":226,"Ubat":212,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[http  ] TRACE 2024/08/23 21:05:14 GET http://192.168.189.38:8080/api/v1/status
[http  ] TRACE 2024/08/23 21:05:15 {"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":975,"Consumption_W":904,"Fac":50.01599884033203,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-43,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":843,"Production_W":0,"RSOC":86,"RemainingCapacity_Wh":35404,"SystemStatus":"OnGrid","Timestamp":"2024-08-23 21:05:14","USOC":85,"Uac":226,"Ubat":212,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site  ] DEBUG 2024/08/23 21:05:15 battery soc: 85%
[site  ] DEBUG 2024/08/23 21:05:15 battery power: 843W
[http  ] TRACE 2024/08/23 21:05:15 GET http://192.168.189.38:8080/api/v1/status
[http  ] TRACE 2024/08/23 21:05:15 {"BackupBuffer":"0","BatteryCharging":false,"BatteryDischarging":true,"Consumption_Avg":974,"Consumption_W":904,"Fac":50.018001556396484,"FlowConsumptionBattery":true,"FlowConsumptionGrid":true,"FlowConsumptionProduction":false,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":false,"GridFeedIn_W":-43,"IsSystemInstalled":1,"OperatingMode":"2","Pac_total_W":886,"Production_W":0,"RSOC":86,"RemainingCapacity_Wh":35404,"SystemStatus":"OnGrid","Timestamp":"2024-08-23 21:05:15","USOC":85,"Uac":226,"Ubat":212,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site  ] DEBUG 2024/08/23 21:05:15 grid meter: 43W
[site  ] DEBUG 2024/08/23 21:05:15 site power: 986W
[lp-1  ] DEBUG 2024/08/23 21:05:15 charger status: A
[eebus ] DEBUG 2024/08/23 21:05:19 mdns: stop announcement
[eebus ] DEBUG 2024/08/23 21:05:19 ski d9a203d9e918f950ec97e5937251e17979a3905e disconnected
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.2

DerAndereAndi, 2024-08-24:
The log shows now error in evcc or EEBUS. It shows that the wallbox gets disconnected early in the communication. This may be due to the wallbox EEBUS software often crashing on the first connection attempt and then restarts its service. You also see this in the web interface showing lots of errors in the energy management view. After a while it is restarted and a new connection attempt will be done and often succeeds.

The only difference to other logs where this behaviour is seen is, that no log data is shown about the wallbox software stopping the connection because of this behaviour. But the last sent message is identical.

The log is to short to see what happens in the next few minutes. Please let it run for a few minutes and wait what happens.

There is no error in the data being sent by evcc and the connection is not stopped by evcc.

DerAndereAndi, 2024-08-24:
One addition: in one log file the following error is seen:

```
[eebus ] ERROR 2024/08/23 21:08:50 websocket server error: http: Server closed
```

This hints at the local web server being stopped. Are you running another service on the same port?

ilion1, 2024-08-24:
Thanks for looking into the topic although you advised to use an external meter. I thought that the initial startup phase of EVCC would be sufficient to identify the difference because the wallbox shows status connected a few seconds after EVCC starts (eebus1.log) or will not connect (eebus2.log). Attached is now a longer log of more than 30 minutes including also 10 minutes of charging without EVCC being able to control the wallbox. Maybe there is more information. For now I will revert back to version 0.128.4 which is working fine.

Sequence of changes:
1. Start of EVCC at 8:57
2. Wallbox idle for more than 20 minutes but status of wallbox stays on "not connected"
3. Plugging in of car at 9:21
4. EVCC shows car connected but no charging
5. Wallbox is charging at 6A with 3P (fallback behaviour without active energy manager control)
6. EVCC is logging "charge power: not available" every 10 seconds during charging
7. Unplugging car at 9:31
8. Stopping EVCC at 9:32:58

Detailed log file [eebus3.zip](https://github.com/user-attachments/files/16735837/eebus3.zip)

DerAndereAndi, 2024-08-24:
The evcc2.log file shows that evcc IS connected to the Wallbox! Sorry, but your claim of the opposite is wrong. If there would be no connection, you wouldn't see eny `[eebus ]` entry containing `Send: ...` or  `Recv: ...` messages. But they do just perfectly fine.

Now to the new log file eebus3.log:

- 8:57:54: A connection is initiated and successfully established, including initial message exchange. Everything looks fine -> the message in the wallbox web interface is wrong
- 09:22:02. An EV is connected and all messages required are exchanged
- 09:22:05: evcc shows `charger status: B` and `car connected`
- 09:22:05: the wallbox reports that the EV does NOT charge, there are no measurements provided, also there are no current limits provided -> see the discussion containing the known issues. This is a bug in the wallbox!
- 09:22:05: evcc sends a charging limit of 6A and the wallbox acknowledges it
- The wallbox does NOT send any updated measurements -> This is a bug in the wallbox!
- The log continuously reports `charge power: not available` -> Because the wallbox does NOT provide any measurements -> known bug in the wallbox

There is NO bug in evcc or the EEBUS code, this behaviour is known and documented as a set of bugs in the wallbox. Solution for some of the bugs -> Use an external meter as documented since 2 years.

ilion1, 2024-08-24:
All good and fine with me. I am not claiming an error in the code but want to provide feedback in case it can help. I saw the send/receive messages and also that EVCC recognizes the car so there must be communication ongoing. Still the web interface of the wallbox states "not connected". I have no idea what triggers this status even though communication is established. This may explain why the wallbox does not send any measurements and a bug in the wallbox software as described before.

andig, 2024-08-24:
> Still the web interface of the wallbox states "not connected". I have no idea what triggers this status even though communication is established. This may explain why the wallbox does not send any measurements and may be a bug in the wallbox software as described before.

Since this is the evcc repo (and not the Elli support) and we know that Elli is buggy seems this is out of scope here. Thank you.

For the original issue

> and EVCC cannot control the wallbox.

There is no indication of that?

ilion1, 2024-08-24:
Yes, can be closed.

closed 2024-08-24 by andig