evcc-io/evcc #15635: error executing query: sql: Scan error on column index 0

TobiasHuber1980, 2024-08-24:
### Describe the bug

Error siehe unten, hängt vielleicht mit https://github.com/evcc-io/evcc/commit/4a49b04819dad567613e43dcb2a97ef8b969a69a zusammen.

### Steps to reproduce

1. ./.
2. ./.
3. ./.
...

### Configuration details

```yaml
Fahrezeuge im UI konfiguriert.
```

### Log details

```text
[stats ] ERROR 2024/08/24 08:41:05 error executing query: sql: Scan error on column index 0, name "SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh)": converting NULL to float64 is unsupported
[stats ] ERROR 2024/08/24 08:41:05 error executing query: sql: Scan error on column index 0, name "SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh)": converting NULL to float64 is unsupported
[stats ] ERROR 2024/08/24 08:41:05 error executing query: sql: Scan error on column index 0, name "SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh)": converting NULL to float64 is unsupported
[stats ] ERROR 2024/08/24 08:41:05 error executing query: sql: Scan error on column index 0, name "SUM(charged_kwh * co2_per_kwh) / SUM(charged_kwh)": converting NULL to float64 is unsupported
```

### What type of operating system are you running?

Linux

### Version

0.130.2 (5253d6f9)

StefanSchoof, 2024-08-24:
Ich glaube das wurde mit https://github.com/evcc-io/evcc/pull/15620 gefixt

andig, 2024-08-24:
Sollte im master behoben sein.

closed 2024-08-24 by andig