evcc-io/evcc #15639: 0.130.0-0.130.2 keine Grid, Pv Daten Huawei sun 2000

a0377016, 2024-08-24:
### Describe the bug

Seit dem Umstieg von 0.129.0 auf 0.130.0 oder höher werden keine Daten vom Smartmeter und WR mehr empfangen. timeout.
Retour auf die 0.129.0 Version läuft alles perfekt wie gewohnt!

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Last login: Wed Aug 21 11:18:33 on ttys000
Markuss-iMac-3:~ markus$ ssh admin@192.168.1.37
admin@192.168.1.37's password: 
Linux Rasppi4 6.6.31+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.31-1+rpt1 (2024-05-29) aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Wed Aug 21 11:18:41 2024 from 192.168.1.63
admin@Rasppi4:~ $ sudo apt update
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://deb.debian.org/debian-security bookworm-security InRelease        
Get:3 http://deb.debian.org/debian bookworm-updates InRelease [55.4 kB]        
Get:4 https://dl.cloudsmith.io/public/evcc/stable/deb/debian bookworm InRelease [5,127 B]
Get:5 http://archive.raspberrypi.com/debian bookworm InRelease [39.0 kB]     
Get:6 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages [451 kB]
Get:7 http://archive.raspberrypi.com/debian bookworm/main armhf Packages [453 kB]
Fetched 1,004 kB in 2s (567 kB/s) 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
51 packages can be upgraded. Run 'apt list --upgradable' to see them.
admin@Rasppi4:~ $ sudo apt --only-upgrade install -y evcc
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  gtk2-engines-clearlookspix libcamera-apps libcamera0.1 libcamera0.2
  libpisp0.0.1 libssl1.1 libwpe-1.0-1 libwpebackend-fdo-1.0-1
  linux-headers-6.1.0-rpi4-common-rpi linux-headers-6.1.0-rpi4-rpi-2712
  linux-headers-6.1.0-rpi4-rpi-v8 linux-headers-6.1.0-rpi6-common-rpi
  linux-headers-6.1.0-rpi6-rpi-2712 linux-headers-6.1.0-rpi6-rpi-v8
  linux-headers-6.1.0-rpi7-common-rpi linux-headers-6.1.0-rpi7-rpi-2712
  linux-headers-6.1.0-rpi7-rpi-v8 linux-image-6.1.0-rpi4-rpi-2712
  linux-image-6.1.0-rpi4-rpi-v8 linux-image-6.1.0-rpi6-rpi-2712
  linux-image-6.1.0-rpi6-rpi-v8 linux-image-6.1.0-rpi7-rpi-2712
  linux-image-6.1.0-rpi7-rpi-v8
Use 'sudo apt autoremove' to remove them.
The following packages will be upgraded:
  evcc
1 upgraded, 0 newly installed, 0 to remove and 50 not upgraded.
Need to get 22.6 MB of archives.
After this operation, 1,573 kB disk space will be freed.
Get:1 https://dl.cloudsmith.io/public/evcc/stable/deb/debian bookworm/main arm64 evcc arm64 0.130.2 [22.6 MB]
Fetched 22.6 MB in 3s (7,864 kB/s)
apt-listchanges: Reading changelogs...
(Reading database ... 238536 files and directories currently installed.)
Preparing to unpack .../evcc_0.130.2_arm64.deb ...
Unpacking evcc (0.130.2) over (0.129.0) ...
Setting up evcc (0.130.2) ...
admin@Rasppi4:~ $ sudo reboot

[... 137 lines cut ...]
[site ] DEBUG 2024/08/24 11:10:14 pv power: 0W
[modbus] TRACE 2024/08/24 11:10:15 modbus: send 01 03 90 f9 00 02 39 3a
[site ] ERROR 2024/08/24 11:10:20 grid meter: read failed: read tcp 192.168.1.37:37784->192.168.1.28:502: i/o timeout
[site ] DEBUG 2024/08/24 11:10:25 ----
[http ] TRACE 2024/08/24 11:10:25 GET http://192.168.1.12/status
[http ] TRACE 2024/08/24 11:10:26 {"wifi_sta":{"connected":true,"ssid":"Haselberg4","ip":"192.168.1.12","rssi":-88},"cloud":{"enabled":true,"connected":true},"mqtt":{"connected":false},"time":"11:10","unixtime":1724490626,"serial":1079,"has_update":false,"mac":"485519DB5EB6","cfg_changed_cnt":1,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"http"}],"emeters":[{"power":0.00,"pf":0.03,"current":0.01,"voltage":236.59,"is_valid":true,"total":111494.9,"total_returned":0.0},{"power":0.00,"pf":0.02,"current":0.01,"voltage":232.79,"is_valid":true,"total":109733.3,"total_returned":0.0},{"power":0.00,"pf":-0.02,"current":0.01,"voltage":235.19,"is_valid":true,"total":111167.7,"total_returned":0.0}],"total_power":0.00,"emeter_n":{"current":0.00,"ixsum":0.02,"mismatch":false,"is_valid":false},"fs_mounted":true,"v_data":1,"ct_calst":0,"update":{"status":"idle","has_update":false,"new_version":"20230913-114244/v1.14.0-gcb84623","old_version":"20230913-114244/v1.14.0-gcb84623","beta_version":"20231107-165007/v1.14.1-rc1-g0617c15"},"ram_total":49920,"ram_free":31680,"fs_size":233681,"fs_free":154616,"uptime":1281159}
[lp-1 ] DEBUG 2024/08/24 11:10:26 charge power: 0W
[lp-2 ] DEBUG 2024/08/24 11:10:26 charge power: 0W
[lp-3 ] DEBUG 2024/08/24 11:10:26 charge power: 0W
[shelly] TRACE 2024/08/24 11:10:26 POST http://192.168.1.13/rpc/Shelly.GetStatus
[shelly] TRACE 2024/08/24 11:10:26 {"id":0,"on":false,"src":"evcc","method":"Shelly.GetStatus"} -- {"ble":{},"cloud":{"connected":true},"input:0":{"id":0,"state":false},"mqtt":{"connected":true},"switch:0":{"id":0, "source":"init", "output":true, "apower":2.5, "voltage":235.9, "current":0.000, "aenergy":{"total":9667.807,"by_minute":[41.289,41.289,41.289],"minute_ts":1724491020},"temperature":{"tC":41.7, "tF":107.1}},"sys":{"mac":"D4D4DA7E0A58","restart_required":false,"time":"11:17","unixtime":1724491065,"uptime":402673,"ram_size":253592,"ram_free":136688,"fs_size":393216,"fs_free":110592,"cfg_rev":30,"kvs_rev":0,"schedule_rev":0,"webhook_rev":0,"available_updates":{},"reset_reason":3},"wifi":{"sta_ip":"192.168.1.13","status":"got ip","ssid":"Haselberg4","rssi":-72},"ws":{"connected":false}}
[lp-4 ] DEBUG 2024/08/24 11:10:26 charge power: 0W
[modbus] TRACE 2024/08/24 11:10:27 modbus: send 01 03 7d 40 00 02 dd b3
[site ] ERROR 2024/08/24 11:10:32 pv 1 power: read failed: read tcp 192.168.1.37:52638->192.168.1.28:502: i/o timeout
[modbus] TRACE 2024/08/24 11:10:33 modbus: send 02 03 7d 40 00 02 dd 80
[site ] ERROR 2024/08/24 11:10:38 pv 2 power: read failed: read tcp 192.168.1.37:52652->192.168.1.28:502: i/o timeout
```

### What type of operating system are you running?

Linux

### Version

0.130.0

a0377016, 2024-08-24:
interval: 15s # control cycle interval

meters:

#Netz Smart Meter
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: 192.168.1.28  
  port: 502  
  usage: grid  
  storageunit: 1  
  modbus: tcpip  
  timeout: 5s  
  name: grid1

#PV 1
- type: template
  template: huawei-sun2000-rs485 
  id: 1  
  host: 192.168.1.28  
  port: 502  
  usage: pv  
  modbus: rs485tcpip  
  #storageunit: 1  
  name: pv1

#PV 2
- type: template
  template: huawei-sun2000-rs485 
  id: 2  
  host: 192.168.1.28  
  port: 502  
  usage: pv  
  modbus: rs485tcpip  
  #storageunit: 1  
  name: pv2

#PV 3
- type: template
  template: huawei-sun2000-rs485 
  id: 3  
  host: 192.168.1.28  
  port: 502  
  usage: pv  
  modbus: rs485tcpip  
  name: pv3

#Leistungsmessung Boiler 1 6KW
- name: 3EMBoiler1
  type: custom
  power:
    source: http
    uri: http://192.168.1.12/status
    jq: .total_power  
 
########################################################
chargers:
- type: template
  template: shelly 
  host: 192.168.1.12  
  user: a0377016@gmail.com  
  password: '06906788'  
  channel: 0  
  standbypower: 15  
  heating: true  
  icon: waterheater  
  name: boiler1
 

- type: template
  template: tapo 
  host: 192.168.1.43  
  user: a0377016@gmail.com 
  password: 'zahna7-typdod-poffYc'  
  standbypower: 15  
  heating: true  
  icon: waterheater  
  name: boiler2
  integrateddevice: true

- type: template
  template: tapo 
  host: 192.168.1.5  
  user: a0377016@gmail.com
  password: 'zahna7-typdod-poffYc'  
  standbypower: 15  
  heating: true  
  icon: waterheater  
  name: teststeckdose
  integrateddevice: true

- type: template
  template: shelly 
  host: 192.168.1.13  
 user: a0377016@gmail.com 
  password: '06906788'  
  channel: 0  
  standbypower: 15  
  integrateddevice: false
  heating: false
  icon: tractor
  name: kompressor

####################################
vehicles:
- name: Markus
  type: custom
  title: Boiler 500L
  icon: waterheater 
  capacity: 60
#  soc:
 #  source: http
  # uri: http://192.168.1.23/rpc/Temperature.GetStatus?id=100
  # jq: .temperature:100.tC
  

####################################
loadpoints:
- title: Boiler 1
  charger: boiler1
  mode: off
  meter: 3EMBoiler1
  vehicle: Markus  
#  soc:
#   source: http
#   uri: http://192.168.1.96/rpc/Switch.GetStatus?id=0
#   jq: .temperature:100.tC
#  onldentify:
#   targetSoC: 74   

- title: Boiler 2
  charger: boiler2
  mode: off

- title: Test_Steckdose
  charger: teststeckdose
  mode: off

- title: Kompressor
  charger: kompressor
  mode: off
 

#######################################################
site:
 - title: Jason
   residualPower: 7000

   meters:
     grid: grid1
     pv:
       - pv1
       - pv2
       - pv3

tariffs:
    currency: EUR
    grid:
      type: fixed
      price: 0.22
    feedin:
      type: fixed
      price: 0.05

#mqtt:
 # broker: 192.168.1.37:1882
 # topic: evcc

andig, 2024-08-24:
Bitte vollständiges, formatiertes Issue. Danke.

Johsnon, 2024-08-24:
Jap, ich hab das gleiche Problem mit meinem Huawei Zeug.
(deshalb bleibe ich noch auf 0.129.0; läuft auf RasPi)

`[site ] ERROR 2024/08/18 18:22:15 pv 1 power: read failed: read tcp 192.168.0.114:50288->192.168.0.192:502: i/o timeout`

Log
```
[main ] INFO 2024/08/18 18:20:12 evcc 0.130.0
[main ] INFO 2024/08/18 18:20:12 using config file: /storage/evcc/evcc.yaml
[db ] INFO 2024/08/18 18:20:13 using sqlite database: /storage/.evcc/evcc.db
[main ] INFO 2024/08/18 18:20:28 listening at :7070
[lp-1 ] WARN 2024/08/18 18:20:33 poll mode '{always 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1 ] DEBUG 2024/08/18 18:20:35 charge total import: 929.348kWh
[lp-2 ] DEBUG 2024/08/18 18:20:35 charge total import: 0.000kWh
[lp-3 ] DEBUG 2024/08/18 18:20:35 charge total import: 0.000kWh
[lp-4 ] DEBUG 2024/08/18 18:20:36 charge total import: 0.000kWh
[site ] WARN 2024/08/18 18:20:36 vehicle 'ID3 Offline' does not support automatic detection
[site ] INFO 2024/08/18 18:20:36 site config:
[site ] INFO 2024/08/18 18:20:36 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/08/18 18:20:36 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/08/18 18:20:36 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/18 18:20:36 pv 2: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/18 18:20:36 pv 3: power ✓ energy ✓ currents ✗
[site ] INFO 2024/08/18 18:20:36 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/08/18 18:20:36 vehicles:
[site ] INFO 2024/08/18 18:20:36 vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site ] INFO 2024/08/18 18:20:36 vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1 ] INFO 2024/08/18 18:20:36 loadpoint 1:
[lp-1 ] INFO 2024/08/18 18:20:36 mode: pv
[lp-1 ] INFO 2024/08/18 18:20:36 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2024/08/18 18:20:36 meters: charge ✓
[lp-1 ] INFO 2024/08/18 18:20:36 charge: power ✓ energy ✓ currents ✓
[lp-2 ] INFO 2024/08/18 18:20:36 loadpoint 2:
[lp-2 ] INFO 2024/08/18 18:20:36 mode: off
[lp-2 ] INFO 2024/08/18 18:20:36 charger: power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2 ] INFO 2024/08/18 18:20:36 meters: charge ✓
[lp-2 ] INFO 2024/08/18 18:20:36 charge: power ✓ energy ✓ currents ✗
[lp-3 ] INFO 2024/08/18 18:20:36 loadpoint 3:
[lp-3 ] INFO 2024/08/18 18:20:36 mode: off
[lp-3 ] INFO 2024/08/18 18:20:36 charger: power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-3 ] INFO 2024/08/18 18:20:36 meters: charge ✓
[lp-3 ] INFO 2024/08/18 18:20:36 charge: power ✓ energy ✓ currents ✗
[lp-4 ] INFO 2024/08/18 18:20:36 loadpoint 4:
[lp-4 ] INFO 2024/08/18 18:20:36 mode: off
[lp-4 ] INFO 2024/08/18 18:20:36 charger: power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-4 ] INFO 2024/08/18 18:20:36 meters: charge ✓
[lp-4 ] INFO 2024/08/18 18:20:36 charge: power ✓ energy ✓ currents ✗
[site ] DEBUG 2024/08/18 18:20:36 set buffer soc: 85
[site ] DEBUG 2024/08/18 18:20:36 set buffer start soc: 100
[site ] DEBUG 2024/08/18 18:20:36 set priority soc: 65
[site ] DEBUG 2024/08/18 18:20:36 set battery discharge control: false
[lp-1 ] DEBUG 2024/08/18 18:20:36 set smart cost limit: 0.1
[lp-1 ] DEBUG 2024/08/18 18:20:36 phase timer inactive
[lp-1 ] DEBUG 2024/08/18 18:20:36 pv timer inactive
[lp-1 ] INFO 2024/08/18 18:20:36 vehicle updated: unknown -> ID3
[lp-2 ] DEBUG 2024/08/18 18:20:37 set smart cost limit: 0.1
[lp-2 ] DEBUG 2024/08/18 18:20:37 phase timer inactive
[lp-2 ] DEBUG 2024/08/18 18:20:37 pv timer inactive
[lp-3 ] DEBUG 2024/08/18 18:20:37 set smart cost limit: 0.1
[lp-3 ] DEBUG 2024/08/18 18:20:37 phase timer inactive
[lp-3 ] DEBUG 2024/08/18 18:20:37 pv timer inactive
[lp-4 ] DEBUG 2024/08/18 18:20:37 set smart cost limit: 0.1
[lp-4 ] DEBUG 2024/08/18 18:20:37 phase timer inactive
[lp-4 ] DEBUG 2024/08/18 18:20:37 pv timer inactive
[site ] WARN 2024/08/18 18:20:38 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/08/18 18:20:38 ----
[lp-1 ] DEBUG 2024/08/18 18:20:39 charge power: 0W
[lp-1 ] DEBUG 2024/08/18 18:20:39 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2024/08/18 18:20:39 charge power: 0W
[lp-3 ] DEBUG 2024/08/18 18:20:39 charge power: 0W
[... 198 lines cut ...]
- title: 2️⃣Boiler
  charger: shellypro3_ch3
  #mode: pv
  priority: 4 # Je höher der Wert, desto größer die Priorität. Regelt NICHT die Einschaltreihenfolge!
  guardduration: 5s #(default:5m) Zeitlicher Mindestabstand zwischen 2 Schaltvorgängen.
  enable:
    threshold: -3000 #(default:0) Positiv=Netzbezug, Negativ=Export. "0": 'mincurrent' muss als Export erreicht sein.
    delay: 240s #(default:1m) Wie lange muss Einschaltschwelle erfüllt sein
  disable:
    threshold: -500 #(default:0) maximum import power (W)
    delay: 5s #(default:3m)

site:
  title: Zuhause
  meters:
    grid: grid_huawei
   #grid: grid_tibberpulse  # benötigt Internet
    pv:
    - pv_ost
    - pv_west
    - pv_garten
    battery:
    - battery1
  residualPower: 100 # in Watt; Positiver Wert: Verbleibende Netzeinspeiseleistung
```

Johsnon, 2024-08-25:
siehe https://github.com/evcc-io/evcc/discussions/15627

andig, 2024-08-25:
Könnte jemand zum Testen ein VPN für [info@evcc.io](mailto:info@evcc.io) einrichten (Fritzbox, Wireguard oder SSH)?

a0377016, 2024-08-27:
ich check mal ab ob eine vpn bei mir möglich ist

andig, 2024-08-30:
Let's reopen when VPN is established.

closed 2024-08-30 by andig

andig, 2024-09-01:
Resolution: the configuration contains a mix of Modbus TCP and RTU over TCP for the same host/port combination. That is not possible. This did work in the past only by chance depending on the order of device definitions. I will add a check to highlight such errors:

Johsnon, 2024-09-09:
Jetzt mit der v0.130.8 bekomme ich diesen **ERROR**  😢 
```log
[main ] INFO 2024/09/09 21:50:26 evcc 0.130.8
[main ] INFO 2024/09/09 21:50:27 using config file: /storage/evcc/evcc.yaml
[db ] INFO 2024/09/09 21:50:27 using sqlite database: /storage/.evcc/evcc.db
[main ] INFO 2024/09/09 21:50:45 listening at :7070
[main ] FATAL 2024/09/09 21:50:46 cannot create meter 'pv_ost': cannot create meter type 'template': cannot create meter type 'custom': power: connection already registered with different protocol: 192.168.0.192:502
[main ] FATAL 2024/09/09 21:50:46 will attempt restart in: 15m0s 
```

Meine yaml config hab ich schon oben gepostet.
In der `evcc.dist.yaml` hat sich im Vergleich nix getan, deshalb dachte ich dass ich nix umstellen muss 🤔

andig, 2024-09-09:
Perfekt. Erklärung s.o.