evcc-io/evcc #15645: Juice Me 3 Charger: charger logic error

JulianOhm, 2024-08-24:
### Describe the bug

since update from 0.129.x to 0.130.2 i got error messages for my charger:

```
vor 12 Sekunden
Parkplatz: charger logic error: current mismatch (got 9.62A measured, expected 6A)
vor 3 Minuten
Parkplatz: charger logic error: current mismatch (got 14.4A measured, expected 10A)
vor 4 Minuten
Parkplatz: charger logic error: current mismatch (got 10.6A measured, expected 6A)
```

### Steps to reproduce

charge my car

### Configuration details

```yaml
chargers:
  - name: juiceme
    type: template
    template: bender
    host: 192.168.178.35

loadpoints:
  - title: Parkplatz
    circuit: main
    charger: juiceme
    mode: pv
    enable:
      threshold: 0
      delay: 2m
    disable:
      threshold: 200
      delay: 5m
    guardduration: 15m
    priority: 10
```

### Log details

```text
Aug 24 13:14:27 evcc systemd[1]: Stopped evcc.service - evcc.
Aug 24 13:14:27 evcc systemd[1]: Started evcc.service - evcc.
Aug 24 13:14:28 evcc evcc[92885]: [main  ] INFO 2024/08/24 13:14:28 evcc 0.130.2
Aug 24 13:14:28 evcc evcc[92885]: [main  ] INFO 2024/08/24 13:14:28 using config file: /etc/evcc.yaml
Aug 24 13:14:28 evcc evcc[92885]: [db    ] INFO 2024/08/24 13:14:28 using sqlite database: /var/lib/evcc/evcc.db
Aug 24 13:14:28 evcc evcc[92885]: [main  ] INFO 2024/08/24 13:14:28 listening at :7070
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30 site config:
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30   meters:      grid ✓ pv ✓ battery ✓
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30     grid:      power ✓ energy ✓ currents ✓
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30     pv 1:      power ✓ energy ✓ currents ✗
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30     pv 2:      power ✓ energy ✓ currents ✗
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30     pv 3:      power ✓ energy ✓ currents ✗
Aug 24 13:14:30 evcc evcc[92885]: [site  ] INFO 2024/08/24 13:14:30     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Aug 24 13:14:30 evcc evcc[92885]: [lp-1  ] INFO 2024/08/24 13:14:30 loadpoint 1:
Aug 24 13:14:30 evcc evcc[92885]: [lp-1  ] INFO 2024/08/24 13:14:30   mode:        pv
Aug 24 13:14:30 evcc evcc[92885]: [lp-1  ] INFO 2024/08/24 13:14:30   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Aug 24 13:14:30 evcc evcc[92885]: [lp-1  ] INFO 2024/08/24 13:14:30   meters:      charge ✓
Aug 24 13:14:30 evcc evcc[92885]: [lp-1  ] INFO 2024/08/24 13:14:30     charge:    power ✓ energy ✓ currents ✓
Aug 24 13:14:44 evcc evcc[92885]: [lp-1  ] INFO 2024/08/24 13:14:44 car connected
Aug 24 13:18:08 evcc evcc[92885]: [site  ] ERROR 2024/08/24 13:18:08 pv 3 energy: strconv.ParseFloat: parsing "\ufeff<!DOCTYPE html PUBLIC \"-//W3C//DTD XHTML 1.0 Transitional//EN\" \"http://www.w3.org/TR/xhtml1/DTD/xhtml1-transitional.dtd\">\r\n\r\n<html xmlns=\"http://www.w3.org/1999/xhtml\">\r\n<head>\r\n<style type=\"text/css\">\r\n.in_body\r\n{\r\n\tmargin-top:0px;\r\n\tmargin-left:0px;\r\n\tmargin-right:0px;\r\n\tmargin-bottom:0px;\r\n\tbackground-color:transparent;\r\n}\r\n.div_c\r\n{\r\n\tmargin-left:50px;\r\n\tmargin-right:50px;\r\n\tmargin-top:50px;\r\n\tmargin-bottom:50px;\r\n}\r\n.cu\r\n{\r\n\tcursor:pointer;\r\n}\r\n.b\r\n{\r\n\tfont-weight:bold;\r\n}\r\n.lab_5\r\n{\r\n\tfont-size:16px;\r\n\tcolor:#666666;\r\n\tmargin-left:-20px;\r\n}\r\n.lab_l2\r\n{\r\n\tfloat:left;\r\n\twidth:32%;\r\n\tcolor:#666666;\r\n\tmargin-bottom:-2px;\r\n\tfont-size:14px;\r\n}\r\n.lab_r2\r\n{\r\n\tfloat:left;\r\n\twidth:68%;\r\n\tcolor:#666666;\r\n\ttext-align:right;\r\n\tfont-size:14px;\r\n}\r\n.cl\r\n{\r\n\tclear:left;\r\n}\r\n.line\r\n{\r\n\theight:1px;\r\n\tbackground-color:#666666;\r\n\twidth:100%;\r\n\tmargin-top:5px;\r\n\tmargin-bottom:5px;\r\n}\r\n.sp_5\r\n{\r\n\theight:5px;\r\n\twidth:500px;\r\n}\r\n.sp_20\r\n{\r\n\theight:20px;\r\n\twidth:500px;\r\n}\r\n.label\r\n{\r\n\tfloat:left;\r\n\twidth:50%;\r\n\tcolor:#666666;\r\n\tmargin-bottom:-2px;\r\n\tfont-size:14px;\r\n}\r\n.lab_r\r\n{\r\n\tfloat:left;\r\n\twidth:50%;\r\n\tcolor:#666666;\r\n\ttext-align:right;\r\n\tfont-size:14px;\r\n}\r\n.lab_l\r\n{\r\n\tfloat:left;\r\n\twidth:40%;\r\n\tcolor:#666666;\r\n\tmargin-bottom:-2px;\r\n\tmargin-left:10%;\r\n\tfont-size:14px;\r\n}\r\n.line_l\r\n{\r\n\theight:1px;\r\n\tbackground-color:#666666;\r\n\twidth:450px;\r\n\tmargin-top:5px;\r\n\tmargin-bottom:5px;\r\n\tmargin-left:50px;\r\n}\r\n.sub\r\n{\r\n    display:inline-block;\r\n    width:16px;\r\n    text-align:center;\r\n}\r\n</style>\r\n<script type=\"text/javascript\">\r\nvar height=0;function fileText(id,value){if(document.getElementById(id)){document.getElementById(id).innerHTML=value}}function changeFont(){reCon(\"main_div\").style.fontFamily=window.parent.reFont()}function child_getH(){var nh=document.body.offsetHeight+100;if(nh<500||nh==null){nh=500}if(height!=nh){height=nh;window.parent.child_height(height)}}function reCon(id){return document.getElementById(id)}function ready(){try{window.parent.show_ifr()}catch(e){}child_getH()}function show(v){var c=document.getElementById(v);if(c!=null){c.style.display=\"\"}}function hide(v){var c=document.getElementById(v);if(c!=null){c.style.display=\"none\"}};\r\n</script>\r\n<script type=\"text/javascript\">\r\nvar webdata_sn = \"\";\r\nvar webdata_msvn = \"\";\r\nvar webdata_ssvn = \"\";\r\nvar webdata_pv_type = \"\";\r\nvar webdata_rate_p = \"\";\r\nvar webdata_now_p = \"\";\r\nvar webdata_today_e = \"\";\r\nvar webdata_total_e = \"\";\r\nvar webdata_alarm = \"\";\r\nvar webdata_utime = \"\";\r\nvar cover_mid = \"4063152792\";\r\nvar cover_ver = \"MW3_SSL_5408_1.0B\";\r\nvar cover_wmode = \"APSTA\";\r\nvar cover_ap_ssid = \"AP_4063152792\";\r\nvar cover_ap_ip = \"10.10.100.254\";\r\nvar cover_ap_mac = \"30:EA:E7:C2:64:98\";\r\nvar cover_sta_ssid = \"Chapp\";\r\nvar cover_sta_rssi = \"76%\";\r\nvar cover_sta_ip = \"192.168.178.29\";\r\nvar cover_sta_mac = \"34:EA:E7:C2:64:98\";\r\nvar status_a = \"1\";\r\nvar status_b = \"0\";\r\nvar status_c = \"0\";\r\n\r\nfunction initPageText(){var list=window.parent.reList(\"status\");fileText(\"st1\",list[\"t1\"]);fileText(\"st2\",list[\"t2\"]);fileText(\"st3\",list[\"t3\"]);for(var i=1;i<=27;i++){if(i!=14){fileText(\"tx\"+i,list[i])}}changeFont();child_getH()}function upfold(v){if(document.getElementById(\"up_\"+v+\"_div\").style.display==\"none\"){show(\"up_\"+v+\"_div\");reCon(\"p_\"+v).innerHTML=\"-\"}else{hide(\"up_\"+v+\"_div\");reCon(\"p_\"+v).innerHTML=\"+\"}}function init_main_page(){var on=window.parent.reTip(\"1\");var off=window.parent.reTip(\"2\");document.getElementById(\"cover_mid\").innerHTML=cover_mid;document.getElementById(\"cover_ver\").innerHTML=cover_ver;document.getElementById(\"cover_ap_status\").innerHTML=off;document.getElementById(\"cover_sta_status\").innerHTML=off;if(cover_wmode!=\"STA\"){document.getElementById(\"cover_ap_status\").innerHTML=on;document.getElementById(\"cover_ap_ssid\").innerHTML=cover_ap_ssid;document.getElementById(\"cover_ap_ip\").innerHTML=cover_ap_ip;document.getElementById(\"cover_ap_mac\").innerHTML=cover_ap_mac}if(cover_wmode!=\"AP\"){document.getElementById(\"cover_sta_status\").innerHTML=on;document.getElementById(\"cover_sta_ssid\").innerHTML=cover_sta_ssid;document.getElementById(\"cover_sta_rssi\").innerHTML=cover_sta_rssi;document.getElementById(\"cover_sta_ip\").innerHTML=cover_sta_ip;document.getElementById(\"cover_sta_mac\").innerHTML=cover_sta_mac}if(webdata_sn==\"\"){webdata_sn=\"---\"}fileText(\"webdata_sn\",webdata_sn);if(webdata_msvn==\"\"){webdata_msvn=\"---\"}fileText(\"webdata_msvn\",webdata_msvn);if(webdata_ssvn==\"\"){webdata_ssvn=\"---\"}fileText(\"webdata_ssvn\",webdata_ssvn);if(webdata_pv_type==\"\"){webdata_pv_type=\"---\"}fileText(\"webdata_pv_type\",webdata_pv_type);if(webdata_rate_p==\"\"){webdata_rate_p=\"---\"}fileText(\"webdata_rate_p\",webdata_rate_p+\" W\");if(webdata_now_p==\"\"||webdata_now_p==0){webdata_now_p=\"---\"}fileText(\"webdata_now_p\",webdata_now_p+\" W\");if(webdata_today_e==\"\"){webdata_today_e=\"---\"}fileText(\"webdata_today_e\",webdata_today_e+\" kWh\");if(webdata_total_e==\"\"){webdata_total_e=\"---\"}fileText(\"webdata_total_e\",webdata_total_e+\" kWh\");if(webdata_alarm==\"\"){webdata_alarm=\"---\"}fileText(\"webdata_alarm\",webdata_alarm);if(webdata_utime==\"\"){if(document.getElementById(\"webdata_sn\").innerHTML==\"---\"){webdata_utime=\"---\"}else{webdata_utime=value+window.parent.reTip(\"5 [... cut ...]
Aug 24 13:24:50 evcc evcc[92885]: [lp-1  ] WARN 2024/08/24 13:24:50 charger logic error: current mismatch (got 10.6A measured, expected 6A)
Aug 24 13:25:57 evcc evcc[92885]: [lp-1  ] WARN 2024/08/24 13:25:57 charger logic error: current mismatch (got 14.4A measured, expected 10A)
Aug 24 13:28:21 evcc evcc[92885]: [lp-1  ] WARN 2024/08/24 13:28:21 charger logic error: current mismatch (got 9.62A measured, expected 6A)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.2

andig, 2024-08-24:
Bitte trace log für site, loadpoint und bender.

JulianOhm, 2024-08-24:
mit evcc -l trace site usw komme ich nicht weiter. Wenn es hier einen besseren Weg gibt, bitte sagen, dann mach ich das neu. Ansonsten anbei das komplette log auf Level trace.

[evcc-20240824-141912-trace.log](https://github.com/user-attachments/files/16736601/evcc-20240824-141912-trace.log)

andig, 2024-08-24:
Siehe Doku- kannst du im ui machen

JulianOhm, 2024-08-24:
so hier denn noch einmal

[evcc-20240824-152720-trace.log](https://github.com/user-attachments/files/16736794/evcc-20240824-152720-trace.log)

andig, 2024-08-25:
Sieht das aus wie ein Timingproblem:

```
[lp-1  ] DEBUG 2024/08/24 14:43:32 max charge current: 6A
[site  ] DEBUG 2024/08/24 14:43:32 ----
[lp-1  ] DEBUG 2024/08/24 14:43:32 charge power: 6195W
[lp-1  ] DEBUG 2024/08/24 14:43:32 charge currents: [8.71 8.72 8.64]A
[lp-1  ] DEBUG 2024/08/24 14:43:54 charge voltages: [237 238 238]V
[lp-1  ] DEBUG 2024/08/24 14:43:55 charger status: C
[lp-1  ] WARN 2024/08/24 14:43:55 charger logic error: current mismatch (got 8.72A measured, expected 6A)
```

Es werden 6A vorgegeben und _sofort_ wieder gemessen. So schnell ist die WB nicht. Hier scheint irgendwo eine Totzeit zu sein oder die Updates gehen zu schnell. Ohne vollständige Config nicht zu sagen, sonst sieht alles korrekt aus.

closed 2024-08-25 by andig