evcc-io/evcc #15647: Huawei -> Problem Batteriespeere

Geri-G93, 2024-08-24:
### Describe the bug

Hallo
kurz zum Setup:
1 LoadPoint mit Wallbox
2. Loadpoint Wärmepumpe Pool
Huawei Sun2000 System mit Batteriesteuerung

wenn ich auf schnell Laden schalten sperrt es ja die Batterie, soweit richtig ist auch gewollt
ein Nachteil bzw Bug: die Batterie lädt dann nicht mehr!! genauso bei Tarif bezogenem laden (experimentiere damit zz. rum)
wenn ich unseren Pool zb. auf schnell schalte sperrt es die Batterie, PV seitig kommen 8Kw an. soweit eigentlich richtig, allerdings geht alles ins Netz und die Batterie bleibt leer obwohl genug zum laden vorhanden ist

vorschlag
anstatt 47100 (also Forcible charge/discharge) zu ändern einfach
47077 auf 0 setzen (max DischargePower) dann sollte auch ruhe sein und laden aber noch gehen

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: debug
levels:
  cache: error
  #mqtt: trace
interval: 30s # control cycle interval
# unique installation id
plant: **
sponsortoken: *HAB ICH*
# mqtt message broker
mqtt:
  broker: 127.0.0.1:1889
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: EVCC
  user: EVCC
  password: evcc2023
modbusproxy:
  - port: 503
    uri: 192.168.155.9:502    #Huawei sun2000_1
  #- port: 504
   # uri: 192.168.154.11:502   #Orno Zähler Wallbox
  
meters:
  - name: ChargerMeter
    type: template
    template: orno
    usage: charge
    
    # Modbus TCP Waveshare_1 ORNO
    # RS485 via adapter (Modbus RTU)
    modbus: rs485serial
    id: 1
    device: COM3 # USB-RS485 Adapter Adresse
    baudrate: 9600 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8E1" # Kommunikationsparameter für den Adapter  
    
  - name: GridMeter
    type: template
    template: huawei-dongle-powersensor
    usage: grid
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.155.9 # Hostname
    port: 502 # Port
    timeout: 15s # optional 
    
    
  - name: Sun2000_10KTL
    type: template
    template: huawei-dongle-powersensor
    usage: pv
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.155.9 # Hostname
    port: 502 # Port
    timeout: 15s # optional 
    
  - name: Luna2000
    type: template
    template: huawei-dongle-powersensor
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.155.9 # Hostname
    port: 502 # Port
    timeout: 15s # optional 
    capacity: 15
    
    
  - name: PoolWPMeter
    type: template
    template: shelly-1pm
    usage: charge
    host: 192.168.154.71 # IP-Adresse oder Hostname
    user: admin # Standard-User ist admin (optional)
    password: admin # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    channel: 1 # Meterkanal (1-8) 

    
chargers:

  - name: ABL_EMH01_Isabell
    type: template
    template: abl
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485serial
    id: 3
    device: COM2 # USB-RS485 Adapter Adresse
    baudrate: 38400 # Prüfe die Geräteeinstellungen, typische Werte sind 9600, 19200, 38400, 57600, 115200
    comset: "8E1" # Kommunikationsparameter für den Adapter
 
  - name: Pool_WP
    type: template
    template: shelly
    host: 192.168.154.71 # IP-Adresse oder Hostname
    user: admin # Standard-User ist admin (optional)
    password: admin # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)
    channel: 1 # Schaltkanal (1-8)
    standbypower: 15 # Leistung oberhalb des angegebenen Wertes wird als Ladeleistung gewertet (optional) 
    heating: true
    integrateddevice: true
    icon: waterheater

loadpoints:

  - title: CarPort # display name for UI
    charger: ABL_EMH01_Isabell 
    vehicle: OpelCorsaElektric # Referenz auf Standardfahrzeug
    mode: off # charge mode (off, now, minpv, pv) 
    circuit: carport
    meter: ChargerMeter
    priority: 0
    soc:
      poll:
        mode: connected
        interval: 1m
      estimate: true
  - title: Garten # display name for UI
    #phases: 1
    meter: PoolWPMeter
    charger: Pool_WP 
    #vehicle: Pool # Referenz auf Standardfahrzeug
    mode: off
    disable:
      threshold : -300
      delay: 1m 
    enable:
      threshold : -500
      delay: 1m 

vehicles:

  - name: OpelCorsaElektric
    type: custom
    title: Opel Corsa Elektric 
    capacity: 50
    #vin: VXKUHZKXZP4261182
    soc: # battery soc (%)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/soc
      timeout: 24h # don't accept values older than timeout
      #scale: 0.001 # floating point factor applied to result, e.g. for Wh to kWh conversion
      # ...
      
    status: # optional charge status (A..F)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/status
      timeout: 24h # don't accept values older than timeout
      
    range: # optional electric range (km)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/range
      timeout: 24h # don't accept values older than timeout
      
    odometer: # optional electric range (km)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/odometer
      timeout: 24h # don't accept values older than timeout
      
    climater: # optional electric range (km)
      source: mqtt
      #clientid: EVCC
      user: EVCC
      password: evcc2023
      topic: evcc/corsaSOCFlobZFix/climater
      timeout: 24h # don't accept values older than timeout

   
site:
  - title: Zuhause # display name for UI
    meters:
      grid: GridMeter # grid meter reference
      pv: Sun2000_10KTL
      battery: Luna2000  
    residualPower: 100
    maxGridSupplyWhileBatteryCharging: 50
    #bufferSoc: 80
    #prioritySoc: 66
   # smartCostLimit
    
circuits:
  - name: SLS # eindeutiger Name, wird als Referenz für andere Circuits verwendet
    title: SLS # Anzeige in der UI (später)
    maxCurrent: 35 # 63A (optional)
    #maxPower: 30000 # 30kW (optional)
    meter: GridMeter # optional
  - name: carport
    title: Carport
    meter: ChargerMeter
    maxCurrent: 20
    parent: SLS
    
tariffs:
  currency: EUR # (default EUR)

  grid:
    #type: fixed
    #price: 0.32 # EUR/kWh
    type: custom
    forecast:
      source: mqtt
      user: EVCC
      password: evcc2023
      topic: evccTEST/tarrifs
      timeout: 24h # don't accept values older than timeout
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh
```

### Log details

```text
keine
```

### What type of operating system are you running?

Windows

mucki12, 2024-08-24:
Wurde schon mal diskutiert. Das "Problem" dabei ist, dass der vorherige Entladewert in irgendeine Variable geschrieben werden müsste und dieser Wert dann anschließend wieder reingeschrieben wird, da die Entladeleistungen der unterschiedlichen LUNAS ja nicht immer identisch sind.

mdkeil, 2024-08-24:
ist der Wert für jede LUNA variablel vom Nutzer einstellbar oder ist der Wert statisch in Abhängigkeit von der "Luna-Ausführung"?

mucki12, 2024-08-24:
Sowohl als auch. Für die alte Generation gilt:
1 Akkupack max. 2500W
2-3 Akkupacks max. 5000W

Und diese Werte könnte man noch einmal entsprechend nach unten verändern:
[image]

Geri-G93, 2024-08-25:
vorher den alten Wert speichern?
davon ab, du kannst dem 20.000 entladeleistung sagen, es würde, bei mir mit 15kwH, eine 5000 als Antwort kommen
habs Parallel im iobroker und nodeRed zum probieren

mucki12, 2024-08-25:
> vorher den alten Wert speichern? 

Ja - hatte ich ja auch geschrieben.

> davon ab, du kannst dem 20.000 entladeleistung sagen, es würde, bei mir mit 15kwH, eine 5000 als Antwort kommen habs Parallel im iobroker und nodeRed zum probieren

Interessant - habe das bisher immer nur mit Home Assistant getestet und da kommt lediglich eine Meldung, dass der eingegebene Wert außerhalb des gültigen Bereichs liegt.

premultiply, 2024-08-25:
Das Speichern irgendwelcher Werte geht spätestens dann schief wenn die Batterie bei einem Neustart von evcc nicht im "Normal"-Modus war.
Daher machen wir das grundsätzlich nicht.
Es muss stateless funktionieren.

premultiply, 2024-08-25:
Dann müsste das mal jemand(TM) mit einem "normalen" Modbustool testen.

andig, 2024-09-07:
Backlog label entfernt da wir hier auf externe Tests angewiesen sind.

mucki12, 2024-09-07:
Sorry, habe es mit einem externen Modbus Tool versucht nachzustellen, aber meine Kenntnisse haben leider nicht ausgereicht und alle Tests meinerseits sind schon im Vorfeld gescheitert. Vielleicht testet das ja noch jemand mit mehr Kenntnissen...

andig, 2024-09-14:
> wenn ich auf schnell Laden schalten sperrt es ja die Batterie, soweit richtig ist auch gewollt
ein Nachteil bzw Bug: die Batterie lädt dann nicht meh

Es ist nicht klar geworden, durch welche Einstellungen sich das verhindern ließe. Daher closing mangels konkretem Todo.

closed 2024-09-14 by andig