evcc-io/evcc #15651: Elli Wallbox not reacheable via mdns -> Timeout

mbjw77, 2024-08-24:
### Describe the bug

I have a problem when starting evcc. It seems that my wallbox (ID. Charger Pro) is not reachable.
The problem is displayed as “`mdns: avahi - no IPv4 addresses available`”.

My last successful configuration was with version 0.127.3. and I would like to update to 0.130.2.
Since 0.127.3 I can no longer reach my ID. Charger Pro. Neither with a previously working evcc.yaml nor after replacing the certificate during configuration. I have tried it with every version.

Is there a way to enter a fixed IP address as before? Or another workaround?
I know that this wallbox is defective, but unfortunately I have to use it. That's why I bought an external electricity meter, hoping that the latest optimization for the Elli wallboxes would solve my problem. But unfortunately not.

Tried macOS, Linux and a native home assistant.

### Steps to reproduce

Start evcc

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: trace
  eebus: trace
  core: trace

# unique installation id
plant: ***

interval: 30s # control cycle interval

sponsortoken: ***
telemetry: true

meters:
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.178.240
  port: 1502
  usage: pv
  modbus: tcpip
  name: pv_kostal
- type: template
  template: kostal-plenticore
  id: 71
  host: 192.168.178.240
  port: 1502
  usage: battery
  modbus: tcpip
  name: battery_byd
- type: template
  template: kostal-ksem-inverter
  id: 71
  host: 192.168.178.240
  port: 1502
  usage: grid
  modbus: tcpip
  name: grid_kostal_ksem
- type: template
  template: shelly-pro-3em
  usage: charge
  host: 192.168.178.146
  name: meter_elli
[... 220 lines cut ...]
[sunspec] TRACE 2024/08/24 17:55:17 modbus: recv 00 0b 00 00 00 67 47 03 64 06 e2 02 48 02 4b 02 4e ff fe ff ff ff ff ff ff 09 17 09 0e 09 35 ff ff 10 0b 00 00 13 82 ff fe 10 1f 00 00 01 92 00 00 03 e8 ff fd 01 8a 26 b4 00 00 ff ff 00 00 ff ff 00 00 10 b0 00 00 02 3e 01 cc 01 e7 00 00 ff ff 00 04 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[db    ] TRACE 2024/08/24 17:55:17 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[eebus ] TRACE 2024/08/24 17:55:17 registering ski: 4d168768a168bd4d1515a82b80ad2960d074e718
[sunspec] TRACE 2024/08/24 17:56:16 modbus: closing connection due to idle timeout: 1m0.000091284s
[cache ] TRACE 2024/08/24 17:56:47 fatal: <nil>
[cache ] TRACE 2024/08/24 17:56:47 interval: 30s
[cache ] TRACE 2024/08/24 17:56:47 network: {http evcc.local 7070}
[cache ] TRACE 2024/08/24 17:56:47 mqtt: { evcc   false}
[cache ] TRACE 2024/08/24 17:56:47 influx: {   }
[cache ] TRACE 2024/08/24 17:56:47 sponsor: {mbjw77 2026-07-01 13:08:45 +0000 UTC false}
[cache ] TRACE 2024/08/24 17:56:47 auth: {map[]}
[main  ] FATAL 2024/08/24 17:56:47 cannot create charger 'wallbox_elli': cannot create charger type 'template': cannot create charger type 'eebus': timeout
[main  ] FATAL 2024/08/24 17:56:47 will attempt restart in: 15m0s
[main  ] TRACE 2024/08/24 17:56:47 GET https://api.github.com/repos/evcc-io/evcc/releases/latest
[cache ] TRACE 2024/08/24 17:56:47 version: 0.130.2
[cache ] TRACE 2024/08/24 17:56:47 fatal: {0xc0013bf878}
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.2

andig, 2024-08-24:
Does 130.2 still work for you? Current nightly?

mbjw77, 2024-08-24:
> Does 130.2 still work for you? Current nightly?

0.127.3 is working. Later released Version not.
I will give the latest nightly a try.

andig, 2024-08-24:
/cc @DerAndereAndi avahi issue

DerAndereAndi, 2024-08-24:
Well, v0.127.3 is also using avahi. If the mDNS entry provided by the wallbox, does not provide an IPv4 address, then no connecting can be provided. Usually shortly after another mDNS entry is provided which includes an IPv4.

The log entry is NOT an error, just an  info. Also, it is NOT the case that only the HEMS shall initiate a connection, the wallbox shall does so as well, but as far as I know hardly ever does.

But please provide a log with eebus on TRACE level to see more details.

DerAndereAndi, 2024-08-24:
Could you please run on your linux system where evcc is running the following command and provide the output (may require to install the avahi-utils apt package):

```sh
avahi-browse -d local _ship._tcp --resolve -t
```

mbjw77, 2024-08-24:
> Could you please run on your linux system where evcc is running the following command and provide the output (may require to install the avahi-utils apt package):
> 
> ```shell
> avahi-browse -d local _ship._tcp --resolve -t
> ```
> 
> On macOS you can use the `Discovery` tool, available in the App Store. And there the "folder" `_ship._tcp.` is relevant. Open the entry of the Elli wallbox and provide a screenshot with all the data present.

I can only provide the log while running 0.127.3:
[avahi-0.127.3.log](https://github.com/user-attachments/files/16737774/avahi-0.127.3.log)

0.130.2 now crashes with
```
[eebus ] DEBUG 2024/08/24 21:25:55 starting websocket server on :4712
panic: send on closed channel

goroutine 41 [running]:
github.com/holoplot/go-avahi.(*ServiceBrowser).dispatchSignal(0xc001066540, 0xc0002fca50)
        github.com/holoplot/go-avahi@v1.0.1/service-browser.go:51 +0x459
github.com/holoplot/go-avahi.ServerNew.func1()
        github.com/holoplot/go-avahi@v1.0.1/server.go:58 +0x1ac
created by github.com/holoplot/go-avahi.ServerNew in goroutine 1
        github.com/holoplot/go-avahi@v1.0.1/server.go:47 +0x219
```

mbjw77, 2024-08-24:
> Well, v0.127.3 is also using avahi. If the mDNS entry provided by the wallbox, does not provide an IPv4 address, then no connecting can be provided. Usually shortly after another mDNS entry is provided which includes an IPv4. I never got a connection with IPv6 working, so I do not try.
> 
> The log entry is NOT an error, just an info. Also, it is NOT the case that only the HEMS shall initiate a connection, the wallbox shall does so as well, but as far as I know hardly ever does.
> 
> But please provide a log with eebus on TRACE level to see more details.

Is there more to configure for TRACE-Level than in the config above?

DerAndereAndi, 2024-08-24:
The crash is fixed and will be part of the next release. evcc accidentally included the wrong version of the mdns avahi library.

No, set the log level in the configuration file and run evcc. Then grab the output as you did before.

Could you please provide the `avahi-browse` or Discovery output? evcc doesn't have to run for that, as the data from the Elli wallbox is what's relevant.

DerAndereAndi, 2024-08-24:
Could you please try to set add the network interface `` to the eebus configuration in evcc.yaml? See https://docs.evcc.io/docs/reference/configuration/eebus#interfaces

```yaml
eebus:
  interfaces:
    - eno1
```

This way evcc only uses that network interface for its mDNS announcement and all other IPs are ignored.

The log file shows that the IPv4 address is provided for the wallbox, but somehow not visible to the eebus code. I really would need the evcc log with eebus set to level trace and hope I'll see a hint on why it is not reported.

mbjw77, 2024-08-24:
> Could you please try to set add the network interface `` to the eebus configuration in evcc.yaml? See https://docs.evcc.io/docs/reference/configuration/eebus#interfaces
>
> ```yaml
> eebus: [...]

I retried with the interface 0.130.2 under Linux. eno1 is the right interface (checked ifconfig).
evcc didn't crash, but is not running.
Setting:
```
log: trace
levels:
  eebus: trace
```
[startup.log](https://github.com/user-attachments/files/16737960/startup.log)

sebastianohl, 2024-08-25:
me too. looking forward to the next release...

DerAndereAndi, 2024-08-25:
Could you guys please run this tool for a minute and post the output?

Windows Intel/AMD CPU:
```sh
mdns_windows_amd.exe zeroconf
```

Linux ARM CPU (run both):
```sh
mdns_linux_arm avahi
mdns_linux_arm zeroconf
```

Linux Intel/AMD CPU (run both):
```sh
mdns_linux_amd avahi
mdns_linux_amd zeroconf
```

macOS ARM CPU:
```sh
mdns_macos_arm zeroconf
```

[mdns.zip](https://github.com/user-attachments/files/16740399/mdns.zip)

DerAndereAndi, 2024-08-25:
@sebastianohl some notes:

1. even avahi never receives an IPv4 address which is very weird
2. Can you ping `wallbox-2135A1LQJU.local` or `fe80::cb61:a6b2:cfe3:c78` ?

mDNS is needed anyway, since we need to know the port address.

DerAndereAndi, 2024-08-25:
@mbjw77 Can you try to ping `wallbox-2040A0ZSTF.local` and `fe80::b389:a1d7:80d:53c9` ?

Did you quit the app directly or did you let it run a bit? Especially with `zeroconf` ?

mbjw77, 2024-08-25:
> @mbjw77 Can you try to ping wallbox-2040A0ZSTF.local and fe80::b389:a1d7:80d:53c9 ?

```
user@server:~ $ ping wallbox-2040A0ZSTF.local
PING wallbox-2040A0ZSTF.local (192.168.178.182) 56(84) bytes of data.
64 bytes from 192.168.178.182 (192.168.178.182): icmp_seq=1 ttl=64 time=0.628 ms
...
```
```
user@server:~ $ ping fe80::b389:a1d7:80d:53c9
PING fe80::b389:a1d7:80d:53c9(fe80::b389:a1d7:80d:53c9) 56 data bytes
64 bytes from fe80::b389:a1d7:80d:53c9%eno1: icmp_seq=1 ttl=64 time=1.77 ms
...
```
> Did you quit the app directly or did you let it run a bit? Especially with zeroconf ?

They ran for about 2 minutes. But I can add some more... running 6 minutes each.
[mdns_linux_amd-ahavi.log](https://github.com/user-attachments/files/16741413/mdns_linux_amd-ahavi.log)
[mdns_linux_amd-zeroconf.log](https://github.com/user-attachments/files/16741414/mdns_linux_amd-zeroconf.log)

DerAndereAndi, 2024-08-25:
Thanks. Sometimes fe80 addresses work, but not always depending on the network setup.

Could you guys please give this updated version a try? The zeroconf option should hopefully now also provide an mDNS entry for the wallbox.

[mdns.zip](https://github.com/user-attachments/files/16741452/mdns.zip)

sebastianohl, 2024-08-25:
@DerAndereAndi i can ping the fqdn and the ipv6 address. I let the zeroconf run for a few minutes (<5m)

DerAndereAndi, 2024-08-25:
@sebastianohl thanks. could you retry with the new build and post the outputs ?

sebastianohl, 2024-08-25:
By New buidl you mean 130.4 or master? I can try tomorrow...

DerAndereAndi, 2024-08-25:
The new build of the test tool attached here: https://github.com/evcc-io/evcc/issues/15651#issuecomment-2308945481

mbjw77, 2024-08-25:
> Thanks. Sometimes fe80 addresses work, but not always depending on the network setup.
> 
> Could you guys please give this updated version a try? The zeroconf option should hopefully now also provide an mDNS entry for the wallbox.
> 
> [mdns.zip](https://github.com/user-attachments/files/16741452/mdns.zip)

10 minutes each
[mdns_linux_amd-ahavi.log](https://github.com/user-attachments/files/16741596/mdns_linux_amd-ahavi.log)
[mdns_linux_amd-zeroconf.log](https://github.com/user-attachments/files/16741597/mdns_linux_amd-zeroconf.log)

DerAndereAndi, 2024-08-25:
Sadly zeroconf still doesn't show the wallbox. I can't reproduce this over here. What kind of setup do you guys have? WLAN, LAN, both, ... ?

Is anyone maybe able to provide a wireshark trace of the mDNS packages?

jomach, 2024-08-25:
@DerAndereAndi  I just tried the version 0.126.1 and I still get the timeout but it works. 
[image]

on the network config from evcc I removed the .local and I only enter the IP address. Then I get this: 

```
> $ /Users/jorge/Downloads/evcc_0.126.1_macOS-all/evcc -c evcc.yaml
[main  ] INFO 2024/08/25 22:09:28 evcc 0.126.1
[main  ] INFO 2024/08/25 22:09:28 using config file: evcc.yaml
[main  ] INFO 2024/08/25 22:09:28 starting ui and api at :7070
[db    ] INFO 2024/08/25 22:09:28 using sqlite database: /Users/jorge/.evcc/evcc.db
[mqtt  ] INFO 2024/08/25 22:09:28 connecting evcc-1725378163 at tcp://192.168.1.216:1883
[eebus ] INFO 2024/08/25 22:09:28 Local SKI:  9a46bcb42045efb7a5f449a89482621ca87eb4dc
[eebus ] DEBUG 2024/08/25 22:09:28 mdns: announce
[eebus ] DEBUG 2024/08/25 22:09:28 mdns: using zeroconf
[eebus ] DEBUG 2024/08/25 22:09:28 starting websocket server on :4712
[eebus ] DEBUG 2024/08/25 22:09:28 mdns: start search
[eebus ] DEBUG 2024/08/25 22:09:32 incoming connection request from ae82347bb2de22aa304c83fb1d8f1d94aeb29b4f
[eebus ] DEBUG 2024/08/25 22:09:32 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/08/25 22:09:32 Recv: d:_i:47859_EVBox-Livo-EVB-500-015-503: to NodeManagement read 675 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/08/25 22:09:32 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 reply 2 675 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/08/25 22:09:32 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 result 3 675 ResultData 0
[eebus ] DEBUG 2024/08/25 22:09:32 Recv: d:_i:47859_EVBox-Livo-EVB-500-015-503: to NodeManagement reply 677 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/08/25 22:09:32 Send:  call 4 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/08/25 22:09:32 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/08/25 22:09:32 ev connected
[eebus ] DEBUG 2024/08/25 22:09:32 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 call 6 NodeManagementSubscriptionRequestCall
```

With 0.130.x it does not work and I get: 

```
[eebus ] DEBUG 2024/08/25 22:13:02 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 86 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:04 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 87 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:06 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 88 DeviceDiagnosisHeartbeatData
[lp-1  ] ERROR 2024/08/25 22:13:06 charge power: not available
[lp-1  ] ERROR 2024/08/25 22:13:07 charge currents: not available
[lp-1  ] ERROR 2024/08/25 22:13:07 charge rater: not available
[lp-1  ] ERROR 2024/08/25 22:13:07 charge power: not available
[lp-1  ] ERROR 2024/08/25 22:13:08 charge currents: not available
[eebus ] DEBUG 2024/08/25 22:13:08 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 89 DeviceDiagnosisHeartbeatData
[lp-1  ] ERROR 2024/08/25 22:13:08 charge rater: not available
[eebus ] DEBUG 2024/08/25 22:13:10 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 90 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:12 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 91 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:14 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 92 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:16 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 93 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:18 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 94 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:20 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 95 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:22 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 96 DeviceDiagnosisHeartbeatData
[eebus ] DEBUG 2024/08/25 22:13:24 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 97 DeviceDiagnosisHeartbeatData
[lp-1  ] ERROR 2024/08/25 22:13:25 charge power: not available
[lp-1  ] ERROR 2024/08/25 22:13:26 charge currents: not available
[eebus ] DEBUG 2024/08/25 22:13:26 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 98 DeviceDiagnosisHeartbeatData
[lp-1  ] ERROR 2024/08/25 22:13:26 charge rater: not available
c[eebus ] DEBUG 2024/08/25 22:13:28 Send: d:_i:47859_EVBox-Livo-EVB-500-015-503 notify 99 DeviceDiagnosisHeartbeatData
```

So something has changed between 0.126.x and 0.130.x

DerAndereAndi, 2024-08-25:
@jomach please create a different ticket, you do have a different issue than the one discussed over here. This gets way too confusing for me to track.

mbjw77, 2024-08-26:
> Sadly zeroconf still doesn't show the wallbox. I can't reproduce this over here. What kind of setup do you guys have? WLAN, LAN, both, ... ?

Wallbox, server running evcc, a FritzBox and the Kostal-stuff are directly connected to an unmanaged gigabit switch via LAN.
The Wallbox gets its IP-address via dhcp from the FritzBox.
DNS is a Pi-Hole. Its IP-address is distributed via dhcp.
Server has a static IP-adress and uses the Fritz-Box direct as DNS

> Is anyone maybe able to provide a wireshark trace of the mDNS packages?

I would try to generate one. But I'm not that experienced with Wireshark. Do you have a filter-configuration I could apply?

jomach, 2024-08-26:
@mbjw77 Can you disable IPV6 from your FritzBox? can you run `ifconfig  eno1` ?

mbjw77, 2024-08-26:
> @mbjw77 Can you disable IPV6 from your FritzBox? can you run `ifconfig eno1` ? @DerAndereAndi Looks to me some network issue on the users network

I disabled IPv6 in my FritzBox
- Router Advertisement im LAN aktiv: off
- DNSv6-Server auch über Router Advertisement bekanntgeben: off
- DHCPv6-Server in der FRITZ!Box deaktivieren: Es sind keine anderen DHCPv6-Server im Heimnetz vorhanden.

and tried 0.130.4

works :-)
Is there anything I could change, to get IPv6 running correctly (actually not in use, but maybe in future)?

[startup.log](https://github.com/user-attachments/files/16746222/startup.log)

```
user@server:~ $  ifconfig eno1
eno1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.178.254  netmask 255.255.255.0  broadcast 192.168.178.255
        inet6 2003:dd:af1a:b100:4e52:62ff:fe28:e68f  prefixlen 64  scopeid 0x0<global>
        inet6 fda7:dea9:e24e:0:4e52:62ff:fe28:e68f  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::4e52:62ff:fe28:e68f  prefixlen 64  scopeid 0x20<link>
        ether 4c:52:62:28:e6:8f  txqueuelen 1000  (Ethernet)
        RX packets 2551978769  bytes 2006405165806 (1.8 TiB)
        RX errors 0  dropped 398691  overruns 0  frame 0
        TX packets 2608060934  bytes 3651453488924 (3.3 TiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 16  memory 0xa1200000-a1220000
```

jomach, 2024-08-26:
> inet6

I see that you have ipv6 still active on your ethernet card. You can try to disable it. So the issue is confirmed to be related to mdns and ipv6 ?

sebastianohl, 2024-08-28:
sorry for the delay. here are my logs. 

```
ohli@magadan:~/Downloads$ ./mdns_linux_amd avahi
2024-08-28 20:00:33 DEBUG mdns: announce
2024-08-28 20:00:33 DEBUG mdns: using avahi
2024-08-28 20:00:33 DEBUG mdns: start search
2024-08-28 20:00:33 DEBUG mdns: stop announcement
2024-08-28 20:00:33 TRACE mdns: avahi - process add service: EVCC_HEMS_01 _ship._tcp local tiksi.ohli.de.local 10.42.0.1 4712 map[brand:EVCC id:EVCC-3531636530656434 model:HEMS path:/ship/ register:false ski:ff32ea1788126d3085a51d0ce3c98a5472db1cb3 txtvers:1 type:EnergyManagementSystem]
2024-08-28 20:00:33 DEBUG mdns: new - ski: ff32ea1788126d3085a51d0ce3c98a5472db1cb3 name: EVCC_HEMS_01 brand: EVCC model: HEMS typ: EnergyManagementSystem identifier: EVCC-3531636530656434 register: false host: tiksi.ohli.de.local port: 4712 addresses: [10.42.0.1]
2024-08-28 20:00:33 INFO  Currently known entries:
2024-08-28 20:00:33 INFO    &{EVCC_HEMS_01 ff32ea1788126d3085a51d0ce3c98a5472db1cb3 EVCC-3531636530656434 /ship/ false EVCC EnergyManagementSystem HEMS tiksi.ohli.de.local 4712 [10.42.0.1]}
2024-08-28 20:00:33 TRACE mdns: avahi - process add service: Elli-Wallbox-2135A1LQJU _ship._tcp local wallbox-2135A1LQJU.local fe80::cb61:a6b2:cfe3:c78 4712 map[brand:Elli id:Elli-Wallbox-2135A1LQJU model:Wallbox org.freedesktop.Avahi.cookie:2134517928 path:/ship/ register:false ski:991e0fcb2c17c1e372a95178c6408490279e3859 txtvers:1 type:Wallbox]
2024-08-28 20:00:33 DEBUG mdns: new - ski: 991e0fcb2c17c1e372a95178c6408490279e3859 name: Elli-Wallbox-2135A1LQJU brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-2135A1LQJU register: false host: wallbox-2135A1LQJU.local port: 4712 addresses: [fe80::cb61:a6b2:cfe3:c78]
2024-08-28 20:00:33 INFO  Currently known entries:
2024-08-28 20:00:33 INFO    &{EVCC_HEMS_01 ff32ea1788126d3085a51d0ce3c98a5472db1cb3 EVCC-3531636530656434 /ship/ false EVCC EnergyManagementSystem HEMS tiksi.ohli.de.local 4712 [10.42.0.1]}
2024-08-28 20:00:33 INFO    &{Elli-Wallbox-2135A1LQJU 991e0fcb2c17c1e372a95178c6408490279e3859 Elli-Wallbox-2135A1LQJU /ship/ false Elli Wallbox Wallbox wallbox-2135A1LQJU.local 4712 [fe80::cb61:a6b2:cfe3:c78]}
2024-08-28 20:00:33 TRACE mdns: avahi - process add service: EVCC_HEMS_01 _ship._tcp local tiksi.ohli.de.local 10.42.0.1 4712 map[brand:EVCC id:EVCC-3531636530656434 model:HEMS path:/ship/ register:false ski:ff32ea1788126d3085a51d0ce3c98a5472db1cb3 txtvers:1 type:EnergyManagementSystem]
2024-08-28 20:00:33 TRACE mdns: avahi - process add service: EVCC_HEMS_01 _ship._tcp local tiksi.ohli.de.local 10.42.0.1 4712 map[brand:EVCC id:EVCC-3531636530656434 model:HEMS path:/ship/ register:false ski:ff32ea1788126d3085a51d0ce3c98a5472db1cb3 txtvers:1 type:EnergyManagementSystem]
2024-08-28 20:00:33 TRACE mdns: avahi - process add service: Elli-Wallbox-2135A1LQJU _ship._tcp local wallbox-2135A1LQJU.local fe80::cb61:a6b2:cfe3:c78 4712 map[brand:Elli id:Elli-Wallbox-2135A1LQJU model:Wallbox org.freedesktop.Avahi.cookie:2134517928 path:/ship/ register:false ski:991e0fcb2c17c1e372a95178c6408490279e3859 txtvers:1 type:Wallbox]
2024-08-28 20:00:33 TRACE mdns: avahi - process add service: EVCC_HEMS_01 _ship._tcp local tiksi.ohli.de.local 10.42.0.1 4712 map[brand:EVCC id:EVCC-3531636530656434 model:HEMS path:/ship/ register:false ski:ff32ea1788126d3085a51d0ce3c98a5472db1cb3 txtvers:1 type:EnergyManagementSystem]

```

```
2024-08-28 20:02:28 DEBUG mdns: announce
2024-08-28 20:02:28 DEBUG mdns: using zeroconf
2024-08-28 20:02:28 DEBUG mdns: start search
2024-08-28 20:02:28 DEBUG mdns: stop announcement
2024-08-28 20:02:28 DEBUG mdns: new - ski: ff32ea1788126d3085a51d0ce3c98a5472db1cb3 name: EVCC_HEMS_01 brand: EVCC model: HEMS typ: EnergyManagementSystem identifier: EVCC-3531636530656434 register: false host: tiksi.ohli.de.local. port: 4712 addresses: [192.168.0.3 10.42.0.0 10.42.0.1 2001:4091:a243:827c:da3a:ddff:fe66:a33]
2024-08-28 20:02:28 INFO  Currently known entries:
2024-08-28 20:02:28 INFO    &{EVCC_HEMS_01 ff32ea1788126d3085a51d0ce3c98a5472db1cb3 EVCC-3531636530656434 /ship/ false EVCC EnergyManagementSystem HEMS tiksi.ohli.de.local. 4712 [192.168.0.3 10.42.0.0 10.42.0.1 2001:4091:a243:827c:da3a:ddff:fe66:a33]}
```

also letting it run for some time did not change the logs.

DerAndereAndi, 2024-08-28:
The next nightly build will introduce a few updates:
- IPv6 addresses will be used to establish a connection
- It is again possible to assign a fixed IPv4 address if nothing else works

andig, 2024-08-29:
Has anyone verified the fix? Can this issue be closed?

sebastianohl, 2024-08-29:
I just upgraded my installation to the latest nightly container (sha256:a9dea5584fa4b0990205061fdd9b4575fbc69cd657d73f3bc973e3b7b2eb0407). It is working with an ip in the config. If I remove the IP, it is looking up the announced DNS entry but do not get a reply (and than it is failing). For me the solution with it IP in the config works. So, the ticket can be closed (for my part).

closed 2024-08-29 by andig

mbjw77, 2024-09-01:
Vielen Dank für euren Support :-)