evcc-io/evcc #15671: OCPP: Reinit after BootNotification received

premultiply, 2024-08-25:
Currently we initialize stations (and connectors) only once after initial connection.
If the charger is restarted it will just reconnect but it will never be configured again until evcc is restarted too.
In many cases like simple network interruption this should be fine.

But there are other cases where the station is rebooted (configuration change, power cycle, RCD tripped, firmware update, manual reboot, watchdog reset, ...) but evcc keeps running. This can be especially the case when several stations are connected.

In this case operation may continue with the internal (factory) default settings and may not respond as expected according to the initial configuration. This may lead to unusual und unexpected behavior.

To fix this we need to monitor for `BootNotification` requests and run the initialization routine again for each station and connector where they occured.

andig, 2024-08-25:
This is not OCPP specific. Currently wontfix.

andig, 2024-08-29:
Not currently a topic. For time being might need restart of evcc.

closed 2024-08-29 by andig

premultiply, 2024-08-29:
I am pretty sure that we could simply move the relevant configuration calls out of the charger init procedure itself so it could be recalled each time a BootNotification is received without the need for a complete restart.