evcc-io/evcc #15677: OCPP: Mennekes Smart Charging Station seit 0.130.0 Fehlerhaft

benesolar, 2024-08-25:
### Describe the bug

Hallo,

seit dem Update 0.130.0 startet die Ladung nicht mehr.
Habe auch schon 0.130.3 getestet.

- Anzeige falsche Leistung: 11W
- charge power: not available
- charger out of sync: expected enabled, got disabled

[image]
[evcc-20240825-152823-debug.log](https://github.com/user-attachments/files/16740432/evcc-20240825-152823-debug.log)
[evcc-20240825-152707-ocpp trace.log](https://github.com/user-attachments/files/16740433/evcc-20240825-152707-ocpp.trace.log)

### Steps to reproduce

1. Funktionierende Config mit 0.129.0
2. Update auf 0.130.3
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 5s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# telemetry: true

# log settings
log: error
levels:
  core: error
  lp-1: error
  lp-2: error
  lp-3: error
  cache: error
  db: error
  ocpp: error

# meter definitions
meters:
  - name: grid
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.1.248
    port: 502
  - name: pv1
    type: template
    template: fronius-gen24
    usage: pv
    host: 192.168.1.248
    port: 502
  - name: pv2
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.1.249
    port: 502
  - name: pv3
[... 209 lines cut ...]
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} disconnected of {{.title}}"
      msg: |
        {{.vehicleTitle}} disconnected of {{.title}} nach {{.connectedDuration}}.
        Geladen {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
  services:
    - type: telegram
      token: xxx # bot id
      chats:
      - -xxx # list of chat ids
```

### Log details

```text
Siehe Anhang
```

### What type of operating system are you running?

Linux

### Version

0.130.3

andig, 2024-08-25:
Bitte 0.130.3 probieren. Sollte behoben sein.

benesolar, 2024-08-25:
ich habe bereits 0.130.3 laufen - Fehler tritt auch da auf.

Komlettes Log:
[evcc-20240825-155251-trace.log](https://github.com/user-attachments/files/16740473/evcc-20240825-155251-trace.log)

andig, 2024-08-25:
Dann brauchts bitte ein Log mit site, loadpoint, ocpp auf trace. Sonst nix. Danke.

benesolar, 2024-08-25:
Log mit beiden Anschlüssen der Ladestation:

[evcc-20240825-171151-trace.log](https://github.com/user-attachments/files/16740847/evcc-20240825-171151-trace.log)

andig, 2024-08-25:
Bevor ich da versuche weiter einzusteigen: Du zwingst der Wallbox die Messwerte auf:

    metervalues: Energy.Active.Import.Register

wunderst Dich dann aber, dass Leistung nicht dabei ist...

benesolar, 2024-08-25:
Stand nach dem Update auf 0.130.4
und geänderter Config:
```
  - name: SmartTRechts
    stationid: Webergasse
    connector: 1
    type: template
    template: ocpp
    connecttimeout: 5m
    timeout: 2m
    bootnotification: false
  - name: SmartTLinks
    stationid: Webergasse
    connector: 2
    type: template
    template: ocpp
    connecttimeout: 5m
    timeout: 2m
    bootnotification: false
```

werden keine Fehlermeldungen mehr angezeigt, Ladung funktioniert.

[evcc-20240825-204406-trace.log](https://github.com/user-attachments/files/16741556/evcc-20240825-204406-trace.log)

Bei dem einen Ladepunkt werden im `Aus` und `PV` Modus 12 W Ladeleistung angezeigt.
Hier misst der Zähler den Eigenverbrauch vom Controller der Ladestation.

premultiply, 2024-08-25:
@andig Irgendwas stimmt hier noch nicht mit der Initialisierung bei mehreren Connectoren.
Lass uns da nochmal gemeinsam nach schauen.

@benesolar 
```yaml
  - name: SmartTRechts
    stationid: Webergasse
    connector: 1
    type: template
    template: ocpp
  - name: SmartTLinks
    stationid: Webergasse
    connector: 2
    type: template
    template: ocpp
```
ist mehr als ausreichend.

benesolar, 2024-08-27:
zu früh gefreut, ich habe die Einstellungen oben übernommen und auf 0.130.4+1724669427 aktualisiert:

```
[lp-2 ] ERROR 2024/08/27 12:39:52 charge power: not available
[lp-3 ] ERROR 2024/08/27 12:39:52 charge power: not available 
```

[evcc-20240827-124057-trace.log](https://github.com/user-attachments/files/16760811/evcc-20240827-124057-trace.log)

andig, 2024-08-27:
Ja, sind wir dran...

andig, 2024-08-29:
Test des angehängten PR wäre Klasse. Der ist ein bisschen zu groß, um ihn mal eben so zu mergen.

benesolar, 2024-09-07:
Erst jetzt wieder möglich, Ladestation war defekt.

0.130.7: weiterhin `charge power: not available`

0.130.7+1725675906: cannot create charger 'SmartTLinks': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[evcc-20240907-170440-trace.log](https://github.com/user-attachments/files/16918821/evcc-20240907-170440-trace.log)

andig, 2024-09-15:
@benesolar der timeout sollte jetzt auch behoben sein. Charge power bin ich unsicher, ggf. bräuchte es da nochmal ein Logfile.

benesolar, 2024-09-15:
evcc 0.130.11 (81a31592)
cannot create charger type 'ocpp': timeout

[evcc-20240915-204603-trace.log](https://github.com/user-attachments/files/17007025/evcc-20240915-204603-trace.log)

andig, 2024-09-15:
Deine WB trennt die Verbindung:

    charge point disconnected: Webergasse
    
Wäre eine Frage an den Herstellersupport möglich?

/cc @premultiply

benesolar, 2024-09-15:
Ich starte immer die Ladestation und EVCC gleichzeitig neu, hängt eventuell damit zusammen?
Hier ein neuer Log ohne disconnect.

[evcc-20240915-213704-trace.log](https://github.com/user-attachments/files/17007140/evcc-20240915-213704-trace.log)

Mit 0.130.7 funktioniert es.

Was genau soll ich den Hersteller fragen?

premultiply, 2024-09-15:
> Ich starte immer die Ladestation und EVCC gleichzeitig neu, hängt eventuell damit zusammen?

Mach mal bitte erst die Ladestation aus, starte dann evcc (neu) und dann die Ladestation wieder an.

benesolar, 2024-09-17:
So kann ich es erst am Wochenende machen wenn ich wieder vorort bin.

Inzwischen ein log mit 0.130.7 falls es hilft:
[evcc-20240917-173511-trace.log](https://github.com/user-attachments/files/17030935/evcc-20240917-173511-trace.log)

andig, 2024-09-17:
```
[Webergasse-1] DEBUG 2024/09/15 21:31:29 waiting for chargepoint: 5m0s
[ocpp  ] TRACE 2024/09/15 21:31:29 error dispatching request [341681420, TriggerMessage] to Webergasse: cannot send request 341681420, no client Webergasse exists
[Webergasse-2] DEBUG 2024/09/15 21:31:29 failed triggering StatusNotification: cannot send request 341681420, no client Webergasse exists
```

Das ist maximal merkwürdig. -2 kommt völlig aus dem Nichts, ohne dass auch nur ein Ladepunkt verbunden wäre (@premultiply wo kommt das her?)

Dann fällt auf, dass die Box lower-case Config Keys benutzt, aber auf `ChangeConfiguration` in richtiger Schreibweise reagiert (@premultiply sollten wir das Einlesen case insensitive machen?).

Leider ergibt das Log (zumindest der Start) nicht viel Sinn für mich. Vielleicht liegts tatsächlich am gleichzeitigen Restart.

benesolar, 2024-09-17:
Habe den Bereich in der .py gefunden der für ChangeConfiguration zuständig ist.
.lower steht dort überall dabei

Ich denke ein Zitat daraus ist erlaubt:
`elif ConfigKey.lower() == 'HeartbeatInterval'.lower():`

andig, 2024-09-17:
Das heisst ja nicht, dass das irgendwie ein Standard wäre. Aber ja...

andig, 2024-09-17:
See https://github.com/evcc-io/evcc/pull/16186

premultiply, 2024-09-17:
In der Spec steht es eigentlich ganz klar als fester String mit Groß- und kleinbuchstaben drin.
Aber warum einfach kopieren wenn man wieder irgendeine Sonderlocke in die Firmware bauen kann...

andig, 2024-09-20:
@benesolar könntest Du dem Nightly nochmal eine Chance geben? Bitte erst evcc, dann WB starten damit wir ein sauberes Log bekommen.

benesolar, 2024-09-20:
Hier die Logs mit dem gewünschten Prozedere:

[0.130.7.log](https://github.com/user-attachments/files/17078383/0.130.7.log)
[0.130.11+1726846520.log](https://github.com/user-attachments/files/17078384/0.130.11%2B1726846520.log)

Weiterhin gleicher Fehler.

mfuchs1984, 2024-09-20:
Kann es sein, dass evcc die verfugbaren MeterValuesSampledData nicht verarbeitet, da die wallbox diese in Hochkommata packt?

`{"readonly": false, "key": "metervaluessampleddata", "value": "'Power.Active.Import', 'Energy.Active.Import.Register'"}`

andig, 2024-09-20:
Ähhhh…. Was soll denn der Mist jetzt wieder?! Es ist wirklich unglaublich, welchen Sch*** die sich da zusammen braten. Das lässt sich aber lösen 👍🏻

premultiply, 2024-09-21:
🤦🏻‍♂️🙄

andig, 2024-09-21:
@benesolar @premultiply ich verstehe das Log nicht. Der Anfang fehlt. Ich sehe

```
[Ladestation-1] DEBUG 2024/09/20 18:57:53 waiting for chargepoint: 5m0s
[ocpp  ] TRACE 2024/09/20 18:57:53 error dispatching request [3589246882, TriggerMessage] to Ladestation: cannot send request 3589246882, no client Ladestation exists
[Ladestation-2] DEBUG 2024/09/20 18:57:53 failed triggering StatusNotification: cannot send request 3589246882, no client Ladestation exists
```

Hier spricht evcc schon fröhlich mit der WB. Wo kommt das her wenn noch gar keine verbunden ist??? Das kann nicht sein.

andig, 2024-09-21:
@benesolar ansonsten zeigt das Log keinen Fehler mehr. Was klappt denn jetzt noch nicht?

benesolar, 2024-09-21:
das ist nur im `main` Log drinnen: `cannot create charger type 'ocpp': timeout`

Und dann ist zumindest in der funktionierenden 0.130.7 der Fehler `charge power: not available` auf den Ladepunkten wo kein Ladevorgang läuft.

andig, 2024-09-21:
> das ist nur im main Log drinnen: cannot create charger type 'ocpp': timeout

Ich seh da nix. Was ist das main log?

> Und dann ist zumindest in der funktionierenden 0.130.7 der Fehler

...aber um das gehts ja nun schon seit 5 Releases nicht mehr.

Ich mache mal vorsichtig zu. Gerne nochmal komplettes Log fürs Nightly ab Start falls es noch einen Fehler gibt. https://github.com/evcc-io/evcc/issues/15677#issuecomment-2365099832 erweckt bei mir den Eindruck, als würden da Teile vom Log fehlen, warum auch immer.

benesolar, 2024-09-21:
evcc 0.130.11 (4f51534b) 
```
[main ] FATAL 2024/09/21 13:12:57 cannot create charger 'SmartTLinks': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main ] FATAL 2024/09/21 13:12:57 will attempt restart in: 15m0s 
```

neue Log mit der gewohnten Vorgangsweise (Ladestation stromlos, EVCC neustart, Stromversorgung Ladestation ein):
[evcc-20240921-131707-trace.log](https://github.com/user-attachments/files/17083842/evcc-20240921-131707-trace.log)
deaktiviert sind nur `db`, `modbus` und `sunspec`

andig, 2024-09-21:
Es bleibt leider dabei:

```
[main  ] INFO 2024/09/21 13:09:24 evcc 0.130.11 (4f51534b)
[main  ] INFO 2024/09/21 13:09:24 using config file: /etc/evcc.yaml
[main  ] INFO 2024/09/21 13:09:24 listening at :7070
[Ladestation-2] DEBUG 2024/09/21 13:09:32 waiting for chargepoint: 5m0s
[ocpp  ] TRACE 2024/09/21 13:09:32 error dispatching request [3813203135, TriggerMessage] to Ladestation: cannot send request 3813203135, no client Ladestation exists
[Ladestation-1] DEBUG 2024/09/21 13:09:32 failed triggering StatusNotification: cannot send request 3813203135, no client Ladestation exists
```

In deiner ganzen Config taucht die Vokabel `Ladestation` nicht auf. Woher kommt das also???

benesolar, 2024-09-21:
Sorry hab es in der EVCC Config auf `Ladestation` umbenannt.
 ```
 - name: SmartTLinks
    stationid: Ladestation
    connector: 1
    type: template
    template: ocpp
  - name: SmartTRechts
    stationid: Ladestation
    connector: 2
    type: template
    template: ocpp
```

Die Ladestation braucht offenbar eine gute Minute bis sie nach dem Start erreichbar ist.
Und ich auch ca. bis ich beim Verteiler bin und den Leitungsschutzschalter wieder eingeschalten habe.

andig, 2024-09-21:
> Die Ladestation braucht offenbar eine gute Minute bis sie nach dem Start erreichbar ist.

Ok. Ich verstehe trotzdem nicht, wie es sein kann dass evcc da hin schickt bevor sich überhaupt etwas verbunden hat. Entweder fehlt uns ein Teil des Logs oder ich weiss es auch nicht :(

andig, 2024-09-22:
In 30min gibts nochmal neues nightly. Mir ist weiterhin unklar, warum vor Setup kommuniziert werden kann. Es scheint ein Teil vom Log zu fehlen.

benesolar, 2024-09-22:
[evcc-20240922-202447-trace.log](https://github.com/user-attachments/files/17090773/evcc-20240922-202447-trace.log)
`[main ] FATAL 2024/09/22 20:22:42 cannot create charger 'SmartTLinks': cannot create charger type 'template': cannot create charger type 'ocpp': timeout`

andig, 2024-09-22:
Also das Log zeigt den Fehler nicht?

benesolar, 2024-09-22:
dafür das:
```
[ocpp  ] TRACE 2024/09/22 20:22:06 received JSON message from Ladestation: [2, "947e0a64-790f-11ef-8f09-000a1484c32e", "BootNotification", {"chargePointSerialNumber": "000a1484c32e", "chargePointVendor": "Mennekes", "meterType": "IEC61107", "meterSerialNumber": "", "chargePointModel": "ACUV4", "firmwareVersion": "3633"}]
[ocpp  ] TRACE 2024/09/22 20:22:06 handling incoming CALL [947e0a64-790f-11ef-8f09-000a1484c32e, BootNotification] from Ladestation
```
`[Ladestation-1] DEBUG 2024/09/22 20:22:25 BootNotification timeout`

mfuchs1984, 2024-09-22:
@andig der Fehler ist im log aus Betrag https://github.com/evcc-io/evcc/issues/15677#issuecomment-2366908347

`[main  ] FATAL 2024/09/22 20:22:42 cannot create charger 'SmartTLinks': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/09/22 20:22:42 will attempt restart in: 15m0s`

Allerdings sieht es aus, als ginge es danach einfach weiter.

andig, 2024-09-22:
Bingo. Ich denke ich habe den Fehler möglicherweise gefunden. 

@premultiply schau mal bitte: 

```
func (cs *CS) OnBootNotification(id string, request *core.BootNotificationRequest) (*core.BootNotificationConfirmation, error) {
	cp, err := cs.ChargepointByID(id)
	if err != nil {
		return nil, err
	}

	return cp.OnBootNotification(request)
}
```

Ich vermute, dass wir dem Chargepoint hier gar nicht antworten solange er noch nicht registriert ist. M.E. sollten auch alle CS Handler- so der CP nicht gefunden wird- immer eine Antwort zurück geben so wie wir das an anderer Stelle mit den Connections auch machen? PR to follow.

benesolar, 2024-09-23:
Danke auch für die Geduld mit meiner Ladestation 😀

Leider weiterhin:
cannot create charger 'SmartTLinks': cannot create charger type 'template': cannot create charger type 'ocpp': timeout

[evcc-20240923-111623-trace.log.txt](https://github.com/user-attachments/files/17095172/evcc-20240923-111623-trace.log.txt)

andig, 2024-09-25:
Nightly in 20min. Neue Problem (danke Mennekes) ist, dass die WB den Status Request für den Connector zwar akzeptiert, aber nie antwortet...

benesolar, 2024-09-26:
[main  ] FATAL 2024/09/25 22:44:01 cannot create charger 'SmartTRechts': cannot create charger type 'template': cannot create charger type 'ocpp': timeout

[evcc-20240925-224845-trace.log](https://github.com/user-attachments/files/17151115/evcc-20240925-224845-trace.log)

premultiply, 2024-09-26:
@andig:

Ursache: Innerhalb von `Timeout` bzw. `Timeout/2` bekommen die Konnektoren keine `StatusNotification` nach der Initialiserung des CP. Weder indirekt über CP noch über die Konnektoren.

Einfache Dinge die noch zu probieren wären, in der Hoffnung dass Mennekes die angetriggerte Nachricht nur erst viel später schickt:
- Init-Timeout der Konnectoren (auch) auf `ConnectTimeout` statt fest auf 30s legen. - 
- Bei der Initialisierung der Konnektoren **immer** gleich eine `StatusNotification` antriggern (statt nur als Rettungsanker nach Timeout/2).

Erweiterter möglicher Lösungsweg:
- Letzte eingegangene `StatusNotification` pro Konnektor immer am CP cachen. Egal ob Charger verbunden oder nicht und beim Konnektor-Verbinden nur passiv prüfen ob Connector-ID plausibel ist (<= Gesamtzahl der vom CP gemeldeten Konnektoren) und schon mal für den gewünschten Konnektor jemals ein valider Status eingangen ist.

andig, 2024-09-26:
Letzteres. Ich mache Vorschlag.

benesolar, 2024-10-25:
Über eine Lösung wäre ich dankbar.
Kann ich hier irgendwie unterstützen?

andig, 2024-10-25:
Ich hab nochmal einen Versuch gemacht. Könntest Du den lokal bauen (oder in Docker ausprobieren)?

benesolar, 2024-10-25:
Damit kenne ich mich nicht aus.
Mit einem nightly könnte ich es testen.

andig, 2024-10-26:
@benesolar könntest Du heute testen?

benesolar, 2024-10-26:
kann ich heute machen

andig, 2024-10-26:
In 20min hast Du passendes Nightly.

closed 2024-10-26 by andig

benesolar, 2024-10-26:
```
Fehler beim Starten. Überprüfe deine Konfiguration und starte neu.
cannot create charger 'SmartTRechts': cannot create charger type 'template': cannot create charger type 'ocpp': cannot send request 2314568884, no client Ladestation exists
```
[evcc-20240825-204406-trace.log](https://github.com/user-attachments/files/17530605/evcc-20240825-204406-trace.log)

benesolar, 2024-10-26:
waiting for chargepoint: 5m0s
Fehler kommt aber bereits nach 30 Sekunden?

andig, 2024-10-26:
Das Thema hatten wir schon mal:

```
[main  ] INFO 2024/10/26 14:22:35 evcc 0.131.1 (ce14d937)
[main  ] INFO 2024/10/26 14:22:35 using config file: /etc/evcc.yaml
[main  ] INFO 2024/10/26 14:22:35 listening at :7070
[Ladestation-2] DEBUG 2024/10/26 14:22:37 waiting for chargepoint: 5m0s
[ocpp  ] TRACE 2024/10/26 14:22:37 error dispatching request [1086470933, ChangeAvailability] to Ladestation: cannot send request 1086470933, no client Ladestation exists
```

Hier ist das allererste im Log überhaupt ein OCPP Fehler. Da scheint der Charger (Annahme) schon mit evcc zu reden ohne dass OCPP überhaupt konfiguriert wäre. Mir scheint als würde ein Teil vom Log fehlen, kann ich aber nur raten. Auf jeden Fall fehlt jegliche OCPP Konaktaufnahme. Lässt sich so leider nicht analysieren.

andig, 2024-10-26:
Du denkst an https://github.com/evcc-io/evcc/issues/15677#issuecomment-2351768817?

benesolar, 2024-10-26:
ich habe es auch auf diese Weise probiert:
[evcc-20241026-161147-trace.log](https://github.com/user-attachments/files/17530867/evcc-20241026-161147-trace.log)

Da ist das Problem, dass evcc bereits nach 30 Sekunden die Fehlermeldung ausgibt, die Ladestation aber mindestens 2 Minuten zum starten braucht.
Bei früheren evcc Versionen war das nicht so.

andig, 2024-10-26:
Ich schaue nochmal ob ich das in einem Testfall nachstellen kann. Mir ist aber immer noch rätselhaft, warum evcc da an eine Ladestation was schicken will die sich noch nicht mal verbunden hat. Das kann egtl. nicht sein (tm). Melde mich...

andig, 2024-10-26:
Gleiches Spiel: In 20min hast Du passendes Nightly.

benesolar, 2024-10-26:
übliche Startprozedur:
[evcc-20241026-184228-trace.log](https://github.com/user-attachments/files/17531346/evcc-20241026-184228-trace.log)

[34 more replies]

PR #18468 (2025-02-01): Fix/ocpp initial status 3
@andig I took your branch fix/ocpp-initial-status-3 made to resolve #15677 and performed some debugging to resolve #18443. It was successfully initializing with the eProwallbox that does not support remote trigger after my changes.

Basically, the changes are:
1. Fix a panic in https://github.com/evcc-io/evcc/commit/8adb8feb6e94775a046f23fc042ef2dcfae0c4fe. The original code was accessing cs.regs[chargePoint.ID()].cp which was not existing yet.
2.  Change the order of caching status and calling cp.OnStatusNotification(request) in https://github.com/evcc-io/evcc/commit [... cut]