evcc-io/evcc #15678: Victron: Grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'

Hofyyy, 2024-08-25:
### Describe the bug

Nach dem Update auf 0.130.3 bekomme ich eine Fehlermeldung:
[site  ] ERROR 2024/08/25 15:43:54 grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'

### Steps to reproduce

1. Update auf 0.130.3
2. Docker restart
3.
...

### Configuration details

```yaml
meters:
- type: template
  template: victron-energy
  usage: grid
  host: 192.168.0.10
  port: 502
  name: EM24
- type: template
  template: victron-energy
  usage: pv
  host: 192.168.0.10
  port: 502
  name: solarpanels
- type: template
  template: victron-energy
  usage: battery
  host: 192.168.0.10
  port: 502
  name: varta
  capacity: 5
```

### Log details

```text
[site ] DEBUG 2024/08/25 15:47:14 pv power: 1195W
[site ] DEBUG 2024/08/25 15:47:14 battery soc: 80%
[site ] DEBUG 2024/08/25 15:47:14 battery power: -851W
[site ] DEBUG 2024/08/25 15:47:14 grid meter: -317W
[site ] ERROR 2024/08/25 15:47:14 grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'
[site ] DEBUG 2024/08/25 15:47:14 battery has priority at soc 80% (< 85%)
[site ] DEBUG 2024/08/25 15:47:14 site power: -367W
```

### What type of operating system are you running?

Docker container

### Version

0.130.3

andig, 2024-08-25:
Siehe https://github.com/evcc-io/evcc/pull/15625

closed 2024-08-25 by andig

Hofyyy, 2024-08-25:
Passt.

Ich habe wie beschrieben die VRM Instanz in der Geräteliste für den Grid Zähler nachgeschaut. In meinem Fall 40. und die Konfiguration abgeändert.

- type: template
  template: victron-energy
  usage: grid
  host: 192.168.0.10
  port: 502
  meterid: 40
  name: EM24

Hiermit funktioniert es jetzt.

Hofyyy, 2024-08-25:
Man könnte überlegen die Fehlermeldung zu ändern. ggf.