evcc-io/evcc #15686: system down after update 0.130.4

raspb1, 2024-08-25:
### Describe the bug

After I updated on 0.130.4 (and installation only the upgrade) My system is down and is not able to get started. The system message is: data base disk image malformed (11).    The curios thing is:  I made the update like always I made it svereal times before.

### Steps to reproduce

1.
2.
3.
...
look to the description above

### Configuration details

```yaml
d
```

### Log details

```text
d
```

### What type of operating system are you running?

Linux

### Version

d

0x3d13f, 2024-08-25:
See also https://github.com/evcc-io/evcc/discussions/15685 and https://github.com/evcc-io/evcc/discussions/15684
A downgrade to 0.130.3 was reported as helping for the moment there.

pdonovan, 2024-08-26:
I'm seeing the same message on 0.130.4:

`Aug 26 10:00:58 raspberrypi3 evcc[16317]: [main  ] FATAL 2024/08/26 10:00:58 database disk image is malformed (11)`
This is the 'armhf' build.

RTTTC, 2024-08-26:
@andig

jgantenberg, 2024-08-26:
> I'm seeing the same message on 0.130.4:
> 
> `Aug 26 10:00:58 raspberrypi3 evcc[16317]: [main ] FATAL 2024/08/26 10:00:58 database disk image is malformed (11)` This is the 'armhf' build.

same here, 0.130.2 starts fine with same DB.

StefanSchoof, 2024-08-26:
I don't think any me too are needed at this point.

If someone who is affected, wants to help and is able to build evcc: A `git bisect` to find the first bad commit can be helpful for the team.

andig, 2024-08-26:
Better idea: could someone please email me a "corrupted" database to info@evcc.io? Might be an issue with the upgraded Gorm library but I'm only guessing.

n0cturne, 2024-08-26:
> Better idea: could someone please email me a "corrupted" database to [info@evcc.io](mailto:info@evcc.io)? Might be an issue with the upgraded Gorm library but I'm only guessing.

I have send you a mail.

andig, 2024-08-26:
For time being, I have remove the 0.130.4 release packages and Docker image.

/cc @naltatis not sure if Gorm has anything to do with it?

andig, 2024-08-26:
> This is the 'armhf' build.

I've found the claimed corrupted database to be working on `arm64`. Which platforms are experiencing the problem?

andig, 2024-08-26:
I can indeed repro on `armhf`:

```
❯ GOOS=linux GOARCH=armv6 make build
❯ docker run -it --platform linux/armhf -v $(pwd):/app -v ~/Download/evcc.db:/root/.evcc/evcc.db alpine /app/evcc
[main  ] INFO 2024/08/26 08:12:50 evcc 0.130.3 (91f0e534b)
[main  ] INFO 2024/08/26 08:12:50 missing config file - switching into demo mode
[db    ] INFO 2024/08/26 08:12:50 using sqlite database: /root/.evcc/evcc.db
[db    ] ERROR 2024/08/26 08:12:50 failed to initialize database, got error unable to open database file: out of memory (14)
[main  ] INFO 2024/08/26 08:12:50 listening at :7070
[main  ] FATAL 2024/08/26 08:12:50 unable to open database file: out of memory (14)
[main  ] FATAL 2024/08/26 08:12:50 will attempt restart in: 15m0s
[main  ] INFO 2024/08/26 08:12:51 new version available: 0.130.4
```

andig, 2024-08-26:
Opened https://gitlab.com/cznic/sqlite/-/issues/191, upvoted might help.

Robman1, 2024-08-26:
> > This is the 'armhf' build.
> 
> I've found the claimed corrupted database to be working on `arm64`. Which platforms are experiencing the problem?

Same issue at my end. Use a Raspberry 4 with 64 Bit Raspberry OS -> AArch64 platform.

Michel83, 2024-08-26:
My System:

`Linux version 6.1.21-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1642 SMP Mon Apr  3 17:20:52 BST 2023`

hoppewf, 2024-08-26:
Linux raspi-dns 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"

closed 2024-08-26 by andig

andig, 2024-08-26:
So, here's what I've found:

- issue appears on any `linux/arm` build but not `arm64`
- db is apparent not corruped since it can be opened on `arm64` in my tests
- issue is introduced with modernc.org/sqlite >= v1.29.0  downstream dependency

homuser2003, 2024-08-26:
Also have the corruption problem (Pi2B). Appears that 1.30.5 is not yet listed in the repository? Apt Upgrade says with 1.30.4 I had the latest release...

andig, 2024-08-26:
Indeed. The downgraded version does not compile on Windows: https://github.com/evcc-io/evcc/actions/runs/10564539950/job/29267316724

RTTTC, 2024-08-27:
New version 0.130.6 works fine.

hardoverflow, 2024-08-27:
It works again with v0.130.6.