evcc-io/evcc #15687: after upgrading from v0.129.0 to v0.130.3: "grid current modbus expection 10"

angeldelivery86, 2024-08-25:
### Describe the bug

see title. no other changes made while updating

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
[main  ] INFO 2024/08/25 21:52:46 evcc 0.130.3 (eff595c6)
[main  ] INFO 2024/08/25 21:52:46 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/25 21:52:46 using sqlite database: /home/florian/.evcc/evcc.db
[lp-1  ] ERROR 2024/08/25 21:53:09 charge total import: read tcp 192.168.1.101:36088->192.168.1.89:1502: i/o timeout
[lp-2  ] ERROR 2024/08/25 21:53:19 charge total import: read tcp 192.168.1.101:50178->192.168.1.88:1502: i/o timeout
[site  ] WARN 2024/08/25 21:53:19 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 8s # control cycle interval

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: 192.168.1.233  
  port: 502  
  minsoc: 5
  maxsoc: 100  
  name: grid
- type: template
  template: victron-energy 
  usage: pv  
  host: 192.168.1.233  
  port: 502  
  minsoc: 5 
  maxsoc: 100  
  name: pv
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.1.233  
  port: 502  
  minsoc: 5  
  maxsoc: 100  
  name: battery

chargers:
- type: template
  template: openwb-2.0
  id: 1
  host: 192.168.1.88
  port: 1502
[... 186 lines cut ...]
[site ] DEBUG 2024/08/25 21:58:59 battery power: 1526W
[site ] DEBUG 2024/08/25 21:58:59 grid meter: 8W
[site ] ERROR 2024/08/25 21:58:59 grid currents: read failed: modbus: exception '10' (gateway path unavailable), function '4'
[site ] DEBUG 2024/08/25 21:58:59 site power: 1534W
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: send 02 13 00 00 00 06 01 04 27 76 00 02
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: recv 02 13 00 00 00 07 01 04 04 00 4c f2 65
[lp-2 ] DEBUG 2024/08/25 21:58:59 charge total import: 5042.789kWh
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: send 02 14 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: recv 02 14 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: send 02 15 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: recv 02 15 00 00 00 05 01 04 02 00 00
[lp-2 ] DEBUG 2024/08/25 21:58:59 charger status: A
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: send 02 16 00 00 00 06 01 04 27 83 00 01
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: recv 02 16 00 00 00 05 01 04 02 00 00
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: send 02 17 00 00 00 06 01 04 27 82 00 01
[openwb-2.0] TRACE 2024/08/25 21:58:59 modbus: recv 02 17 00 00 00 05 01 04 02 00 00
```

### What type of operating system are you running?

Linux

### Version

0.130.3 (eff595c6)

CHB68, 2024-08-26:
Bei mir nach dem Update auf 0.130.4 die identische Fehlermeldung. Evcc scheint aber trotzdem zu funktionieren. Es werden die korrekten Leistungen für das Netz (über Gridmeter Victron) angezeigt. Nur die für Gridmeter von Victron neu hinzugekommenen Ströme werden nicht angezeigt.
[image]

andig, 2024-08-26:
See https://github.com/evcc-io/evcc/pull/15625

closed 2024-08-26 by andig