evcc-io/evcc #15688: eebus works on 0.126.1 but does not on 0.130.4

jomach, 2024-08-25:
### Describe the bug

Hello, 

I'm working with evcc since a couple of months now. I have a wallbox from EVBox Livo which "supports" eebus. 
with the version 0.126.1 it works aldo I get some errors, but the car get's connected and the button start and off over the UI works. On 0.130.x it does not work at all.... Config was not changed. 
 
I will attach the logs from both evvc versions. 
[evcc0126.txt](https://github.com/user-attachments/files/16741826/evcc0126.txt)
[evcc01304.txt](https://github.com/user-attachments/files/16741834/evcc01304.txt)

@DerAndereAndi  FYI. Hier nochmal detaliert. Wenn ich was noch tun kann sag bescheid. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
chargers:
  - type: template
    template: eebus
    ski: ae82347bb2de22aa304c83fb1d8f1d94aeb29b4f #a9cfa5fac0e5631cb87f09ce726103519eff0fa6
    ip: 192.168.1.230
    name: wallbox3

vehicles:
- name: ev1
  type: template
  template: mg
  title: MG
  user: xxxxxxxxx
  password: xxxxxxx
  vin: xxxxxxx
  capacity: 70
  phases: 3
  icon: car
  mode: pv
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - MG

loadpoints:
  - title: Garage
    charger: wallbox3
    vehicle: ev1
    mode: pv
    enable:
      # einschalten, wenn 1 Minute lang mindestens 2000 W Überschuss vorhanden ist
      delay: 2m
      threshold: -3400
    disable:
      threshold: 200 # maximum import power (W)
      delay: 10m
site:
  title: Mein Zuhause
  residualPower: 350
  meters:
    grid: grid1
    pv:
      - Huawei-SUN2000
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

DerAndereAndi, 2024-08-26:
Kannst du bitte beides nochmal mit log level trace für EEBUS laufen lassen? So erkenne ich leider keine Unterschiede. Danke.

jomach, 2024-08-26:
Hi @DerAndereAndi, hier die logs in trace mode. Dazu ist mir was aufgefallen. 
Beim 0.126 sehe ich die meldung: "charger status: C", beim 0.130 nicht.  
Beim 0.126 hat es geklappt nur wenn ich den stecker kurz raus mach und wieder rein. 
Beim 0.130 geht es garnicht egal was ich versuche... 

[evcc126.log](https://github.com/user-attachments/files/16752592/evcc126.log)
[evcc130.log](https://github.com/user-attachments/files/16752596/evcc130.log)

DerAndereAndi, 2024-08-27:
These are identical issues as with the Elli Connect/Pro wallboxes, no wonder as it is the same developer EVBox.

The box simply does not provide measurements and hence EVCC does not know if the EV is charging and can not detect state C. I analyzed the message exchange and there are no errors or invalid messages. The code requests the wallbox to provide measurements, but it doesn't do so. Why it sometimes works with 0.126 and doesn't in 0.130 seems to be pure "luck". There is no logical error in the messages, and the current code works just fine with EEBUS wallboxes from other vendors.

Maybe ask EVBox why they still have the same bugs as in the Elli wallbox. Workaround: use an external meter.

andig, 2024-08-30:
Seems there's nothing to do here? Closing, thank you!

closed 2024-08-30 by andig

jomach, 2024-09-02:
Hey @andig  is there a way do disable ipv6 ? I'm getting: 

[eebus ] DEBUG 2024/09/02 22:30:45 mdns: new - ski: ae82347bb2de22aa304c83fb1d8f1d94aeb29b4f name: Livo-EVB-500-015-503 brand: EVBox model: Livo typ: ChargingStation identifier: EVBox-Livo-EVB-500-015-503 register: false host: EVB-500-015-503.local. port: 4712 addresses: [192.168.1.230 fe80::383f:62d9:6d40:d69b]

DerAndereAndi, 2024-09-02:
why is this a problem?

andig, 2024-09-03:
Recently, specifying the ip4 address was re-added. Does doing so help?