evcc-io/evcc #15699: EVCC HA add-on failed to install

drojasvela, 2024-08-26:
### Describe the bug

my EVCC add-on was working fine until the last couple of updates (0.130.4). It used to be able to unlock my Wallbox Pulsar Pro charger without any issues but then it stopped unlocking it automatically so I had to do it manually. Then, it simply stopped adjusting the power limit, so I uninstall the add-on but when trying to install it again, I'm facing the following message:

Failed to install add-on
Can't install evcc/evcc:0.130.4: 404 Client Error for http+docker://localhost/v1.45/images/create?tag=0.130.4&fromImage=evcc%2Fevcc&platform=linux%2Farm64: Not Found ("manifest for evcc/evcc:0.130.4 not found: manifest unknown: manifest unknown")

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
n
```

### Log details

```text
Can't access log anymore
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.4

Koltschin, 2024-08-26:
can´t update EVCC HA add-on from 0.130.2 to 0.130.4

Error:
Can't install evcc/evcc:0.130.4: 404 Client Error for http+docker://localhost/v1.45/images/create?tag=0.130.4&fromImage=evcc%2Fevcc&platform=linux%2Farm64: Not Found ("manifest for evcc/evcc:0.130.4 not found: manifest unknown: manifest unknown")

mfuchs1984, 2024-08-26:
The release was pulled back, see
https://github.com/evcc-io/evcc/issues/15686#issuecomment-2309577464

closed 2024-08-26 by andig