evcc-io/evcc #15705: Odometer wird in API (mqtt) scheinbar nicht aktualisiert (Peugeot)!?

dermobb, 2024-08-26:
### Describe the bug

Der Peugeot e208 wurde über die GUI hinzugefügt. SoC, Reichweite und Odometer werden in der GUI (unter Konfiguration) ständig erfolgreich aktualisiert. Soweit, so gut.

Ich reiche die Daten für eine eigene Ladesteuerung mittels mqtt an den ioBroker weiter. SoC und Reichweite werden auch über die API erfolgreich aktualisiert, nur das Odometer eben leider nicht, obwohl es in der evcc GUI wie gesagt der Fall ist.

Im Odometer steht dann der Wert wie bei der letzten Aktualisierung von evcc. Das heißt, jedes mal, wenn ich ein Update mache, wird der Wert aktualisiert. Nur eben leider nicht dauerhaft, wie ich es gerne hätte.

Im log sieht man auch nur SoC und Reichweite sich aktualisieren. Odometer findet nicht statt.

Vielleicht mache ich etwas falsch. Bin für jede Hilfe dankbar.

### Steps to reproduce

1. Update evcc
2. Fahre das Auto
3. Erkenne stehendes Odometer über mqtt

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 10s

javascript:
  - vm: shared
    script: |
      state = {
        residualpower: 0,
        pvpower: 0,
        gridpower: 0,
        loadpoints: [
          { enabled: false, vehicleSoc: 0, maxcurrent: 16, phases: 1, chargepower: 0 },
          { enabled: false, vehicleSoc: 0, maxcurrent: 16, phases: 1, chargepower: 0 }
        ]
      };
      function logState() {
        console.log("state:", JSON.stringify(state));
      }

meters:
  - name: grid
    type: custom
    power:
      source: js
      vm: shared
      script: |
        state.gridpower = 0;
        state.gridpower;

  - name: meter_charger_1
    type: custom
    power:
      source: js
      vm: shared
      script: state.loadpoints[0].chargepower;

chargers:
  - name: charger_1
    type: custom
    enable:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[0];
        lp.enabled = enable;
        enable;
      out:
        - name: enable
          type: bool
          config:
            source: js
            vm: shared
            script: |
              if (enable) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;
    enabled:
      source: js
      vm: shared
      script: |
        state.loadpoints[0].enabled;
    status:
      source: js
      vm: shared
      script: |
        if (state.loadpoints[0].enabled) "C"; else "B";
    maxcurrent:
      source: js
      vm: shared
      script: |
        logState();
        var lp = state.loadpoints[0];
        lp.maxcurrent = maxcurrent;
        if (lp.enabled) lp.chargepower = lp.maxcurrent * 230 * lp.phases; else lp.chargepower = 0;

site:
  title: Zuhause
  meters:
    grid: grid

loadpoints:
  - title: Carport
    charger: charger_1
    mode: off
    meter: meter_charger_1
    soc:
      poll:
        mode: always
        interval: 60s
      estimate: false
```

### Log details

```text
[site ] DEBUG 2024/08/26 17:29:21 ----
[lp-1 ] DEBUG 2024/08/26 17:29:21 charge power: 0W
[site ] DEBUG 2024/08/26 17:29:21 grid meter: 0W
[site ] DEBUG 2024/08/26 17:29:21 pv power: 0W
[site ] DEBUG 2024/08/26 17:29:21 site power: 0W
[lp-1 ] DEBUG 2024/08/26 17:29:21 charger status: B
[lp-1 ] DEBUG 2024/08/26 17:29:21 next soc poll remaining time: 9s
[site ] DEBUG 2024/08/26 17:29:31 ----
[lp-1 ] DEBUG 2024/08/26 17:29:31 charge power: 0W
[site ] DEBUG 2024/08/26 17:29:31 grid meter: 0W
[site ] DEBUG 2024/08/26 17:29:31 pv power: 0W
[site ] DEBUG 2024/08/26 17:29:31 site power: 0W
[lp-1 ] DEBUG 2024/08/26 17:29:31 charger status: B
[lp-1 ] DEBUG 2024/08/26 17:29:31 vehicle soc: 69%
[lp-1 ] DEBUG 2024/08/26 17:29:31 vehicle range: 208km
[site ] DEBUG 2024/08/26 17:29:41 ----
[lp-1 ] DEBUG 2024/08/26 17:29:41 charge power: 0W
[site ] DEBUG 2024/08/26 17:29:41 grid meter: 0W
[site ] DEBUG 2024/08/26 17:29:41 pv power: 0W
[site ] DEBUG 2024/08/26 17:29:41 site power: 0W
[lp-1 ] DEBUG 2024/08/26 17:29:41 charger status: B
[lp-1 ] DEBUG 2024/08/26 17:29:41 next soc poll remaining time: 49s
[site ] DEBUG 2024/08/26 17:29:51 ----
```

### What type of operating system are you running?

Docker container

### Version

0.130.2

andig, 2024-08-26:
Warum sollte man bei verbundenem Auto das Odometer aktualisieren? Mit Kabel kannst du ja nicht rum fahren 😎

dermobb, 2024-08-26:
> Warum sollte man bei verbundenem Auto das Odometer aktualisieren? Mit Kabel dran kannst du ja nicht rum fahren 😎

Das Problem scheint für mich zu sein, dass der Wert in der API (hier mqtt) überhaupt nicht aktualisiert wird. Egal ob verbunden oder nicht. Ich würde erwarten, dass der Wert in der API identisch ist mit dem Wert, der mir in evcc angezeigt wird.

andig, 2024-08-27:
Das ist auch so. Aktualsieirung 1x bei Erkennung.

closed 2024-08-27 by andig

dermobb, 2024-08-27:
> Das ist auch so. Aktualsieirung 1x bei Erkennung.

Was genau meinst du bitte mit "Erkennung"? Ich bin mir immer noch nicht sicher, ob mein Anliegen hier verstanden wird.
In evcc (dem Webserver) wird mir der korrekte Kilometerstand von 25.585 km angezeigt. Über mqtt wird ein Kilometerstand von 25.443 km übertragen. Das ist der Wert, der bei der letzten Aktualisierung von evcc aktuell war. Ich bin seit dem mehrfach gefahren. Soll das wirklich so sein? Warum???

andig, 2024-08-27:
>In evcc (dem Webserver) wird mir der korrekte Kilometerstand von 25.585 km angezeigt. 

Wo?

dermobb, 2024-08-27:
[image]

In der GUI.

andig, 2024-08-29:
Achso. Das sind Abrufe außerhalb der Verarbeitungslogik. Die werden auch nicht in die DB geschrieben.

dermobb, 2024-08-29:
Mag sein.
Würdest du es nicht als Mehrwert ansehen, wenn der korrekte Kilometerstand über die API übertragen werden könnte? Ich würde das gut und sinnvoll finden. Warum taucht der Kilometerstand überhaupt in der API auf, wenn er veraltet/falsch ist? An mangelnder Datenverfügbarkeit kann es ja nicht liegen, wie man sieht.

Bitte mal drüber nachdenken, ob das möglich wäre. Ganz, ganz bitte.