evcc-io/evcc #15710: Min+PV lädt nicht mehr

gimmel, 2024-08-26:
### Describe the bug

Konstellation:
11 kWp Panels mit Huawei Wechselrichter und 10 kWh Speicher, Zaptec Go Wallbox, Tesla Model Y, Renault Zoe. evcc läuft als Add-On in Homeassistant.
Den Sommer über stand die Ladefunktion meistens auf PV und ich bin ziemlich sicher, dass ich des öfteren auch mal erfolgreich auf Min+PV geschaltet habe, um das abendliche Sonnenlicht noch zu nutzen.

Nun startet der Ladevorgang bei Min+PV gar nicht mehr, die 2,5 kWh Einspeisung, die ich zur Sekunde habe, verschwinden im Netz. Ich weiß nun gar nicht, wo ich den Fehler suchen soll: es gab ein Update des Huawei WR und mehrere von evcc.

Ich habe jetzt schon einige Dinge ausprobiert, aber das einzige, was funktioniert hat, war die residualPower auf -4140 (3 Phasen * 6 A * 230 V) zu setzen. Das erscheint mir aber keine saubere Lösung zu sein - das würde ja auch den PV-Modus betreffen. Außerdem hat mir evcc vorhin damit den Speicher leergezogen - weit über die eingestellten Grenzen hinaus. Ich habe den Eindruck, evcc fehlen Werte aus dem Huawei-Wechselrichter. Der Zeitpunkt der Veränderung könnte mit dem Huawei-Update vom 1.8.2024 zusammenhängen.

Hat jemand das gleiche Problem?

### Steps to reproduce

1. Auto an Wallbox anschließen
2. In evcc bei ca. 2000 W Solarleistung auf Min+PV schalten
3. Auto lädt nicht (Ladebereit. Warte auf Fahrzeug...)
4. 

### Configuration details

```yaml
sponsortoken: ***
plant: ***

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

interval: 30s # control cycle interval

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

modbusproxy:
  - port: 5021
    uri: 192.168.172.168:502
  # rtu: true
  # readonly: true # use `deny` to raise modbus errors

meters:
  - name: jrk39_grid
    type: template
    template: huawei-dongle-powersensor
    usage: grid

    modbus: tcpip
    id: 1
    host: 192.168.172.168
    port: 502
    timeout: 15s

  - name: jrk39_pv
    type: template
    template: huawei-dongle-powersensor
    usage: pv

    modbus: tcpip
    id: 1
    host: 192.168.172.168
    port: 502
    timeout: 15s

  - name: jrk39_battery
    type: template
    template: huawei-dongle-powersensor
    usage: battery

    modbus: tcpip
    id: 1
    host: 192.168.172.168
    port: 502
    timeout: 15s

chargers:
  - name: walle
    type: template
    template: zaptec
    user: dreikaisers@icloud.com
    password: ys200&Zaptec

vehicles:
  - name: elmo
    title: Tesla Model Y
    type: template
    template: tesla-command
    icon: car
    accessToken: ***
    refreshToken: ***
    vin: ***
    capacity: 60
    mode: minpv # enable PV-charging when vehicle is identified
    mincurrent: 5
    maxcurrent: 16
    
  - name: zoe
    title: Renault Zoe
    type: template
    template: renault
    icon: car
    vin: ***
    capacity: 50
    user: ***
    password: ***
    mode: minpv # enable PV-charging when vehicle is identified
    mincurrent: 6
    maxcurrent: 16

# site describes the EVU connection, PV and home battery
site:
  title: JRK39
  meters:
    grid: jrk39_grid
    pv:
    - jrk39_pv
    battery:
    - jrk39_battery
  residualPower: 100 # additional household usage margin

loadpoints:
  - title: Wallbox # display name for UI
    charger: walle # Wallbox Referenz
    mode: pv # charge mode (off, now, minpv, pv)
    resetOnDisconnect: true # set defaults when vehicle disconnects
    enable: # pv mode enable behavior
      delay: 1m # threshold must be exceeded for this long
      threshold: -2000 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 3m # threshold must be exceeded for this long
      threshold: 3000 # maximum import power (W)
    guardDuration: 5m # switch charger contactor not more often than this (default 5m)

tariffs:
  grid:
    type: tibber
    token: "***"
    homeid: "***"
```

### Log details

```text
[site ] DEBUG 2024/08/26 17:05:15 ----
[lp-1 ] DEBUG 2024/08/26 17:05:15 charge power: 0W
[lp-1 ] DEBUG 2024/08/26 17:05:15 charge currents: [0.078 0.041 0.041]A
[site ] DEBUG 2024/08/26 17:05:16 pv power: 3025W
[site ] DEBUG 2024/08/26 17:05:19 battery soc: 100%
[site ] DEBUG 2024/08/26 17:05:19 battery power: -15W
[site ] DEBUG 2024/08/26 17:05:20 grid meter: -2545W
[site ] DEBUG 2024/08/26 17:05:23 grid currents: [-3.3 -3.43 -4.04]A
[site ] DEBUG 2024/08/26 17:05:24 site power: -2460W
[lp-1 ] DEBUG 2024/08/26 17:05:24 !! session: chargeRater.chargedEnergy=7.5 - chargedAtStartup=7.5
[lp-1 ] DEBUG 2024/08/26 17:05:24 charger status: B
[lp-1 ] DEBUG 2024/08/26 17:05:24 pv charge current: min 5A > 3.57A (-2460W @ 3p, battery: true)
[site ] DEBUG 2024/08/26 17:05:45 ----
[lp-1 ] DEBUG 2024/08/26 17:05:45 charge power: 0W
[lp-1 ] DEBUG 2024/08/26 17:05:45 charge currents: [0.078 0.041 0.041]A
[site ] DEBUG 2024/08/26 17:05:46 pv power: 2974W
[site ] DEBUG 2024/08/26 17:05:49 battery soc: 100%
[site ] DEBUG 2024/08/26 17:05:49 battery power: -13W
[site ] DEBUG 2024/08/26 17:05:50 grid meter: -2488W
[site ] DEBUG 2024/08/26 17:05:53 grid currents: [-3.22 -3.31 -3.97]A
[site ] DEBUG 2024/08/26 17:05:54 site power: -2401W
[lp-1 ] DEBUG 2024/08/26 17:05:54 !! session: chargeRater.chargedEnergy=7.5 - chargedAtStartup=7.5
[lp-1 ] DEBUG 2024/08/26 17:05:54 charger status: B
[lp-1 ] DEBUG 2024/08/26 17:05:54 pv charge current: min 5A > 3.48A (-2401W @ 3p, battery: true)
[site ] DEBUG 2024/08/26 17:06:15 ----
[lp-1 ] DEBUG 2024/08/26 17:06:15 charge power: 0W
[lp-1 ] DEBUG 2024/08/26 17:06:15 charge currents: [0.078 0.041 0.041]A
[site ] DEBUG 2024/08/26 17:06:18 pv power: 2915W
[site ] DEBUG 2024/08/26 17:06:21 battery soc: 100%
[site ] DEBUG 2024/08/26 17:06:21 battery power: -13W
[site ] DEBUG 2024/08/26 17:06:22 grid meter: -2451W
[site ] DEBUG 2024/08/26 17:06:25 grid currents: [-3.18 -3.32 -3.94]A
[site ] DEBUG 2024/08/26 17:06:27 site power: -2364W
[lp-1 ] DEBUG 2024/08/26 17:06:27 !! session: chargeRater.chargedEnergy=7.5 - chargedAtStartup=7.5
[lp-1 ] DEBUG 2024/08/26 17:06:27 charger status: B
[lp-1 ] DEBUG 2024/08/26 17:06:27 pv charge current: min 5A > 3.43A (-2364W @ 3p, battery: true)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

v0.130.4

OlliLu, 2024-08-26:
Bitte Zugangsdaten wie Mail und Passworte immer auskommentieren. Die obigen sind verbrannt, sofort ändern!

futurini, 2024-08-26:
zur Info, ZOE kann erst ab 8 Ampere laden. (zumindest bei mir) bitte folgendes ausprobieren:     mincurrent: 6 -> auf 8 stellen und testen.