evcc-io/evcc #15717: Docker Compose - volume mount / evcc.db location has changed

gklank, 2024-08-26:
### Describe the bug

the migration to the evcc.db is storing the db in /.evcc/evcc.db
Therefore the volume mount should change to       - /home/[user]/.evcc:/.evcc

In the documentation it is still the /root/.evcc

Regards

Gerhard

### Steps to reproduce

1. check with docker exec -ti evcc sh
2. ls -al /.evcc

### Configuration details

```yaml
docker-compose.yaml:

volumes:
  - /home/[user]/.evcc:/root/.evcc
  - /home/[user]/etc:/etc
```

### Log details

```text
na
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.6

andig, 2024-08-26:
What migration?

gklank, 2024-08-26:
Experimental! 🧪 Only use these features if you are in the mood for adventure and not afraid of debugging. Unexpected things and data loss may happen.

Migration and repair. Run "evcc migrate" to copy configuration from evcc.yaml to the database. 

It's just the experimental part - sorry.

andig, 2024-08-26:
If you run that you‘ll need to make sure that it catches the correct db:

    EVCC_DATABASE_DSN=… evcc migrate

closed 2024-08-26 by andig