evcc-io/evcc #15724: Wallbox Pulsar Plus issues on 130.0-130.6

flepp4ever, 2024-08-27:
### Describe the bug

I'm posting an issue with my Wallbox Pulsar Plus on evcc versions v0.130.0-v0.130.4 with "charger out of sync" and "waiting for local authentication". The OCPP connection doesn't get authorized and hence charging cannot start when the car is connected. The LED on the charger stays yellow instead of switching to light blue.
I had to downgrade to v0.129 to make everything work again.

Here's a screenshot from the log when the car is not connected:
[image]

Here's a screenshot from the log when the car is connected:
[image]

Here's my config of the evcc.yaml file:

Thanks in advance for your help!
Filipe

### Steps to reproduce

1. Installed v0.130.4
2. Got "charger out of sync" and "waiting for local authentication" warning messages in the log
3. Downgraded to v0.129 to make everything work again

### Configuration details

```yaml
type: template
template: pulsarplus
timeout: 5m
name: wallbox5
stationid: XXXXX
```

### Log details

```text
[lp-1  ] WARN 2024/08/27 07:18:10 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:19:13 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:20:17 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:21:40 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:23:11 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:24:44 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:26:09 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:27:40 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:29:10 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:30:14 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:31:40 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:33:13 charger out of sync: expected disabled, got enabled
[lp-1  ] WARN 2024/08/27 07:34:40 charger out of sync: expected disabled, got enabled
```

### What type of operating system are you running?

Docker container

### Version

evcc v0.130.6

premultiply, 2024-08-27:
Exactly what is logged.

Set `remotestart: true` in the config if your charger does not have an option like "autostart", "free vending", "plug&charge", ...

flepp4ever, 2024-08-27:
Hi @premultiply!
Thanks for the hint with remotestart: true . Now the car is successfully loading, BUT the "charger out of sync: expected disabled, got enabled" stays.

[image]

[image]

andig, 2024-08-30:
@flepp4ever please provide trace log for site, loadpoint, ocpp. Thank you!

andig, 2024-08-30:
Could you please remove modbus and other unrelated stuff?

flepp4ever, 2024-08-30:
Sorry, I thought I had unselected these 😅

Second try with lp-1 , OCPP, Site only selected:
[evcc-20240830-115048-trace.txt](https://github.com/user-attachments/files/16815182/evcc-20240830-115048-trace.txt)

mfuchs1984, 2024-08-31:
Looks like the wallbox accepts the command to set to 0 Ampere but keeps reporting an offered current and power greater than 0.

andig, 2024-09-01:
How do you expect us to find the cause of the the error here (first lines)?

```
[site  ] DEBUG 2024/08/30 10:07:31 site power: -82W
[lp-1  ] DEBUG 2024/08/30 10:07:31 charge voltages: [220 241 237]V
[lp-1  ] DEBUG 2024/08/30 10:07:31 detected connected phases: 3p
[lp-1  ] DEBUG 2024/08/30 10:07:31 charge total import: 2048.968kWh
[lp-1  ] DEBUG 2024/08/30 10:07:31 charger status: A
[lp-1  ] WARN 2024/08/30 10:07:31 charger out of sync: expected disabled, got enabled
```

The log file is useless. As usual, please: provide enough context _before_ the error appears.

flepp4ever, 2024-09-02:
So, another try @andig, as I don't have a clue what exactly to look for. There's a new issue that came up as you'll see on the trace log with lp-1 , OCPP, Site: 
[evcc-20240902-235147-trace.txt](https://github.com/user-attachments/files/16841429/evcc-20240902-235147-trace.txt)

Here's the full trace log:
[evcc-20240903-000215-trace.txt](https://github.com/user-attachments/files/16841441/evcc-20240903-000215-trace.txt)

Here's what I did to try to reproduce the error:

I restarted evcc to have a clean log at around 23:00. After 5 minutes only, the warning "charger out of sync: expected disabled, got enabled"  started to be produced every +- 90 seconds.

Then, I connected the charger to the car at exactly  23:24:39 and waited to see if the warning would still be produced. As the charger in evcc was set to "min+solar", it started to charge pretty immediately at 23:25:11

What is weird, is that I setup in evcc the min charging power to 6A and the max to 7A, but it was using 16A all the way. Then the error message "charger logic error: current mismatch (got 15.6A measured, expected 6A) " started being produced every +-30 secs.

I stopped charging the car at 23:31:47 through the Hyundai app, but left the car connected until 23:39:38. No error message was produced during the timespan.

After disconnecting the car, the warning also did not reappear.

I hope that the log is now of use to you.

andig, 2024-09-05:
@premultiply any idea how to proceed given https://github.com/evcc-io/evcc/issues/15724#issuecomment-2322933655? Does using offered current really provide a benefit?

premultiply, 2024-09-05:
What you can try is setting `metervalues: Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage`.
Maybe it helps here.

andig, 2024-09-05:
Nice. That removes offered current from usage 👍🏻

andig, 2024-09-06:
@flepp4ever test would be great. If that works we could also add this as template.

flepp4ever, 2024-09-08:
Hi @andig and @premultiply !

I've finally tested your config including metervalues. EVCC was started up at 23:30:50 and no signs of the the warning "charger out of sync: expected disabled, got enabled", which is good. I connected the car at 23:32:05.

Then, I set evcc to "min+solar" and it started to charge pretty immediately at 23:35:34, but again it was using 16A, although I had setup in evcc the min charging power to 6A and the max to 7A. 

Then,  the error message "charger logic error: current mismatch (got 15.4A measured, expected 6A)" started being produced again at 23:36:34 every +-30 secs.

Here's the log: [evcc-20240908-235321-trace.log](https://github.com/user-attachments/files/16923771/evcc-20240908-235321-trace.log).
I've noticed that there is ocpp and ocpp-1

Here's also my "new" evcc.yaml
`# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: da169f647156c853397bfc2401d9e5bcb336a3b871cb0985aec019305b998bba

interval: 30s # control cycle interval

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: SUN2000M1  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 15s  
  name: grid1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: SUN2000M1  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 15s  
  name: pv1
- type: template
  template: huawei-dongle-powersensor 
  id: 1  
  host: SUN2000M1  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 15s  
  name: battery1

chargers:
- type: template
  template: ocpp
  name: wallbox1
  remotestart: true
  metervalues: Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage

vehicles:
- type: template
  template: hyundai 
  title: KONA EV  
  user: XXXX
  password: XXXX 
  vin: XXXX
  capacity: 64  
  language: de  
  name: ev1

loadpoints:
- title: Garage
  charger: wallbox1
  mode: minpv

site:
  title: Maison
  meters:
    grid: grid1
    pv:
    - pv1
    battery:
    - battery1
  residualPower: 100`

Have I setup anything wrong or where is this bug coming from?

Thanks in advance for your help!

premultiply, 2024-09-08:
```
[ocpp  ] TRACE 2024/09/08 23:35:18 send 433878: [2,"1509059245","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-09-08T21:35:18Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/09/08 23:35:19 recv 433878: [3,"1509059245",{"status": "Accepted"}]
[lp-1  ] DEBUG 2024/09/08 23:35:19 max charge current: 6A
```

premultiply, 2024-09-08:
Try
```yaml
chargers:
- type: template
  template: ocpp
  name: wallbox1
  remotestart: true
  metervalues: Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage
  stacklevelzero: true
```
?

flepp4ever, 2024-09-09:
Well ocpp as a template and stacklevelzero produce fatal errors.

[image]

Here's my config that makes evcc run again, but the "charger logic error: current mismatch (got 15.4A measured, expected 6A)" comes back again when selecting PV+min

mrhund, 2024-09-09:
> Well ocpp as a template and stacklevelzero produce fatal errors.

it needs to look like that:

`chargers:
- type: ocpp
  name: wallbox1
  remotestart: true
  metervalues: Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage
  stacklevelzero: true`

mrhund, 2024-09-09:
> Here's my config that makes evcc run again, but the "charger logic error: current mismatch (got 15.4A measured, expected 6A)" comes back again when selecting PV+min

Same for me, seems a firmware bug. Serveral support cases with the vendor (wallbox.com) are open :(