evcc-io/evcc #15726: [OCPP] EVC04 timeout beim Meter

Chris45671, 2024-08-27:
### Describe the bug

Seit ich von 0.129 auf den 0.130 Zweig gewechselt bin, habe ich dauerhaft die Fehlermeldung:
[EU_EDR_CB12001332-1] WARN 2024/08/27 08:03:30 meter timeout 
Auch wenn ich jetzt wieder auf die 0.129 zurück gehe, bleibt dieser Fehler.
Liegt dies vielleicht an der aktuellen EVCC Version oder ist hier vielleicht bei meiner Wallbox etwas kaputt?

Ausstattung:
EVC04 Wallbox (EON vBox smart) - Firmware v3.118.0
2x PLENTICORE plus 10
1x KSME
1x BYD Akkus 16,6 kWh
1x Rasberry 5 8GB für EVCC

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
  lp-1: trace
  opcc: trace
  cache: error
  db: error

# unique installation id
plant: 26xxx

# gÃ¼ltig bis, Samstag, 28. August 2027
sponsortoken:  xxxxx

interval: 30s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.36 # EUR/kWh
  feedin:
    type: fixed
    price: 0.065 # EUR/kWh
  co2:
    type: grÃ¼nstromindex
    zip: 85652

vehicles:
  - name: Genesis
    title: Genesis G80
    type: template
    template: offline
    icon: car # Icon in der BenutzeroberflÃ¤che (optional)
    capacity: 87 # AkkukapazitÃ¤t in kWh (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden kÃ¶nnen (optional)
    mode: PV # MÃ¶glich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
  - name: Scotty
    type: template
    template: enyaq
[... 1147 lines cut ...]
		SupportedFeatureProfiles (r/o): Core,FirmwareManagement,LocalAuthListManagement,Reservation,SmartCharging,RemoteTrigger
		SupportedFeatureProfilesMaxLength (r/o): 120
		TransactionMessageAttempts (r/w): 3
		TransactionMessageRetryInterval (r/w): 20
		UKSmartChargingEnabled (r/w): FALSE
		UnbalancedLoadDetection (r/w): True
		UnbalancedLoadDetectionMaxCurrent (r/w): 6
		UnlockConnectorOnEVSideDisconnect (r/w): TRUE
		WebSocketPingInterval (r/w): 30
		apnInfo (r/w): ,,
		installationErrorEnable (r/w): TRUE
		publicKey (r/o): 
		randomisedDelayAtOffPeakEnd (r/w): False
		timeZone (r/w): UTC

Allgemeines Log aus der Oberfläche:
```

### What type of operating system are you running?

Linux

### Version

0.130.6 (e97ebfc5)

Chris45671, 2024-08-27:
Allgemeines Log aus der Oberfläche:

```
[main  ] INFO 2024/08/27 08:00:26 evcc 0.130.6 (e97ebfc5)
[main  ] INFO 2024/08/27 08:00:26 using config file: /etc/evcc.yaml
[db    ] INFO 2024/08/27 08:00:26 using sqlite database: /var/lib/evcc/evcc.db
[db    ] TRACE 2024/08/27 08:00:26 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/08/27 08:00:26 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/27 08:00:26 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/27 08:00:26 SELECT * FROM `settings` 24 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/08/27 08:00:27 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main  ] INFO 2024/08/27 08:00:27 listening at :7070
[db    ] TRACE 2024/08/27 08:00:27 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[modbus] TRACE 2024/08/27 08:00:27 modbus: send 00 01 00 00 00 06 47 03 9c 40 00 02
[modbus] TRACE 2024/08/27 08:00:27 modbus: recv 00 01 00 00 00 07 47 03 04 53 75 6e 53
[modbus] TRACE 2024/08/27 08:00:27 modbus: send 00 02 00 00 00 06 47 03 9c 42 00 02
[modbus] TRACE 2024/08/27 08:00:27 modbus: recv 00 02 00 00 00 07 47 03 04 00 01 00 41
[modbus] TRACE 2024/08/27 08:00:27 modbus: send 00 03 00 00 00 06 47 03 9c 85 00 02
[modbus] TRACE 2024/08/27 08:00:27 modbus: recv 00 03 00 00 00 07 47 03 04 00 cb 00 69
[modbus] TRACE 2024/08/27 08:00:27 modbus: send 00 04 00 00 00 06 47 03 9c f0 00 02
[modbus] TRACE 2024/08/27 08:00:27 modbus: recv 00 04 00 00 00 07 47 03 04 ff ff 00 00
[modbus] TRACE 2024/08/27 08:00:27 modbus: send 00 05 00 00 00 06 47 03 9c 44 00 42
[modbus] TRACE 2024/08/27 08:00:27 modbus: recv 00 05 00 00 00 87 47 03 84 4b 4f 53 54 41 4c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4b 4f 53 54 41 4c 20 53 6d 61 72 74 20 45 6e 65 72 67 79 20 4d 65 74 65 72 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32 2e 34 2e 31 00 00 00 00 00 00 00 00 00 00 00 37 36 32 37 33 31 36 31 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ff 00 cb
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 01 00 00 00 06 47 03 9c 40 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 01 00 00 00 07 47 03 04 53 75 6e 53
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 02 00 00 00 06 47 03 9c 42 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 02 00 00 00 07 47 03 04 00 01 00 42
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 03 00 00 00 06 47 03 9c 86 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 03 00 00 00 07 47 03 04 00 67 00 32
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 04 00 00 00 06 47 03 9c ba 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 04 00 00 00 07 47 03 04 00 71 00 3c
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 05 00 00 00 06 47 03 9c f8 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 05 00 00 00 07 47 03 04 00 78 00 1a
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 06 00 00 00 06 47 03 9d 14 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 06 00 00 00 07 47 03 04 00 7b 00 18
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 07 00 00 00 06 47 03 9d 2e 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 07 00 00 00 07 47 03 04 00 a0 00 44
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 08 00 00 00 06 47 03 9d 74 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 08 00 00 00 07 47 03 04 00 cb 00 69
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 09 00 00 00 06 47 03 9d df 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 09 00 00 00 07 47 03 04 03 22 00 3e
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 0a 00 00 00 06 47 03 9e 1f 00 02
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 0a 00 00 00 07 47 03 04 ff ff 00 00
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 0b 00 00 00 06 47 03 9c 44 00 42
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 0b 00 00 00 87 47 03 84 4b 4f 53 54 41 4c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 50 4c 45 4e 54 49 43 4f 52 45 20 70 6c 75 73 31 30 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 20 30 32 2e 31 32 2e 31 36 32 31 39 00 00 00 00 00 39 33 35 35 38 57 46 4a 30 30 30 37 33 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 47 5a 5a
[sunspec] TRACE 2024/08/27 08:00:27 modbus: send 00 0c 00 00 00 06 47 03 9d 30 00 08
[sunspec] TRACE 2024/08/27 08:00:27 modbus: recv 00 0c 00 00 00 13 47 03 10 ff fe ff ff 00 00 00 00 00 00 00 00 00 03 00 00
[... cut ...]
[ocpp  ] TRACE 2024/08/27 08:04:58 recv EU_EDR_CB12001332: [3,"2522847499",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/27 08:05:00 send EU_EDR_CB12001332: [2,"4141704262","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/27 08:05:00 recv EU_EDR_CB12001332: [3,"4141704262",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/27 08:05:02 send EU_EDR_CB12001332: [2,"3162630228","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/27 08:05:02 recv EU_EDR_CB12001332: [3,"3162630228",{"status":"Accepted"}]
```

premultiply, 2024-08-27:
> habe ich dauerhaft die Fehlermeldung:
[EU_EDR_CB12001332-1] WARN 2024/08/27 08:03:30 meter timeout

Auch während eines laufenden Ladevorgangs?

NikhilKumar77777, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ESMELSTEPHANE24, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Chris45671, 2024-08-27:
DIese Fehlermeldung kommt direkt nach dem Start von EVCC. Danach kommt diese Meldung in dieser Form nicht mehr.
Erst weider beim nächsten Start. Unter dem verlinkten Thread habe ich alle Logs auch während dem Laden abgelegt.
Diese Meldung hatte ich wie geschrieben vorher nicht, aber seit neuestem auch wenn ich eben auf die alte Version wieder zurück gehe.

Denke dies hängt vielleicht auch mit den anderen Problemen zusammen, das ich das Auto nicht mehr laden kann.

premultiply, 2024-08-27:
> Beim Ladevorgang und danach verschwinden diese dann.

Funktioniert soweit wie erwartet.

> Erst weider beim nächsten Start erscheint die Meldung mit dem timeout.

Das hängt damit zusammen, dass die Box selbst auf explizite Anweisung (TriggerMessage) ohne laufende Transaktion keine Messwerte herausrückt. Das ist bereits der Workaround gegen fehlende Messwerte, die normalerweise bei laufender Transaktion ohnehin automatisch im festgelegten Intervall (10s) von der Box gesendet werden.
Dagegen können wir erstmal offensichtlich nichts tun. Dies ist im ungünstigen Verhalten der aktuellen Box-Firmware selbst begründet.

Wir planen aber aktuell noch eine interne Codeumstrukturierung, vielleicht ergibt sich daraus nochmal ein kleiner Optimierungsansatz in dieser Sache wenn wir die Messwerte nicht für einen bestimmten Anschluss anfragen sondern allgemein für die ganze Station. Ich halte es mal im Hinterkopf und mache hier aber erstmal zu, da wir akut nichts dagegen tun können.

Prinzipiell könntest du natürlich auch mal beim Herstellersupport anfragen weshalb die Firmware sich so dagegen wehrt ohne Transaktion mittels 
```
[ocpp  ] TRACE 2024/08/27 07:54:08 send EU_EDR_CB12001332: [2,"3197583823","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/27 07:54:08 recv EU_EDR_CB12001332: [3,"3197583823",{"status":"Accepted"}]
```
die angefragten Messwerte zu liefern und ob sie das nicht bitte fixen können.

Darüber hinaus könntest du einmal probieren ob es ggf. hilft wenn du in der Box-Konfiguration bei `MeterValuesAlignedData` den Wert `Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage` einträgst und bei `ClockAlignedDataInterval` den Wert `10` angibst. Sollte `10` nicht funktionieren alternativ `30`, `45` und `60` (in dieser Reihenfolge) probieren.

Gerne auch mal rückmelden ob das funktioniert und wie das ggf. etwas gebracht hat.

closed 2024-08-27 by premultiply

Chris45671, 2024-08-27:
Dem Hersteller habe ich dies soeben geschrieben, mal sehen ob ich eine Antwort bekomme.
Die anderen beiden Punkte werde ich ausprobieren, sobald der Ladevorgang abgeschlossen ist.

@premultiply  : Ich weiss nicht ob es ein neues Problem ist. Während dem Ladevorgang hat scheinbar der Rasby das Netzwerksignal verloren und konnte keine Komponenten mehr erreichen. Nach dem Ausfall hatte er dann alle Komponenten inkl. der Wallbox wieder, aber jetzt fehlen dann auf einmal wieder Werte so das er auch nicht mehr steuern kann oder den Bezug anzeigen kann.

```
[lp-1 ] DEBUG 2024/08/27 13:32:41 charge power: 4857W
[lp-1 ] DEBUG 2024/08/27 13:32:41 charge currents: [7 6.6 7]A
[lp-2 ] DEBUG 2024/08/27 13:32:41 charge power: 0W
[site ] ERROR 2024/08/27 13:32:51 pv 1 power: add[0]: model 160 block 1 point DCW: read tcp 192.168.178.79:50074->192.168.178.60:1502: i/o timeout
[site ] ERROR 2024/08/27 13:33:01 pv 2 power: add[0]: model 160 block 1 point DCW: read tcp 192.168.178.79:41858->192.168.178.61:1502: i/o timeout
[site ] DEBUG 2024/08/27 13:33:01 pv power: 0W
[site ] ERROR 2024/08/27 13:33:04 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:33:11 ----
[lp-1 ] DEBUG 2024/08/27 13:33:11 charge power: 4873W
[lp-1 ] DEBUG 2024/08/27 13:33:11 charge currents: [7 6.6 7]A
[lp-2 ] DEBUG 2024/08/27 13:33:11 charge power: 0W
[site ] ERROR 2024/08/27 13:33:12 pv 1 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] ERROR 2024/08/27 13:33:13 pv 2 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.61:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:33:13 pv power: 0W
[site ] ERROR 2024/08/27 13:33:16 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:33:41 ----
[lp-1 ] DEBUG 2024/08/27 13:33:41 charge power: 4873W
[lp-1 ] DEBUG 2024/08/27 13:33:41 charge currents: [7 6.6 7]A
[lp-2 ] DEBUG 2024/08/27 13:33:41 charge power: 0W
[site ] ERROR 2024/08/27 13:33:44 pv 1 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] ERROR 2024/08/27 13:33:46 pv 2 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.61:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:33:46 pv power: 0W
[site ] ERROR 2024/08/27 13:33:49 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:34:11 ----
[lp-1 ] DEBUG 2024/08/27 13:34:11 charge power: 4873W
[lp-1 ] DEBUG 2024/08/27 13:34:11 charge currents: [7 6.6 7]A
[lp-2 ] DEBUG 2024/08/27 13:34:11 charge power: 0W
[ocpp ] DEBUG 2024/08/27 13:34:14 charge point disconnected: EU_EDR_CB12001332
[site ] ERROR 2024/08/27 13:34:15 pv 1 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] ERROR 2024/08/27 13:34:17 pv 2 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.61:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:34:17 pv power: 0W
[site ] ERROR 2024/08/27 13:34:18 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:34:41 ----
[lp-1 ] ERROR 2024/08/27 13:34:42 charge power: timeout
[lp-1 ] ERROR 2024/08/27 13:34:43 charge currents: timeout
[lp-2 ] DEBUG 2024/08/27 13:34:43 charge power: 0W
[site ] ERROR 2024/08/27 13:34:46 pv 1 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] ERROR 2024/08/27 13:34:49 pv 2 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.61:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:34:49 pv power: 0W
[site ] ERROR 2024/08/27 13:34:52 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:35:11 ----
[lp-1 ] ERROR 2024/08/27 13:35:12 charge power: timeout
[lp-1 ] ERROR 2024/08/27 13:35:12 charge currents: timeout
[lp-2 ] DEBUG 2024/08/27 13:35:12 charge power: 0W
[site ] ERROR 2024/08/27 13:35:15 pv 1 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] ERROR 2024/08/27 13:35:18 pv 2 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.61:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:35:18 pv power: 0W
[site ] ERROR 2024/08/27 13:35:21 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:35:41 ----
[lp-1 ] ERROR 2024/08/27 13:35:41 charge power: timeout
[lp-1 ] ERROR 2024/08/27 13:35:42 charge currents: timeout
[lp-2 ] DEBUG 2024/08/27 13:35:42 charge power: 0W
[site ] ERROR 2024/08/27 13:35:45 pv 1 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] ERROR 2024/08/27 13:35:48 pv 2 power: add[0]: model 160 block 1 point DCW: dial tcp 192.168.178.61:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:35:48 pv power: 0W
[site ] ERROR 2024/08/27 13:35:51 battery 1 power: model 802 block 0 point W: dial tcp 192.168.178.60:1502: connect: no route to host
[site ] DEBUG 2024/08/27 13:36:11 ----
[lp-1 ] ERROR 2024/08/27 13:36:12 charge power: timeout
[lp-1 ] ERROR 2024/08/27 13:36:12 charge currents: timeout
[lp-2 ] DEBUG 2024/08/27 13:36:12 charge power: 0W
[... cut ...]
[lp-1 ] ERROR 2024/08/27 13:47:14 charge rater: charge total import: timeout
[lp-1 ] ERROR 2024/08/27 13:47:14 charge total import: timeout
[lp-1 ] DEBUG 2024/08/27 13:47:14 charger status: C
[lp-1 ] DEBUG 2024/08/27 13:47:14 pv charge current: 20.3A = 6A + 14.3A (-9871W @ 3p)
[lp-1 ] DEBUG 2024/08/27 13:47:15 max charge current: 10A
[lp-1 ] DEBUG 2024/08/27 13:47:23 set max current: 9 
```

Das komplette Trace 
[evcc-20240827-135738-trace.log](https://github.com/user-attachments/files/16761734/evcc-20240827-135738-trace.log)

Gruß
Christian

premultiply, 2024-08-27:
WiFi aktiv?

Das Neuverbinden ist noch einer der Schwachpunkte bei OCPP da die Boxen dabei ggf. ihre von evcc initialisierte Konfig verlieren und dabei nicht auch wieder neu initialisiert werden. Ist aber auch boxabhängig.
Hier hilft momentan leider nur ein evcc-Neustart. Wir haben das aber schon auf dem Schirm. Ist aber leider nicht kurzfristig lösbar.

Chris45671, 2024-08-27:
WiFi aktiv, aber schon die ganze Zeit. Ich war nicht zu Hause als es abgebrochen hat.
Rasby und WiFi Access Point stehen fast nebeneinander. Ab Donnerstag/Freitag soll aber ein Kabel folgen.

Werde nach dem Ladevorgang die Box neu starten. Die Wallbox gegebenenfalls vorher auch?

Chris45671, 2024-09-07:
@premultiply : Das Meter funktioniert jetzt mit 130.7, aber nur wenn ich die von EVCC gesetzten Werte anpasse.

MeterValuesAlignedData
EVCC (alt): Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered

Deine Werte zusätzlich (Neu): Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage

Dann bekomme ich derzeit keinen Timeout während dem normalen Betrieb.

Gruß
Christian

premultiply, 2024-09-07:
Wo passt du da was an?

Chris45671, 2024-09-07:
Ich passe es in der Wallbox an, wo Du es damals gesagt hast.

Wie geschrieben ist es das Feld "MeterValuesAlignedData" in den OCPP Einstellungen meiner Wallbox. Die von EVCC Gesetzen Werte passen nicht und ich ergänze es durch die von Dir genannten Werte und dann geht es aktuell bei mir.

premultiply, 2024-09-07:
Dann ist doch prima.

cjander, 2025-02-22:
@Chris45671 ich habe ein ganz ähnliches problem wie du, habe die gleiche vBox und sie läuft auf der selben firmware wie hier beschrieben. ich hatte zwar nicht dein timeout problem, aber diese Logmeldungen kommen ständig:
_[lp-1 ] ERROR 2025/02/22 10:43:22 charge voltages: not available
[lp-1 ] ERROR 2025/02/22 10:43:22 charge total import: not available_

...wenn lein Auto verbunden ist.

Ich hatte auch MeterValuesAlignedData  im Verdacht (darüber bin ich auf diesen thread hier gestoßen) und hab diese Konfiguration jetzt so angepasst wie von dir vorgeschlagen, dennoch bekomme ich weiterhin diese Meldungen im log. 
Sind diese Fehlermeldungen bei dir jetzt vollständig verschwunden?