evcc-io/evcc #15728: Monday abbreviation (Mo) does not get interpreted correctly in tariff section

nejcsu, 2024-08-27:
### Describe the bug

I made a script that updates evcc.yml everyday with next day prices for https://www.bsp-southpool.com, since it's not supported by api.

So i get partial config like:
tariffs:
  currency: EUR
  grid:
    price: 0.99
    type: fixed
    zones:
    - days: Mo
      hours: 0-1
      price: 0.115
    - days: Mo
      hours: 1-2
      price: 0.10294
    - days: Mo
      hours: 2-3
      price: 0.08634

checkconfig runs fine:

evcc checkconfig
[main  ] INFO 2024/08/27 09:40:07 evcc 0.130.6
[main  ] INFO 2024/08/27 09:40:07 using config file: /home/nejcs/evcc.yaml

tarrif switch not so much:

evcc tariff
[main  ] INFO 2024/08/27 09:40:39 evcc 0.130.6
[main  ] INFO 2024/08/27 09:40:39 using config file: /home/nejcs/evcc.yaml
[db    ] INFO 2024/08/27 09:40:39 using sqlite database: /home/nejcs/.evcc/evcc.db
grid:
[main  ] FATAL 2024/08/27 09:40:39 cannot create tariff type 'fixed': invalid day: tu

### Steps to reproduce

use partial config above and run 

evcc tariff

### Configuration details

```yaml
tariffs:
  currency: EUR
  grid:
    price: 0.99
    type: fixed
    zones:
    - days: Fr
      hours: 0-1
      price: 0.09755
    - days: Fr
      hours: 1-2
      price: 0.08768000000000001
    - days: Fr
      hours: 2-3
      price: 0.08734
    - days: Fr
      hours: 3-4
      price: 0.08648
    - days: Fr
      hours: 4-5
      price: 0.08702
    - days: Fr
      hours: 5-6
      price: 0.09223999999999999
    - days: Fr
      hours: 6-7
      price: 0.13069999999999998
    - days: Fr
      hours: 7-8
      price: 0.136
    - days: Fr
      hours: 8-9
      price: 0.13
    - days: Fr
      hours: 9-10
      price: 0.121
    - days: Fr
      hours: 10-11
      price: 0.10157
    - days: Fr
      hours: 11-12
      price: 0.09004000000000001
    - days: Fr
      hours: 12-13
      price: 0.09006
    - days: Fr
      hours: 13-14
      price: 0.09572
    - days: Fr
      hours: 14-15
      price: 0.09706999999999999
    - days: Fr
      hours: 15-16
      price: 0.1147
    - days: Fr
      hours: 16-17
      price: 0.12168000000000001
    - days: Fr
      hours: 17-18
      price: 0.13141
    - days: Fr
      hours: 18-19
      price: 0.16
    - days: Fr
      hours: 19-20
      price: 0.25001
    - days: Fr
      hours: 20-21
      price: 0.17
    - days: Fr
      hours: 21-22
      price: 0.19473
    - days: Fr
      hours: 22-23
      price: 0.14851
    - days: Fr
      hours: 23-0
      price: 0.1316
    - days: Mo
      hours: 0-1
      price: 0.115
    - days: Mo
      hours: 1-2
      price: 0.10294
    - days: Mo
      hours: 2-3
      price: 0.08634
    - days: Mo
      hours: 3-4
      price: 0.08466
    - days: Mo
      hours: 4-5
      price: 0.08636
    - days: Mo
      hours: 5-6
      price: 0.08683
    - days: Mo
      hours: 6-7
      price: 0.126
    - days: Mo
      hours: 7-8
      price: 0.14172
    - days: Mo
      hours: 8-9
      price: 0.12382
    - days: Mo
      hours: 9-10
      price: 0.125
    - days: Mo
      hours: 10-11
      price: 0.08026000000000001
    - days: Mo
      hours: 11-12
      price: 0.08723
    - days: Mo
      hours: 12-13
      price: 0.115
    - days: Mo
      hours: 13-14
      price: 0.116
    - days: Mo
      hours: 14-15
      price: 0.11741
    - days: Mo
      hours: 15-16
      price: 0.1226
    - days: Mo
      hours: 16-17
      price: 0.11004000000000001
    - days: Mo
      hours: 17-18
      price: 0.13301
    - days: Mo
      hours: 18-19
      price: 0.1708
    - days: Mo
      hours: 19-20
      price: 0.32921
    - days: Mo
      hours: 20-21
      price: 0.27302
    - days: Mo
      hours: 21-22
      price: 0.15506999999999999
    - days: Mo
      hours: 22-23
      price: 0.12637
    - days: Mo
      hours: 23-0
      price: 0.10778
    - days: Tu
      hours: 0-1
      price: 0.11741
    - days: Tu
      hours: 1-2
      price: 0.11688
    - days: Tu
      hours: 2-3
      price: 0.11537
    - days: Tu
      hours: 3-4
      price: 0.11081
    - days: Tu
      hours: 4-5
      price: 0.09952
    - days: Tu
      hours: 5-6
      price: 0.09937
    - days: Tu
      hours: 6-7
      price: 0.12519
    - days: Tu
      hours: 7-8
      price: 0.1368
    - days: Tu
      hours: 8-9
      price: 0.11270000000000001
    - days: Tu
      hours: 9-10
      price: 0.08843000000000001
    - days: Tu
      hours: 10-11
      price: 0.07176
    - days: Tu
      hours: 11-12
      price: 0.09311
    - days: Tu
      hours: 12-13
      price: 0.11711
    - days: Tu
      hours: 13-14
      price: 0.118
    - days: Tu
      hours: 14-15
      price: 0.12691
    - days: Tu
      hours: 15-16
      price: 0.13479
    - days: Tu
      hours: 16-17
      price: 0.19
    - days: Tu
      hours: 17-18
      price: 0.24683000000000002
    - days: Tu
      hours: 18-19
      price: 0.195
    - days: Tu
      hours: 19-20
      price: 0.27542
    - days: Tu
      hours: 20-21
      price: 0.26602
    - days: Tu
      hours: 21-22
      price: 0.157
    - days: Tu
      hours: 22-23
      price: 0.12764
    - days: Tu
      hours: 23-0
      price: 0.11855
telemetry: true
```

### Log details

```text
No special logs are recorded
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.130.6

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

StefanSchoof, 2024-08-27:
You need the first 3 chars for englisch weekdays and two for german, so `thu` or `do` for thursday. See

https://github.com/evcc-io/evcc/blob/e6d3ad3f9f5e22b417d6472c99d2f41822cda53f/tariff/fixed/day.go#L26-L42

nejcsu, 2024-08-27:
Thanx!

it works now.

closed 2024-08-27 by andig