evcc-io/evcc #15735: Kein recovery nach mqtt-Server-Ausfall

maf-soft, 2024-08-27:
### Describe the bug

Evcc erhält bei mir einige Werte von einem mqtt-Server, und der fällt leider manchmal kurz oder auch länger aus. Leider ist das Verhalten von evcc weder während des Ausfalls noch danach brauchbar.

Während des Ausfalls scheint mit dem letzten grid-meter-Wert weitergemacht zu werden, so dass ich im PV-Modus mehrmals die Situation vorfand, dass er schon längere Zeit mit voller Leistung geladen hatte, anstatt des wenigstens irgendwann zu erkennen und damit aufzuhören.

Und nachdem der mqtt-Server wieder läuft, wird offenbar keine neue Verbindung wiederhergestellt (ich meine das ging in einer früheren Version schonmal) und es bleibt bei ständigen timeout-Meldungen.

### Steps to reproduce

mqtt-Server im laufenden Betrieb stoppen

### Configuration details

```yaml
network:
  schema: http
  host: maf10 # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  easee: trace
  cache: error

# unique installation id
plant: xxx

interval: 10s # control cycle interval

sponsortoken: eyJhbGxxx # your token from sponsor.evcc.io

mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing

tariffs:
  grid:
    type: tibber
    token: "xxx" # access token
    homeid: "xxx" # optional if multiple homes associated to account
  feedin:
    type: fixed
    price: 0.2501 # [currency]/kWh

meters:
- type: custom
  name: gridspo
  power:
    source: mqtt
    topic: spo/gridPower
- type: custom
  name: pv
  power:
    source: mqtt
    topic: spo/pvPower
- type: custom
  name: battery
  power:
    source: mqtt
    topic: spo/batteryPower
  soc:
    source: mqtt
    topic: spo/batterySoc

vehicles:
- type: template
  name: tesla
  template: tesla
  #template: tesla-command
  title: Tesla Model 3
  accessToken: xxx
  refreshToken: xxx
  capacity: 60
  identifiers:
  - xxx
  mode: pv

chargers:
- type: template
  name: easee_carport
  template: easee
  user: xxx
  password: xxx
  charger: xxx
  authorize: true
- type: template
  name: easee_front
  template: easee
  user: xxx
  password: xxx
  charger: xxx
  authorize: true

loadpoints:
- title: Carport (rot) # display name for UI
  charger: easee_carport # Wallbox Referenz
  maxCurrent: 25
  phases: 0 # 0=Automatik eingeschaltet
  #vehicle: tesla # Referenz auf Standardfahrzeug
  mode: off # charge mode (off, now, minpv, pv)
- title: Vorne (blau) # display name for UI
  charger: easee_front # Wallbox Referenz
  maxCurrent: 32
  phases: 0 # 0=Automatik eingeschaltet
  #vehicle: tesla # Referenz auf Standardfahrzeug
  mode: off # charge mode (off, now, minpv, pv)

site:
  title: Mein Zuhause
  meters:
    grid: gridspo
    pv: pv
    battery: battery
  residualPower: 100
```

### Log details

```text
[site  ] DEBUG 2024/08/27 12:02:21 ----
[lp-1  ] DEBUG 2024/08/27 12:02:21 charge power: 11347W
[lp-1  ] DEBUG 2024/08/27 12:02:21 charge currents: [15.4 15.5 15.4]A
[lp-2  ] DEBUG 2024/08/27 12:02:21 charge power: 0W
[lp-2  ] DEBUG 2024/08/27 12:02:21 charge currents: [0 0 0]A
[site  ] DEBUG 2024/08/27 12:02:21 pv power: 2752W
[site  ] DEBUG 2024/08/27 12:02:21 battery soc: 81%
[site  ] DEBUG 2024/08/27 12:02:21 battery power: -735W
[site  ] DEBUG 2024/08/27 12:02:21 grid meter: -239W
[site  ] DEBUG 2024/08/27 12:02:21 site power: -874W
[lp-1  ] DEBUG 2024/08/27 12:02:21 detected active phases: 3p
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/1/chargeCurrents: timeout
[lp-1  ] DEBUG 2024/08/27 12:02:21 !! session: chargeRater.chargedEnergy=11.1 - chargedAtStartup=0.0
[lp-1  ] DEBUG 2024/08/27 12:02:21 charge total import: 4343.566kWh
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/pv/1/power: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/updated: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/batteryCapacity: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargeCurrent: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/sessionEnergy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/1/chargePower: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargeCurrents: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/1/chargeCurrents/l1: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/1/chargeCurrents/l2: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargePower: timeout
[lp-1  ] DEBUG 2024/08/27 12:02:21 charger status: C
[lp-1  ] DEBUG 2024/08/27 12:02:21 soc estimated: 84.50% (vehicle: 83.00%)
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/pvPower: timeout
[lp-1  ] DEBUG 2024/08/27 12:02:21 vehicle soc: 85%
[lp-1  ] DEBUG 2024/08/27 12:02:21 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/08/27 12:02:21 vehicle range: 338km
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargeCurrents/l1: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargeCurrents/l2: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargeCurrents/l3: timeout
[lp-1  ] DEBUG 2024/08/27 12:02:21 pv charge current: 17.3A = 16A + 1.27A (-874W @ 3p)
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/batteryGridChargeActive: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/pv/1/energy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/1/chargeCurrents/l3: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/pvEnergy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/batterySoc: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/pv: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargeTotalImport: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/effectivePriority: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/chargedEnergy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/planProjectedEnd: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/effectivePlanSoc: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/effectivePlanTime: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/site/greenShareHome: timeout
[mqtt  ] ERROR 2024/08/27 12:02:21 send: evcc/loadpoints/2/planProjectedStart: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/2/chargeCurrents/l1: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/battery/1/controllable: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/battery/1/capacity: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/battery/1/soc: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/battery/1/energy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/chargeCurrents: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/vehicleWelcomeActive: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/2/chargePower: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/2/chargeCurrents: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/chargeCurrents/l3: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/chargeCurrents/l2: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/chargeCurrents/l1: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/updated: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/chargePower: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/battery/1/power: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/battery: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/batteryGridChargeActive: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/batterySoc: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/batteryPower: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/2/chargeCurrents/l2: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/2/chargeCurrents/l3: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/batteryEnergy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/pvEnergy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/pv/1/energy: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/pv: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/pv/1/power: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/batteryCapacity: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/site/pvPower: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/charging: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/connected: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/effectiveMaxCurrent: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/effectiveLimitSoc: timeout
[mqtt  ] ERROR 2024/08/27 12:02:31 send: evcc/loadpoints/1/enabled: timeout
[site  ] DEBUG 2024/08/27 12:02:31 ----
[lp-1  ] DEBUG 2024/08/27 12:02:31 charge power: 11347W
```

### What type of operating system are you running?

Windows

### Version

0.130.4

Maschga, 2024-08-30:
Wie sollte evcc den Ausfall des mqtt-Servers handhaben?

maf-soft, 2024-08-30:
> Wie sollte evcc den Ausfall des mqtt-Servers handhaben?

Das wäre zu diskutieren, was da die eleganteste Lösung für die aktuell dadurch aufretenden Probleme wäre: wenn der letzte gültige grid-meter-Wert sagt, dass Überschuss da ist, wird die Ladeleistung immer weiter bis zum Maximum erhöht, jedenfalls verstehe ich es so. Vermutlich müsste man dafür eher direkt bei diesem Wert ansetzen, denn er kann ja auch aus anderen Quellen veraltet sein.

Zumindest sollte ein Ausfall erkannt werden und alle darüber empfangenen Werte ungültig werden, und natürlich sollte (evtl. in größer werdenen Abständen) eine neue Verbindung versucht werden, und wenn die wieder funktioniert, sollte der normale Betrieb wieder aufgenommen werden.

Bei erkanntem Ausfall ergibt es auch keinen Sinn, die an mqtt zu sendenen Werte ständig weiter zu senden und das Log mit timeout-Meldungen für jeden einzelnen Wert zu überschwemmen.

andig, 2024-09-01:
Kann ich nicht nachvollziehen:

```
❯ gor main.go --log error,mqtt:trace -c demo.yaml
[main  ] INFO 2024/09/01 14:23:42 evcc 0.0.0
[main  ] INFO 2024/09/01 14:23:42 using config file: demo.yaml
[mqtt  ] INFO 2024/09/01 14:23:42 connecting evcc-292687436 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/09/01 14:23:42 tcp://localhost:1883 connected
[mqtt  ] ERROR 2024/09/01 14:24:21 tcp://localhost:1883 connection lost: EOF
[mqtt  ] ERROR 2024/09/01 14:24:33 send: evcc/loadpoints/1/planOverrun: timeout
...
[mqtt  ] DEBUG 2024/09/01 14:24:36 tcp://localhost:1883 connected
```

closed 2024-09-01 by andig

maf-soft, 2024-09-02:
> Geht wunderbar.

Wärst du evtl. bereit noch einen zweiten Test zu machen, bei dem der Ausfall viel länger dauert, vielleicht eine halbe oder ganze Stunde? Ich hatte das jetzt schon mehrmals dass es sich nicht wieder fing, vielleicht frisst sich da was fest, läuft was über, was weiß ich, wenn es länger dauert...

andig, 2024-09-02:
Kann ich mal laufen lassen. Prüfst du mal, ob kurzes Recovery funktioniert?

maf-soft, 2024-09-03:
Habe auf Version 130.7 aktualisiert.
1. Test: mqtt-Server knapp über 5min gestoppt. send-timeouts laufen zunächst weiter, aber nach ca. 3 weiteren minuten kommt connected/subscribe und es scheint wieder alles zu laufen

Wird die Retry-Zeitspanne immer länger? Bis zu welchem Maximum?

maf-soft, 2024-09-05:
@andig bitte zurück auf bug... Ich hoffe es lässt sich nicht nur bei mir reproduzieren. Log-Dateien sind gespeichert.

maf-soft, 2024-09-11:
> Kann ich mal laufen lassen. Prüfst du mal, ob kurzes Recovery funktioniert?

Siehe oben, da sich das Problem reproduzieren lässt, bitte ich um Wiedereröffnung.

andig, 2024-09-11:
Logfile fehlt weiter...