evcc-io/evcc #15736: Nachträgliche Erkennung von grid meter: outdated

maf-soft, 2024-08-27:
Ich habe festgestellt, dass die Meldung "grid meter: outdated" erscheint, wenn seit dem Start von evcc noch nie Daten empfangen wurden (gut+richtig). Sobald einmal Daten kamen, kommt diese Meldung aber anscheinend nie wieder (oder ist es ein sehr langer timeout?) wenn die Daten ausfallen. Es wäre sinnvoll, wenn nach einem gewissen timeout nicht mehr mit veralteten Daten gearbeitet wird.

Hintergrund in meinem Fall:
bei mir kommen die grid-meter-Daten von einem mqtt-Server, der sie von Discovergy bekommt. Bei Discovergy kommt es leider zu Ausfällen - manchmal kommen keine Daten, manchmal kommen ständig die gleichen veralteteten Daten. Für letzteren Fall habe ich in meinem mqtt-Server eingebaut, dass er sie einfach nicht mehr weiterleitet, wenn sie zu alt sind. Ich hätte dann gehofft, dass evcc dann wieder outdated meldet, das passiert aber wohl nicht.

StefanSchoof, 2024-08-27:
Man kann am mqtt plugin ein timeout hinterlegen, siehe https://docs.evcc.io/docs/reference/plugins#mqtt-lesenschreiben

maf-soft, 2024-08-27:
Danke für den Hinweis, leider ist dort nicht erklärt was dann genau damit passiert und was der Defaultwert für den Timeout ist, ich hätte jetzt 30s vermutet. Ich befürchte, dass trotzdem der zuletzt korrekt empfangene Wert einfach "aktuell" bleibt und beim grid meter ist das halt tödlich...