evcc-io/evcc #15745: Elvi EVBox: Cable not released after charging is complete (after upgrade to 0.130.6)

jpwenzel, 2024-08-27:
### Describe the bug

Upgraded from 0.129.6 to 0.130.6.

Since, the (detachable) cable is not released from the socket in the charger after the charging session is complete. The lock remains active and the only way to release the cable is to power-cycle the charger or reboot it via the EVBox app.

I remember I had the same issue after https://github.com/evcc-io/evcc/pull/12459 was introduced?
Previously, switching the charging mode to "off" released the cable.

### Steps to reproduce

1. Start charging
2. Complete or end charging session (by switching the charging mode to "off")
3. Charging session is stopped but cable remains locked.

### Configuration details

```yaml
chargers:
  - name: evbox-ocpp
    type: template
    template: elvi
    stationId: EVB-P20......
    connector: "1"
    remotestart: true
    #idtag: evcc
    connecttimeout: 5m
    timeout: 30s
    meter: true
    metervalues: Energy.Active.Import.Register,Power.Active.Import,SoC,Current.Offered,Power.Offered,Current.Import,Voltage
```

### Log details

```text
[lp-1  ] INFO 2024/08/27 16:22:01 stop charging <-
[lp-1  ] INFO 2024/08/27 16:28:01 car disconnected
[lp-1  ] INFO 2024/08/27 16:28:01 vehicle updated: Hyundai Kona -> unknown
[lp-1  ] WARN 2024/08/27 16:28:01 charger out of sync: expected disabled, got enabled
```

### What type of operating system are you running?

Docker container

### Version

v0.130.6

jpwenzel, 2024-08-27:
[image]

jpwenzel, 2024-08-27:
(The car was definitely already disconnected on vehicle side)

premultiply, 2024-08-27:
This is a charger user setting.
Normally named something like UnlockCableOnEVsideDisconnect true/false

Not everybody likes to have the cable unlocked.
Many users want to keep it always locked like a fixed cable.

jpwenzel, 2024-08-27:
IMHO that setting is for the vehicle not the charger end of the cable.

It used to work fine before 0.130.x

I am going to downgrade to 0.126.6 and will follow up if it works there.

jpwenzel, 2024-08-27:
Here's the old issue from a few versions ago when the problem appeared first: https://github.com/evcc-io/evcc/issues/12821

I forgot exactly what it was but some OCPP change was reverted back then and made it work again.

(I still have to switch charging mode to "Off" before pulling the cable every time but that's a workaround I could live with, given the crappy OCPP implementation in EVBox...)

premultiply, 2024-08-27:
It was a failure before.

Car has to be disconnected first.
Charger side will unlock afterwards if it is configured in the charger to do so.
Otherwise the cable at the charger side will keep locked until an admin user tells the charger to unlock the cable.

This has been cleaned up now.

jpwenzel, 2024-08-27:
Nice to hear that! Thanks!
Is this a configuration setting that can be/needs to be configured in evcc?

I don't seem to find anything related to that in the EVBox app, not even in the admin/installer settings.

mfuchs1984, 2024-08-27:
Maybe it is only configurabe through the ocpp configuration key `UnlockConnectorOnEVSideDisconnect`?

premultiply, 2024-08-27:
You can query the the charger setup using `evcc charger --diagnose`.

andig, 2024-08-30:
We're not doing anything specifically to forbid disconnecting. Sounds like this is a charger issue where contacting support might help.

closed 2024-08-30 by andig

jpwenzel, 2024-08-30:
As helpful as the Elvi support was in the past, my hopes are not too high.

I'll go through the changelog to see if I can spot anything between the 2 versions.

After downgrading, the cable is unlocked and can be removed from the station side.

I am wondering if any of the recent changes had an (unintentional) side effect on charger lock/unlock.

andig, 2024-08-30:
Nothing that we'd be aware of, especially no config we're changing in that regard :(

jpwenzel, 2024-08-30:
I will try to grab some trace logs for both versions.

I remember the EVBox charger was not properly supporting profile changes and acted similarly dumb (red ring indicator and cable not unlocked).

andig, 2024-08-30:
Looking at https://github.com/evcc-io/evcc/pull/15271/files#diff-1433a5ce8c4d120c8c5246bc37c68ac913eb9717daf6cb5c2817a1cc4920c666 it seems that `metervalues` has been removed from the template. 

> I remember the EVBox charger was not properly supporting profile changes and acted similarly dumb (red ring indicator and cable not unlocked).

Please try:

```yaml
type: ocpp
metervalues: Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage
```

/cc @premultiply

jpwenzel, 2024-08-30:
I have not been able to reproduce the issue properly but I do in fact now seem to have found a working configuration.

(I think the issue once again is in the implementation on EVBox side...)

In evcc, this config works for me:

```
chargers:
  - name: evbox-ocpp
    type: template
    template: ocpp
    stationid: EVB-P20......
    connector: 1
    idtag: '1'
    connecttimeout: 5m
    timeout: 2m
    metervalues: Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage
    meterinterval: 4s
```

In the box, even though I am not using authentication via chip or card, I **do have to add one card** and enable it for **autostart**.

Then, I can control the charging modes via evcc UI.

Even removing the cable from the car without changing the mode to "off" now works! **AND** I can also remove the cable from the box, I can hear that the lock is disengaged as soon as the charging process is ended on vehicle side.

<3

andig, 2024-09-01:
@jpwenzel can we dissect your config to allow us to update the template for other users?

- idtag: '1' -> is this necessary? evcc would only use it when remote-starting transaction which you've not configured
- connecttimeout: 5m -> that's already the default
- timeout: 2m -> same
- metervalues: Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage -> these seem necessary
-  meterinterval: 4s -> is this necessary?

premultiply, 2024-09-01:
> metervalues: Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage -> these seem necessary

Can you explain why?
Until no measurands is autodetected which does not work correcty this should be unnecessary.

andig, 2024-09-01:
See https://github.com/evcc-io/evcc/issues/15745#issuecomment-2320347320. Plain config did not work.

But I agree. From this issue it is unclear why that should be the case. Maybe we should not do this change until better diagnosis is available. Seems we're missing a full logfile here.

jpwenzel, 2024-09-01:
Let me do the follwing:
- I'll strip the unneeded (default) parts from the configuration as recommended
- I'll verify the behavior with the `ocpp` template
- I'll then change the configuration to use the `evbox` template and try to verify the (correct) behavior here as well

jpwenzel, 2024-09-01:
This configuration fully works (incl. autostart and unlock/release cable without the need to prior switch the mode to "off"):
```
chargers:
  - name: evbox-ocpp
    type: template
    template: ocpp
    stationid: EVB-P20......
    connector: 1
    metervalues: Current.Import,Energy.Active.Import.Register,Power.Active.Import,Voltage

```

jpwenzel, 2024-09-01:
This configuration is working as well:

```
chargers:
  - name: evbox-ocpp
    type: template
    template: elvi
    stationId: EVB-P20......
    connector: "1"
```

andig, 2024-09-01:
... and now please try `template: ocpp`

premultiply, 2024-09-01:
Yes. And this should work exactly the same:

```yaml
chargers:
  - name: evbox-ocpp
    type: template
    template: ocpp
```

jpwenzel, 2024-09-01:
> Yes. And this should work exactly the same:
> 
> ```yaml
> chargers:
>   - name: evbox-ocpp
>     type: template
>     template: ocpp
> ```

This works as well!