evcc-io/evcc #15752: OCPP: evcc unable to charge wants to unplug replug car always (EN+ firmware)

jeffborg, 2024-08-27:
### Describe the bug

evcc is now unworkable when your ocpp evse decides to jump to finishing status upon plugging in

Prior to 0.130.0 evcc was quite happy to break ocpp spec and jump from Finishing back to SuspendedEVSE state (IE tell the evse to start charging!)

This happens even if the mode is set to min + solar or it's off (no solar) - I spent almost 1/2 hour trying to evcc into a working state this morning!

Only way to get going now seems to be to restart both the evse & evcc & unplug replug car! & make sure evcc is in min+solar or fast mode before plugging the car in! And not do this too quickly either otherwise evcc crashed see

```
[ocpp  ] TRACE 2024/08/28 08:23:02 send SN10052307203612: [2,"1810363461","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"30"}]
[ocpp  ] TRACE 2024/08/28 08:23:02 recv SN10052307203612: [3,"1810363461",{"status":"NotSupported"}]
panic: send on closed channel

goroutine 52 [running]:
github.com/evcc-io/evcc/charger.(*OCPP).configure.func1(0x0?, {0x0, 0x0})
	github.com/evcc-io/evcc/charger/ocpp.go:375 +0xc0
github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).ChangeConfiguration.func1({0x344b480?, 0x40014a63b0?}, {0x0?, 0x0?})
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:78 +0x54
created by github.com/lorenzodonini/ocpp-go/ocpp1%2e6.(*centralSystem).handleIncomingConfirmation in goroutine 87
	github.com/lorenzodonini/ocpp-go@v0.18.0/ocpp1.6/central_system.go:545 +0xc8
```

Note this evse can't be set to free vend mode as this disables ocpp!

I have tried other combos to try and get going

- unplug replug car with evcc enabled for charging - no go ends in finishing state
- unplug car, turn off evse, turn on evse, plug car in no go ends in finishing state
- restart evcc in enabled mode - doesn't even boot up timeout for ocpp configure

relying upon vendors to fix issues is really a non starter, my only real option is evcc isn't fixed is to implement ocpp in node red and just connect this to evcc.

### Steps to reproduce

1. unplug car
2. plug car back in
3. evcc still wants to unplug and plug car back in
4. jump to 1 evcc still wants to unplug and replug car in!

### Configuration details

```yaml
chargers:
  - name: solax
    type: template
    template: ocpp
    timeout: 2m
    # added for 0.130.0
    remotestart: true
    # commented out 19/8 for 0.130.0 update from 0.129.0
    # getconfiguration: false
    # metervalues: Energy.Active.Import.Register,Power.Active.Import,Current.Import
    # bootnotification: false
    # meterinterval: 10s
```

### Log details

```text
[ocpp  ] DEBUG 2024/08/28 08:07:55 charge point connected: SN10052307203612
[ocpp  ] TRACE 2024/08/28 08:07:57 send SN10052307203612: [2,"56388924","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/28 08:07:57 recv SN10052307203612: [3,"56388924",{"status":"Accepted"}]
[site  ] DEBUG 2024/08/28 08:07:58 ----
[lp-1  ] DEBUG 2024/08/28 08:07:58 charge power: 0W
[lp-1  ] DEBUG 2024/08/28 08:07:58 charge currents: [0 0 0]A
[circuit-main] DEBUG 2024/08/28 08:07:58 power: 558.75W
[circuit-main] DEBUG 2024/08/28 08:07:58 current: 0A
[site  ] DEBUG 2024/08/28 08:07:58 pv power: 807W
[... 85 lines cut ...]
[lp-1  ] DEBUG 2024/08/28 08:08:58 charge total import: 0.000kWh
[lp-1  ] DEBUG 2024/08/28 08:08:58 charger vehicle id: evcc
[lp-1  ] DEBUG 2024/08/28 08:08:58 vehicle status: A
[lp-1  ] DEBUG 2024/08/28 08:08:58 vehicle soc: 83%
[lp-1  ] DEBUG 2024/08/28 08:08:58 vehicle range: 335km
[lp-1  ] DEBUG 2024/08/28 08:08:58 pv charge current: 0A = 0A + -2.16A (496W @ 1p)
[lp-1  ] DEBUG 2024/08/28 08:08:58 pv enable timer reset
[lp-1  ] DEBUG 2024/08/28 08:08:58 pv timer inactive
[ocpp  ] TRACE 2024/08/28 08:09:04 recv SN10052307203612: [2,"18","Heartbeat",{}]
[ocpp  ] TRACE 2024/08/28 08:09:04 send SN10052307203612: [3,"18",{"currentTime":"2024-08-27T22:09:04Z"}]
[ocpp  ] TRACE 2024/08/28 08:09:05 send SN10052307203612: [2,"3269614396","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/28 08:09:05 recv SN10052307203612: [3,"3269614396",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/28 08:09:05 recv SN10052307203612: [2,"19","MeterValues",{"connectorId":1,"meterValue":[{"timestamp":"2024-08-27T22:09:04Z","sampledValue":[{"measurand":"Current.Offered","unit":"A","value":"32.0","context":"Trigger"},{"measurand":"Current.Import","unit":"A","phase":"L1","value":"0.0"},{"measurand":"Voltage","unit":"V","phase":"L1","value":"245.3","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"0.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}]}]
[ocpp  ] TRACE 2024/08/28 08:09:05 send SN10052307203612: [3,"19",{}]
[ocpp  ] TRACE 2024/08/28 08:09:06 recv SN10052307203612: [2,"20","StatusNotification",{"connectorId":1,"status":"Finishing","errorCode":"NoError","timestamp":"2024-08-27T22:09:03Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}"}]
[ocpp  ] TRACE 2024/08/28 08:09:06 send SN10052307203612: [3,"20",{}]
```

### What type of operating system are you running?

Linux

### Version

0.130.6

premultiply, 2024-08-28:
The final reason why it jumps to "Finishing" state is an internal timeout of the charger waiting for a transaction to be started after vehicle connection (Available->Preparing) either by the user (RFID or Autostart) or by CS (RemoteStartTransaction).

`remotestart: true` is set here so evcc SHOULD have started a new transaction but did not do this, at least not in time.

So we need to find out why. Charger signaling itself looks good so far.

What have you set for `interval` in evcc configuration?

This value might be just too large for the internal transaction timeout trap.

Can you please try to reduce it to `interval: 10s` and try again?

Additionally please attach the output of `evcc charger --diagnose` so that we can see all internal configuration settings of the charger.

jeffborg, 2024-08-28:
@premultiply interval was set to 30 seconds, Found 10 seconds was too bouncy but this was quite some time ago when I did the initial setup. I have changed it to 10 can see how this goes tomorrow.

Here is the output from `evcc charger --diagnose` just seems to be the same as the normal trace ocpp level.

```
/app # evcc charger --diagnose --config /config/evcc-min.yaml
[main  ] INFO 2024/08/28 15:18:56 evcc 0.130.6
[main  ] INFO 2024/08/28 15:18:56 using config file: /config/evcc-min.yaml
[db    ] INFO 2024/08/28 15:18:56 using sqlite database: /config/.evcc/evcc.db

[ocpp-1] DEBUG 2024/08/28 15:18:58 waiting for chargepoint: 5m0s

[ocpp  ] INFO 2024/08/28 15:19:25 charge point connected, registering: SN10052307203612
[ocpp  ] TRACE 2024/08/28 15:19:25 recv SN10052307203612: [2,"1755","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":0,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp  ] TRACE 2024/08/28 15:19:25 send SN10052307203612: [3,"1755",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/28 15:19:26 send SN10052307203612: [2,"1221330925","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2024/08/28 15:19:26 recv SN10052307203612: [3,"1221330925",{"status":"Rejected"}]
[ocpp-1] DEBUG 2024/08/28 15:19:26 failed configuring availability: Rejected
[ocpp  ] TRACE 2024/08/28 15:19:26 send SN10052307203612: [2,"2840335103","GetConfiguration",{}]
[ocpp  ] TRACE 2024/08/28 15:19:27 recv SN10052307203612: [3,"2840335103",{"configurationKey":[{"key":"vendorId","value":"EN+","readonly":false},{"key":"chargePointSN","value":"","readonly":false},{"key":"AuthorizationKey","value":"ABCDEFGHIJ123456","readonly":false},{"key":"ChargingParametersEnabled","value":"true","readonly":false},{"key":"AllowOfflineTxForUnknownId","value":"false","readonly":false},{"key":"AuthorizationCacheEnabled","value":"false","readonly":false},{"key":"AuthorizeRemoteTxRequests","value":"false","readonly":false},{"key":"BlinkRepeat","value":"0","readonly":false},{"key":"ClockAlignedDataInterval","value":"0","readonly":false},{"key":"ConnectionTimeOut","value":"30","readonly":false},{"key":"GetConfigurationMaxKeys","value":"50","readonly":true},{"key":"HeartbeatInterval","value":"60","readonly":false},{"key":"LocalAuthorizeOffline","value":"false","readonly":false},{"key":"LocalPreAuthorize","value":"false","readonly":false},{"key":"MaxEnergyOnInvalidId","value":"3000","readonly":false},{"key":"MeterValuesAlignedData","value":"Energy.Active.Import.Register","readonly":false},{"key":"MeterValuesSampledData","value":"Power.Active.Import,Energy.Active.Import.Register,Current.Import,Voltage,Current.Offered,Power.Offe","readonly":false},{"key":"MeterValueSampleInterval","value":"10","readonly":false},{"key":"MinimumStatusDuration","value":"0","readonly":false},{"key":"ResetRetries","value":"0","readonly":false},{"key":"ConnectorPhaseRotation","value":"","readonly":false},{"key":"StopTransactionOnEVSideDisconnect","value":"false","readonly":false},{"key":"StopTransactionOnInvalidId","value":"false","readonly":false},{"key":"StopTxnAlignedData","value":"","readonly":false},{"key":"StopTxnSampledData","value":"Energy.Active.Import.Register","readonly":false},{"key":"TransactionMessageAttempts","value":"0","readonly":false},{"key":"TransactionMessageRetryInterval","value":"15","readonly":false},{"key":"UnlockConnectorOnEVSideDisconnect","value":"false","readonly":false},{"key":"WebSocketPingInterval","value":"0","readonly":false},{"key":"NumberOfConnectors","value":"1","readonly":true},{"key":"MeterValuesAlignedDataMaxLength","value":"120","readonly":true},{"key":"MeterValuesSampledDataMaxLength","value":"120","readonly":true},{"key":"ConnectorPhaseRotationMaxLength","value":"120","readonly":true},{"key":"StopTxnAlignedDataMaxLength","value":"120","readonly":true},{"key":"StopTxnSampledDataMaxLength","value":"120","readonly":true},{"key":"SupportedFeatureProfiles","value":"Core,Reservation,FirmwareManagement,LocalAuthListManagement,RemoteTrigger,SmartCharging","readonly":true},{"key":"SupportedFeatureProfilesMaxLength","value":"120","readonly":true},{"key":"ReserveConnectorZeroSupported","value":"false","readonly":true},{"key":"LocalAuthListEnabled","value":"true","readonly":false},{"key":"LocalAuthListMaxLength","value":"10","readonly":true},{"key":"SendLocalListMaxLength","value":"10","readonly":true},{"key":"ChargeProfileMaxStackLevel","value":"5","readonly":true},{"key":"ChargingScheduleAllowedChargingRateUnit","value":"Current","readonly":true},{"key":"ChargingScheduleMaxPeriods","value":"5","readonly":true},{"key":"ConnectorSwitch3to1PhaseSupported","value":"false","readonly":true},{"key":"MaxChargingProfilesInstalled","value":"3","readonly":true}],"unknownKey":[]}]
[ocpp  ] TRACE 2024/08/28 15:19:27 send SN10052307203612: [2,"1375653863","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp  ] TRACE 2024/08/28 15:19:28 recv SN10052307203612: [3,"1375653863",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/28 15:19:28 recv SN10052307203612: [2,"1756","BootNotification",{"chargePointModel":"23","chargePointSerialNumber":"SN10052307203612","chargePointVendor":"EN+","firmwareVersion":"1.3.53"}]
[ocpp  ] TRACE 2024/08/28 15:19:28 send SN10052307203612: [3,"1756",{"currentTime":"2024-08-28T05:19:28Z","interval":60,"status":"Accepted"}]
[... cut ...] urand":"Voltage","unit":"V","phase":"L1","value":"243.2","context":"Trigger"},{"measurand":"Energy.Active.Import.Register","unit":"Wh","value":"2518.0","context":"Trigger"},{"measurand":"Power.Active.Import","unit":"W","value":"0.0","context":"Trigger"}]}],"transactionId":4}]
[ocpp  ] TRACE 2024/08/28 15:21:36 send SN10052307203612: [3,"1772",{}]
[main  ] FATAL 2024/08/28 15:21:37 cannot create charger 'solax': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
```

jeffborg, 2024-08-28:
So tomorrow morning,
Usual things
1 - because ocpp channel is too chatty see all my other bug reports and it was far less chatty in 0.129.0 the evse of course is disconnected
2 - just power cycled the evse, no unplug replug but only change is the interval set to 10 seconds from yesterday so I think evcc sent the messages faster than the evse deciding to timeout and ending in finishing state

The only thing in evcc is showing waiting for Authorisation which is a bit weird it's working!

[image]

Here is the logs of the evse connecting something might be in here to show why it's saying asking for auth but of course it's charging.

The other weird thing is why is the evse sending the `MeterValues` and `DataTransfer` messages twice every 10 seconds. if somehow this wasn't doubled up it may help the issue of the evse deciding to disconnect and being very unstable from 0.130 onwards

```
[ocpp  ] DEBUG 2024/08/29 08:49:32 charge point connected: SN10052307203612
[ocpp  ] TRACE 2024/08/29 08:49:32 send SN10052307203612: [2,"1159558247","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/29 08:49:32 recv SN10052307203612: [3,"1159558247",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 08:49:34 send SN10052307203612: [2,"2717001353","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/29 08:49:34 recv SN10052307203612: [3,"2717001353",{"status":"Accepted"}]
[site  ] DEBUG 2024/08/29 08:49:35 ----
[lp-1  ] ERROR 2024/08/29 08:49:35 charge power: timeout
[ocpp  ] TRACE 2024/08/29 08:49:36 send SN10052307203612: [2,"1015783284","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[lp-1  ] ERROR 2024/08/29 08:49:36 charge currents: timeout
[circuit-main] DEBUG 2024/08/29 08:49:36 power: -527.9W
[circuit-main] DEBUG 2024/08/29 08:49:36 current: 0A
[site  ] DEBUG 2024/08/29 08:49:36 pv power: 1806W
[site  ] DEBUG 2024/08/29 08:49:36 grid meter: -528W
[site  ] DEBUG 2024/08/29 08:49:36 site power: -528W
[lp-1  ] ERROR 2024/08/29 08:49:36 charge voltages: timeout
[lp-1  ] ERROR 2024/08/29 08:49:36 charge total import: timeout
[lp-1  ] DEBUG 2024/08/29 08:49:36 charger status: B
[lp-1  ] DEBUG 2024/08/29 08:49:36 pv charge current: min 6A > 2.3A (-528W @ 1p, battery: false)
[ocpp  ] TRACE 2024/08/29 08:49:36 recv SN10052307203612: [3,"1015783284",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 08:49:36 send SN10052307203612: [2,"1537393320","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-28T22:49:36Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/08/29 08:49:37 recv SN10052307203612: [3,"1537393320",{"status":"Accepted"}]
[lp-1  ] DEBUG 2024/08/29 08:49:37 max charge current: 6A
[ocpp  ] TRACE 2024/08/29 08:49:37 send SN10052307203612: [2,"276879150","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":3,"stackLevel":5,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2024-08-28T22:49:37Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp  ] TRACE 2024/08/29 08:49:38 recv SN10052307203612: [3,"276879150",{"status":"Accepted"}]
[lp-1  ] DEBUG 2024/08/29 08:49:38 charger enable
[ocpp  ] TRACE 2024/08/29 08:49:38 send SN10052307203612: [2,"1076851147","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[lp-1  ] DEBUG 2024/08/29 08:49:38 wake-up timer: start
[ocpp  ] TRACE 2024/08/29 08:49:39 recv SN10052307203612: [3,"1076851147",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 08:49:40 send SN10052307203612: [2,"3495524807","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2024/08/29 08:49:40 recv SN10052307203612: [3,"3495524807",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 08:49:40 recv SN10052307203612: [2,"7","BootNotification",{"chargePointModel":"23","chargePointSerialNumber":"SN10052307203612","chargePointVendor":"EN+","firmwareVersion":"1.3.53"}]
[ocpp  ] TRACE 2024/08/29 08:49:40 send SN10052307203612: [3,"7",{"currentTime":"2024-08-28T22:49:40Z","interval":60,"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 08:49:41 recv SN10052307203612: [2,"8","DataTransfer",{"vendorId":"EN+","messageId":"gatewayInfo","data":"{\"SN\":\"SN10052307203612\",\"fwVer\":\"V43.0.552\",\"gateCode\":107,\"measureType\":1,\"disconnectReason\":4,\"maxCurr\":320,\"phase\":1,\"sdCardStatus\":0}"}]
[ocpp  ] TRACE 2024/08/29 08:49:41 send SN10052307203612: [3,"8",{"status":"Accepted"}]
[ocpp  ] TRACE 2024/08/29 08:49:41 recv SN10052307203612: [2,"offline-1","StopTransaction",{"meterStop":32,"transactionId":2,"transactionData":[{"sampledValue":[{"measurand":"Energy.Active.Import.Register","context":"Transaction.End","value":"0.032","unit":"kWh"}],"timestamp":"2024-08-28T22:48:24Z"}],"timestamp":"2024-08-28T22:48:24Z","idTag":"evcc","reason":"SoftReset"}]
[ocpp  ] TRACE 2024/08/29 08:49:41 send SN10052307203612: [3,"offline-1",{"idTagInfo":{"status":"Accepted"}}]
[ocpp  ] TRACE 2024/08/29 08:49:42 recv SN10052307203612: [2,"9","StatusNotification",{"connectorId":0,"status":"Preparing","errorCode":"NoError","timestamp":"2024-08-28T22:49:41Z","vendorId":"EN+","info":"{\"reason\":\"plugInGun\",\"cpv\":0,\"rv\":0}"}]
[... cut ...]
[ocpp  ] TRACE 2024/08/29 08:50:09 recv SN10052307203612: [2,"24","DataTransfer",{"vendorId":"EN+","messageId":"cpChargingParameterRpt","data":"{\"duty\":100,\"cpVolt\":615,\"transactionId\":3,\"totalPowerFactor\":250,\"aPhasesPowerFactor\":0,\"bPhasesPowerFactor\":6,\"cPhasesPowerFactor\":1,\"ccVolt\":250,\"offset0\":2015,\"offset1\":2018,\"leakcurr\":20,\"AMBTemp\":275,\"lockStatus\":1,\"ctCurrent\":12}"}]
[ocpp  ] TRACE 2024/08/29 08:50:09 send SN10052307203612: [3,"24",{"status":"Accepted"}]
[site  ] DEBUG 2024/08/29 08:50:15 ----
[lp-1  ] DEBUG 2024/08/29 08:50:15 charge power: 1504W
```

jeffborg, 2024-08-29:
Correction on the 10 second interval, evcc is too bouncy once solar was above the minimum 6amps, will try 15 seconds

restart evcc with interval setting changed and back to square 1 had to do the following to get working again

- turn off evse
- unplug car
- restart evcc
- turn on evse
- wait till evcc said disconnected don't plug in car too quick
- plug in car

Also evcc has a warning about interval less and 30 seconds can lead to issues, 30 seconds worked pretty well it didn't need to be any quicker, the ocpp stuff shouldn't be doing commands for connection etc.. on the interval schedule.

Or could just go back to 0.129.0 and it was perfect 🤷

premultiply, 2024-08-29:
So is it working now with the reduced interval or not?

Log looks fine to me now.

jeffborg, 2024-08-29:
No I had to do the whole 6 step process in previous comment to get it working again and that is with a 15 second interval.

jeffborg, 2024-08-30:
Today the interval at 15 seconds I'm getting 
```
[lp-1 ] WARN 2024/08/30 13:43:18 charger logic error: current mismatch (got 16.2A measured, expected 14A) 
```

it's oscillating because the interval is really too quick to get stable power measurements between the evse grid meter and solar inverter and if a household load switches on/off and a major adjustment is required in the evse for solar tracking this occurs.

Lowering the interval to fix this isn't really a solution

andig, 2024-08-30:
> evcc is too bouncy once solar was above the minimum 6amps, will try 15 seconds

Sounds like your car consumes reactive power.

> Lowering the interval to fix this isn't really a solution

It shouldn't be. In this case: please contact vendor how charger can start session if it can't be started immediately.

Also: it's totally unclear to me why this should have worked on 129. We were also using RemoteStart then and only did this in the regular intervals.

> Today the interval at 15 seconds I'm getting

The log doesn't show any of that.

What it shows is this, so there is still a different issue that needs be fixed:

> charge power: timeout

andig, 2024-08-30:
Since there are more problems with this charger that do not seem contributable to evcc (https://github.com/evcc-io/evcc/issues/15552) I'm closing here. If there's an actual issue with evcc feel free to reopen.

closed 2024-08-30 by andig

jeffborg, 2024-08-31:
@andig there were a lot less problems before 0.130.0. I have gone back to 0.129.0 just turn EVSE on off and it started working no funny unplugging car etc required.