evcc-io/evcc #15754: Seat Mii electric API Anbindung funktioniert nicht mehr

moritzmair, 2024-08-28:
### Describe the bug

Habe gestern ein upgrade von Version v.0.126.x auf Version v.0.130.6 durchgeführt. Seitdem funktioniert die API Anbindung zu meinem Seat Mii electric nicht mehr. Davor ging alles problemlos. An der Config hab ich auch nichts verändert, es muss also mit dem upgrade zu tun haben.
Neustarten und eine Nacht warten und nochmal neustarten habe ich bereits probiert ohne Erfolg.

### Steps to reproduce

1. Seat Mii einrichten
2. Upgrade von v.0.126.x auf Version v.0.130.6
3. EVCC Webinterface besuchen

### Configuration details

```yaml
vehicles:
- name: seat_mii
  type: template
  template: seat
  title: Seat Mii
  user: <meine email>
  password: <mein passwort>
  vin: <meine vin>
  capacity: 36
  phases: 2
```

### Log details

```text
[main ] ERROR 2024/08/28 06:28:02 creating vehicle seat_mii failed: cannot create vehicle type 'template': cannot create vehicle type 'seat': missing id_token
```

### What type of operating system are you running?

Linux

### Version

0.130.6

StefanSchoof, 2024-08-28:
Gibt es neue AGB in App oder auf der Webseite die man akzeptieren muss?

StefanSchoof, 2024-08-28:
evcc macht nur beim Start ein Login und zwischen durch wird der Token erneuert und das dazu das man denkt es gibt einen Zusammenhang zu dem Update und in den aller meisten Fällen ist es aber nicht das Update. Kannst ja mal testweise zurück auf die alte Version gehen und sehr wahrscheinlich geht es da auch nicht mehr.

moritzmair, 2024-08-28:
Tatsache, ich hatte erst nur in der App geschaut, jetzt aber nochmal über https://cupraid.vwgroup.io/landing-page dort gab es tatsächlich neue AGB. Einmal akzeptieren, EVCC neustarten und alles geht wieder.
Danke.

closed 2024-08-28 by moritzmair