evcc-io/evcc #15766: OpenWB - Improve Documentation

Stroell, 2024-08-28:
Hi,
I'm new to EVCC and it was really frustrating to get my OpenWB Wallbox with Software 2.0 working with EVCC.
It would be great if the following could be added to the Documentation:
https://docs.evcc.io/docs/devices/chargers#software-2x

> In den OpenWB Einstellungen - Allgemein müssen folgende zwei Punkte aktiviert werden:
> Steuerungsmodus: secondary
> Steuerung über Modbus als secondary: An

[image]

Thank you for your fantastic product and keep up the good work!
Thanks,
Stefan

andig, 2024-08-28:
Happy to take a PR- just add it to the openwb template.

StefanSchoof, 2024-08-28:
Best under https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/openwb-2.0.yaml click on edit write an text that you would helped.

closed 2024-08-31 by naltatis