evcc-io/evcc #15768: Project EV

wonkydog, 2024-08-28:
### Describe the bug

Hi,

Using OCPP and EVCC to monitor and control a Project EV charger. Previously working correctly on V0.129.# however the upgrade to 0.130.# is producing an errors regarding voltage and charge total import.

I can't see anything obvious in the logs so wondering if this is a bug? 

See below configuration and log output - please let me know if there is anything else I can do to try and diagnose or solve the issue. 
    
Thank you for any help or support - FYI rolling back to 129 works perfectly (With the required change to config regarding remote start)

  

### Steps to reproduce

1. Upgrade previous 129 version to 130

### Configuration details

```yaml
sponsortoken: ###

site:
  title: 3 Firth Close
  meters:
    grid: grid
    
    
    
mqtt:
  broker: 192.168.3.26:1883
  topic: evcc # root topic for publishing, set empty to disable publishing

network:
  port: 7070

log: trace

interval: 10s

meters:
  - name: grid
    type: custom    
    power:
      source: mqtt
      topic: glow/4417937B0/SENSOR/electricitymeter
      timeout: 30s
      jq: .electricitymeter.power.value
      scale: 1000

circuits:
  - name: main
    title: 'Main Circuit'
    maxCurrent: 95
    maxPower: 20000
  - name: external
    title: 'External'
    maxCurrent: 45
    maxPower: 10000
    parent: main

chargers:
  - name: garage
    type: template
    template: ocpp
    stationid: TAG0D1024B
    connector: 1
    timeout: 2m # optional 
    remotestart: true
[... 151 lines cut ...]
[lp-1 ] INFO 2024/08/28 12:40:12 meters: charge ✓
[lp-1 ] INFO 2024/08/28 12:40:12 charge: power ✓ energy ✓ currents ✓
[lp-1 ] DEBUG 2024/08/28 12:40:12 phase timer inactive
[lp-1 ] DEBUG 2024/08/28 12:40:12 pv timer inactive
[lp-1 ] INFO 2024/08/28 12:40:12 vehicle updated: unknown -> JET73D
[ocpp ] TRACE 2024/08/28 12:40:12 recv TAG0D1024B: [3,"1868551218",{"status":"Rejected"}]
[ocpp ] TRACE 2024/08/28 12:40:12 send TAG0D1024B: [2,"1290609915","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2024/08/28 12:40:12 send TAG0C41E03: [2,"680918305","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2024/08/28 12:40:12 recv TAG0D1024B: [3,"1290609915",{"status":"Accepted","connectorId":1,"chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}]
[ocpp ] TRACE 2024/08/28 12:40:12 recv TAG0C41E03: [3,"680918305",{"status":"Rejected"}]
[ocpp ] TRACE 2024/08/28 12:40:12 send TAG0C41E03: [2,"233931057","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2024/08/28 12:40:13 recv TAG0C41E03: [3,"233931057",{"status":"Accepted","connectorId":1,"chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}]
[lp-1 ] DEBUG 2024/08/28 12:40:13 charge power: 0W
[lp-1 ] DEBUG 2024/08/28 12:40:13 charge currents: [0 0 0]A
[lp-1 ] ERROR 2024/08/28 12:40:13 charge voltages: not available
[lp-1 ] ERROR 2024/08/28 12:40:13 charge total import: not available
```

### What type of operating system are you running?

Linux

### Version

V0.130.6

premultiply, 2024-08-29:
The charger does not send any metering data (yet?) but accecpted the measurands configuration.
It also refuses to send metering data after triggered by watchdog. There is nothing left we could do.

Just try to start charging.

wonkydog, 2024-08-29:
Charging does work on 130 and most of the data is there.

I had to change the device change mode to 'Plug and Charge' but also needed to set remote start in EVCC to true in order to get a connection established.

All data is there apart from the total charge Watts. I can't work out why this would be?

The GUI also says that it is 'waiting for authorisation' despite the 'Plug and Charge' mode being enabled?

Does the 'idtag' setting refer to the RFID card the charger is looking for? The device does not have a RFID reader but it is an option for one of the other models, however I can still specify this in the charger itself? Maybe this could help authorise?

I have two of these chargers, the above relates to the one that has connected. There is no difference in any of the setups on the devices however the second refuses to connect regardless of how many restarts of EVCC and the charger I try! 

I also found this relating to the some of the blank queries: (https://community.home-assistant.io/t/project-ev-car-charger/399106/17)

Would there be any way to have an option that assumes the last values in case of a blank response?

I appreciate that the newer OCPP integration is more conformant but I'm a little disheartened as the old version worked seamlessly!!

Thanks for any help!

Rolled back to 129 for now.

WD

premultiply, 2024-08-29:
> I had to change the device charge mode to 'Plug and Charge' but also needed to set remote start in EVCC to true in order to get a connection established.

None of this setting is in any way related to the initial connection.

Apart from this the OCPP implementation of this charger looks quite incomplete. Many of the mandatory configuration settings are missing.

wonkydog, 2024-08-30:
Thank you very much for your reply. 

What about the 'id tag' field. Does that have any impact on authentication with the charge device?

I have managed to get some limited data as explained above from one of the devices. Would another trace log help in diagnosis now some data is being captured?

I'm also in contact with the vendor of the charger to try and install a later firmware that may help also - it's slow progress with them though.

Is there anyway to customise what messages are sent/received via OCCP or restore the previous functionality as that worked with this charger flawlessly? Is there anything I can help with in this regard? 

Maybe this could be done with a template?

Thanks for anything further!

WD

andig, 2024-09-01:
Its is totally unclear what problem the box might have with totally valid evcc messages. We can neither receive meter values nor set charging profile. The latter might potentially be fixed by setting

    stacklevelzero: true

That said there diesn‘t seem to be any apprent issue with evcc, hence closing.

closed 2024-09-01 by andig

wonkydog, 2025-04-23:
Sorry to reopen this. 

I don't know if you would appreciate it under a new issue, however I seem to have made some progress with this particular charger.

I can now get the charger to connect to EVCC and a charge can be initiated. There is a slight delay in the start of charging but I do have some limited control over the device in EVCC. I still need to dig into the charger to see if there is a delay start setting active but that's for another day. I'm taking the ability to charge as a positive step forward today.

In the logs I can see that meter values are rejected if the device is not actively charging. While charging the logs do specify seemingly correct meter values although EVCC appears to not like them and use them.

I have attached the logs I have got for today in the hopes that someone may be able to interpret them and come up with some suggestions.

My next step will be to look at specifying an extremely large meter values interval in the hopes it may avoid some of the messages.

I have looked through the docs and can also see that meter values can be set in the config. Could anyone provide a list of these or any insight to how this part of the config works. Am I specifying the values that should work or instead values that should be omitted?

Keen to provide this feedback as EVCC is great to use overall and this particular charger, although it's awful in setup and configuration is widely used across the UK and other will want to use.

Thanks all for your time!

[

[lp-1 ] DEBUG 2025/04/23 17:46:31 charge power: 0W
[lp-1 ] DEBUG 2025/04/23 17:46:31 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/23 17:46:31 charge voltages: [238 0 0]V
[lp-1 ] DEBUG 2025/04/23 17:46:31 detected connected phases: 1p
[lp-1 ] DEBUG 2025/04/23 17:46:31 charge total import: 0.000kWh
[lp-1 ] DEBUG 2025/04/23 17:46:31 charger status: B
[ocpp ] TRACE 2025/04/23 17:46:31 send TAG0D1024B: [2,"722717107","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2025/04/23 17:46:31 recv TAG0D1024B: [3,"722717107",{"status":"Accepted","connectorId":1,"chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}]
[ocpp ] TRACE 2025/04/23 17:46:31 send TAG0D1024B: [2,"709132359","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:32 recv TAG0D1024B: [3,"709132359",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:33 recv TAG0D1024B: [2,"523","Heartbeat",{}]
[ocpp ] TRACE 2025/04/23 17:46:33 send TAG0D1024B: [3,"523",{"currentTime":"2025-04-23T16:46:33Z"}]
[ocpp ] TRACE 2025/04/23 17:46:33 send TAG0D1024B: [2,"1183729924","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:34 recv TAG0D1024B: [3,"1183729924",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:35 send TAG0D1024B: [2,"1794949719","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:36 recv TAG0D1024B: [3,"1794949719",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:37 send TAG0D1024B: [2,"1384892134","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:38 recv TAG0D1024B: [3,"1384892134",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:39 send TAG0D1024B: [2,"3033435340","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:40 recv TAG0D1024B: [3,"3033435340",{"status":"Rejected"}]
[lp-1 ] DEBUG 2025/04/23 17:46:41 charge power: 0W
[lp-1 ] DEBUG 2025/04/23 17:46:41 charge currents: [0 0 0]A
[ocpp ] TRACE 2025/04/23 17:46:41 send TAG0D1024B: [2,"1119100386","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:42 recv TAG0D1024B: [3,"1119100386",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:43 send TAG0D1024B: [2,"782671055","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:44 recv TAG0D1024B: [3,"782671055",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:45 send TAG0D1024B: [2,"3912515356","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:46 recv TAG0D1024B: [3,"3912515356",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:47 send TAG0D1024B: [2,"1070357143","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:48 recv TAG0D1024B: [3,"1070357143",{"status":"Rejected"}]
[ocpp ] TRACE 2025/04/23 17:46:49 send TAG0D1024B: [2,"4004396678","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/04/23 17:46:50 recv TAG0D1024B: [3,"4004396678",{"status":"Rejected"}]
[lp-1 ] DEBUG 2025/04/23 17:46:51 charge power: 0W
[lp-1 ] DEBUG 2025/04/23 17:46:51 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/04/23 17:46:51 charge voltages: [238 0 0]V
[lp-1 ] DEBUG 2025/04/23 17:46:51 detected connected phases: 1p
[lp-1 ] DEBUG 2025/04/23 17:46:51 charge total import: 0.000kWh
[lp-1 ] DEBUG 2025/04/23 17:46:51 charger status: B
[ocpp ] TRACE 2025/04/23 17:46:51 send TAG0D1024B: [2,"584189080","GetCompositeSchedule",{"connectorId":1,"duration":60}]
[ocpp ] TRACE 2025/04/23 17:46:51 recv TAG0D1024B: [3,"584189080",{"status":"Accepted","connectorId":1,"chargingSchedule":{"chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":0}]}}]
[... cut ...] context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L1","unit":"V"},{"value":"3906","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L1","unit":"W"},{"value":"30.0","context":"Sample.Periodic","format":"Raw","measurand":"Temperature","unit":"Celsius"}]}]}]
[ocpp ] TRACE 2025/04/23 18:00:34 send TAG0D1024B: [3,"599",{}]
[lp-1 ] DEBUG 2025/04/23 18:00:41 charge power: 3906W
[lp-1 ] DEBUG 2025/04/23 18:00:41 charge currents: [16.4 0 0]A

]()

premultiply, 2025-04-23:
> While charging the logs do specify seemingly correct meter values although EVCC appears to not like them and use them.

> [ocpp ] TRACE 2025/04/23 18:00:23 send TAG0D1024B: [2,"3956942422","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
> [ocpp ] TRACE 2025/04/23 18:00:24 recv TAG0D1024B: [...]

I can no see any error here receiving the meter values.