evcc-io/evcc #15769: Trial token not working

Flirina, 2024-08-28:
### Describe the bug

The trial token stopped working today although stated to work until 31.08.

# trial token, valid until 2024-08-31
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjUwOTQ4MDAsImlhdCI6MTcyNDIzMDgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.61RAc8c7vXkgX_INZgM5Cjbq9nta6XR-7Aodmle6gQ8

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# trial token, valid until 2024-08-31
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJldmNjLmlvIiwic3ViIjoidHJpYWwiLCJleHAiOjE3MjUwOTQ4MDAsImlhdCI6MTcyNDIzMDgwMCwic3BlIjp0cnVlLCJzcmMiOiJtYSJ9.61RAc8c7vXkgX_INZgM5Cjbq9nta6XR-7Aodmle6gQ8
```

### Log details

```text
[main  ] INFO 2024/08/28 13:48:10 evcc 0.130.6
[main  ] INFO 2024/08/28 13:48:10 using config file: /config/evcc.yaml
[main  ] INFO 2024/08/28 13:48:11 listening at :7070
[main  ] FATAL 2024/08/28 13:48:11 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/08/28 13:48:11 will attempt restart in: 15m0s
```

### What type of operating system are you running?

Linux

andig, 2024-08-28:
Please don't double-post. Thank you!

/cc @naltatis

closed 2024-08-28 by andig

Falke56, 2024-08-31:
The same today 2024/08/31. Do we get a new one tomorrow?

naltatis, 2024-08-31:
> Do we get a new one tomorrow?

Yes, new token is available.

Flirina, 2024-09-01:
could it be that the new token is not working?

[main  ] INFO 2024/09/01 02:27:09 evcc 0.130.6
[main  ] INFO 2024/09/01 02:27:09 using config file: /config/evcc.yaml
[main  ] INFO 2024/09/01 02:27:10 listening at :7070
[main  ] FATAL 2024/09/01 02:27:10 sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2024/09/01 02:27:10 will attempt restart in: 15m0s

or EVEN there is a bug atm? I copied the template file evcc.dist.io for testing and I get still the same error. sponsorship is commented out. 

also does not work when the chargers are deleted. still I get the same error.

naltatis, 2024-09-01:
Just double checked the new trial token. It's fine and should work. @Flirina did you play around with Config UI? Maybe you have set a token there. Config UI settings take precedence over evcc.yaml settings. See warning info box :)

Flirina, 2024-09-04:
hi! I couldn't remember that I inserted a token in the ui, but when I did insert the current token now it started to work again.