evcc-io/evcc #15770: EVCC stürzt ab. Yaml Anpassung bei Nutzung von TelsaBleHTTPProxy nötig?

dodo1912, 2024-08-28:
Hallo Zusammen.

ich komme nicht weiter: 
Tesla M Y 
TWC 3
Fronius Gen24 
Fronius Symo 7
BYD HVM 11
Tibber
Sponsorship Evcc vorhanden (token funktioniert)
Installiert auf nem Rasp 2w (der direkt über dem Auto hängt (am Querträger vom Garagentor) 
WLan AP direkt daneben...

ip:7070 öffnet EVCC. (check) Werte Solar und Tibber stimmen auch.
ip:8080/dashboard öffnet dieses Proxy config Fenster....(check) Schlüssel wurde mittels Schlüsselkarte wohl auch erfolgreich hinzugefügt....

Allerdings wird das Auto immer als Gastfahrzeug erkannt. Das Auto steht in der Liste zur Auswahl - Wähle ich das aus, stürzt gefühlt alles ab. 

Muss ich der Yaml (Fahrzeug) noch irgendwas anpassen? Während der Konfiguration hab die ip: als Proxy angegeben....müssen die access und refresh token raus aus der yaml? Wir scheint da läuft was doppelt!?

```
vehicles:
- type: template
  template: tesla 
  icon: car  
  accessToken: eyJutzi76hgfYz798uitjikgeaendert4V1B>
  refreshToken: EU_ageaendert2d7afda7865538dad864976>
  vin: LRausgext4  
  capacity: 77  
  phases: 3  
  commandProxy: IP:8080  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev5
```

_Originally posted by @dodo1912 in https://github.com/evcc-io/evcc/discussions/15741_

closed 2024-08-28 by andig

dodo1912, 2024-08-28:
Hmm hab ich was verpasst? Wieso geschlossen?

andig, 2024-08-28:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

dodo1912, 2024-08-29:
Anbei mal ein Log-Eintrag. Auto angesteckt...lädt als Gastfahrzeug. (automatische Änderungen nicht möglich) 
Dann EV5 aus Liste gewählt (Seltsam: am ersten Tag Stand da noch der der Name vom Auto - meine ich zumindest)

ich werde evcc jetzt mal neustarten und noch einen Test machen. 

```
Aug 29 19:12:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:22 charge currents: [0.3 0.1 0]A
Aug 29 19:12:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:22 pv power: 831W
Aug 29 19:12:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:22 battery soc: 100%
Aug 29 19:12:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:22 battery power: 0W
Aug 29 19:12:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:22 grid meter: -70W
Aug 29 19:12:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:22 grid currents: [0.835 1.17 0.891]A
Aug 29 19:12:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:22 site power: -69W
Aug 29 19:12:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:22 charge voltages: [5.2 0 0]V
Aug 29 19:12:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:22 !! session: chargeRater.chargedEnergy=1.5 - chargedAtStartup=0.0
Aug 29 19:12:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:22 charger status: A

Aug 29 19:12:51 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:51 ----
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 charge power: 507W
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 charge currents: [1.7 0.3 0.2]A
Aug 29 19:12:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:52 pv power: 830W
Aug 29 19:12:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:52 battery soc: 100%
Aug 29 19:12:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:52 battery power: -0W
Aug 29 19:12:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:52 grid meter: -39W
Aug 29 19:12:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:52 grid currents: [1.72 1.37 1.12]A
Aug 29 19:12:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:12:52 site power: -39W
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 charge voltages: [230 232 230]V
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 detected connected phases: 3p
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 charger status: C
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] INFO 2024/08/29 19:12:52 car connected
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 vehicle api refresh
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 pv timer elapse
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 pv timer inactive
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] INFO 2024/08/29 19:12:52 start charging ->
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 wake-up timer: stop
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] WARN 2024/08/29 19:12:52 charger out of sync: expected disabled, got enabled
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 charger enable
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:12:52 pv charge current: min 6A > 3.76A (-39W @ 3p, battery: true)
Aug 29 19:12:52 laden evcc[2403]: [lp-1  ] ERROR 2024/08/29 19:12:52 max charge current 6A: vehicle not capable of current control
Aug 29 19:13:21 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:21 ----
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 charge power: 4908W
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 charge currents: [7.1 7.1 7.1]A
Aug 29 19:13:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:22 pv power: 883W
Aug 29 19:13:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:22 battery soc: 100%
Aug 29 19:13:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:22 battery power: 4654W
Aug 29 19:13:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:22 grid meter: 323W
Aug 29 19:13:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:22 grid currents: [1.31 1.86 1.62]A
Aug 29 19:13:22 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:22 site power: 4977W
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 charge voltages: [231 231 229]V
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 detected connected phases: 3p
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 detected active phases: 3p
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 !! session: chargeRater.chargedEnergy=0.0 - chargedAtStartup=0.0
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 charger status: C
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:22 pv charge current: min 6A > 1.89A (4977W @ 3p, battery: true)
Aug 29 19:13:22 laden evcc[2403]: [lp-1  ] ERROR 2024/08/29 19:13:22 max charge current 6A: vehicle not capable of current control
Aug 29 19:13:27 laden evcc[2403]: [lp-1  ] INFO 2024/08/29 19:13:27 vehicle updated: unknown -> Ev5
Aug 29 19:13:51 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:51 ----
Aug 29 19:13:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:52 charge power: 4912W
Aug 29 19:13:52 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:13:52 charge currents: [7.1 7.1 7.1]A
Aug 29 19:13:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:52 pv power: 868W
Aug 29 19:13:52 laden evcc[2403]: [site  ] DEBUG 2024/08/29 19:13:52 battery soc: 99%
[... cut ...]
Aug 29 19:16:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:16:22 wake-up timer: start
Aug 29 19:16:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:16:22 !! session: chargedEnergy=0.3 > chargedEnergy=0.0
Aug 29 19:16:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:16:22 !! session: start=0.000 stop=0.000 chargedEnergy=0.263
Aug 29 19:16:22 laden evcc[2403]: [lp-1  ] WARN 2024/08/29 19:16:22 charger out of sync: expected enabled, got disabled
Aug 29 19:16:22 laden evcc[2403]: [lp-1  ] DEBUG 2024/08/29 19:16:22 charger disable
```