evcc-io/evcc #15795: v0.130.6 does not start - wallbox is connected via OCPP

timostr, 2024-08-29:
### Describe the bug

Version 0.130.6 does not start.
Going  back to version 0.130.3 fixes the start problem.

The web interface of the walllbox shows connected for the occp connection.

Latest version of HomeAssistant is installed:
* Core 2024.8.3
* Supervisor 2024.08.0
* Operating System 13.1
* Frontend 20240809.0

### Steps to reproduce

1. Update to evcc 0.130.6
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: https
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: *******.de
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 30s # control cycle interval (default 30s)

# sponsor token enables optional features (request at https://cloud.evcc.io)
sponsortoken: "****"

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true
plant: ****

# log settings
log: debug # info
levels:
  site: debug
  lp-1: trace
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
#  - port: 502
#    uri: 192.168.188.56:1502 # Vom Wechselrichter
#    # rtu: true
#    # readonly: true

[... 199 lines cut ...]
  #   tags: <tags>
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/08/29 11:19:27 evcc 0.130.6
[main  ] INFO 2024/08/29 11:19:27 using config file: /config/evcc.yaml
[mqtt  ] INFO 2024/08/29 11:19:27 connecting evcc-556216010 at tcp://192.168.188.42:1883
[mqtt  ] DEBUG 2024/08/29 11:19:27 tcp://192.168.188.42:1883 connected
[main  ] INFO 2024/08/29 11:19:27 listening at :7070
[99009919-1] DEBUG 2024/08/29 11:19:37 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2024/08/29 11:19:42 charge point connected: 99009919
[main  ] FATAL 2024/08/29 11:21:54 cannot create charger 'SolaredgeWallbox': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2024/08/29 11:21:54 will attempt restart in: 15m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.6

andig, 2024-08-29:
Needs trace log containing site, loadpoint, ocpp.

timostr, 2024-08-29:
Well this is actually a problem. 
ecpp v0.130.6 is starting with following log settings:
log: trace 
levels:
  site: trace 
  lp-1: trace
  lp-2: trace
  cache: trace
  db: trace
  ocpp: trace

Anyhow log file is attached:
[49686a9f_evcc_2024-08-29T10-38-32.129Z.log](https://github.com/user-attachments/files/16797705/49686a9f_evcc_2024-08-29T10-38-32.129Z.log)

timostr, 2024-08-29:
Reproduceable as well with:
log: trace # info
levels:
  site: trace # debug
  lp-1: trace # debug
  lp-2: debug # debug
  cache: error # error
  db: error # error
  ocpp: trace
[49686a9f_evcc_2024-08-29T11-24-09.302Z.log](https://github.com/user-attachments/files/16798234/49686a9f_evcc_2024-08-29T11-24-09.302Z.log)

andig, 2024-08-29:
Closing until requested log available.

closed 2024-08-29 by andig

timostr, 2024-08-29:
I've provided requested log files. What is missing?

andig, 2024-08-29:
which one is the _single_ log requested?