evcc-io/evcc #15809: EVCC suddenly shows incorrect values: In=Out

Baaaazingaaaa, 2024-08-30:
### Describe the bug

Without a change to the system, EVCC suddenly displays incorrect IN, OUT values. Everything that is fed into the grid is displayed for IN and OUT. 
Charging processes are no longer started, despite feeding into the grid
I have already tested several versions. 
Grid meter is a sunny home manager 2.0

In this topic there is also the problem.
#15744 

[image]

Thanks for help

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
meters:

type: template
template: sma-home-manager
usage: grid
host: *
name: grid1
type: template
template: victron-energy
usage: battery
host: *
port: 502
name: battery2
chargers:

type: template
template: daheimladen-mb
host: *
port: 502
name: wallbox4
vehicles:

name: MG
type: template
template: mg
title: MG
user: *
password: *
vin: *
capacity: 64
loadpoints:

title: Einfahrt
charger: wallbox4
vehicle: MG
mode: pv
tariffs:
grid:
type: tibber
token: *

site:
title: Home
meters:
grid: grid1
battery:
- battery2

mqtt:
broker: 1*
topic: evcc
user: *
password: *

hems:
type: sma
AllowControl: false
```

### Log details

```text
will follow later
```

### What type of operating system are you running?

Linux

### Version

0.130.6

naltatis, 2024-08-30:
> Log details: will follow later

Please make sure we have everything we need to analyze when opening an issue. Without log files we can't do anything meaningful, here. I'll reopen once everything is there.

closed 2024-08-30 by naltatis

Baaaazingaaaa, 2024-08-30:
alles klar, danke.

Baaaazingaaaa, 2024-08-30:
Hallo,
Logfile ist hinzugefügt