evcc-io/evcc #15812: Easee Charge Wallbox Charger Logic, Current Mismatch

mibo-wb-gh, 2024-08-30:
### Describe the bug

When setting evcc into "quick" / "schnell" Mode Easee Charge Wallbox does not switch to 3phase 32A charging and stays in 1phase charging 6A charging. Problem (if i understodd correctly) was addressed with the version update 0.130.0
XXX
https://github.com/evcc-io/evcc/commit/9056fc60bda9f815b0f7c025a7d71060a7099a54 Easee: always pause/resume when phase switching on charger level (https://github.com/evcc-io/evcc/pull/15306)
XXX
problem still remains. charger logic error: current mismatch (got 6A, expected 32A)

Also tried with 1phase charging 8A setup, still the same
Or the charger switches to 1phase 32A max. 7.4KW

If I pause charging in the Easee App and resume charging then charging with 3phases 32A and 22KW is possible

### Steps to reproduce

Set Charging to "now / schnell"
Set Charging logic to auto
hook up car
...

### Configuration details

```yaml
excerpt:

interval: 120s

chargers:
  - type: template
    template: easee
    user: 
    password: 
    charger: EC6CAE5D
    timeout: 20s
    name: WB98-EASEE

vehicles:
  - type: template
    template: bmw
    title: WB98-iX50
    user: 
    password: 
    capacity: 105  
    name: WB98-iX50

loadpoints:
  - title: Garage
    charger: WB98-EASEE
    vehicle: WB98-iX50
    mode: minpv
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true

site:
  - title: WB98-EVCC
    meters:
      grid: WB98-GRID
```

### Log details

```text
[site  ] DEBUG 2024/08/29 19:59:48 grid meter: 480W
[site  ] DEBUG 2024/08/29 19:59:48 pv power: 0W
[site  ] DEBUG 2024/08/29 19:59:48 site power: 480W
[lp-1  ] DEBUG 2024/08/29 19:59:48 charge total import: 2778.580kWh
[lp-1  ] DEBUG 2024/08/29 19:59:48 charger status: A
[... 226 lines cut ...]
[easee ] TRACE 2024/08/29 20:03:48 ProductUpdate EC6CAE5D: (2024-08-29 18:03:46 +0000 UTC) OUT_VOLT_PIN1_3 236.93899536132812
[easee ] TRACE 2024/08/29 20:03:48 ProductUpdate EC6CAE5D: (2024-08-29 18:03:46 +0000 UTC) IN_CURRENT_T2 14.477999687194824
[easee ] TRACE 2024/08/29 20:03:48 ProductUpdate EC6CAE5D: (2024-08-29 18:03:46 +0000 UTC) TOTAL_POWER 3.502000093460083
[easee ] TRACE 2024/08/29 20:03:48 ProductUpdate EC6CAE5D: (2024-08-29 18:03:46 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 20.541000366210938
[easee ] TRACE 2024/08/29 20:03:53 ProductUpdate EC6CAE5D: (2024-08-29 18:03:51 +0000 UTC) CIRCUIT_TOTAL_PHASE_CONDUCTOR_CURRENT_L1 31.194000244140625
[easee ] TRACE 2024/08/29 20:03:53 ProductUpdate EC6CAE5D: (2024-08-29 18:03:51 +0000 UTC) TOTAL_POWER 7.422999858856201
[easee ] TRACE 2024/08/29 20:03:53 ProductUpdate EC6CAE5D: (2024-08-29 18:03:51 +0000 UTC) IN_CURRENT_T3 31.19499969482422
[easee ] TRACE 2024/08/29 20:03:53 ProductUpdate EC6CAE5D: (2024-08-29 18:03:51 +0000 UTC) IN_CURRENT_T2 31.19499969482422
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_INPUT_T2 40
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_INPUT_T4 32
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_INPUT_T3 38
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_INPUT_T5 32
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_OUTPUT_L1 34
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_OUTPUT_N 34
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_OUTPUT_L3 33
[easee ] TRACE 2024/08/29 20:04:31 ProductUpdate EC6CAE5D: (2024-08-29 18:04:29 +0000 UTC) TEMP_OUTPUT_L2 33
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.0

andig, 2024-09-01:
/cc @GrimmiMeloni is there anything to do here?

GrimmiMeloni, 2024-09-01:
Don't think so.

@mibo-wb-gh few remarks.

1. As you are only running a single Easee, #15306 is irrelevant for you.
2. The log clearly shows that evcc sends the respective commands for 3p switch and 32A. The Easee also confirms both commands, so it is not clear why it does not work as expected.
3. I have seen a similar case recently, were also the max Circuit Current was set to 1000A. I _believe_ (but do not know for sure!) that this is an installation error. My working theory is, that since 1000A is an impossible value, that it gets confirmed by the API, but technically ignored by the charger. I assume the correct value would be 32A - so you might want to try adjusting the configuration of our site on the Easee portal.

Closing here for now - please feel free to reopen if you have anything showing an actual error.

closed 2024-09-01 by GrimmiMeloni