evcc-io/evcc #15834: Easee Wallbox / Error at startup

Toppen77, 2024-09-01:
### Describe the bug

Error message at startup ->
_"Error during startup. Check your configuration and restart.
cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'easee': unexpected status: 400 (Bad Request)"_

- EVCC has been working before. No change on evcc.yaml  has been made beside update of sponsor token.
- No issues when controlling wallbox via Easee App or Tibber. 

### Steps to reproduce

1. start EVCC 

### Configuration details

```yaml
sponsortoken: 

site:
  - title: m59 # display name for UI
    meters:
      grid: my_grid # grid meter reference
      pv: my_pv
    residualPower: 100

loadpoints:
  - title: Garage # display name for UI
    charger: my_charger
    mode: minpv # charge mode (off, now, minpv, pv)
    enable:
      # einschalten, wenn 1 Minute lang mindestens 2000 W Überschuss vorhanden ist
      delay: 1m
      threshold: -3000
    disable:
      # ausschalten, wenn 30 Minuten lang mehr als 2000 W aus dem Netz bezogen werden
      delay: 30m
      threshold: 4500

meters:
  - name: my_grid
    type: template
    template: shelly-3em
    usage: grid
    host: 192.168.178.101
  - name: my_pv
    type: template
    template: shelly-3em
    usage: pv
    host: 192.168.178.77

chargers:
  - name: my_charger
    type: template
    template: easee
    user: 
    password: 
    charger: EHURXH9H
    timeout: 30s # optional
    authorize: # Steuert ob evcc die Authentifizierung am Charger vornimmt. Vorteil ist ein kontrollierter Ladestart. Nicht kompatibel mit RFID Identifikation von Fahrzeugen. (optional)

vehicles:
  - name: my_car
    type: template
    template: tesla
    title: # Wird in der Benutzeroberfläche angezeigt (optional)
    accessToken: 
    refreshToken: 
    #vin: W... # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)

tariffs:
  grid:
    type: template
    template: tibber
    token: 
    #homeid:  # Nur erforderlich, wenn du mehrere Häuser in deinem Tibber-Konto hast. (optional)
    #charges: # Zusätzlicher fester Aufschlag pro kWh (z.B. 0.05 für 5 Cent) (optional)
    #tax: # Zusätzlicher prozentualer Aufschlag (z.B. 0.2 für 20%) (optional)

# mqtt message broker
mqtt:
  broker: 192.168.178.100:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  # user:
  # password:
```

### Log details

```text
"Error during startup. Check your configuration and restart.
cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'easee': unexpected status: 400 (Bad Request)"
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.130.6

naltatis, 2024-09-01:
I dont see a general problem with Easee right now. My connection works.

Can you please extend the logging for Easee by adding this to you config file and post the details.

```
levels:
  easee: trace
```

Note: On trace you'll have to manually remove security relevant details like accessToken strings from the log output before posting it here.

Toppen77, 2024-09-01:
[main  ] INFO 2024/09/01 14:27:01 evcc 0.130.6
[main  ] INFO 2024/09/01 14:27:01 using config file: /config/evcc.yaml
[db    ] INFO 2024/09/01 14:27:01 using sqlite database: /data/evcc.db
[db    ] TRACE 2024/09/01 14:27:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT * FROM `settings` 2 <nil>
[mqtt  ] INFO 2024/09/01 14:27:01 connecting evcc-1822808070 at tcp://192.168.178.100:1883
[mqtt  ] DEBUG 2024/09/01 14:27:01 tcp://192.168.178.100:1883 connected
[db    ] TRACE 2024/09/01 14:27:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[main  ] INFO 2024/09/01 14:27:01 listening at :7070
[db    ] TRACE 2024/09/01 14:27:01 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db    ] TRACE 2024/09/01 14:27:01 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[easee ] TRACE 2024/09/01 14:27:01 POST https://api.easee.com/api/accounts/login
[main  ] FATAL 2024/09/01 14:27:01 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'easee': unexpected status: 400 (Bad Request)
[main  ] FATAL 2024/09/01 14:27:01 will attempt restart in: 15m0s
[main  ] TRACE 2024/09/01 14:27:01 GET https://api.github.com/repos/evcc-io/evcc/releases/latest

naltatis, 2024-09-01:
Just to verify. Your credentials from the evcc.yaml work on https://portal.easee.com/ right?

Your password does not contain special characters like `#` or a leading zero? And if it does make sure it's wrapped in quotes.

Toppen77, 2024-09-02:
Password was wrong. :-P 

Now it works. Password was wrong for unknown reason. 
I changed the password at portal.easee.com and updated the on stored in cvcc.yaml.

Thank you for your support!

closed 2024-09-02 by Toppen77