evcc-io/evcc #15836: GUI iPhone 15 Pro MAX

batmanvane, 2024-09-01:
### Describe the bug

GUI and iOS infos overlap in the top on iPhone 15 Pro MAX running iOS 17+ See attachment. 
[image]

### Steps to reproduce

open evcc webgui on ip 15 pro max.

### Configuration details

```yaml
i think not important. otherwise i can provide.
```

### Log details

```text
i think not important. otherwise i can provide.
```

### What type of operating system are you running?

Linux

### Version

130.6

StefanSchoof, 2024-09-01:
Try readd it to the home screen. This is caused by the iOS update.

naltatis, 2024-09-01:
Yes, looks like a bug apple introduced in one of its last iOS updates.
Let me know if readding it to the homescreen as @StefanSchoof suggested didn't work for you.

closed 2024-09-01 by naltatis

TobiasHuber1980, 2024-09-02:
Or try the iOS Beta App:

https://testflight.apple.com/join/8Y4elMpv evcc.io

batmanvane, 2024-09-02:
so yes, re-adding works - i forgot about this issue... thanks!