evcc-io/evcc #15845: Sonnenanteil sinkt stetig (PV Anlage mit Balkonkraftwerk)

phL9r, 2024-09-01:
### Describe the bug

Hallo!

Ich betreibe eine PV Anlage. Hinter dieser hängen noch Balkonkraftwerke. Derzeit ist es so, dass der Sonnenanteil beim Laden meines E-Fahrzeuges stetig sinkt, obwohl kein Netzbezug stattfindet. Ich konnte es mittlerweile darauf eingrenzen, dass z. B. zum jetzigen Zeitpunkt etwa 1 kW neben der Haupt-PV produziert wird. Dieses erkennt evcc richtigerweise und nimmt diesen Zusatzüberschuss auch zum Laden, betrachtet diesen jedoch nicht als Sonnenanteil, sondern als Netzbezug. Ich habe zur Verdeutlichung noch Screenshots angehängt. Ein Netzbezug in der Übersicht wird nicht angezeigt, lediglich am Ladepunkt wird Co2-Verbrauch ermittelt und der Sonnenanteil sinkt stetig. 
[image]

### Steps to reproduce

1. PV Anlage
2. Balkonkraftwerk
3. Ladung eines E-Fahrzeuges
...

### Configuration details

```yaml
X
```

### Log details

```text
X
```

### What type of operating system are you running?

Linux

closed 2024-09-01 by andig

phL9r, 2024-09-01:
Warum wurde direkt geschlossen?

CarG70, 2024-09-01:
ich schätze, weil Du weder Deine Config noch ein Log angehängt hast und hier keine Hellseher mitlesen.